espressif/esp-idf #13681: my buffer is not filling up

arvind55555, 2024-04-23:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

hi all,
i am currently working on a ESP-IDF reciever SPI code, where it receives data from transmitter, stores in a buffer and sends it to a external TCP socket via Wi-Fi. everything in this code works fine but, the problem i am facing is that for some reason my buffer is never filling up and i want to know why. below is my full code:

```
#include <errno.h>
#include <stddef.h>
#include <stdint.h>
#include <stdio.h>
#include <stdlib.h>
#include <string.h>

#include "freertos/FreeRTOS.h"
#include "freertos/queue.h"
#include "freertos/semphr.h"
#include "freertos/task.h"

#include "lwip/dns.h"
#include "lwip/igmp.h"
#include "lwip/netdb.h"
#include "lwip/sockets.h"

#include "driver/gpio.h"
#include "driver/ledc.h"
#include "driver/spi_slave.h"
#include "esp_event.h"
#include "esp_log.h"
#include "esp_system.h"
#include "esp_wifi.h"
#include "nvs_flash.h"
#include "soc/rtc_periph.h"
#include "spi_flash_mmap.h"

// WiFi configuration
const char *ssid = "ESP";
const char *pass = "dynamicroad704";

// Server configuration (replace with your server details)
#define SERVER_IP "192.168.0.3"
#define SERVER_PORT 8080

// Pins in use
#define GPIO_MOSI 23
#define GPIO_MISO 19
#define GPIO_SCLK 18
#define GPIO_CS 5

// Buffer size
#define BUFFER_SIZE 16384 // 16K buffer size
#define HALF_BUFFER_SIZE (BUFFER_SIZE / 2)

// Socket variable
int sock = -1;

// Buffers for double buffering
uint32_t buffer1[BUFFER_SIZE];
uint32_t buffer2[BUFFER_SIZE];

// Pointers to the current buffers being filled and transmitted
volatile uint32_t *currentTxBuffer;
volatile uint32_t *currentRxBuffer;

// Function to initialize the socket
[... 148 lines cut ...]

    // Send data from the filled buffer to the TCP server
    send_data_to_socket((char *)currentTxBuffer, HALF_BUFFER_SIZE);

    // Swap the buffers for next transaction
    volatile uint32_t *temp = currentTxBuffer;
    currentTxBuffer = currentRxBuffer;
    currentRxBuffer = temp;

    // Print the contents of the currentTxBuffer after swapping
    print_buffer("Rx Buffer", currentRxBuffer, HALF_BUFFER_SIZE);

    // Add a delay or yield to prevent stack overflow
    // vTaskDelay(pdMS_TO_TICKS(1)); // Delay for 1 millisecond
  }
  // Close the socket before exiting the loop
  close_socket();
}
```

below you can see the printed outputs that i am receivinng from the transmitter. i am transmitted 8 kilobyte worth of data through SPI but for some reason the buffer only fills 1/3rd and rest all goes to zero, like shown below (**note that i am printing it as a signed**)

`-1082525832 1953966715 -1988321407 -1451638638 -75825772 930755735 -1535554153 -1434758249 -495218282 -1434695785 343561623 -2139496554 -2005199209 -142904170 712785302 -1468414570 1048286613 897331095 -1988539753 1434148503 545036950 2038037142 1736130198 1149023126 -1585863018 -1568977770 1249694101 -2139527273 1903910550 561780887 679072662 326849687 -1602367110 762795387 1903940993 -75762286 -713332331 561777303 695849622 -377872746 -1569074027 1752909206 276616598 578379671 343638422 -1854205289 -1250222441 561831830 1098487444 695970965 -1334100586 276617622 142172054 -931465579 -1434880618 1434131862 -92537963 1769778327 1434056086 1115385751 -293901929 1501335703 1920589206 1115371927 1518721665 -175440243 1351608468 -1349473386 -611374442 1301401238 512831127 -275714154 1603291030 1536218774 1569752470 1804664726 1435530391 1083199383 143665303 -896577641 -644830827 -812626537 -812613737 -1483669610 -1047493738 -1903113066 1586552726 1469079447 1536196501 1167084950 714115478 546365590 395321239 596702102 378558359 361824150 -445131657 -1955020932 -1770498689 1685693570 -897950319 41567382 1803115926 1971041686 -92545643 -1786999401 -1082556521 -931422313 -1535327081 -344334441 -1283731305 981262231 1920540053 -1233409642 -1870898539 -176614761 92053141 897184406 1350294166 830241430 -1451671403 -1971609706 1350309270 -1267096170 -696542314 -2005122922 -1468430954 1652290965 -175471222 1452024464 -342963308 -1517239146 -544243562 1217464471 -7278698 -191856745 -1466894186 2005964695 -1886339177 1871769494 1922086807 1435564439 445697943 345030039 59810709 -1081102442 1334914967 798051222 395419541 730944406 -175073643 2140155031 -1752138091 1603297174 1167120022 1418737303 1284538006 647030422 -141589098 1100072854 -1350522753 1602879357 1334776962 -409970798 630232980 1418769558 -1886326890 -1886326891 1116762006 1334855831 -1701819241 -661619305 764538263 1603361431 1435525015 2140160151 1150309524 -1785703018 -1617898091 43081623 361862038 747746454 1066476951 1754354582 -1533946730 -1483647081 -1752082538 -1148067947 -544121449 613598871 378606743 898708887 645320823 1500964732 360363391 1970920834 -8721519 1132115350 -2005277290 1467698582 -1602528363 293355671 1752784791 -1451527017 1098678679 1887006614 -1418020714 -226929001 1534799253 1501351318 -1585881705 1853595542 695941014 -629410667 1434240406 -1871052906 -1434763113 1434228630 1903775382 -2139375466 897319831 -2072380778 -1250196841 930902679 797821835 -762571375 1569670549 -1617888618 -544219242 663859095 328311703 -359617642 -2104449643 -1919947370 -560967786 -963575914 -2087697515 395320725 294658964 -896536683 546377108 294769044 -1081024619 2140176279 -1383119721 43079575 -2054135913 1116766102 1100017045 160459671 -930133865 9535383 -1064297833 1989165719 730912662 -712041065 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 `

i am pretty sure it is the way i am using the two buffer but i am not sure how exactly to implement it, can anyone help me in this?

wanckl, 2024-04-24:
@arvind55555 

Please note `t.length` is in **BIT** not BYTE

arvind55555, 2024-04-24:
> @arvind55555 
> 
> Please note `t.length` is in **BIT** not BYTE

Yeah I'm receiving 2048 Kilobit data right? So if that's the case why am I seeing zeros?

kayzen9, 2024-05-15:
@arvind55555  Try to multiply length by 8 - convert  to bytes... that shud fix the problem

wanckl, 2025-01-06:
@arvind55555 are you still facing confusing? If no please close this issue, thanks

closed 2025-01-06 by arvind55555