espressif/esp-idf #13682: Clarify the Release 5.3 if there is no official release yet?

filzek, 2024-04-24:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

Clarify the Release 5.3 if there is no official release yet?

dobairoland, 2024-04-24:
The testing of v5.3 has been started and there is a release/v5.3 branch where you can follow the progress and start to use it. v5.3 will be released when it passes all the tests (approximately in 2-3 Months). 

The development of v5.4 is started now in the master branch.

filzek, 2024-04-24:
Perfect, we will set one developer to start port it to it and follow

closed 2024-04-24 by filzek