espressif/esp-idf #13683: esp32c6 gpio wakeup stop working if CONFIG_PM_POWER_DOWN_PERIPHERAL_IN_LIGHT_SLEEP=y

AxelLin, 2024-04-24:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.3-dev-3593-g64a1ed07d9, v5.4-dev-12-g3d813afa01

### Espressif SoC revision.

ESP32C6

### Operating System used.

Linux

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### Development Kit.

Custom Board

### Power Supply used.

USB

### What is the expected behavior?

esp32c6 gpio wakeup stop working if CONFIG_PM_POWER_DOWN_PERIPHERAL_IN_LIGHT_SLEEP=y
Note, this was working in v5.2.1-443-g2ba5320112

### What is the actual behavior?

press the gpio button seems has no effect.(device does not wake up from sleep)

### Steps to reproduce.

I test it with my application code which I cannot share here.
The gpio wakeup related code are as below:
```
#if CONFIG_PM_POWER_DOWN_PERIPHERAL_IN_LIGHT_SLEEP
gpio_hold_en(MY_GPIO_BUTTON);
#endif

esp_sleep_enable_gpio_wakeup();
esp_light_sleep_start();

#if CONFIG_PM_POWER_DOWN_PERIPHERAL_IN_LIGHT_SLEEP
gpio_hold_dis(MY_GPIO_BUTTON);
#endif

esp_sleep_wakeup_cause_t wakeup_reason = esp_sleep_get_wakeup_cause();
// check wakeup reason..
```
It should be easy to reproduce it with some modification of  examples/system/light_sleep/ example code.

esp-lis, 2024-04-24:
@AxelLin In the design of the esp32c6 chip, GPIO (Digital IO is located in the TOP power domain), Configuring CONFIG_PM_POWER_DOWN_PERIPHERAL_IN_LIGHT_SLEEP=y means that the TOP domain will power down during sleep, it means that Digital IO cannot be used as a wake source, In this case you can use LPIO or EXT1 as the wake source, behaving in the same way as GPIO (Digital IO).

Use of EXT1 or LPIO wake configurations can be found [sleep_modes.html#external-wakeup-ext1](https://docs.espressif.com/projects/esp-idf/en/latest/esp32c6/api-reference/system/sleep_modes.html#external-wakeup-ext1)

AxelLin, 2024-04-24:
@esp-lis 
I don't get it.
It works if build with esp-idf-5.2 branch. It stop working after switch to 5.3+.

esp-lis, 2024-04-24:
@AxelLin Our recently optimized power domain management strategy (which should be included in version 5.3+), I guess on esp-idf-5.2 branch, CONFIG_PM_POWER_DOWN_PERIPHERAL_IN_LIGHT_SLEEP=y doesn't really power down TOP domain.

You can you the follow code to check the TOP domain state during system go to sleep.
`esp_rom_printf(DRAM_STR("pd_flags:%x\n"), pd_flags)`
put it into [sleep_modes.c#L707](https://gitlab.espressif.cn:6688/espressif/esp-idf/-/blob/master/components/esp_hw_support/sleep_modes.c?ref_type=heads#L707), the TOP domain is powered down during sleep if BIT(0) is set of **pd_flags**.

AxelLin, 2024-04-24:
@esp-lis 

v5.2: pd_flags:2c15
master (v5.4-dev-12-g3d813afa01) : pd_flags:2c15

so the pd_flags is the same for v5.2 and master.

BTW, I have apply this fix for my v5.2. (https://github.com/espressif/esp-idf/issues/11858#issuecomment-1640244734 to make modem_domain_pd_allowed() return true)

esp-lis, 2024-04-24:
> @esp-lis
> 
> v5.2: pd_flags:2c15 master (v5.4-dev-12-g3d813afa01) : pd_flags:2c15
> 
> so the pd_flags is the same for v5.2 and master.
> 
> BTW, I have apply this fix for my v5.2. ([#11858 (comment)](https://github.com/espressif/esp-idf/issues/11858#issuecomment-1640244734) to make modem_domain_pd_allowed() return true)

This is strange, **pd_flags:2c15** means that the TOP domain will be powered down during sleep.

```
#if CONFIG_PM_POWER_DOWN_PERIPHERAL_IN_LIGHT_SLEEP
gpio_hold_en(MY_GPIO_BUTTON);
#endif

esp_sleep_enable_gpio_wakeup();
esp_light_sleep_start();

#if CONFIG_PM_POWER_DOWN_PERIPHERAL_IN_LIGHT_SLEEP
gpio_hold_dis(MY_GPIO_BUTTON);
#endif

esp_sleep_wakeup_cause_t wakeup_reason = esp_sleep_get_wakeup_cause();
// check wakeup reason..
```
Based on the above code, I can't see any more problems, Can you share more information about GPIO wakeup configuration?

AxelLin, 2024-04-25:
There is nothing special for the GPIO wakeup configuration.
As I said the same code works if build with IDF-5.2, there must be some thing in newer IDF causes this regression.

esp-lis, 2024-04-28:
> There is nothing special for the GPIO wakeup configuration. As I said the same code works if build with IDF-5.2, there must be some thing in newer IDF causes this regression.

I'm unable to construct a case like this, could you please provide the simplest project that has this issue? It can help me locate the problem.

AxelLin, 2024-05-10:
@esp-lis 

I summary the issue:
1. Build with CONFIG_PM_POWER_DOWN_PERIPHERAL_IN_LIGHT_SLEEP=y using the test code.
   gpio wakeup works if build with esp-idf-v5.2 branch.
   gpio wakeup does tno work if build with v5.3 branch.

2. Test with v5.3 branch:
   If setting CONFIG_PM_POWER_DOWN_PERIPHERAL_IN_LIGHT_SLEEP=n, gpio wakeup works.
   If setting CONFIG_PM_POWER_DOWN_PERIPHERAL_IN_LIGHT_SLEEP=y but comment out esp_sleep_cpu_retention_init() call, gpio wakeup also works.
   i.e. Calling esp_sleep_cpu_retention_init() + CONFIG_PM_POWER_DOWN_PERIPHERAL_IN_LIGHT_SLEEP=y makes the gpio wakeup stop working in v5.3+.

Below is the testing code.

```
#define GPIO_WAKEUP_BUTTON GPIO_NUM_5
static void gpio_wakeup_test()
{
    esp_err_t err;

#if 1
    err = esp_sleep_cpu_retention_init();
    if (err)
        ESP_LOGE(TAG, "Failed to init cpu retention: %d", err);
#endif

    do {
        ESP_LOGI(TAG, "To enter sleep");
        vTaskDelay(pdMS_TO_TICKS(1 * 1000));

        //prepare_lightsleep
        gpio_wakeup_enable(GPIO_WAKEUP_BUTTON, wakeup_level == 0 ? GPIO_INTR_LOW_LEVEL : GPIO_INTR_HIGH_LEVEL);
        #if CONFIG_PM_POWER_DOWN_PERIPHERAL_IN_LIGHT_SLEEP
        gpio_hold_en(GPIO_WAKEUP_BUTTON);
        #endif
        esp_sleep_enable_gpio_wakeup();

        err = esp_light_sleep_start();

        //unprepare_lightsleep
        #if CONFIG_PM_POWER_DOWN_PERIPHERAL_IN_LIGHT_SLEEP
        gpio_hold_dis(GPIO_WAKEUP_BUTTON);
        #endif
        gpio_wakeup_disable(GPIO_WAKEUP_BUTTON);

        if (err) {
            ESP_LOGE(TAG, "esp_light_sleep_start err = %d", err);
        }

        if (gpio_get_level(GPIO_WAKEUP_BUTTON) == wakeup_level) {
            ESP_LOGW(TAG, "wakeup button pressed");

            //wait_until_wakeup_gpio_release
            if (gpio_get_level(GPIO_WAKEUP_BUTTON) == wakeup_level) {
            ESP_LOGD(TAG, "Waiting for GPIO%d to go high...", GPIO_WAKEUP_BUTTON);
            do {
                vTaskDelay(pdMS_TO_TICKS(30));
            } while (gpio_get_level(GPIO_WAKEUP_BUTTON) == wakeup_level);
            }
        }
        /* Check wake up reason */
        esp_sleep_wakeup_cause_t wakeup_reason = esp_sleep_get_wakeup_cause();
        ESP_LOGI(TAG, "wakeup_reason=%d", wakeup_reason);

        vTaskDelay(pdMS_TO_TICKS(1 * 1000));
    } while (1);
}
```

esp-lis, 2024-05-10:
> Build with CONFIG_PM_POWER_DOWN_PERIPHERAL_IN_LIGHT_SLEEP=y using the test code.
> gpio wakeup works if build with esp-idf-v5.2 branch.

If you are using the esp-idf version before commit [470d5406](https://github.com/espressif/esp-idf/commit/470d5406ec3040e84c8ce126fcc130ce1f797083) of release/v5.2, this result is as expected, otherwise, the result same as release/v5.3

> gpio wakeup does tno work if build with v5.3 branch.
>> If setting CONFIG_PM_POWER_DOWN_PERIPHERAL_IN_LIGHT_SLEEP=n, gpio wakeup works.

the result is as expected.

> gpio wakeup does tno work if build with v5.3 branch.
>> If setting CONFIG_PM_POWER_DOWN_PERIPHERAL_IN_LIGHT_SLEEP=y but comment out esp_sleep_cpu_retention_init() call, gpio wakeup also works.

the result is as expected, refer to the implement of [top_domain_pd_allowed](https://github.com/espressif/esp-idf/blob/release/v5.2/components/esp_hw_support/sleep_modes.c#L1948), because the all modules of CPU, MODEM _(WiFi, BLE and 802.15.4)_ and Digital peripherals are located in TOP power domain, **comment out esp_sleep_cpu_retention_init() call** means the CPU domain will not allow the TOP domain to power down during sleep.

> gpio wakeup does tno work if build with v5.3 branch.
>> i.e. Calling esp_sleep_cpu_retention_init() + CONFIG_PM_POWER_DOWN_PERIPHERAL_IN_LIGHT_SLEEP=y makes the gpio wakeup stop working in v5.3+.

the result is as expected, All conditions for the TOP domain are met, TOP domain will power down during sleep.

AxelLin, 2024-05-10:
> If you are using the esp-idf version before commit [470d5406](https://github.com/espressif/esp-idf/commit/470d5406ec3040e84c8ce126fcc130ce1f797083) of release/v5.2, this result is as expected,

No, I test with v5.2.1-507-gf8319f08cc and it still works.

esp-lis, 2024-05-11:
> No, I test with v5.2.1-507-gf8319f08cc and it still works.

I have trouble shoot for it, it was introduced in [60e985e7](https://github.com/espressif/esp-idf/commit/60e985e7afd8cfd905cdf823ec76661469564a51) and in this commit we optimized the clock configuration (disabling the clock for all modules), If you want to use it correctly, call the [rtc_gpio_init](https://github.com/espressif/esp-idf/blob/master/components/esp_driver_gpio/src/rtc_io.c#L55) interface, which will reenable the clock of LP IO.

AxelLin, 2024-05-13:
> > > > Build with CONFIG_PM_POWER_DOWN_PERIPHERAL_IN_LIGHT_SLEEP=y using the test code.
> > > > gpio wakeup works if build with esp-idf-v5.2 branch.
> > > 
> > > 
> > > If you are using the esp-idf version before commit [470d5406](https://github.com/espressif/esp-idf/commit/470d5406ec3040e84c8ce126fcc130ce1f797083) of release/v5.2, this result is as expected,
> > 
> > 
> > No, I test with v5.2.1-507-gf8319f08cc and it still works.
> 
> I have trouble shoot for it, it was introduced in [60e985e7](https://github.com/espressif/esp-idf/commit/60e985e7afd8cfd905cdf823ec76661469564a51) and in this commit we optimized the clock configuration (disabling the clock for all modules), If you want to use it correctly, call the [rtc_gpio_init](https://github.com/espressif/esp-idf/blob/master/components/esp_driver_gpio/src/rtc_io.c#L55) interface, which will reenable the clock of LP IO (GPIO5 is a LP IO, [gpio_wakeup_enable](https://github.com/espressif/esp-idf/blob/master/components/esp_driver_gpio/src/gpio.c#L634) checks and calls [rtc_gpio_wakeup_enable](https://github.com/espressif/esp-idf/blob/master/components/esp_driver_gpio/src/rtc_io.c#L260) to enable LPIO5 wakeup).

@esp-lis 

Calling rtc_gpio_init(GPIO_WAKEUP_BUTTON) before gpio_wakeup_enable() does not work.
(Looks like it fails to enter lightsleep)
E (4395) LightSleep: esp_light_sleep_start err = 259
W (4395) LightSleep: wakeup button pressed

I also tried comment out the 60e985e7  changes in components/esp_system/port/soc/esp32c6/clk.c,
but it still does not work.

BTW, I want to point out it's important to make existing application work in newer esp-idf versions.
If there are some changes need to update application code, you must document it in migration guide.

AxelLin, 2024-05-14:
@esp-lis 

I tried git bisect, but there are build failure for the commits between fc9484785b..0c68b9b06c, so I stop bisect.
Now I can confirm v5.3-dev-3006-gfc9484785b is good, but v5.3-dev-3049-g0c68b9b06c is bad.
Maybe you can try to figure out what is wrong between above commits, I believe you can reproduce the issue using above test code.

closed 2024-06-17 by espressif-bot