espressif/esp-idf #13687: Panic in I2C driver when reading from I2C EEPROM chip

sfzhi, 2024-04-24:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

5.2.1

### Espressif SoC revision.

ESP32-C3 revision 0.4

### Operating System used.

Linux

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### Development Kit.

Seeed XIAO ESP32C3

### Power Supply used.

USB

### What is the expected behavior?

It should be possible to read a longer sequence of bytes from an external EEPROM over I2C.

### What is the actual behavior?

```
Guru Meditation Error: Core  0 panic'ed (Store access fault). Exception was unhandled.

Core  0 register dump:
MEPC    : 0x40382350  RA      : 0x40382332  SP      : 0x3fc97970  GP      : 0x3fc94a00
TP      : 0x3fc7b4a4  T0      : 0x0000001f  T1      : 0x600c2140  T2      : 0x00000002
S0/FP   : 0x3fcae2c4  S1      : 0x3fcae2e8  A0      : 0x3fcae2c4  A1      : 0x00000020
A2      : 0x60013000  A3      : 0x00000020  A4      : 0x00000053  A5      : 0x00000001
A6      : 0xa0000000  A7      : 0x0000000a  S2      : 0x3fcae1ac  S3      : 0x3fcae1ac
S4      : 0xffffffff  S5      : 0x00000001  S6      : 0x3fc9e858  S7      : 0x3fca0ce4
S8      : 0x00000075  S9      : 0x00000001  S10     : 0x0000005a  S11     : 0x00000010
T3      : 0x00000038  T4      : 0x3fca11cc  T5      : 0x00001800  T6      : 0x00000000
MSTATUS : 0x00001881  MTVEC   : 0x40380001  MCAUSE  : 0x00000007  MTVAL   : 0x00000000
MHARTID : 0x00000000

Stack memory:
3fc97970: 0x00000000 0x3fcae1c4 0x00000053 0x40383fbc 0x000000ff 0x3fcae1c4 0x3fc9aaf8 0x40383f7a
3fc97990: 0x00000000 0x80000007 0x60023000 0x00000001 0x00001881 0x8000000b 0x3fcae590 0x40380f5a
3fc979b0: 0x00001881 0x80000007 0x00000009 0x403801e8 0x60023000 0x4038665c 0x40386668 0x00000000
3fc979d0: 0x3fc979cc 0x00000000 0x00000000 0x00000000 0x3fc979e4 0xffffffff 0x3fc979e4 0x3fc979e4
3fc979f0: 0x00000000 0x3fc979f8 0xffffffff 0x3fc979f8 0x3fc979f8 0x00000001 0x00000001 0x00000000
3fc97a10: 0x0001ffff 0x00000000 0x00000000 0x00000000 0x00000000 0x3fc97a20 0x00000000 0x00000000
3fc97a30: 0x00000000 0x3fc97a38 0xffffffff 0x3fc97a38 0x3fc97a38 0x00000000 0x3fc97a4c 0xffffffff
3fc97a50: 0x3fc97a4c 0x3fc97a4c 0x00000001 0x00000001 0x00000000 0x0001ffff 0x00000000 0x00000000
3fc97a70: 0x00000000 0x3fc9e60c 0x3fc9e674 0x3fca1ed8 0x3fca32f8 0x00000000 0x00000000 0x00000000
3fc97a90: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
```

### Steps to reproduce.

[... 57 lines cut ...]
                                                                                                                                                                                                
613             ptr[i] = HAL_FORCE_READ_U32_REG_FIELD(hw->fifo_data, data);                                                                                                                     
   0x40382348 <+444>:   lw      a4,28(a2)                                                                                                                                                       
   0x4038234a <+446>:   addi    a5,a5,1                                                                                                                                                         
   0x4038234c <+448>:   sw      a4,8(sp)                                                                                                                                                        
   0x4038234e <+450>:   lw      a4,8(sp)                                                                                                                                                        
   0x40382350 <+452>:   sb      a4,-1(a5)                                                                                                                                                       
                                                                                                                                                                                                
612         for (int i = 0; i < len; i++) {                                                                                                                                                     
   0x40382354 <+456>:   bne     a3,a5,0x40382348 <i2c_master_isr_handler_default+444>                                                                                                           
                                                                                                                                                                                                
/COMPONENT_DRIVER_DIR/i2c/i2c_master.c:                                                                                                                                                         
535             i2c_master->w_r_size = i2c_master->rx_cnt;                                                                                                                                      
   0x40382358 <+460>:   lw      a1,20(s0)                                                                                                                                                       
                                                                                                                                                                                                
536             i2c_operation->bytes_used += i2c_master->rx_cnt;                                                                                                                                
   0x4038235a <+462>:   lhu     a4,8(s1)                                                                                                                                                        
                                                                                                                                                                                                
535             i2c_master->w_r_size = i2c_master->rx_cnt;                                                                                                                                      
   0x4038235e <+466>:   add     a4,a4,a1                                                                                                                                                        
   0x40382360 <+468>:   slli    a4,a4,0x10                                                                                                                                                      
```
As can be seen from the register dump above, the exact spot where the problem occurs is
`0x40382350 <+452>:   sb      a4,-1(a5)`, which is the assignment to `ptr[i]` inside `i2c_ll_read_rxfifo(...)`.
Apparently the destination address (stored in `a5` and already incremented for the next loop iteration) is NULL.

mythbuster5, 2024-04-25:
@sfzhi I tried to reproduce with our i2c_eeprom_main example but failed. Could you please provide your sdkconfig?

mythbuster5, 2024-04-25:
IIRC，power-on reset will reset all condition, so after power-on reset, everything should begin as start. Some panic happen after power-on reset looks very strange to me even if GDB shows something. So 1. I want to see your configuration 2. See your reset reason
```
ESP-ROM:esp32c3-api1-20210207
Build:Feb  7 2021
rst:0x1 (POWERON),boot:0xc (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
```

sfzhi, 2024-04-25:
Here you go:
```
ESP-ROM:esp32c3-api1-20210207                                                    
Build:Feb  7 2021                                                                
rst:0x1 (POWERON),boot:0xe (SPI_FAST_FLASH_BOOT)                                 
SPIWP:0xee                                                                       
mode:DIO, clock div:1                                                            
load:0x3fcd5820,len:0x1880                                                       
load:0x403cc710,len:0xd50                                                        
load:0x403ce710,len:0x2f50                                                       
entry 0x403cc71a                                                                 
```
The `sdkconfig` file: [sdkconfig.txt](https://github.com/espressif/esp-idf/files/15108162/sdkconfig.txt).

sfzhi, 2024-04-25:
I have conducted a few more experiments and the results are even more puzzling than before. When the panic occurs, if I don't wait until the device is restarted automatically (after 3 seconds, as per the configuration), but press the reset button again immediately, the device boots fine and the problem does not occur. I can't think of any reasonable explanation of what's happening.

eriksl, 2024-08-01:
Test again with larger stack.

pawel6843, 2024-10-08:
Seems that this thread explains why my debugger was having a problem with the same line of code.
I unfortunately do not have such expertise as sfzhi.
I can only tell that the debugger gets again at this line after hitting "Continue". If I hit "Step Over", it goes to _UserExceptionVector: in the Assembler, and then here again.
The pointer is 0.
I really dunno what a strange reason can be behind that..
I have stumbled on this 2 times only, and I do not know how could I reproduce it.
I have not noticed such problem before - well maybe because as sfzhi has written, that strange thing is happening depending on power source. And before I was not powering my ESP through an external ESP-PROG like now.
Hm maybe under debug some more memory was used...
I was not using wifi.
I had some problem that looked like lack of memory caused apparently by the primary console not being choosen in sdkcongig as the "USB Serial/JTAG Controller", but don't know if it could be connected. https://esp32.com/viewtopic.php?f=13&t=31032

ESP32S3
IDF 5.2.2

eriksl, 2024-10-08:
> Test again with larger stack.

Or if not possible, make variables > sizeof(int) static (so they're not on the stack).

pawel6843, 2024-10-08:
For me it happened when I just lanuched the debugger for the first time on ESP.
Reducing stack sizes of tasks did not bring help in reproducing it.
But I dunno maybe reducing ESP_SYSTEM_EVENT_TASK_STACK_SIZE could? Or ESP_MAIN_TASK_STACK_SIZE? ESP_IPC_TASK_STACK_SIZE? ESP_MINIMAL_SHARED_STACK_SIZE?
[image]

eriksl, 2024-10-08:
Using static type buffers for your I2C API calls can be interesting to try, regardless of the stack size.

I have had a similar issue with the SPI API and there it turned out, for this particular function, the pointer was used asynchronously, so when my function returned, the related storage was  already cleaned up but the pointer was still used. Using static buffers will make a difference here, although you will loose the ability to do re-entrant en multi-threading stuff, but that's for later worry ;-).

Interestingly I am currently doing a little project with I2C sensors on ESP32-S3. Until now no real problems, but I haven't yet have any transfer sizes > 8 bytes.

Other issues I ran into:
- use 7 as value for dev_addr_length instead of I2C_ADDR_BIT_LEN_7, which makes the interface go into 10-bit mode with all kinds of trouble.
- irritating logging from the IDF I2C master code about unexpected behaviour when you don't connect pull-up resistors. It's I2C, I think that's kind of obvious???

pawel6843, 2024-10-08:
And bum, just after that I have that pointer in i2c_ll_read_rxfifo equal to 0.

eriksl, 2024-10-08:
Don't you have call backtrace? I often get to the cause of the trouble quite quickly following the backtrace. Dereferencing a null pointer normally leads to a "Guru Meditation" with a "load/store prohibited" exception. The point where this happens is usually noted in the backtrace, even if it's in IDF code. I don't have any experience with any debugger because I never had the need to use one, but maybe you're using a different environment? I am using a board connected to USB (while still in development phase), so I simply do `idf.py app-build monitor` and then idf.py shows the backtrace and resolves the addresses to source lines.

pawel6843, 2024-10-09:
Call backtrace? Do you mean functions call stack? Em if yes, then you can see it on my screenshot.
It happens in the same function about which sfzhi was writing: i2c_ll_read_rxfifo.
Eh when I do ifd.py app-build monitor, it wants me to do fullclean :p...

eriksl, 2024-10-09:
Is that a problem, to do a complete fresh build?

My point is that the backtrace doesn't only show the function, but also the line in the source where null pointer is hit. It can give a clue as to what exactly happens and which (pointer) variable is having the null value.

eriksl, 2024-10-09:
> And bum, just after that I have that pointer in i2c_ll_read_rxfifo equal to 0.

Or is what what you mean by this? I was assuming i2c_ll_read_rxfifo was an integer variable where a 0 value is legitimate. If it is a pointer and it is indeed a pointer to a fifi and it is null now, I'd rather think of memory corruption by another issue, I don't think the variable will be assigned a null value anywhere, but you'd have to check.

pawel6843, 2024-10-09:
Well a fullclean takes like a 10 min (somtimes more, even 20 min., when I include more components). And I do not have a particularly slow computer.
Am I have the same situation as sfzhi. The same pointer variable is 0. It is the second argument of the i2c_ll_read_rxfifo function.
Eh looks like I was writing in a hard to read language. Well it happens often even in my native, so I dont't blame you. :)

Your suspicion seems to be what I am also starting to think.
The reason why I wanted to set up the debug for the first time was in fact completely other part of the project, where also a pointer was strangely assigned 0, because it was causing crashes in a way that I was able to quite easily reproduce. It seems to be in situation when a particular third party lib is used.
I will probably write here when I will check it. I have to do it in the first place.

eriksl, 2024-10-09:
The only thing I can add at the moment is my own experiences. I now have support for one I2C sensor. That amount will gradually increase, including testing. I will also add some I/O extenders. So I may run into the same issue at some point, who knows. I am not using EEPROM's though, that might make a difference.

FWIW I am using a ESP32-S3 with 16 Mb of flash and 8 Mb of PSRAM (so RAM is never an issue ;-)). I am thinking, maybe the send/receive buffers should be allocated from internal, DMA-capable RAM? For SPI that's true, but in that case, the SPI code will copy the buffers if it's not. I am not sure about I2C. If you're allocating the buffers from the stack and you have not selected the option "allocate stack from SPIRAM", this should be fine. Otherwise I'd try with this option "off".

BTW I am so happy right now with my decadent i9-13900K, which I bought for a completely different purpose, but now it serves very well for IDF compilation. Indeed it's nice that IDF has all the fancy cmake and ninja stuff, so you can never have an out-of-date source file (including those from IDF) not in the image. BUT the whole has become very heavy this way. I am not too keen myself, too. For my previous projects (ESP8266) I was using a custom-crafted makefile which did exactly the right thing and nothing more, so compilation generally took just a few seconds. Now when I say "idf.py build", it starts to check for ALL IDF source directories and all components. Every single time. It's frustrating. What's frustrating most, is that actually makefiles are created by CMake, but you can't use them, they don't work without the proper (magic) incantation from the CMake environment. Otherwise it would be all simple.

eriksl, 2024-10-09:
I am curious what's here in the source: 
`0x40382350: i2c_ll_read_rxfifo at /COMPONENT_HAL_DIR/esp32c3/include/hal/i2c_ll.h:613`
The exact statement that causes the exception.

Also please note that I'm using an S3, not a C3, which is a completely different thing, so I may never encounter this issue. This means my IDF sources version may be different too.

eriksl, 2024-10-09:
For documentation purposes: my version says (at a different source code line):

```
__attribute__((always_inline))
static inline void i2c_ll_read_rxfifo(i2c_dev_t *hw, uint8_t *ptr, uint8_t len)
{
    for (int i = 0; i < len; i++) {
        ptr[i] = HAL_FORCE_READ_U32_REG_FIELD(hw->fifo_data, data);
    }
}
```
There are just two pointers there, "ptr" and "hw".  It would be interesting to place some assert()s there. Not much that can go wrong here, the code in itself is correct. The macro HAL_FORCE_READ_U32_REG_FIELD is something quite weird, but doesn't look suspicious.

Interesting that the parameter "len" is specified as "uint8_t". I've seen this more often in IDF and SDK code. This is bad practice. As it's not a pointer, due to C parameter promotion rules, it will get expanded to an "unsigned int" (32 bit here) and passed as such. So you're losing quite a bit of range (len can never be bigger than 255) and actually it requires the same amount of stack space and **more** code (calling function needs to expand to 32 bit, the called function must chop 24 bits off, check the code, it really happens!). So don't do it guys (from Espressif), just use unsigned int. These are not 8 bit microcontrollers where it mattered.

pawel6843, 2024-10-09:
I think there is rather no issue in the ESP IDF.
In my case it really turned out that I had a library causing pointers to turn into null. I see this no longer happens in the other part of the code where I was easily able to replicated a null pointer dereference. And it is probably the same here in this case of i2c_ll_read_rxfifo. If it would be different, and I would spot it, I will write, but I doubt it now when I fixed the third party library. Is probably won't happen anymore. Though to be honest it was not fault of the authors of that library, but my deobfuscation done on it. Since sfzhi went silent, maybe in his case it was also user to balme, and he was not especially keen to admit that... :)

It just happened that this part of IDF is not checking the pointer against it being null, and quite prone to be disturbed by the users. :)

I am also using ESP32S3 (sfzhi was using C3).

And about your last remarks: ah you are probably correct. But well it is not an 8-bit microcontroller, in this behemoth people do not care so much apparently...

eriksl, 2024-10-09:
Yep, an assert (pointer not null) here and there can seriously help.

sfzhi, 2024-10-09:
> Since sfzhi went silent, maybe in his case it was also the user to balme, and he was not especially keen to admit that... :)

@pawel6843 How nice of you to assume the worst of people, thank you ;)
I went silent because I had nothing to say further, but now that you have twisted my arm... :)

It has been a while since I spent any time looking into this issue. A few months back I rearranged my code to run WiFi initialization (the main suspect) a bit later. That made the problem to _almost_ go away. Now it happens very rarely, I fact it only happened one ever since. But the problem is far from solved. All I did was sweep it under the rug. I expect it to be back as soon as my project reaches the state when it starts using I2C a lot more, which will happen at some point.

Regarding backtraces, as I mentioned before, in some cases the problem occurs at the point that would have been impossible to reach, had the faulty pointer been NULL all along. And I didn't find a spot in the code that could plausibly set the pointer to NULL in between. That led me to the conclusion that another task must have been interfering. If that is true, this is where the usefulness of backtraces ends.

Regarding stack size, I have been building my project with `CONFIG_FREERTOS_CHECK_STACKOVERFLOW_CANARY=y` and `CONFIG_ESP_SYSTEM_HW_STACK_GUARD=y` from the beginning. Also various stack sizes in the config are slightly increased compared to their default values. So I think if there had been any stack overrun there would have been some kind of evidence pointing straight to that. The memory corruption looks very narrowly "targeted", as if someone somewhere was doing the right thing (setting the value to 0) with the wrong pointer (inadvertently pointing to the internal data of the I2C driver). This would probably be better diagnosed with hardware data breakpoints (a.k.a. watchpoints). According to the docs, ESP32C3 does support those, but I haven't tried to use them. At this time, I don't know when I will get around to debugging this issue further, but it certainly hasn't "magically" disappeared.

pawel6843, 2024-10-09:
OMG, sorry, at least it made your arm twisted (I might not understand what does it mean, but well at least you talk more :P)

pawel6843, 2024-10-10:
I was wrong that the third party lib was the culprit. It was only making the problem visible, but the pointers themselves were messed up by something else.
Do you guys think that it could be caused by a problem described in this topic: https://www.esp32.com/viewtopic.php?t=2653. There are others similar reports. Seems that at least 7 years ago ESP was not supporting shared poitners. In my project people indeed use them.
In my previous ESP project there were no such pointeres, and no such problems.
I have tried CONFIG_FREERTOS_UNICORE, but it did not help. But maybe rewriting all of the shared pointers would help....

eriksl, 2024-10-10:
I think the C++ concept of shared pointers in itself shouldn't be a problem, but IIRC they use a concept "pointer aliasing" which compilers hate. It doesn't even have to do with multi-threading. It's where you have two (or more) pointers to the same object and randomly use one or another to access the object. This makes some types of optimisation hard to do for the compiler. At least in C, but probably in C++ too, the compiler assumes all pointer can be aliased and disables this optimisation. You can (using some -f option) have the compiler assume that no pointer is aliased and force the optimisation. This is done on a regular basis, I also do it myself. But if you're not careful, it can raise interesting, hard to find problems. So that's something you might want to try, disable, wherever possible, this option and see if it helps.

Just thinking aloud, you never know...

pawel6843, 2024-10-10:
Well I did not mean the C++ concept of shared pointers itself.
But in that thread the ESP team member admited that they had a problem with them in the IDF. He said at the end, that he was going to try "try to find someone to work on this", but I dunno if he did find such person.
It was something about not using atomics, not having implemented pthread wrapper

eriksl, 2024-10-10:
Ok, in the department "very hard to track down issues" :-(

pawel6843, 2024-10-10:
Well ok, they sey it was done in https://github.com/espressif/esp-idf/issues/690
But I am still curious if maybe sfzhi uses them also :)

pawel6843, 2024-10-11:
They should not be the same mutex. That's the thing. I do not yet think that they are the same mutex. But why would main crash through calling abort in the innter workings of printing due to some mutex being recursive in ISR context? Just it was strange that in the meantime the other task was using a mutex created in this project. But it was created using xSemaphoreCreateMutex, not xSemaphoreCreateRecursiveMutex. We are using recursive mutexes also though. Pointers to that other recursive mutexes is in fact what crashes my app most of the time. They are assigned zeros.
Although I still sometimes have also a carsh of the type sfzhi posted in this thread (in which i2c_ll_read_rxfifo has a null pointer).
I have to make sure we are not using recursive mutexes in ISR code at least. Even though this debug does not apparently show them, but some other mutexes, rather two. This event made me think.

pawel6843, 2024-10-11:
OK I finally have found out what was messing up my project all around by memory corruption.
So that shared pointer usage lead turned out to point in a meaningful direction.

There was a shared pointer pointing to a display layout. It was a class containing pointers to multiple screen objects. Each of them had also a shared pointer pointing to its mutex.
So first the main screen was assigned to the primary pointer. Then another screen was assigned to it, and when I went back to the main screen, then it was being reasigned again with a pointer to a newly created main screen. But right then always the new main screen had one of its objects that had a mutex pointer equal to 0 (while another one was having a mutex pointer of 0xffffffff).
Well getting rid of that single top shared pointer, and using a regular pointer fixes the crash. :)
I dunno if this has something to do with a problem to allocate again a certain size of memory chunk or what

sfzhi, 2024-10-11:
> But I am still curious if maybe sfzhi uses them also :)

I am not using any C++ at all (except for what might be inside ESP-IDF, which is beyond my control).

pawel6843, 2024-10-12:
Aaa, goot to know
I have not yet seen any C++ in the IDF

mythbuster5, 2025-04-11:
For original issue about `isr_receive_xxx` perhaps can be solved by https://github.com/espressif/esp-idf/commit/f8ecc4885d720c3a36b25097774691ca6d0e46b3 according to a race condition. Maybe you can check this on latest master.