espressif/esp-idf #13688: memcpy failure

Jurand1, 2024-04-24:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

Hi, I am using my own component to send and receive data using espnow. I can not find problem what makes memcpy failure. The same component is used at two projects-sending receiving data. One is working without any problems and but in a second app the same component line with 'memcpy' crashing. 

Guru Meditation Error: Core  0 panic'ed (StoreProhibited). Exception was unhandled.

Core  0 register dump:
PC      : 0x4008e396  PS      : 0x00050033  A0      : 0x4008e253  A1      : 0x3ffca620  
0x4008e396: vPortYieldFromInt at C:/Users/Janek/esp/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/portasm.S:647
0x4008e253: _frxt_int_exit at C:/Users/Janek/esp/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/portasm.S:245

A2      : 0xffffffa0  A3      : 0x00000000  A4      : 0x800859a5  A5      : 0x4008e1d6  
0x4008e1d6: _frxt_int_enter at C:/Users/Janek/esp/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/portasm.S:139

A6      : 0x3ffb4118  A7      : 0x00000002  A8      : 0x80085950  A9      : 0x3ffb1ce0  
A10     : 0x3ffb4108  A11     : 0x00000001  A12     : 0x00000000  A13     : 0x00060023  
A14     : 0xb33fffff  A15     : 0xb33fffff  SAR     : 0x00000020  EXCCAUSE: 0x0000001d  
EXCVADDR: 0xffffffa0  LBEG    : 0x4000c2e0  LEND    : 0x4000c2f6  LCOUNT  : 0xffffffff
0x4000c2e0: memcpy in ROM
0x4000c2f6: memcpy in ROM

Backtrace: 0x4008e393:0x3ffca620 0x4008e250:0x3ffca630 0x0005001e:0x00000000 |<-CORRUPTED
0x4008e393: vPortYieldFromInt at C:/Users/Janek/esp/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/portasm.S:646
0x4008e250: _frxt_int_exit at C:/Users/Janek/esp/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/portasm.S:244

Thanks for any help.

hyuan-kamuda, 2024-04-28:
I have similar issue:
Guru Meditation Error: Core  0 panic'ed (StoreProhibited). Exception was unhandled.

Core  0 register dump:
PC      : 0x400d4948  PS      : 0x00060333  A0      : 0x800d4a30  A1      : 0x3ffb9710  
0x400d4948: timer_process_alarm at C:/Users/hyuan/esp/esp-idf/components/esp_timer/src/esp_timer.c:422 (discriminator 3)

A2      : 0x00000000  A3      : 0x00000000  A4      : 0x00152caf  A5      : 0x00000000
A6      : 0x00000000  A7      : 0x3ffcc9d8  A8      : 0x00000000  A9      : 0x00000000
A10     : 0x00000000  A11     : 0x00000000  A12     : 0x00152caf  A13     : 0x00060523
A14     : 0x00000081  A15     : 0x0000cdcd  SAR     : 0x0000001d  EXCCAUSE: 0x0000001d  
EXCVADDR: 0x00000000  LBEG    : 0x4000c2e0  LEND    : 0x4000c2f6  LCOUNT  : 0xffffffff
0x4000c2e0: memcpy in ROM
0x4000c2f6: memcpy in ROM

Backtrace: 0x400d4945:0x3ffb9710 0x400d4a2d:0x3ffb9740 0x40089b9d:0x3ffb9760
0x400d4945: timer_process_alarm at C:/Users/hyuan/esp/esp-idf/components/esp_timer/src/esp_timer.c:422 (discriminator 3)
0x400d4a2d: timer_task at C:/Users/hyuan/esp/esp-idf/components/esp_timer/src/esp_timer.c:479 (discriminator 1)
0x40089b9d: vPortTaskWrapper at C:/Users/hyuan/esp/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/port.c:134

ELF file SHA256: 9568f6bb9

CPU halted.

Jurand1, 2024-05-05:
Hello, I found issue it is not memcpy but mbed rsa decrypting is making crashing.

closed 2024-05-05 by Jurand1