espressif/esp-idf #13689: esp_timer软件定时器

heyangxinmei, 2024-04-25:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

软件定时器的回调函数，有时候会停止运行一段时间。之后有正常回调。请问什么原因会引起这个问题呢？谢谢
[image]

o-marshmallow, 2024-04-25:
Hello @heyangxinmei ,

In order to help you, I have a few questions:
* Which target (ESP32, S2, S3.. ?) are you running this on? 
* Which version of IDF are you using?
* How did you notice that it stopped sometimes?
* What about the output on the UART?
* Do you have a small reproducible example?

heyangxinmei, 2024-04-26:
esp32s3

heyangxinmei, 2024-04-26:
1. esp32s3  2. idf v4.4.5  3. 通过日志观察到的计数值不增加了。4. 通过串口输出日志

heyangxinmei, 2024-04-26:
1. esp32s3 2. idf v4.4.5 3. 通过日志观察到的计数值不增加了。4. 通过串口输出日志  是有比软件定时器优先级的任务运行引起的吗？

ESP-Marius, 2024-04-30:
Yes, the callback is called from a task with `ESP_TASK_TIMER_PRIO`, so there could be tasks with a higher priority (e.g. BT)

> 会停止运行一段时间

How long is 一段时间? Is the callback delayed a few ms, or is it multiple seconds? Are you using power-management/sleep in your application?

It is also possible to dispatch the callback directly from the ISR: https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/system/esp_timer.html#using-the-interrupt-dispatch-method

ESP-Marius, 2024-05-31:
Closing this issue due to lack of activity. Feel free to re-open if there is any new information

closed 2024-05-31 by ESP-Marius