espressif/esp-idf #13692: can't flash esp32-devkitS-V1.1

loraadh, 2024-04-25:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

esp-idf 5.1 powershell

### Operating System used.

Windows

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

PowerShell

### What is the expected behavior?

i expect to flash esp32-devkitS-V1.1 (with esp32-c6-wroom-1U) without any issues. Knowing that, with the same steps, I am succesfully flashing esp32-c6-devkitC-1 V1.2.

### What is the actual behavior?

I use esp-idf 5.1 powershell to build and select C6 target. this is done succesfully. Then I get the error message below when I try to flash.

[image]

### Steps to reproduce.

I follow the same steps as on ESP website: https://docs.espressif.com/projects/esp-idf/en/stable/esp32/get-started/windows-setup.html#get-started-windows-first-steps

the only difference is that I choose "iperf" instead of "Hello_world" example.
here are the commands I use:
cd wifi\iperf
idf.py set-target esp32c6       (this is done successfully)
idf.py menuconfig	     (I enter "ESC" to keep it as by default)		
idf.py build         (this is done successfully)
idf.py -p COM54 flash
then i get the error message below.

[image]

dobairoland, 2024-04-25:
Please provide the output in text format. I can see nothing on that screenshot.

radimkarnis, 2024-04-25:
Hello @loraadh,
Your issue is that the ESP32 is not entering the download mode automatically. You can forcing it manually by holding the boot button when flashing (keep holding the button and run the process again).
You can read about the error and related solutions in the [troubleshooting guide](https://docs.espressif.com/projects/esptool/en/latest/esp32/troubleshooting.html). Let us know if that helped!

radimkarnis, 2024-04-25:
Also, please see the documentation of the [esp32-devkitS-V1.1](https://docs.espressif.com/projects/esp-idf/en/stable/esp32/hw-reference/esp32/user-guide-devkits-r-v1.1.html). It is not fully compatible with the `esp32-c6-wroom-1U` module. With that in mind, you might expect some things to not work - e.g. the auto-reset circuitry. 

If the boot button trick doesn't work, you might need to pull down the `IO9` pin of the C6 module LOW manually (connecting it to `GND`).

loraadh, 2024-04-25:
thanks @radimkarnis 
the boot bottom trick is not working.
I can't find the I09 pin, i've looked in the schamatics, would you please help me more on that?
i've looked to the troubleshooting guide and the documentation but got no solution so far.
i'am trying the alternative method by using the "flash_download_tool_3.9.6" but I'am still looking for the bin file that correspespond to the iperf application.  
BR

here is the error that i am having:

PS C:\Users\nlv18071\wifi\iperf> idf.py -p COM54 flash
Executing action: flash
Running ninja in directory C:\Users\nlv18071\wifi\iperf\build
Executing "ninja flash"...
[1/5] cmd.exe /C "cd /D C:\Users\nlv18071\wifi\iperf\build\esp-idf\esptool_py && C:\Espressif\python_env\idf5.1_py3.11_env\Scripts\python.exe C:/Espressif/frameworks/esp-idf-v5.1.2/components/partition_table/check_sizes.py --offset 0x8000 partition --type app C:/Users/nlv18071/wifi/iperf/build/partition_table/partition-table.bin C:/Users/nlv18071/wifi/iperf/build/iperf.bin"
iperf.bin binary size 0xde460 bytes. Smallest app partition is 0x100000 bytes. 0x21ba0 bytes (13%) free.
[2/5] Performing build step for 'bootloader'
[1/1] cmd.exe /C "cd /D C:\Users\nlv18071\wifi\iperf\build\bootloader\esp-idf\esptool_py && C:\Espressif\python_env\idf5.1_py3.11_env\Scripts\python.exe C:/Espressif/frameworks/esp-idf-v5.1.2/components/partition_table/check_sizes.py --offset 0x8000 bootloader 0x0 C:/Users/nlv18071/wifi/iperf/build/bootloader/bootloader.bin"
Bootloader binary size 0x5770 bytes. 0x2890 bytes (32%) free.
[2/3] cmd.exe /C "cd /D C:\Espressif\frameworks\esp-idf-v5.1.2\components\esptool_py && C:\Espressif\tools\cmake\3.24.0\bin\cmake.exe -D IDF_PATH=C:/Espressif/frameworks/esp-idf-v5.1.2 -D SERIAL_TOOL=C:/Espressif/python_env/idf5.1_py3.11_env/Scripts/python.exe;;C:/Espressif/frameworks/esp-idf-v5.1.2/components/esptool_py/esptool/esptool.py;--chip;esp32c6 -D SERIAL_TOOL_ARGS=--before=default_reset;--after=hard_reset;write_flash;@flash_args -D WORKING_DIRECTORY=C:/Users/nlv18071/wifi/iperf/build -P C:/Espressif/frameworks/esp-idf-v5.1.2/components/esptool_py/run_serial_tool.cmake"
esptool.py --chip esp32c6 -p COM54 -b 460800 --before=default_reset --after=hard_reset write_flash --flash_mode dio --flash_freq 80m --flash_size 2MB 0x0 bootloader/bootloader.bin 0x10000 iperf.bin 0x8000 partition_table/partition-table.bin
esptool.py v4.7.0
Serial port COM54
Connecting......................................

A fatal error occurred: Failed to connect to ESP32-C6: Wrong boot mode detected (0x0)! The chip needs to be in download mode.
For troubleshooting steps visit: https://docs.espressif.com/projects/esptool/en/latest/troubleshooting.html
CMake Error at run_serial_tool.cmake:66 (message):

  C:/Espressif/python_env/idf5.1_py3.11_env/Scripts/python.exe;;C:/Espressif/frameworks/esp-idf-v5.1.2/components/esptool_py/esptool/esptool.py;--chip;esp32c6
  failed.

FAILED: CMakeFiles/flash C:/Users/nlv18071/wifi/iperf/build/CMakeFiles/flash
cmd.exe /C "cd /D C:\Espressif\frameworks\esp-idf-v5.1.2\components\esptool_py && C:\Espressif\tools\cmake\3.24.0\bin\cmake.exe -D IDF_PATH=C:/Espressif/frameworks/esp-idf-v5.1.2 -D SERIAL_TOOL=C:/Espressif/python_env/idf5.1_py3.11_env/Scripts/python.exe;;C:/Espressif/frameworks/esp-idf-v5.1.2/components/esptool_py/esptool/esptool.py;--chip;esp32c6 -D SERIAL_TOOL_ARGS=--before=default_reset;--after=hard_reset;write_flash;@flash_args -D WORKING_DIRECTORY=C:/Users/nlv18071/wifi/iperf/build -P C:/Espressif/frameworks/esp-idf-v5.1.2/components/esptool_py/run_serial_tool.cmake"
ninja: build stopped: subcommand failed.
ninja failed with exit code 1, output of the command is in the C:\Users\nlv18071\wifi\iperf\build\log\idf_py_stderr_output_36208 and C:\Users\nlv18071\wifi\iperf\build\log\idf_py_stdout_output_36208
PS C:\Users\nlv18071\wifi\iperf>

loraadh, 2024-04-25:
here is the error message that I have
[image]

radimkarnis, 2024-04-25:
Your only issue is this part of the output:

```
Connecting......................................

A fatal error occurred: Failed to connect to ESP32-C6: Wrong boot mode detected (0x0)! The chip needs to be in download mode.
```

The hardware needs to be switched to the download mode. Without it, no flashing programs (esptool.py, flash_download_tool, to others) can do anything.

> but I'am still looking for the bin file that correspespond to the iperf application.

Forget about the application for a moment. You have a HW issue. And that is most likely caused by using an incompatible combination of a `esp32-devkitS-V1.1` with a ESP32-C6 module.

I just gave you information how to bypass this. Just pull `IO9` (bottom right pin on the module) LOW when connecting the USB cable and then keep it LOW throughout the whole process of flashing.

[image]

dobairoland, 2024-04-25:
Since the boot pin is at the same place as for ESP32, it could help just pushing the BOOT button while starting `idf.py -p COM54 flash`.

loraadh, 2024-04-26:
thanks @radimkarnis and @dobairoland 
I've checked the functionality of the boot switch and forced the boot pin to GND (IO0 in the case of esp32-c6-wroom, pin diagram attached). I still have the same flashing issue, and i confirm that the boot pin IO0 is pulled down to gnd during the whole flashing process. BR

dobairoland, 2024-04-26:
> IO0 in the case of esp32-c6-wroom

That is the boot pin for ESP32. IO9 is for ESP32-C6.

radimkarnis, 2024-04-26:
I don't know what datasheet you are looking at, but this is not esp32-c6-wroom. 
See [here](https://www.espressif.com/sites/default/files/documentation/esp32-c6-wroom-1_wroom-1u_datasheet_en.pdf). Check the `3.1 Pin Layout` and `3.3 Strapping Pins` sections.

loraadh, 2024-04-26:
you are right @radimkarnis i took the pin diagram from the wrong document. by the way, attached is the board that I am trying to flash.
[image]

radimkarnis, 2024-04-26:
You can try using any serial terminal program to see the output from the ESP (e.g. `python -m serial.tools.miniterm <PORT> 115200`). If you then hold the BOOT button and press the EN button, you should see a `waiting for download` output. If that doesn't happen, you won't be able to flash this module with this carrier board (without having to manually tinker with the strapping pins).

loraadh, 2024-04-26:
@radimkarnis 
here is what happens on the terminal when I hold boot and press EN.
.
.
.
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0xffffffff
invalid header: 0üESP-ROM:esp32c6-20220919
Build:Sep 19 2022
rst:0x1 (POWERON),boot:0x0 (USB_BOOT)
wait usb download

radimkarnis, 2024-04-26:
Ok so that's the issue. The buttons work, but other strapping pins are in a wrong state. I presume it's an issue with GPIO8. You can try pulling it HIGH or LOW and repeating the process until you see `waiting for download` (not `wait usb download`).

Please read about the strapping pins and their valid/invalid states in the linked datasheet.

loraadh, 2024-04-26:
@radimkarnis 
thanks a lot for your great. it worked after pulling up IO8.
BR

closed 2024-04-26 by dobairoland