espressif/esp-idf #13696: When I use ble_spp_server for nimble example,I can't send data more than four pieces of data

HanRuoX, 2024-04-26:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

for (int i = 0; i <= BT_NIMBLE_MAX_CONNECTIONS; i++) {
							/* Check if client has subscribed to notifications */
							if (conn_handle_subs[i]) {
								struct os_mbuf *txom;
								txom = ble_hs_mbuf_from_flat(pTempBuf, sizeof(pTempBuf));
								int rc = ble_gatts_notify_custom(i, ble_spp_svc_gatt_read_val_handle,
															txom);
															ble_gatts_notify_multiple_custom
								printf("%s",txom->om_data);
								if (rc == 0) {
									ESP_LOGI("NIMBLE", "Notification sent successfully");
								} else {
									ESP_LOGI("NIMBLE", "Error in sending notification rc = %d", rc);
								}
							}
						}
					}
			
		for example I send data by uart to ble device “123456”，my ble service only can receive "1234"

rahult-github, 2024-04-26:
Hi @HanRuoX , 

 I tried the ble_spp_server example , with the added change of printing the contents of data received.  Since it is a simple ascii value recieved and i do see the values coming as i typed on client side. 
 
 
 I (3183) NimBLE: GAP procedure initiated: advertise; 
I (3183) NimBLE: disc_mode=2
I (3193) NimBLE:  adv_channel_map=0 own_addr_type=0 adv_filter_policy=0 adv_itvl_min=0 adv_itvl_max=0
I (3203) NimBLE: 

I (6213) NimBLE: Data received in write event,conn_handle = 1,attr_handle = 10
Data received --> 31              
I (9763) NimBLE: Data received in write event,conn_handle = 1,attr_handle = 10
Data received --> 32 
I (26213) NimBLE: Data received in write event,conn_handle = 1,attr_handle = 10
Data received --> 31 32 33 34 35 36 
I (38963) NimBLE: Data received in write event,conn_handle = 1,attr_handle = 10
Data received --> 31 32 33 34 35 36 37 38 39 
I (58913) NimBLE: Data received in write event,conn_handle = 1,attr_handle = 10
Data received --> 22 31 31 22 

31 --> 1
32 --> 2
31 32 33 34 35 36 --> 123456
31 32 33 34 35 36 37 38 39   --> 123456789 
22 31 31 22 -->  "11" 

How are you entering the data ?  Since it reads UART if there is pause or different keypresses, then it will not take it. Or should take it as next set of data

HanRuoX, 2024-04-26:
when I send hex data 31 32 33 34 35 36 37 38 39, I only receive 1234

HanRuoX, 2024-04-26:
I find example code "static uint16_t ble_spp_svc_gatt_read_val_handle;" only declared that there is no definition

rahult-github, 2024-04-26:
> when I send hex data 31 32 33 34 35 36 37 38 39, I only receive 1234

Can you please share your application changes done , if any, on server and client side ?  

> I find example code "static uint16_t ble_spp_svc_gatt_read_val_handle;" only declared that there is no definition

You can check the SPP service details for this. Client will write on that handle, which server then parses. 

[image]

HanRuoX, 2024-04-26:
oh no! I find! 
txom = ble_hs_mbuf_from_flat(ntf, event.size);
                            rc = ble_gatts_notify_custom(i, ble_spp_svc_gatt_read_val_handle,
                                                         txom);
 question is event.size, old version v5.1.2 use size of ntf, I found it fixed in the latest version
I'm sorry that I don't compare the latest version first, Thanks you very much!

closed 2024-04-26 by HanRuoX