espressif/esp-idf #13699: ESP32 watchdog  interrupt

zzff-sys, 2024-04-26:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

I want to use TWDT, enter the interrupt function when a watchdog timeout occurs, and execute some of the logic I want in the interrupt function (without using the print function). The interrupt function is declared using esp_task_wdt_isr_user_handler (). However, it seems that the timeout behavior does not enter the function, and repeated restarts occur. How do I correctly define a watchdog interrupt @svenfuchs @paulreimer @kumekay @

zzff-sys, 2024-04-26:
As I said above, I want to execute the contents of my interrupt function on timeout. I think at this point you should stay in the watchdog interrupt and not reset the behavior (is that understood correctly) 。Attached is my error log and program
[log.txt](https://github.com/espressif/esp-idf/files/15132905/log.txt)
[image]
[image]

nopnop2002, 2024-04-26:
I think you should enable this.
[image]

zzff-sys, 2024-04-27:
> 我认为启用此功能可以解决问题。
> 
> ![测试](https://private-user-images.githubusercontent.com/6020549/326135308-26c95560-567a-43ee-9f50-7af0fab22fcb.jpg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTQxNzY2MzYsIm5iZiI6MTcxNDE3NjMzNiwicGF0aCI6Ii82MDIwNTQ5LzMyNjEzNTMwOC0yNmM5NTU2MC01NjdhLTQzZWUtOWY1MC03YWYwZmFiMjJmY2IuanBnP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI0MDQyNyUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNDA0MjdUMDAwNTM2WiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9OTgwMDQ0ZDc0OTQ1YjQxMjg5MTU0MTY0MzgzOWQ0MDYwOTcwNmRiMTg1YjJiNTNlYTE0MzdiMjJlZmUwMDhmNCZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QmYWN0b3JfaWQ9MCZrZXlfaWQ9MCZyZXBvX2lkPTAifQ.ANA11CDCTCWgtQPRy1JqeTghmbxHw8BwoPbNuPhWAVg)

I've actually enabled this feature. But I wanted it to execute my interrupt program on timeout, and it didn't @nopnop2002

nopnop2002, 2024-04-27:
I confirmed that esp_task_wdt_isr_user_handler() is called in the code below.

```
void esp_task_wdt_isr_user_handler(void)
{
    esp_rom_printf("called esp_task_wdt_isr_user_handler\n");
}
```
[image]

zzff-sys, 2024-04-27:
@nopnop2002 I see that your results did indeed execute the interrupt content, but my results indicate that the interrupt content is still not executed. Can you help me check if there are any design issues with the code snippet above. My phenomenon is still constantly reporting the information shown in the log。

zzff-sys, 2024-04-27:
@nopnop2002 Perhaps what you mean is that when a timeout occurs, it will enter an interrupt, and at the same time, a reset behavior will still occur, rather than staying in an interrupt. Am I correct in understanding this way

zzff-sys, 2024-04-27:
My current phenomenon is that I am able to enter the interrupt to execute the corresponding program, but still experience restart behavior. It seems that no matter what program is in my interrupt program, it will restart when the time is up. Can't we avoid this problem by entering the interrupt execution program instead of constantly restarting.
If the interrupt continues to reset continuously, I think the interrupt will be meaningless

ESP-Marius, 2024-04-28:
> My current phenomenon is that I am able to enter the interrupt to execute the corresponding program, but still experience restart behavior.

Yes this is correct, the callback registered by `esp_task_wdt_isr_user_handler` is just for adding extra functionality, after that function is called the watchdog ISR will continue its execution and do what it is intended to do, restart the chip.

If you disable `CONFIG_ESP_TASK_WDT_PANIC` then you might get a behavior to closer to what you want, then after executing your callback function it will simply print a backtrace and exit the interrupt.

Maybe if you explain in a bit more details what you are trying to achieve and why then we might be able to give some suggestion on how to do this.

zzff-sys, 2024-04-28:
> Yes this is correct, the callback registered by `esp_task_wdt_isr_user_handler` is just for adding extra functionality, after that function is called the watchdog ISR will continue its execution and [...]

1. Firstly, thank you very much for your reply. I'm trying to understand restart behavior.
2. Now that my goal is to perform a loop action in a watchdog interrupt, I use the following form. The final performance is still able to restart, but the restart time is not the timeout time we set 5S, but about 20S. 
`void esp_task_wdt_isr_user_handler(void)
{
       while(1)
     {
        /*I want to do something*/
     }
}`
3. At the same time, I found that reducing the startup timeout time (as shown below) could reduce the 20S time I observed to a certain extent. I wonder if there's a connection. At the same time, I hope you can tell me whether there is dog feeding behavior in the above start-up stage, and where can I  
[image]

ESP-Marius, 2024-04-28:
> Now that my goal is to perform a loop action in a watchdog interrupt

I understand, but I would like to know why you want to do this, and what you want to do in this loop action. Since it is not a super-common use-case I suspect you might be trying to do something that would be better achieved in another way :smile: 

>  The final performance is still able to restart

If you stay in the WDT ISR for too long it will eventually trigger the second stage timeout, which will force a system reset. As shown in the picture above that timeout time is 2x the normal timeout time, so 10s by default

zzff-sys, 2024-04-28:
> > 现在我的目标是在看门狗中断中执行循环操作
> 
> 我明白，但我想知道你为什么要这样做，以及你想在这个循环操作中做什么。由于这不是一个超级常见的用例，我怀疑您可能正在尝试做一些可以通过其他方式更好地实现的事情😄
> 
> > 最终演出仍能重启
> 
> 如果在 WDT ISR 中停留时间过长，最终将触发第二阶段超时，这将强制系统复位。如上图所示，超时时间是正常超时时间的2倍，所以默认10s
1、My goal is: When the timeout behavior occurs, I want to send the error message through the can message, by performing the send behavior in the watchdog interrupt, which is a real-time loop process. The form of the code is shown below
`void esp_task_wdt_isr_user_handler(void)
 {
     while(1) {
            /*send message*/
        send_message(CAN_ERR_ID, Data, CAN_MSG_DATA_SIZE_BYTES);
      } 
}`
2、So because I was in the interrupt for a long time causing phase two to run out of time?

ESP-Marius, 2024-04-28:
1. I see. In this case you will have to be careful with what is happening in `send_message`, it should not have any blocking behavior, etc.
2. Yes, the second stage timeout is there to ensure that even if something goes wrong in the WDT interrupt, the system will eventually be able to reset itself.

Regarding your use-case: do you want the system to reset itself after sending the message (1), or are you just using it to log an error (2) ? If 2. then you could consider just setting a `WDT triggered`-flag, e.g. giving a semaphore from the callback (`xSemaphoreGiveFromISR`) and then handle the logging in a high priority task instead.

zzff-sys, 2024-04-29:
> 1. 我懂了。在这种情况下，您必须小心 中发生的事情`send_message`，它不应该有任何阻塞行为等。
> 2. 是的，第二阶段超时是为了确保即使WDT中断出现问题，系统最终也能够自行复位。
> 
> 关于您的用例：您是否希望系统在发送消息后自行重置（1），或者您只是使用它来记录错误（2）？如果 2. 那么您可以考虑仅设置一个`WDT triggered`-flag，例如从回调 ( `xSemaphoreGiveFromISR`) 中给出信号量，然后在高优先级任务中处理日志记录。

1.Thanks again for your reply
2. My main idea was to send multiple message packets in real time when entering a watchdog interrupt, so I chose to use loops in the interrupt. However, only one packet can be sent, which seems to be stuck in the loop. I want to say whether it is related to the implementation and application mechanism of interrupts.
3. I tried the method of using semaphore for the second point you proposed, but the same phenomenon occurred.
4, a little doubt, when I enter the watchdog interrupt, I wait to receive the semaphore task is still useful? Doesn't seem to fit the freertos principle

ESP-Marius, 2024-04-29:
2. This is possible, but it depends on the implemention of send_message`, it cannot do anything you cannot do in interrupts. E.g. it should not rely on freertos tasks, take locks, block, need interrupts that cannot preempt the current interrupt etc. So this should mostly only be done with very simple bare-metal functions.
4. No, in the WDT interrupt you should not wait to receive any freertos communication like that. But  `xSemaphoreGiveFromISR` should be OK I think (it is intended to be used from interrupts), as long as freerots continues running afterwards

zzff-sys, 2024-04-29:
> 2. 这是可能的，但这取决于 的实现`send_message`，它不应该做任何你在中断中不能做的事情。例如，它不应该依赖于 freertos 任务、获取锁、阻塞、需要无法抢占当前中断的中断等。因此，这通常只能通过非常简单的裸机函数来完成。
> 3. 不，在 WDT 中断中，您不应该等待接收任何像这样的 freertos 通信。但  `xSemaphoreGiveFromISR`我认为应该没问题（它旨在从中断中使用），只要 freerots 之后继续运行

1. I can understand your first point
2. Second, I think that the semaphore established in the interrupt can be sent through xSemaphoreGiveFromISR. But for the reception of the semaphore, I put it in a receive task, but if there is a watchdog timeout entry interrupt, can my receive task continue to run
3.My pseudo-code is as follows, whether my design strategy is correct
[image]
[image]

ESP-Marius, 2024-04-29:
> but if there is a watchdog timeout entry interrupt, can my receive task continue to run

If you use `CONFIG_ESP_TASK_WDT_PANIC` then no, the chip will immediately restart. If you do not use it then after the WDT interrupt freertos will continue running tasks. And if this error log task has a higher priority than whatever task is starving the watchdog then it should get an opportunity to run.

Your example looks correct to me.

nopnop2002, 2024-04-29:
@ESP-Marius 

> If you do not use it then after the WDT interrupt freertos will continue running tasks.

you are right,

zzff-sys, 2024-04-29:
@ESP-Marius 
Why do I feel like there is always only one piece of logic executed in` void esp_task_wdt_isr_user_handler(void)`, and is there a task in the watchdog design mechanism that interrupts it

ESP-Marius, 2024-04-29:
> Why do I feel like there is always only one piece of logic executed in void esp_task_wdt_isr_user_handler(void), and is there a task in the watchdog design mechanism that interrupts it

Not sure I understand what you mean here. I dont think anything should interrupt it (before the second stage timeout triggers, if you are taking too long)

ESP-Marius, 2024-08-08:
I think we can consider the initial question/issue resolved/responded to. 

Feel free to reopen if there is anything that we still need to resolve on the IDF side of things

closed 2024-08-08 by ESP-Marius