espressif/esp-idf #13700: IP101 Using External 50MHz CLK: emac_esp32_init(369): reset timeout

JonRobo, 2024-04-26:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

5.0.4

### Espressif SoC revision.

ESP32-WROVER-IE

### Operating System used.

Windows

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

CMD

### Development Kit.

Custom Board

### Power Supply used.

USB

### What is the expected behavior?

IP101 should have initiated correctly and I should have an operational Ethernet interface.

### What is the actual behavior?

I receive these errors
E (1751) esp.emac: emac_esp32_init(369): reset timeout
E (1751) esp_eth: esp_eth_driver_install(228): init mac failed

I have my schematics attached, this is my first time trying to get the ESP32 and IP101 working on a custom PCB. I'm not sure if everything is correct so if there are any errors please let me know.

I have an external 25MHz clock as per the datasheet, and I have the 50M_CLKO signal connected to the 50M_CLKI pin. GPIO0 is also connected to 50M_CLKO via a 510 ohm resistor. I attached a picture of what that signal looks like on the oscilloscope. It has the correct frequency but I'm not sure if the VPP is enough.

[image]
[image]
[image]

### Steps to reproduce.

I have the following function to initialize Ethernet, trying to ensure the correct configurations are made

uint8_t ethernet_init(void)
{

    // Create new default instance of esp-netif for Ethernet
    esp_netif_config_t cfg = ESP_NETIF_DEFAULT_ETH();
    esp_netif_t *eth_netif = esp_netif_new(&cfg);

    // Init MAC and PHY configs to default
    eth_mac_config_t mac_config = ETH_MAC_DEFAULT_CONFIG();
    eth_phy_config_t phy_config = ETH_PHY_DEFAULT_CONFIG();

    mac_config.sw_reset_timeout_ms = 500;

    phy_config.phy_addr = PHY_ADDRESS;
    phy_config.reset_gpio_num = ETH_RESET_GPIO_NUMBER;
    phy_config.reset_timeout_ms = 500;

    eth_esp32_emac_config_t esp32_emac_config = ETH_ESP32_EMAC_DEFAULT_CONFIG();
    esp32_emac_config.smi_mdc_gpio_num = ETH_MDC_GPIO_NUMBER;
    esp32_emac_config.smi_mdio_gpio_num = ETH_MDIO_GPIO_NUMBER;
    esp32_emac_config.interface = EMAC_DATA_INTERFACE_RMII;
    esp32_emac_config.clock_config.rmii.clock_mode = EMAC_CLK_EXT_IN;
    esp32_emac_config.clock_config.rmii.clock_gpio = EMAC_CLK_IN_GPIO;

    esp_eth_mac_t *mac = esp_eth_mac_new_esp32(&esp32_emac_config, &mac_config);
    //vTaskDelay(500);
    esp_eth_phy_t *phy = esp_eth_phy_new_ip101(&phy_config);

    esp_eth_config_t config = ETH_DEFAULT_CONFIG(mac, phy);
    esp_eth_handle_t eth_handle = NULL;

    ethernet_state = ETHERNET_UNINITIALIZED;

    if(esp_eth_driver_install(&config, &eth_handle) != ESP_OK) return ethernet_state;
    printf("esp_eth_driver_install OK\n");

    /* attach Ethernet driver to TCP/IP stack */
    esp_netif_attach(eth_netif, esp_eth_new_netif_glue(eth_handle));
    printf("ethernet netif attach OK\n");

    // Register user defined event handers
    esp_event_handler_register(ETH_EVENT, ESP_EVENT_ANY_ID, &eth_event_handler, NULL);
    esp_event_handler_register(IP_EVENT, IP_EVENT_ETH_GOT_IP, &got_ip_event_handler, NULL);

    /* start Ethernet driver state machine */
    esp_eth_start(eth_handle);
      
    ethernet_state = ETHERNET_INITIALIZED;
    return ethernet_state;
}

My project configuration has RMII selected and the RMII clock mode is set to  'Input RMII clock from external'
[image]

### More Information.

Any help here will be appreciated. I will continue to dig into any obvious code/board assembly issues in hopes of fixing this.
Thank you

suda-morris, 2024-04-30:
I think the issue is, the Vpp is too small... So the ESP32 EMAC doesn't think it gets a valid RMII clock (50MHz). Then the internal MAC won't work, so you see it even can't reset the EMAC successfully because lack of a correct clock.

I would recommend you to refer to https://docs.espressif.com/projects/esp-idf/en/latest/esp32/hw-reference/esp32/get-started-ethernet-kit.html#esp32-ethernet-kit-v1-2-getting-started-guide it has a hardware reference design there.

JonRobo, 2024-04-30:
Yes, I based this board off of that design.  
Today I found a solution to the clock problem. The DTR pin was influencing GPIO0 too much, I replaced its resistor with a diode from GPIO0 to DTR and also brought the pull up resistance to 47K from GPIO0 to 3V3. This got me connected to the IP101 from the ESP32 using the 50MHz signal from the IP101, and continued to allow me to automatically put the board into flashing mode.

However I'm now facing an issue with link stability. I did not see this kind of behavior on my test board, the link goes up then down several times and sometimes if I'm lucky I get assigned an IP address.

[image]

Here is my event code

static void eth_event_handler(void *arg, esp_event_base_t event_base, int32_t event_id, void *event_data)
{
    uint8_t mac_addr[6] = {0};
    /* we can get the ethernet driver handle from event data */
    esp_eth_handle_t eth_handle = *(esp_eth_handle_t *)event_data;

    switch (event_id) 
    {
        case ETHERNET_EVENT_CONNECTED:
            esp_eth_ioctl(eth_handle, ETH_CMD_G_MAC_ADDR, mac_addr);
            printf("Ethernet Link Up\n");
            set_wifi_led(WIFI_LED_ON_DUTY, 1000);
            printf("Ethernet HW Addr %02x:%02x:%02x:%02x:%02x:%02x\n", mac_addr[0], mac_addr[1], mac_addr[2], mac_addr[3], mac_addr[4], mac_addr[5]);

            ethernet_state = ETHERNET_LINK_UP;
            break;
        case ETHERNET_EVENT_DISCONNECTED:
            printf("Ethernet Link Down\n");
            set_wifi_led(0, 1000);
            ethernet_state = ETHERNET_LINK_DOWN;
            break;
        case ETHERNET_EVENT_START:
            printf("Ethernet Started\n");
            break;
        case ETHERNET_EVENT_STOP:
            printf("Ethernet Stopped\n");
            break;
        default:
            break;
    }
}

static void got_ip_event_handler(void *arg, esp_event_base_t event_base,
                                 int32_t event_id, void *event_data)
{
    ip_event_got_ip_t *event = (ip_event_got_ip_t *) event_data;
    const esp_netif_ip_info_t *ip_info = &event->ip_info;

    printf("Ethernet Got IP Address: "IPSTR, IP2STR(&ip_info->ip));
    printf("\nETHMASK: " IPSTR, IP2STR(&ip_info->netmask));
    printf("\nETHGW: " IPSTR, IP2STR(&ip_info->gw));
    printf("\n");
}

I'm not 100% confident in the Ethernet connector schematics to the IP101, is this causing the issue with link up/down?
Thank you for the help so far.

closed 2024-05-02 by kostaond