espressif/esp-idf #13701: xtaskcreate error

cyberumut, 2024-04-26:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

[image]

every time I try to start a task I get an insufficient memory error 
I have enough memory space as shown in the picture  4.7MB PSRAM   263BYTE INTERNAL RAM

Alvin1Zhang, 2024-04-27:
Would you please help share more details as per [the template](https://github.com/espressif/esp-idf/issues/new/choose)? Thanks.

ESP-Marius, 2024-04-28:
Since PSRAM memory comes with some restrictions, task stacks by default allocate from internal memory. But it is possible provide a PSRAM stack manually with `xTaskCreateStatic` if [CONFIG_SPIRAM_ALLOW_STACK_EXTERNAL_MEMORY](https://docs.espressif.com/projects/esp-idf/en/stable/esp32/api-reference/kconfig.html#config-spiram-allow-stack-external-memory) is enabled.

cyberumut, 2024-04-29:
> Since PSRAM memory comes with some restrictions, task stacks by default allocate from internal memory. But it is possible provide a PSRAM stack manually with `xTaskCreateStatic` if [...]

yes already enabled

ESP-Marius, 2024-04-30:
> yes already enabled

As I said:
> Since PSRAM memory comes with some restrictions, task stacks by default allocate from internal memory. But it is possible provide a PSRAM stack manually with xTaskCreateStatic

ESP-Marius, 2024-08-08:
Marking this one as resolved, please reopen it if there is still any outstanding issues.

closed 2024-08-08 by ESP-Marius