espressif/esp-idf #13702: examples/peripherals/lcd/spi_lcd_touch is not working on ESP32-S3-Touch-LCD-1.28

oshmykov-dev, 2024-04-27:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.3-dev-3225-g5a40bb8746

### Espressif SoC revision.

ESP32-S3 (QFN56) (revision v0.2)

### Operating System used.

Windows

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

CMD

### Development Kit.

ESP32-S3-Touch-LCD-1.28

### Power Supply used.

USB

### What is the expected behavior?

"This example will draw a fancy dash board with the LVGL library."

### What is the actual behavior?

No errors except for the single WARNING but there is no "fancy animation". 
The screen is slightly lit. No image or text or anything is displayed.

### Steps to reproduce.

1. cd %userprofile%\esp
2. xcopy /e /i  %userprofile%\git\esp-idf\examples\peripherals\lcd\spi_lcd_touch spi_lcd_touch
3. cd  %userprofile%\esp\spi_lcd_touch
4. idf.py set-target esp32s3
5. idf.py menuconfig
6. Example Configuration ---> 
i80 LCD controller model (ILI9341) --->
() ILI9341
(X) GC9A01

(S) Save
(Q) Quit

7. idf.py build
8. idf.py -p COM3 flash monitor

### Debug Logs.

```plain
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x1 (POWERON),boot:0x18 (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3810,len:0x178c
load:0x403c9700,len:0x4
load:0x403c9704,len:0xcbc
load:0x403cc700,len:0x2da0
entry 0x403c9914
I (27) boot: ESP-IDF v5.2.1 2nd stage bootloader
I (27) boot: compile time Apr 27 2024 00:51:02
I (27) boot: Multicore bootloader
I (30) boot: chip revision: v0.2
I (34) boot.esp32s3: Boot SPI Speed : 80MHz
I (39) boot.esp32s3: SPI Mode       : DIO
I (43) boot.esp32s3: SPI Flash Size : 2MB
I (48) boot: Enabling RNG early entropy source...
I (53) boot: Partition Table:
I (57) boot: ## Label            Usage          Type ST Offset   Length
I (64) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (72) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (79) boot:  2 factory          factory app      00 00 00010000 00100000
I (87) boot: End of partition table
I (91) esp_image: segment 0: paddr=00010020 vaddr=3c050020 size=14644h ( 83524) map
I (114) esp_image: segment 1: paddr=0002466c vaddr=3fc94500 size=02b44h ( 11076) load
I (117) esp_image: segment 2: paddr=000271b8 vaddr=40374000 size=08e60h ( 36448) load
I (128) esp_image: segment 3: paddr=00030020 vaddr=42000020 size=46fc8h (290760) map
I (181) esp_image: segment 4: paddr=00076ff0 vaddr=4037ce60 size=0759ch ( 30108) load
I (194) boot: Loaded app from partition at offset 0x10000
I (194) boot: Disabling RNG early entropy source...
I (206) cpu_start: Multicore app
I (216) cpu_start: Pro cpu start user code
I (216) cpu_start: cpu freq: 160000000 Hz
I (216) cpu_start: Application information:
I (219) cpu_start: Project name:     spi_lcd_touch
I (225) cpu_start: App version:      1
I (229) cpu_start: Compile time:     Apr 27 2024 00:50:37
I (235) cpu_start: ELF file SHA256:  5871bdc14...
I (240) cpu_start: ESP-IDF:          v5.2.1
I (245) cpu_start: Min chip rev:     v0.0
I (250) cpu_start: Max chip rev:     v0.99
I (255) cpu_start: Chip rev:         v0.2
I (260) heap_init: Initializing. RAM available for dynamic allocation:
I (267) heap_init: At 3FC9FE28 len 000498E8 (294 KiB): RAM
I (273) heap_init: At 3FCE9710 len 00005724 (21 KiB): RAM
I (279) heap_init: At 3FCF0000 len 00008000 (32 KiB): DRAM
I (285) heap_init: At 600FE010 len 00001FD8 (7 KiB): RTCRAM
I (292) spi_flash: detected chip: winbond
I (296) spi_flash: flash io: dio
W (300) spi_flash: Detected size(16384k) larger than the size in the binary image header(2048k). Using the size in the binary image header.
I (313) sleep: Configure to isolate all GPIO pins in sleep state
I (320) sleep: Enable automatic switching of GPIO sleep configuration
I (328) main_task: Started on CPU0
I (338) main_task: Calling app_main()
I (338) example: Turn off LCD backlight
I (338) gpio: GPIO[2]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0
I (348) example: Initialize SPI bus
I (358) example: Install panel IO
I (358) gpio: GPIO[5]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0
I (368) example: Install GC9A01 panel driver
I (368) gpio: GPIO[3]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0
I (378) gc9a01: LCD panel create success, version: 1.2.0
I (508) example: Turn on LCD backlight
I (508) example: Initialize LVGL library
I (508) example: Register display driver to LVGL
I (508) example: Install LVGL tick timer
I (518) example: Create LVGL task
I (518) example: Starting LVGL task
I (568) example: Display LVGL Meter Widget
I (568) main_task: Returned from app_main()
```

### More Information.

I tried different levels of EXAMPLE_LCD_BK_LIGHT_ON_LEVEL with no luck. (1, 10, 100, 1000 etc)
[image]

suda-morris, 2024-04-29:
> Example Configuration --->
i80 LCD controller model (ILI9341) --->
() ILI9341
(X) GC9A01

THis confused me. I think that LCD module uses SPI interface not i80. https://www.waveshare.com/wiki/ESP32-S3-Touch-LCD-1.28

suda-morris, 2024-04-29:
Please also check if the GPIO assignment in your code matches the hardware connection. I just connect my GC9A01 LCD break out board with ESP32S3, it works fine.

oshmykov-dev, 2024-04-29:
@suda-morris you're right the problem was with the wrong GPIO assignments. 
```
#define EXAMPLE_PIN_NUM_SCLK           10
#define EXAMPLE_PIN_NUM_MOSI          11
#define EXAMPLE_PIN_NUM_MISO          12
#define EXAMPLE_PIN_NUM_LCD_DC        8
#define EXAMPLE_PIN_NUM_LCD_RST     14
#define EXAMPLE_PIN_NUM_LCD_CS         9
```

From here the [specification](https://files.waveshare.com/wiki/ESP32-S3-Touch-LCD-1.28/ESP32-S3-Touch-LCD-1.28-Sch.pdf)

closed 2024-04-29 by oshmykov-dev

ostifede02, 2024-12-05:
Hello,
I am facing a similar problem but with the board esp32s3-lcd-1.28 (the one without touch). I can build and flash the sample code /esp/esp-idf/examples/peripherals/lcd/spi_lcd_touch but I can only see a blank screen. I have followed the same steps and double checked al GPIO pins. Am I missing something?