espressif/esp-idf #13706: OpenOCD profiling kills BLE

mickeyl, 2024-04-28:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.2

### Espressif SoC revision.

ESP32-S3 (QFN56) (revision v0.1)

### Operating System used.

Linux

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### Development Kit.

Custom Board

### Power Supply used.

USB

### What is the expected behavior?

I tried to profile using `OpenOCD` and hoped the BLE connection would keep intact.

### What is the actual behavior?

The BLE connection drops down as soon as the profiling starts.

### Steps to reproduce.

1. call openocd.
2. start your program
3. telnet into openocd
4. call `profile 10 gmon.out`

zhp0406, 2024-05-06:
What is the error code when BLE disconnects, can you give me the complete log?

mickeyl, 2024-05-07:
There is no error in the log, that's the thing (not even w/ verbose logs). The BLE layer just hangs and the other side gets a connection timeout:

```
May 07 16:32:12.450  L2CAP Receive    0x0053  F4:12:FA:DF:21:FA  Channel ID: 0x004C  Length: 0x0006 (06) [ 04 00 1F 80 00 00 ]  
May 07 16:32:12.450  L2CAP Receive    0x0053  F4:12:FA:DF:21:FA  LE Flow Control Credit  
May 07 16:32:12.599  HCI Event        0x0053  F4:12:FA:DF:21:FA  Number Of Completed Packets - Handle: 0x0053 - Packets: 0x0001    
May 07 16:32:15.451  HCI Event        0x0053  F4:12:FA:DF:21:FA  Disconnection Complete - Connection Timeout  
```

mickeyl, 2024-05-07:
The nature of the OpenOCD profiler is that it stops and starts the target very rapidly. Perhaps that messes up the internal BLE timers? Interestingly, this does not happen if I do the same on WiFi.

zhp0406, 2024-05-08:
Which specific version are you using? Based on your version, I'll provide you with a controller library for testing.

zhp0406, 2024-05-08:
Is the OpenOCD you're using on the ESP32-S3 from Espressif official or third-party? Could you provide some information about the OpenOCD you're using?

zhp0406, 2024-05-08:
What is the connection timeout you've set?
[image]

mickeyl, 2024-05-26:
@zhp0406 I'm using ESP-IDF cc649ea869b8c24d19ba2d94c4d3f1146cb2947b + some patches on top of it, so basically the 5.2 release.

The OpenOCD is the one from Espressif official.

My connection timeout is set to 2000 ms.

esp-zhp, 2024-12-16:
When a breakpoint is set during program execution, the program will pause when it reaches the breakpoint. The peer device will not receive a response from the ESP, and the ESP will also fail to receive a response from the peer device because the BLE controller cannot schedule properly. This results in the connection being disconnected, which is the expected behavior.

The value 0x08 indicates that the connection timeout has been reached, causing the disconnection due to a timeout.

mickeyl, 2024-12-16:
Well, technically the program does not really pause, since the low level "breakpoint" is just internally used by OpenOCD to measure. Besides, BLE is the only thing that doesn't survive.