espressif/esp-idf #13707: 我无法运行websocket echo server例程中的pytest_ws_server_example.py

FlyyingPiggy2020, 2024-04-29:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

ESP-IDF v5.1.1-dirty

### Operating System used.

Linux

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### What is the expected behavior?

能够正常运行pytest_ws_server_example.py 

### What is the actual behavior?

`ModuleNotFoundError: No module named 'common_test_methods'`

[image]

### Steps to reproduce.

1.python python pytest_ws_server_example.py 

### More Information.

我已经按照Readme去下载了websocket-client

gabsuren, 2024-05-03:
@FlyyingPiggy2020 sorry, but I am not sure what websocket echo server do you use.
Please check the examples in `esp-protocols` https://github.com/espressif/esp-protocols/blob/master/components/esp_websocket_client/examples/target/pytest_websocket.py

And in order to setup ESP-IDF environment please check programming guide 
https://docs.espressif.com/projects/esp-idf/en/latest/esp32/get-started/linux-macos-setup.html

Indastri, 2024-09-16:
Since there was no response from the user since may closing the ticket. If you will encounter this issue again please feel free to reopen this issue or create a new one.

Cheers!

closed 2024-09-16 by Indastri