espressif/esp-idf #13708: 关于蓝牙广播与蓝牙mesh 广播共存问题。

sonwa, 2024-04-29:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

使用SDK v5.2.1-443-g2ba5320112-dirty
使用esp_ble_mesh_start_ble_advertising把相关设置好发送，但另一蓝牙节点没有收到蓝牙广播？请问是什么问题？

forx157, 2024-04-29:
Hi, @sonwa,
I need some more information in order to help you debug this.
Please enable log level to info level and share the log with me so that I can analyse the issue.

sonwa, 2024-04-29:
Hi, @forx157 ,
I've already configured "Support sending normal BLE advertising packets" in menuconfig and I'm using esp_ble_mesh_start_ble_advertising to send the following content:

esp_ble_mesh_ble_adv_param_t adv_param = {
        .interval = 0x0020, 
        .adv_type = BLE_MESH_ADV_IND,
        .own_addr_type = 0,
        .peer_addr_type = 0,
         .peer_addr = {0xFFFFFFFFFFFF},
        .duration = 560,
        .period = 38,
        .count = 5,
        .priority = ESP_BLE_MESH_BLE_ADV_PRIO_HIGH
    };

    esp_ble_mesh_ble_adv_data_t adv_data = {0};
    esp_err_t err = ESP_OK;

    adv_data.scan_rsp_data_len = param->scan_ble_adv_pkt.length;
    memcpy(&adv_data.scan_rsp_data, param->scan_ble_adv_pkt.data, param->scan_ble_adv_pkt.length);

	for (int i = 0; i < 25; i++) {
        adv_data.adv_data[i] = 0x10;
    }

    adv_data.adv_data_len = 25;

    err = esp_ble_mesh_start_ble_advertising(&adv_param, &adv_data);
    
    if (err) {
        ESP_LOGE(TAG, "!!!!!!!~~!!!!!!!Send esp_ble_mesh_start_ble_advertising failed");
        //flag = 0;
        return;
    }

The transmission appears successful without any error messages, but the other Bluetooth node isn't receiving the ADV broadcast packet.

[image]

forx157, 2024-04-29:
Hi, @sonwa,
Would it be convenient to share your log?

sonwa, 2024-04-29:
Hi, @forx157 , log:

--- esp-idf-monitor 1.4.0 on /dev/cu.wchusbserial578E0173861 115200 ---
--- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x1 (POWERON),boot:0x8 (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3810,len:0x239c
load:0x403c9700,len:0x4
load:0x403c9704,len:0xe50
load:0x403cc700,len:0x3270
entry 0x403c995c
I (27) boot: ESP-IDF v5.2.1-443-g2ba5320112-dirty 2nd stage bootloader
I (27) boot: compile time Apr 29 2024 13:45:57
I (28) boot: Multicore bootloader
D (32) bootloader_flash: non-XMC chip detected by SFDP Read (85), skip.
D (38) bootloader_flash: mmu set block paddr=0x00000000 (was 0xffffffff)
I (45) boot: chip revision: v0.2
D (49) boot.esp32s3: magic e9
D (52) boot.esp32s3: segments 04
D (55) boot.esp32s3: spi_mode 02
D (59) boot.esp32s3: spi_speed 0f
D (62) boot.esp32s3: spi_size 01
I (65) boot.esp32s3: Boot SPI Speed : 80MHz
I (70) boot.esp32s3: SPI Mode       : DIO
I (75) boot.esp32s3: SPI Flash Size : 2MB
D (80) boot: Enabling RTCWDT(9000 ms)
I (83) boot: Enabling RNG early entropy source...
D (89) bootloader_flash: rodata starts from paddr=0x00008000, size=0xc00, will be mapped to vaddr=0x3c000000
V (99) bootloader_flash: after mapping, starting from paddr=0x00000000 and vaddr=0x3c000000, 0x10000 bytes are mapped
D (109) boot: mapped partition table 0x8000 at 0x3c008000
D (115) flash_parts: partition table verified, 4 entries
I (120) boot: Partition Table:
I (124) boot: ## Label            Usage          Type ST Offset   Length
D (131) boot: load partition table entry 0x3c008000
D (136) boot: type=1 subtype=2
I (139) boot:  0 nvs              WiFi data        01 02 00009000 00006000
D (147) boot: load partition table entry 0x3c008020
D (152) boot: type=1 subtype=1
I (155) boot:  1 phy_init         RF data          01 01 0000f000 00001000
D (163) boot: load partition table entry 0x3c008040
D (168) boot: type=0 subtype=0
I (171) boot:  2 factory          factory app      00 00 00010000 00177000
I (178) boot: End of partition table
D (182) boot: Trying partition index -1 offs 0x10000 size 0x177000
D (189) esp_image: reading image header @ 0x10000
D (193) bootloader_flash: mmu set block paddr=0x00010000 (was 0xffffffff)
D (200) esp_image: image header: 0xe9 0x05 0x02 0x01 40375408
V (206) esp_image: loading segment header 0 at offset 0x10018
V (212) esp_image: segment data length 0x4c618 data starts 0x10020
V (218) esp_image: segment 0 map_segment 1 segment_data_offs 0x10020 load_addr 0x3c0d0020
I (226) esp_image: segment 0: paddr=00010020 vaddr=3c0d0020 size=4c618h (312856) map
D (235) esp_image: free data page_count 0x00000200
D (240) bootloader_flash: rodata starts from paddr=0x00010020, size=0x4c618, will be mapped to vaddr=0x3c000000
V (250) bootloader_flash: after mapping, starting from paddr=0x00010000 and vaddr=0x3c000000, 0x50000 bytes are mapped
V (317) esp_image: loading segment header 1 at offset 0x5c638
D (317) bootloader_flash: mmu set block paddr=0x00050000 (was 0xffffffff)
V (318) esp_image: segment data length 0x39d8 data starts 0x5c640
V (324) esp_image: segment 1 map_segment 0 segment_data_offs 0x5c640 load_addr 0x3fc98000
I (333) esp_image: segment 1: paddr=0005c640 vaddr=3fc98000 size=039d8h ( 14808) load
D (341) esp_image: free data page_count 0x00000200
D (346) bootloader_flash: rodata starts from paddr=0x0005c640, size=0x39d8, will be mapped to vaddr=0x3c000000
V (356) bootloader_flash: after mapping, starting from paddr=0x00050000 and vaddr=0x3c000000, 0x20000 bytes are mapped
V (370) esp_image: loading segment header 2 at offset 0x60018
D (373) bootloader_flash: mmu set block paddr=0x00060000 (was 0xffffffff)
V (380) esp_image: segment data length 0xccb70 data starts 0x60020
V (386) esp_image: segment 2 map_segment 1 segment_data_offs 0x60020 load_addr 0x42000020
0x42000020: _stext at ??:?
[... 296 lines cut ...]
D (2181) nvs: nvs_get_str_or_blob mesh/sig
D (2191) nvs: nvs_set_blob mesh/s/0004/b 6
D (2201) nvs: nvs_get_str_or_blob mesh/sig
D (2201) nvs: nvs_get_str_or_blob mesh/sig
D (2201) nvs: nvs_set_blob mesh/s/0100/b 6
D (2201) nvs: nvs_get_str_or_blob mesh/sig
D (2211) nvs: nvs_get_str_or_blob mesh/sig
D (2211) nvs: nvs_set_blob mesh/s/0200/b 6
D (2231) nvs: nvs_get_str_or_blob mesh/sig
D (2231) nvs: nvs_get_str_or_blob mesh/sig
D (2231) nvs: nvs_set_blob mesh/v/0000/b 6
D (2231) nvs: nvs_get_str_or_blob mesh/vnd
D (2231) nvs: nvs_get_str_or_blob mesh/vnd
I (3971) EXAMPLE: example_ble_mesh_callback:4, 19
########################################################################
    02 01 05 15 03 1e ff 01 0a db a8 6f 32 b1 8d 00 6d 7f e1 59 10 c3 7d f6 2d 
########################################################################
I (9661) EXAMPLE: example_ble_mesh_callback:4, 19
########################################################################
    02 01 05 15 03 1e ff 01 0b de d0 f8 8c d0 2d 7d da df c4 0f 3e 43 94 30 7c 
########################################################################
I (19371) EXAMPLE: example_ble_mesh_callback:4, 19
########################################################################
    02 01 05 15 03 1e ff 01 0c 54 82 fc 19 62 4d fb a6 6e 36 16 31 e8 a7 a5 bb 
########################################################################

sonwa, 2024-04-29:
########################################################################
xx xx xx xx xx xx xx xx xx
########################################################################

These data are ADV broadcast packets for receiving from a remote controller, and reception is normal. However, other nodes can't receive the ADV broadcast packets when sent.

forx157, 2024-04-30:
Hi, @sonwa 
I tested with your code and the branch you are using and found that the packet is sent out normally.
[image]

sonwa, 2024-05-05:
Hi, @forx157 
Yes, I finally managed to get it to work after trying for a few days. However, I have another question: Do I need to call "esp_ble_mesh_stop_ble_advertising" to stop broadcasting every time after sending the broadcast?

forx157, 2024-05-06:
No need, it will automatically pause when the time and count of packets you set is over, unless you set the `count` to 0xFFFF.

sonwa, 2024-05-06:
I see, thank you.

closed 2024-05-06 by sonwa