espressif/esp-idf #13709: stack/heap memory corruption?

hyuan-kamuda, 2024-04-29:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.2.1

### Espressif SoC revision.

Chip rev: v1.0

### Operating System used.

Windows

### How did you build your project?

VS Code IDE

### If you are using Windows, please specify command line type.

None

### Development Kit.

ESP32-LyraTD-MSC(ESP32-WROVER-E)

### Power Supply used.

USB

### What is the expected behavior?

It should have no stack overflow or heap corruption.

### What is the actual behavior?

with different size definition( #define MAX_HTTP_OUTPUT_BUFFER 512  or 1024 or 2048) , it has different behavior:

512: normal
1024: stack overflow
2048: heap corruption

### Steps to reproduce.

This is a very simple https communication application, just copied and modified from IDF sample application of "esp_http_client".
app_main code copied below, the FreeRTOS task is created with stack size of 8192(8K), should be enough?

//#define MAX_HTTP_OUTPUT_BUFFER 512
//#define MAX_HTTP_OUTPUT_BUFFER 1024
#define MAX_HTTP_OUTPUT_BUFFER 2048

void talk_to_server(void)
{
    char local_response_buffer[MAX_HTTP_OUTPUT_BUFFER+1] = {0};
    esp_http_client_config_t config = {
        .url = SERVER_WEB_URL,
        .event_handler = _http_event_handler,
        .crt_bundle_attach = esp_crt_bundle_attach,
        .user_data = local_response_buffer,
        //.method = HTTP_METHOD_POST,
    };

    esp_http_client_handle_t client = esp_http_client_init(&config);

    ...
    ...
[... 161 lines cut ...]
0x40108c46: ieee80211_send_setup at ??:?

A2 : 0x00000001 A3 : 0x3ffd1208 A4 : 0x3ffb7fbc A5 : 0x00000010
A6 : 0x00000005 A7 : 0x00000005 A8 : 0x3ffb7fbc A9 : 0x3ffc02d0
A10 : 0x00000048 A11 : 0x00000003 A12 : 0x00000018 A13 : 0x00000000
A14 : 0x00000001 A15 : 0x00000000 SAR : 0x00000011 EXCCAUSE: 0x0000001c
EXCVADDR: 0x00000001 LBEG : 0x4000c46c LEND : 0x4000c477 LCOUNT : 0x00000000
0x4000c46c: memset in ROM
0x4000c477: memset in ROM

Backtrace: 0x40108c43:0x3ffc02e0 0x4010a7e5:0x3ffc0310 0x4010a828:0x3ffc0350 0x40116c16:0x3ffc0370 0x4011719f:0x3ffc0390 0x40117c6f:0x3ffc03b0 0x401191d5:0x3ffc03d0 0x40164599:0x3ffc03f0 0x40092216:0x3ffc0410 0x40089b9d:0x3ffc0440
0x40108c43: ieee80211_send_setup at ??:?
0x4010a7e5: ieee80211_encap_null_data at ??:?
0x4010a828: ieee80211_pm_tx_null_process at ??:?
0x40116c16: pm_send_nullfunc at ??:?
0x4011719f: pm_go_to_sleep at ??:?
0x40117c6f: pm_active_timeout_process at ??:?
0x401191d5: dbg_lmac_ps_statis_reset at ??:?
0x40164599: pp_timer_do_process at ??:?
0x40092216: ppTask at ??:?
0x40089b9d: vPortTaskWrapper at C:/Users/hyuan/esp/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/port.c:134

ELF file SHA256: 4a040b880

CPU halted.

hyuan-kamuda, 2024-04-29:
[sdkconfig.txt](https://github.com/espressif/esp-idf/files/15145412/sdkconfig.txt)

Attached with SDKconfig for your information. 

Thanks

mahavirj, 2024-05-06:
@hyuan-kamuda 

It is difficult to comment on this issue without looking at the entire sample application code. Interim, you may increase the stack size and then fine tune it as per this documentation guide: https://docs.espressif.com/projects/esp-idf/en/stable/esp32/api-guides/performance/ram-usage.html#reducing-stack-sizes. This will help eliminate any issues w.r.t. insufficient stack size for the task (given that you have some data buffers placed in stack memory).

Hope this helps!

hyuan-kamuda, 2024-05-06:
> @hyuan-kamuda
>
> It is difficult to comment on this issue without looking at the entire sample application code. Interim, you may increase the stack size and then fine tune it as per this [...]

Thanks. 
Increasing stack size can fix the issue, however, changing code in function to allocate buffer from heap(without increasing task stack size), instead of defining it as char array(from stack), the issue is still there. Guessing ESP lib will copy the data into local var (so using big memory from stack) although user application allocating memory from heap?  Please refer to: https://www.esp32.com/viewtopic.php?f=13&t=39614

mahavirj, 2024-05-13:
@hyuan-kamuda 

Can you please supply minimal application code to recreate this failure?

mahavirj, 2024-05-13:
It will be good if you can attach the entire application with sdkconfig that you are using

hyuan-kamuda, 2024-05-13:
> It will be good if you can attach the entire application with sdkconfig that you are using

Both provided in above.

nileshkale123, 2025-02-26:
Hello @hyuan-kamuda ,

We attempted to reproduce the issue using the esp_http_client_example. Based on our experiments, if a variable is declared on the stack and a stack overflow occurs, you may need to either increase the stack size allocated for the task or declare the variable on the heap.

However, throughout our testing, we could not reproduce any heap corruption issue. If you are able to reproduce this on your end, please provide a complete example or a patch to help us investigate further.

nileshkale123, 2025-03-05:
Hello @hyuan-kamuda ,

I am closing this request for now. Please feel free to reopen if you face any issues regarding this.
Thank you.

closed 2025-03-05 by nileshkale123