espressif/esp-idf #13713: ESP-IDF v5.2.1 LWIP PPP driver didn't work with IPv6

kanjiviroja1991, 2024-04-29:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

pbuf_remove_header: old 0x3ffb9e68 new 0x3ffb9e76 (14)
pbuf_remove_header: old 0x3ffb9e76 new 0x3ffb9e78 (2)
netif_ip6_addr_set: netif address being changed
netif: IPv6 address 0 of interface pp set to FE80::8136:98F9:5863:5798/0x00
netif_ip6_addr_set_state: netif address state being changed
I (60493) pppos_example: PPP IP event! 3
I (60503) pppos_example: GOT IPv6 event!
I (60503) pppos_example: Got IPv6 address fe80:0000:0000:0000:8136:98f9:5863:5798
netif: IPv6 address 0 of interface pp set to FE80::8136:98F9:5863:5798/0x30
I (60523) esp-netif_lwip-ppp: Connected
pbuf_free(0x3ffb9e58)
pbuf_free: deallocating 0x3ffb9e58
pbuf_free(0x3ffb7c18)
pbuf_free: deallocating 0x3ffb7c18
I (60533) PING: PPP Interface  key 4:

I (60543) PING: ppp: 0.0.0.0

I (60543) PING:  STA Interface  key 3:

I (60543) PING: sta: 0.0.0.0

I (60553) PING: ETH Interface  key 2:

I (60553) PING: eth: 0.0.0.0

I (60563) PING: Done listing network interfaces
pbuf_alloc(length=22)
pbuf_alloc(length=22) == 0x3ffb8afc
pbuf_alloc(length=14)
pbuf_alloc(length=14) == 0x3ffb9e58
pbuf_remove_header: old 0x3ffb9e68 new 0x3ffb9e76 (14)
pbuf_remove_header: old 0x3ffb9e76 new 0x3ffb9e78 (2)
pbuf_alloc(length=8)
pbuf_alloc(length=8) == 0x3ffba458
pbuf_alloc(length=0)
pbuf_alloc(length=0) == 0x3ffebb2c
pbuf_free(0x3ffebb2c)
pbuf_free: deallocating 0x3ffebb2c
pbuf_free(0x3ffba458)
pbuf_free: deallocating 0x3ffba458
pbuf_free(0x3ffb9e58)
pbuf_free: deallocating 0x3ffb9e58
pbuf_free(0x3ffb8afc)
pbuf_free: deallocating 0x3ffb8afc
pbuf_alloc(length=12)
pbuf_alloc(length=12) == 0x3ffb8afc
pbuf_alloc(length=14)
pbuf_alloc(length=14) == 0x3ffb9e58
pbuf_remove_header: old 0x3ffb9e68 new 0x3ffb9e76 (14)
pbuf_remove_header: old 0x3ffb9e76 new 0x3ffb9e78 (2)
pbuf_alloc(length=44)
pbuf_alloc(length=44) == 0x3ffba458
pbuf_alloc(length=0)
pbuf_alloc(length=0) == 0x3ffebb2c
pbuf_free(0x3ffebb2c)
pbuf_free: deallocating 0x3ffebb2c
pbuf_free(0x3ffba458)
pbuf_free: deallocating 0x3ffba458
pbuf_free(0x3ffb9e58)
pbuf_free: deallocating 0x3ffb9e58
pbuf_free(0x3ffb8afc)
[... 841 lines cut ...]

I (77323) PING: ETH Interface  key 2:

I (77323) PING: eth: 0.0.0.0

I (77323) PING: Done listing network interfaces
I (77333) PING: DNS lookup success
I (77333) PING: target_addr.type=46
I (77343) PING: target_addr=8.8.8.8
I (77343) PING: initialize_ping success for GSM
dns_tmr: dns_check_entries
dns_check_entry: "csms.arconaconsultancy.com": timeout
E (77463) esp-tls: couldn't get hostname for :csms.arconaconsultancy.com: getaddrinfo() returns 202, addrinfo=0x0
E (77463) transport_base: Failed to open a new connection: 32769
E (77483) transport_ws: Error connecting to host csms.arconaconsultancy.com:8888
I (77483) ocpp-client: WEBSOCKET_EVENT_DISCONNECTED
etharp_timer
dhcp_coarse_tmr()
lwip_recvfrom_udp_raw[UDP/RAW]: netconn_recv err=-3, netbuf=0x0
lwip_recvfrom[UDP/RAW](54): buf == NULL, error is "Timeout."!
W (77823) PING: From 254.128.0.0  PPP Interface icmp_seq=15 timeout
lwip_sendto: invalid addresslwip_getsockopt(54, SOL_SOCKET, SO_ERROR) = 0
E (77833) ping_sock: send error=0
lwip_recvfrom(54, 0x3ffbad20, 64, 0x0, ..)
lwip_recvfrom_udp_raw[UDP/RAW]: top sock->lastdata=0x0

kanjiviroja1991, 2024-04-29:
I have already enabled local link for ipv6:

            **modem_netif_adapter = esp_modem_netif_setup(dte);
            if ((NULL != esp_netif) && (NULL != modem_netif_adapter))
            {
                esp_modem_netif_set_default_handlers(modem_netif_adapter, esp_netif);
                /* attach the modem to the network interface */
                err = esp_netif_attach(esp_netif, modem_netif_adapter);
                ESP_LOGE(TAG,"esp_netif_attach, err: %x ",(int)err);
                esp_netif_create_ip6_linklocal(esp_netif);
            }**

SDK CONFIG:
# CONFIG_LWIP_AUTOIP is not set
CONFIG_LWIP_IPV4=y
CONFIG_LWIP_IPV6=y
CONFIG_LWIP_IPV6_AUTOCONFIG=y
CONFIG_LWIP_IPV6_NUM_ADDRESSES=3
CONFIG_LWIP_IPV6_FORWARD=y
CONFIG_LWIP_IPV6_RDNSS_MAX_DNS_SERVERS=0
CONFIG_LWIP_IPV6_DHCP6=y
CONFIG_LWIP_NETIF_STATUS_CALLBACK=y
CONFIG_LWIP_NETIF_LOOPBACK=y
CONFIG_LWIP_LOOPBACK_MAX_PBUFS=8

PPP CONFIG:
CONFIG_LWIP_TCPIP_TASK_STACK_SIZE=3072
CONFIG_LWIP_TCPIP_TASK_AFFINITY_NO_AFFINITY=y
# CONFIG_LWIP_TCPIP_TASK_AFFINITY_CPU0 is not set
CONFIG_LWIP_TCPIP_TASK_AFFINITY=0x7FFFFFFF
CONFIG_LWIP_PPP_SUPPORT=y
CONFIG_LWIP_PPP_ENABLE_IPV6=y
CONFIG_LWIP_IPV6_MEMP_NUM_ND6_QUEUE=3
CONFIG_LWIP_IPV6_ND6_NUM_NEIGHBORS=3
CONFIG_LWIP_PPP_NOTIFY_PHASE_SUPPORT=y
CONFIG_LWIP_PPP_PAP_SUPPORT=y
# CONFIG_LWIP_PPP_CHAP_SUPPORT is not set
# CONFIG_LWIP_PPP_MSCHAP_SUPPORT is not set
# CONFIG_LWIP_PPP_MPPE_SUPPORT is not set
# CONFIG_LWIP_ENABLE_LCP_ECHO is not set
# CONFIG_LWIP_PPP_DEBUG_ON is not set
# CONFIG_LWIP_SLIP_SUPPORT is not set

kanjiviroja1991, 2024-05-08:
@espressif-bot , Is there any news on this matter? Simply use IPV6 while connecting over PPP to replicate the issue; following that, we receive an IPV6 address, but the internet remains inaccessible.

kanjiviroja1991, 2024-05-09:
@espressif-abhikroy , please find my sdkconfig file that might can help you to debug this issue.
[Uploading sdkconfig - Copy.txt…]()

espressif-abhikroy, 2024-05-15:
@kanjiviroja1991 Can you please provide an example code with which we can reproduce the issue?

kanjiviroja1991, 2024-05-15:
@espressif-abhikroy , Yes, I will provide example code for testing. Do you have GSM module?

espressif-abhikroy, 2024-05-15:
@kanjiviroja1991  Yes I do.

kanjiviroja1991, 2024-05-15:
[PPP_CodeBase.zip](https://github.com/espressif/esp-idf/files/15323311/PPP_CodeBase.zip)

Please configure as per below list of thing:
1) Configure UART for GSM Modem using menuconfig
2) Modem code (AT command some of modified based on N58 NEOWAY module)
3) Set appropriate APN using this object gsm_config_gw gsm_config;
4) Use PDP context with IPV6 only.

kanjiviroja1991, 2024-05-17:
@espressif-abhikroy , Can you reproduce the issue at your end? If you need any help then please ping me.

kanjiviroja1991, 2024-05-22:
@espressif-abhikroy sir, Do we have any update on this issue?, A lot of complaints have been made recently about some SIM cards not working since the SIM service provides IPV6 but driver unable creating an internet gateway bridge, which is a major problem for our product development.

AxelLin, 2024-05-22:
@kanjiviroja1991 Which SIM card/carrier has this issue?

kanjiviroja1991, 2024-05-22:
@AxelLin ,  It functions properly if the SIM service offers IPV4. However, certain SIM service providers—like JIO in India—only permit IPV6. To replicate this problem on all SIM services, simply attempt to establish a PPP link that uses IPV6 exclusively while in a PDP context.

david-cermak, 2024-05-31:
Hi @kanjiviroja1991 

If I understand the problem correctly, you're correctly getting the LL address after the IP6CP negotiation, but are not able to access internet, as that address is not routable?

Could you please just enable `CONFIG_LWIP_PPP_DEBUG_ON` and share the log after getting the link local addresses?
After that point, it's probably up to your provider, how you can obtain routable IPv6 address, that could be either SLAAC, DHCPv6 stateless (in race races maybe even stateful). Could you please check with your operator which one is is?

Note that what you've been doing in https://github.com/espressif/esp-idf/issues/13713#issuecomment-2082704243 won't help, as you've already gotten one LL address and this API won't work for Point to point netifs.

I think I'm able to reproduce the problem, or a similar one (with O2 in CZ); I'm getting supposedly corrupted RA/solicitation packets which are just dropped in lwip. Interestingly the local link communication works seamlessly. Could you please share the logs, just before the IP6CP protocol negotiation, with `IPV6-DEBUG` enabled, so I could tell if you're hit by the same issue?

To reduce some noise you can also try to disable `PPP_IPV4_SUPPORT` (directly in `ppp_opts.h`)

kanjiviroja1991, 2024-06-02:
Hi @david-cermak , Of course, I will share the logs in accordance with the recommended debug log requirements for debugging. 

According to what we have seen, this problem occurred when connecting using IPV6 with all SIM providers. The inability to determine the proper path for data routing through incoming/outgoing message handling may be the cause of several routing-related issues. This is designed specifically for IPV4 and automatically assigns a gateway address and DNS so that users can access the internet.

kanjiviroja1991, 2024-06-03:
Hi @david-cermak , 

**If we disabled PPP_IPV4_SUPPORT direct in ppp_opts.h that generate error due to this api belongs to V4 context => ppp_set_usepeerdns(ppp_obj->ppp, 1);**

Here you go debug log file: 
[teraterm-ipv6-issue_A.log](https://github.com/user-attachments/files/15530140/teraterm-ipv6-issue_A.log)

Please ping me if you require any further logs or anything else.

Thanks & Regards,
Kanji.

david-cermak, 2024-06-03:
Thanks for sharing the log, looks like the same issue on my end, also seeing this weird packet on PPP netif:
```
[2024-06-03 12:38:54.148] ip6_input: packet accepted on interface pp
[2024-06-03 12:38:54.148] pbuf_remove_header: old 0x3ffb9e6a new 0x3ffb9e92 (40)
[2024-06-03 12:38:54.148] ip6_input: 
[2024-06-03 12:38:54.148] IPv6 header:
[2024-06-03 12:38:54.148] +-------------------------------+
[2024-06-03 12:38:54.163] |  8 |   96  |        49073     | (ver, class, flow)
[2024-06-03 12:38:54.163] +-------------------------------+
[2024-06-03 12:38:54.163] |     16448     |   85  |   80  | (plen, nexth, hopl)
[2024-06-03 12:38:54.163] +-------------------------------+
[2024-06-03 12:38:54.163] |     0 |     0 |     0 |     0 | (src)
[2024-06-03 12:38:54.179] |   501 |     0 |     0 |   640 |
[2024-06-03 12:38:54.179] +-------------------------------+
[2024-06-03 12:38:54.179] |   304 |  4040 |  ffff |  ffff | (dest)
[2024-06-03 12:38:54.179] |  ffff |  ffff |     0 |     0 |
[2024-06-03 12:38:54.195] +-------------------------------+
```

> If we disabled PPP_IPV4_SUPPORT direct in ppp_opts.h that generate error due to this api belongs to V4 context => ppp_set_usepeerdns(ppp_obj->ppp, 1);

oh, sorry, need to comment this out. I'll prepare a patch, I think it' would be useful to disable IPv4 (at least on PPP level)

> Please ping me if you require any further logs or anything else.

As mentioned above, I'd like to know how are you supposed to acquire a routable address from your provider (probably something like DHCPv6 relay), but I think you need to check with your operator.

kanjiviroja1991, 2024-06-04:
@david-cermak , Since I have already tested a number of methods, including using the same modem to connect to a laptop and browse the internet to retrieve IPV6, I can say with certainty that the SIM service is operating as intended. Other than this, there isn't a problem. The modem technical team has already given me a lot more support for the same problem.

I used a Vodafone SIM/JIO SIM/AIRTEL SIM card, which offers service everywhere, I don't think there was a problem on their end.

**Here you can see PPP link with N58 using JIO SIM in my laptop, I am able to access internet service without any issues:
PPP adapter N58_RJIO:

   Connection-specific DNS Suffix  . :
   IPv6 Address. . . . . . . . . . . : 2409:40c1:1b:271b:8496:3d47:b1f0:7e5c
   Link-local IPv6 Address . . . . . : fe80::8496:3d47:b1f0:7e5c%50
   Default Gateway . . . . . . . . . : fe80::8ae9:a4ff:fe13:48a4%50**

david-cermak, 2024-06-04:
Yes, there's very likely (at least) one problem on the ESP side, but I just like to know which method your operator use. Is it SLAAC only or DHCPv6 (stateless or statefull)? Or, if it's running on your laptop, are you using `radvd` with `dhcp6c` or without it?

The reason I'm asking is that, in case of DHCPv6 stateless configuration (the most common method IMO), you do have to start the client explicitly with `dhcp6_enable_stateless()` (it's not enough to enable it in menuconfig)

https://docs.espressif.com/projects/esp-idf/en/stable/esp32/api-guides/lwip.html#dhcpv6

kanjiviroja1991, 2024-06-04:
Here you go of some output on windows prompt:
C:\Users\admin>sc query dhcp

SERVICE_NAME: dhcp
        TYPE               : 30  WIN32
        STATE              : 4  RUNNING
                                (STOPPABLE, NOT_PAUSABLE, ACCEPTS_SHUTDOWN)
        WIN32_EXIT_CODE    : 0  (0x0)
        SERVICE_EXIT_CODE  : 0  (0x0)
        CHECKPOINT         : 0x0
        WAIT_HINT          : 0x0

C:\Users\admin>ipconfig /all

Windows IP Configuration

   Host Name . . . . . . . . . . . . : DESKTOP-4VT06RD
   Primary Dns Suffix  . . . . . . . :
   Node Type . . . . . . . . . . . . : Hybrid
   IP Routing Enabled. . . . . . . . : No
   WINS Proxy Enabled. . . . . . . . : No
   DNS Suffix Search List. . . . . . : iballbatonwifi.com

Ethernet adapter Ethernet:

   Media State . . . . . . . . . . . : Media disconnected
   Connection-specific DNS Suffix  . :
   Description . . . . . . . . . . . : Realtek PCIe GbE Family Controller
   Physical Address. . . . . . . . . : 30-D0-42-1D-C2-CB
   DHCP Enabled. . . . . . . . . . . : Yes
   Autoconfiguration Enabled . . . . : Yes

Wireless LAN adapter Local Area Connection* 10:

   Media State . . . . . . . . . . . : Media disconnected
   Connection-specific DNS Suffix  . :
   Description . . . . . . . . . . . : Microsoft Wi-Fi Direct Virtual Adapter #2
   Physical Address. . . . . . . . . : 76-12-B3-94-BB-B5
   DHCP Enabled. . . . . . . . . . . : Yes
   Autoconfiguration Enabled . . . . : Yes

Wireless LAN adapter Local Area Connection* 12:

   Media State . . . . . . . . . . . : Media disconnected
   Connection-specific DNS Suffix  . :
   Description . . . . . . . . . . . : Microsoft Wi-Fi Direct Virtual Adapter #3
   Physical Address. . . . . . . . . : 86-12-B3-94-BB-B5
   DHCP Enabled. . . . . . . . . . . : Yes
   Autoconfiguration Enabled . . . . : Yes

Wireless LAN adapter Wi-Fi:

   Connection-specific DNS Suffix  . : iballbatonwifi.com
   Description . . . . . . . . . . . : Qualcomm QCA9377 802.11ac Wireless Adapter
   Physical Address. . . . . . . . . : 76-12-16-94-D8-19
   DHCP Enabled. . . . . . . . . . . : Yes
   Autoconfiguration Enabled . . . . : Yes
   Link-local IPv6 Address . . . . . : fe80::c94a:4554:e4cb:8049%4(Preferred)
   IPv4 Address. . . . . . . . . . . : 192.169.0.203(Preferred)
   Subnet Mask . . . . . . . . . . . : 255.255.255.0
   Lease Obtained. . . . . . . . . . : 03 June 2024 06:03:42 PM
   Lease Expires . . . . . . . . . . : 05 June 2024 11:18:54 AM
   Default Gateway . . . . . . . . . : 192.169.0.1
   DHCP Server . . . . . . . . . . . : 192.169.0.1
   DHCPv6 IAID . . . . . . . . . . . : 74846742
   DHCPv6 Client DUID. . . . . . . . : 00-03-00-01-76-12-16-94-D8-19
   DNS Servers . . . . . . . . . . . : 192.169.0.1
   NetBIOS over Tcpip. . . . . . . . : Enabled

C:\Users\admin>

kanjiviroja1991, 2024-06-04:
@david-cermak , I am using windows system.

Windows and DHCP:
On a Windows system, the equivalent service for handling IPv6 DHCP client functionality is still managed by the DHCP Client service. Windows does not use radvd and dhcp6c; instead, it integrates the necessary IPv6 functionality into its own DHCP Client service.

The reason I'm asking is that, in case of DHCPv6 stateless configuration (the most common method IMO), you do have to start the client explicitly with dhcp6_enable_stateless() (it's not enough to enable it in menuconfig)
==> I have already tested this thing in past it didn't work at all. I have explicity enabled using this api dhcp6_enable_stateless().

david-cermak, 2024-06-04:
okay, let's assume it's a stateless DHCPv6.

Found one bug with ip6 autoconfig, it's not correctly set to true for point to point interfaces if `CONFIG_LWIP_IPV6_AUTOCONFIG=y`.
I'll create a patch to fix it. Workaround is to set it manually after staring the netif
```cpp
  netif.ip6_autoconfig_enabled = 1;  // this is lwip netif structure
```

kanjiviroja1991, 2024-06-05:
@david-cermak , You're correct—there was a problem with the IPv6 autoconfig flag on the P2P interface. I tested the aforementioned changes, and I was able to receive global IPV6 and correctly ping.

In order to obtain appropriate resolution for modem functioning with IPv6, it would be preferable if this bug fixes were included in the next esp idf release.

Many thanks for the prompt resolution and update, @david-cermak .

Best Regards,
Kanji.

david-cermak, 2024-06-05:
Here's the patch fixing the PPP autoconfig. After applying it you should be able to receive a global IPv6 address and access the internet. DNS servers should be provided via DHCPv6 or directly within RA. 
[fix-esp_netif-Fix-missing-IPv6-autoconfig-for-PPP.txt](https://github.com/user-attachments/files/15574142/fix-esp_netif-Fix-missing-IPv6-autoconfig-for-PPP.txt)

I'll probably also add one Kconfig option to disable IPCP in PPP for users of IPv6 only networks.

There's still one cosmetic bug in lwip, incorrectly displaying the IPv6 headers from PPP netifs, as noted before:
```
[2024-06-03 12:38:54.148] IPv6 header:
[2024-06-03 12:38:54.148] +-------------------------------+
[2024-06-03 12:38:54.163] |  8 |   96  |        49073     | (ver, class, flow)
[2024-06-03 12:38:54.163] +-------------------------------+
[2024-06-03 12:38:54.163] |     16448     |   85  |   80  | (plen, nexth, hopl)
[2024-06-03 12:38:54.163] +-------------------------------+
[2024-06-03 12:38:54.163] |     0 |     0 |     0 |     0 | (src)
[2024-06-03 12:38:54.179] |   501 |     0 |     0 |   640 |
[2024-06-03 12:38:54.179] +-------------------------------+
[2024-06-03 12:38:54.179] |   304 |  4040 |  ffff |  ffff | (dest)
[2024-06-03 12:38:54.179] |  ffff |  ffff |     0 |     0 |
[2024-06-03 12:38:54.195] +-------------------------------+
```

kanjiviroja1991, 2024-06-05:
@david-cermak , As you have created patch with specific location didn't work.

**I'll probably also add one Kconfig option to disable IPCP in PPP for users of IPv6 only networks.**
=> PPP called with IPV4V6 so that case it might doesn't work if you disabled IPCP.

david-cermak, 2024-06-05:
@kanjiviroja1991 Are you saying that the patch posted https://github.com/espressif/esp-idf/issues/13713#issuecomment-2148993609 doesn't work? Like you're not getting the global address after applying it?

Or, that the patch doesn't work in IPv4+IPv6 scenario? or without IPCP?

kanjiviroja1991, 2024-06-05:
@david-cermak ,I tested your fix, and it functioned as intended. Prior to that, it wasn't working because of my code-mixing modifications. 

I'm still not sure if ND6 needs to be set in flag or not(check this config -> ESP_NETIF_INHERENT_DEFAULT_PPP())
netif->flags |= NETIF_FLAG_MLD6; //For PPP default config doesn't enable this flag

david-cermak, 2024-06-05:
Thanks for double checking!

> I'm still not sure if ND6 needs to be set in flag or not(check this config -> ESP_NETIF_INHERENT_DEFAULT_PPP())

No need to set any other flags; Note that `ESP_NETIF_INHERENT_DEFAULT_PPP()` belongs to `esp_netif` (IDF convenient wrapper around lwip), which only has these values:

https://github.com/espressif/esp-idf/blob/8760e6d2a7e19913bc40675dd71f374bcd51b0ae/components/esp_netif/include/esp_netif_types.h#L170-L179

(so when using the default IDF configs, the flags of lwip netifs are set accordingly)

closed 2024-07-05 by espressif-bot

AxelLin, 2024-07-31:
@david-cermak v5.1, v5.2, and v5.3 branches still need fix.

AxelLin, 2024-08-26:
> Yes, there's very likely (at least) one problem on the ESP side, but I just like to know which method your operator use. Is it SLAAC only or DHCPv6 (stateless or statefull)? Or, if it's running on [...]

Hi @david-cermak 
The dhcp6_enable_stateless() takes struct netif * as argument.
Most esp-idf application code uses esp_netif_t * as argement, and cannot access the struct netif *.
I'm wondering if it's necessary to provide an esp_netif API for dhcp6_enable_stateless()?
Also, is it safe to call dhcp6_enable_stateless() in non-tcpip tasks?

If possible, would you also update example code for IPV6 with SLAAC and DHCPv6 use cases?

AxelLin, 2024-08-26:
> I have already enabled local link for ipv6:
>
> ```
>         **modem_netif_adapter = esp_modem_netif_setup(dte);
>         if ((NULL != esp_netif) && (NULL != modem_netif_adapter))
>         { [...]

Hi @david-cermak 
I reference the pppos_client example code: (which does not call esp_netif_create_ip6_linklocal())
I do **NOT** call esp_netif_create_ip6_linklocal() but I still got
Got IPv6 address: fe80:0000:0000:0000:7931:1004:a491:eca0, type: ESP_IP6_ADDR_IS_LINK_LOCAL.
So I'm confused if it's necessary to call esp_netif_create_ip6_linklocal() for PPP interface?

david-cermak, 2024-08-26:
About IPv6 LL:
* No need to call `esp_netif_create_ip6_linklocal()`  for PPP interfaces. The address is acquired differently, during IPV6CP handshake (which is controlled by `CONFIG_LWIP_PPP_ENABLE_IPV6`)

About DHCPv6:
* Yes, we're planning to create an IDF/esp_netif API to start the DHCPv6 server conveniently (and in a thread safe way, and update the example).

AxelLin, 2024-08-28:
Hi @david-cermak 

When I check CONFIG_ESP_NETIF_SET_DNS_PER_DEFAULT_NETIF changes
I found the ESPIP_TO_IP and IP_TO_ESPIP macros are useful for application code.
Would you consider move them to a header file so application code can use it?

AxelLin, 2024-08-28:
> About DHCPv6:
> 
> * Yes, we're planning to create an IDF/esp_netif API to start the DHCPv6 server conveniently (and in a thread safe way, and update the example).

Current problem is I cannot call dhcp6_enable_stateless() from application code
because it needs "struct netif *" but I cannot access mynetif->lwip_netif:

error: invalid use of incomplete typedef 'esp_netif_t' {aka 'struct esp_netif_obj'}
 1889 | dhcp6_enable_stateless(mynetif->lwip_netif);

I think this needs fix first.

david-cermak, 2024-08-28:
About the dhcp server in lwip: a bit clumsy but still all public API:

```cpp
#include "esp_netif_net_stack.h"
...
    struct netif* netif = esp_netif_get_netif_impl(esp_netif);
    dhcp6_enable_stateless(netif);
```
(and if you really want to play by the book you can run it in the lwip context using `esp_netif_tcpip_exec()` callbacks)

---

> I found the ESPIP_TO_IP and IP_TO_ESPIP macros are useful for application code.

A very good point, thanks for suggesting! I was thinking of making them internal functions at least (instead of macros), but making them public makes more sense.

AxelLin, 2024-08-29:
> The reason I'm asking is that, in case of DHCPv6 stateless configuration (the most common method IMO), you do have to start the client explicitly with `dhcp6_enable_stateless()` (it's not enough to [...]

Hi @david-cermak 
Is it required to call dhcp6_enable_stateless(netif) for PPP interface?
esp_netif_get_netif_impl() returns NULL for PPP netif.

david-cermak, 2024-08-29:
> Is it required to call dhcp6_enable_stateless(netif) for PPP interface?

This depends on your cellular provider, how the DNS server info is communicated. For IPv6 it could be either via RA or using DHCPv6 (less common from what I hear).

> esp_netif_get_netif_impl() returns NULL for PPP netif.

The restriction's been lifted in d70716188168d4bc2cab95089eb45c507726b8c9

but point taken on needing a proper API to start it, will prioritize.

AxelLin, 2024-08-29:
> > esp_netif_get_netif_impl() returns NULL for PPP netif.
> 
> The restriction's been lifted in [d707161](https://github.com/espressif/esp-idf/commit/d70716188168d4bc2cab95089eb45c507726b8c9)

Oh, I'm using v5.2 branch.
Could you backport it? Honestly, v5.3 is not yet stable at the moment.

AxelLin, 2024-11-04:
Hi @david-cermak 

I'm testing PDP-type=IPV6 case, with below settings:
 CONFIG_LWIP_IPV6_AUTOCONFIG=y
 CONFIG_LWIP_IPV6_RDNSS_MAX_DNS_SERVERS=2
 Testing with v5.2.3-262-g2b35c55820.

Once PPP connected and got IP6:
I got IP_EVENT_GOT_IP6 twice, one for link_local and one for global address.
esp_netif_get_all_ip6, shows
        IPV6: fe80:0000:0000:0000:f0ff:1630:aab4:f587 (LINK_LOCAL)
        IPV6: 2001:b400:e40e:8bd7:f0ff:1630:aab4:f587 (GLOBAL)
I can reach internet and ping 2001:4860:4860::8888 . (Seems work fine)

However, if I make the PPP disconnect and then reconnect.

I got IP_EVENT_GOT_IP6 twice, one for link_local and one for global address.
esp_netif_get_all_ip6, shows
        IPV6: fe80:0000:0000:0000:a9c9:0445:adb7:5794 (LINK_LOCAL)
        **IPV6: 2001:b400:e40e:8bd7:f0ff:1630:aab4:f587 (GLOBAL)**
        IPV6: 2001:b400:e40e:9215:a9c9:0445:adb7:5794 (GLOBAL)

**It's strange the previous global address is still returned by esp_netif_get_all_ip6.
The key issue is then I cannot reach internet.
i.e. ping 2001:4860:4860::8888 fails after disconnect and reconnect PPP.**

PS. I also test PDP-type=IPV4V6 case
and I observe the same behavior for IPV6 (2 global address after disconect and reconnect).
I cannot ping 2001:4860:4860::8888 but ping IPv4 address 8.8.8.8 still works after reconnect PPP.

Do you have any idea about this issue?

david-cermak, 2024-11-04:
Hi @AxelLin 

Yes, this is an issue in IDF/lwip, we currently have no way of recycling/invalidating older IPv6 records. You can only workaround it by increasing number of IPv6 addresses or clean out the global IPv6 address on PPP disconnect.