espressif/esp-idf #13714: esp32 pdm rx录音不正常。

crl6, 2024-04-29:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

esp32s3, 2通道录音。换了几个pdm麦克风都是这样，这些麦克风模组在rk平台都是正常的。录音wei为16k，16bit，2通道。
表现为有一个声道在0.0标准线以下。

[image]

[录音文件.zip](https://github.com/espressif/esp-idf/files/15149790/default.zip)

crl6, 2024-04-29:
尝试过修改gpio 上下拉，外部上拉。都会这样。如果.clk_inv = false，有问题的声道就会变。

crl6, 2024-04-29:
pdm数字麦型号： SPH0690LM4H

L-KAYA, 2024-04-30:
PDM RX 16K的采样频率+默认的8倍过采样，对应的采样时钟输出为1.024MHz，SPH0690LM4H数据手册里只给出了这几个频率：

[image]

所以可以先试试改变采样频率让采样时钟达到这些频率，比如 24000Hz

crl6, 2024-05-10:
请问，还有什么因素会导致这样？

L-KAYA, 2024-05-20:
如果不是采样问题的话，看这波形有比较大概率是电路上的问题。

> 尝试过修改gpio 上下拉，外部上拉。都会这样。如果.clk_inv = false，有问题的声道就会变。

这个翻转时钟其实就是软件上交换采样通道的顺序，软件上交换会导致有问题的声道也会改变，那对应硬件上就是说有问题的一直是那一路PDM麦克风，另外也能印证软件上其实两路都能正常采样，但是由于硬件上其中一路有问题，导致最终结果有问题。

不知道之前你们用rk平台测的时候硬件电路是否一致，最好对比一下，特别是接地的引脚有没有接错。

crl6, 2024-05-26:
done

closed 2024-05-26 by crl6