espressif/esp-idf #13715: esp_wifi_scan_start API needs improvement

pggh, 2024-04-29:
### Is your feature request related to a problem?

The esp_wifi_scan_start is difficult to use when multiple independent software parts/packages is using it.
There is only one radio and only one scan (or connect) can be running at any given time. The existing API does not offer to coordinate this.

### Describe the solution you'd like.

The first and very easy improvement would be for esp_wifi_scan_start to return the scan_id that is part of WIFI_EVENT_SCAN_DONE/wifi_event_sta_scan_done_t.
This would allow to check whether the received WIFI_EVENT_SCAN_DONE is from the own started scan or from other software components that probably scans for something else. (freeing the memory once and in the correct component is an issue too).

### Describe alternatives you've considered.

An overhaul of the API could be considered; probably something with scan-jobs that could be submitted and queued.

chipweinberger, 2024-05-02:
out of curiosity, what multiple independent software packages are you using that all scan?

do they scan without your knowledge? if so, why?

or can you control when they scan?

filzek, 2024-05-02:
There is no sense of making this as you should control the call fo it, also if you double call the same function the other is aborted ad there is no way to simultaneous run the scan, so, you should use a global populated list and a semaphore to control the call of it, and let the thread stand, and get the call on a list if you want nom blocking, and so return that the routine is in use and will resume later, after it resume you can run the task queue and execute the function you want, so, the semaphore shall work in a nom blocking parade.

Alvin1Zhang, 2025-03-13:
Thanks @filzek  for providing the answer, feel free to reopen.

closed 2025-03-13 by Alvin1Zhang