espressif/esp-idf #13716: ESP32 - GPIO21 & GPIO22 default value GPIO_FUNCn_OUT_SEL

nkaaf, 2024-04-29:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.2.1

### Espressif SoC revision.

ESP32-D0WDQ6 (revision v1.0)

### Operating System used.

macOS

### How did you build your project?

CLion IDE

### If you are using Windows, please specify command line type.

None

### Development Kit.

Dev Kit with ESP32-WROOM-32

### Power Supply used.

USB

### What is the expected behavior?

According to the Technical Reference Manual (5.1), the value of the 'GPIO_FUNCn_OUT_SEL' bits in the register 'GPIO_FUNCn_OUT_SEL_CFG_REG' (n=21 &n=22) is a valid peripheral output signal (Table 4­2. GPIO Matrix Peripheral Signals) or the value '256'.

### What is the actual behavior?

The default value of the bits is 0x101 (257), for both (GPIO 21 && GPIO 22).

### Steps to reproduce.

Code:

```c
#include <esp_log.h>
#include <freertos/FreeRTOS.h>
#include <freertos/task.h>

#define GPIO_FUNCn_OUT_SEL_CFG_REG(n) (0x3FF44000 + 0x530 + 0x4 * n)

void app_main()
{
  ESP_LOGE("GPIO 21 GPIO_FUNCn_OUT_SEL", "0x%lx", REG_READ(GPIO_FUNCn_OUT_SEL_CFG_REG(21)) & 0x1FF);
  ESP_LOGE("GPIO 22 GPIO_FUNCn_OUT_SEL", "0x%lx", REG_READ(GPIO_FUNCn_OUT_SEL_CFG_REG(22)) & 0x1FF);
  while (true)
  {
    vTaskDelay(5000 / portTICK_PERIOD_MS);
  }
}
```

### Debug Logs.

```plain
mode:DIO, clock div:2
load:0x3fff0030,len:7172
load:0x40078000,len:15540
load:0x40080400,len:4
ho 8 tail 4 room 4
load:0x40080404,len:3904
entry 0x40080640
[0;32mI (31) boot: ESP-IDF v5.2.1 2nd stage bootloader[0m
[0;32mI (31) boot: compile time Apr 27 2024 18:19:07[0m
[0;32mI (31) boot: Multicore bootloader[0m
[0;32mI (35) boot: chip revision: v1.0[0m
[0;32mI (39) boot.esp32: SPI Speed      : 40MHz[0m
[0;32mI (44) boot.esp32: SPI Mode       : DIO[0m
[0;32mI (48) boot.esp32: SPI Flash Size : 2MB[0m
[0;32mI (53) boot: Enabling RNG early entropy source...[0m
[0;32mI (58) boot: Partition Table:[0m
[0;32mI (62) boot: ## Label            Usage          Type ST Offset   Length[0m
[0;32mI (69) boot:  0 nvs              WiFi data        01 02 00009000 00006000[0m
[0;32mI (76) boot:  1 phy_init         RF data          01 01 0000f000 00001000[0m
[0;32mI (84) boot:  2 factory          factory app      00 00 00010000 00100000[0m
[0;32mI (91) boot: End of partition table[0m
[0;32mI (96) esp_image: segment 0: paddr=00010020 vaddr=3f400020 size=0957ch ( 38268) map[0m
[0;32mI (117) esp_image: segment 1: paddr=000195a4 vaddr=3ffb0000 size=0220ch (  8716) load[0m
[0;32mI (121) esp_image: segment 2: paddr=0001b7b8 vaddr=40080000 size=04860h ( 18528) load[0m
[0;32mI (130) esp_image: segment 3: paddr=00020020 vaddr=400d0020 size=13414h ( 78868) map[0m
[0;32mI (159) esp_image: segment 4: paddr=0003343c vaddr=40084860 size=07ebch ( 32444) load[0m
[0;32mI (178) boot: Loaded app from partition at offset 0x10000[0m
[0;32mI (178) boot: Disabling RNG early entropy source...[0m
[0;32mI (189) cpu_start: Multicore app[0m
[0;32mI (198) cpu_start: Pro cpu start user code[0m
[0;32mI (198) cpu_start: cpu freq: 160000000 Hz[0m
[0;32mI (198) cpu_start: Application information:[0m
[0;32mI (201) cpu_start: Project name:     IO[0m
[0;32mI (206) cpu_start: App version:      1.0.0[0m
[0;32mI (211) cpu_start: Compile time:     Apr 27 2024 18:19:02[0m
[0;32mI (217) cpu_start: ELF file SHA256:  3c45c323e...[0m
[0;32mI (222) cpu_start: ESP-IDF:          v5.2.1[0m
[0;32mI (227) cpu_start: Min chip rev:     v0.0[0m
[0;32mI (232) cpu_start: Max chip rev:     v3.99 [0m
[0;32mI (236) cpu_start: Chip rev:         v1.0[0m
[0;32mI (241) heap_init: Initializing. RAM available for dynamic allocation:[0m
[0;32mI (248) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM[0m
[0;32mI (254) heap_init: At 3FFB2AD0 len 0002D530 (181 KiB): DRAM[0m
[0;32mI (261) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM[0m
[0;32mI (267) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM[0m
[0;32mI (273) heap_init: At 4008C71C len 000138E4 (78 KiB): IRAM[0m
[0;32mI (281) spi_flash: detected chip: generic[0m
[0;32mI (284) spi_flash: flash io: dio[0m
[0;33mW (288) spi_flash: Detected size(4096k) larger than the size in the binary image header(2048k). Using the size in the binary image header.[0m
[0;32mI (302) main_task: Started on CPU0[0m
[0;32mI (312) main_task: Calling app_main()[0m
[0;31mE (312) GPIO 21 GPIO_FUNCn_OUT_SEL: 0x101[0m
[0;31mE (312) GPIO 22 GPIO_FUNCn_OUT_SEL: 0x101[0m
```
```

suda-morris, 2024-04-30:
I would suggest not rely on the default values that were documented in the TRM, because it may be changed in the ROM code if these GPIOs have another special purpose. So always call the APIs provided by the `driver/gpio.h` for initializing the GPIO.

songruo, 2024-04-30:
Hi @nkaaf,

In the TRM, it didn't say this field has a default value of 256 (shows `x` in the screenshot from the TRM). It only says if you want a GPIOn to be used as a simple GPIO output, then this field needs to be 256.

[image]

nkaaf, 2024-04-30:
Hi @songruo , ok, I realize that. I found it strange that the value is outside the range of a valid value according to TRM. Do you know if the value 257 means anything, or where this value comes from?

Bruce297, 2024-05-16:
Now we add a note that shouldn't rely on default values in [docs](https://docs.espressif.com/projects/esp-idf/en/latest/esp32c6/api-reference/peripherals/gpio.html)

[image]

closed 2024-05-16 by songruo