espressif/esp-idf #13719: ESP32-C3,蓝牙断开后不能被扫描到

monkey-qiu, 2024-04-30:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

ESP32-C3，蓝牙在一个连接断开后，会持续收到ESP_GAP_BLE_EXT_ADV_START_COMPLETE_EVT和ESP_GAP_BLE_ADV_TERMINATED_EVT事件，大概持续2-4分钟，然后蓝牙就不能被扫描到和被连接了

log：
GATTC_COEX: a ESP_GATTC_CONNECT_EVT conn_id 0, if 4
[0m
[0;32mI (4735648) GATTC_COEX: b ESP_GATTC_CONNECT_EVT conn_id 0, if 
5
[0m
[0;32mI (4735658) GATTC_GATTS_COEX: ESP_GAP_BLE_EXT_ADV_START_COMPLETE_EVT, status = 0[0m

[0;33mW (4742098) BT_APPL: gattc
:_conn_cb: if=4 st=0 id=4 rsn=0x8[0m
[0;33mW (4742098) BT_APPL: gattc_conn_cb: if=5 st=0 id=5 rsn=0x8[0m
[0;33mW (4742098) B
T_HCI: hcif disc complete: hdl 0x1, rsn 0x8[0m
[0;32mI (4742098) GATTS_COEX: ESP_GATTS_DISCONNECT_EVT, disconnect reason 0x8
[0m
[0;32mI (4742108) GATTS_COEX: business, conn_id 0, remote 54:03:f0:9b:29:ff[0m
[0;32mI 
(4742118) GATTS_COEX: addr=54:03:f0:9b:29:ff
[0m
[0;32mI (4742118) GATTC_COEX: ESP_GATTC_DISCONNECT_EVT, reason = 8
[0m

[0;32mI (4742128) GATTC_COEX: ESP_GATTC_DISCONNECT_EVT, reason = 8
[0m
[0;32mI (4742138) GATTC_GATTS_COEX: ESP_GAP_BLE_EXT_A
DV_START_COMPLETE_EVT, status = 0[0m

[0;32mI (4774408) GATTC_GATTS_COEX: ESP_GAP_BLE_ADV_TERMINATED_EVT, status = 0[0m
[0;32mI (4774408) GATTC_GATTS_COEX: ADV successfully ended with a connection being created[0m

[0;33mW (4774668) BT_HCI: hcif disc complete: hdl 0x3, rsn 0x3e[0m
[0;32mI (4774668) GATTC_GATTS_COEX: ESP_GAP_BLE_EXT_ADV_START_COMPLETE_EVT, status = 0[0m

[0;32mI (4786468) GATTC_GATTS_COEX: ESP_GAP_BLE_ADV_TERMINATED_EVT, status = 0[0m
[0;32mI (4786468) GATTC_GATTS_COEX: ADV successfully ended with a connection being created[0m

[0;33mW (4786728) BT_HCI: hcif disc complete: hdl 0x1, rsn 0x3e
[0m
[0;32mI (4786728) GATTC_GATTS_COEX: ESP_GAP_BLE_EXT_ADV_START_COMPLETE_EVT, status = 0[0m

[0;32mI (4790598) GATTC_GATTS_COEX: ESP_GAP_BLE_ADV_TERMINATED_:EVT, status = 0[0m
[0;32mI (4790598) GATTC_GATTS_COEX: ADV successfully ended with a connection being created[0m

[0;33mW (4790868) BT_HCI: hcif disc complete: hdl 0x1, rsn 0x3e[0m
[0;32mI (4790868) GATTC_GATTS_COEX: ESP_GAP_BLE_EXT_ADV_START_COMPLETE_EVT, status = 0[0m

[0;32mI (4802428) GATTC_GATTS_COEX: ESP_GAP_BLE_ADV_TERMINATED_EVT, status = 0[0m
[0;32mI (4802428) GATTC_GATTS_COEX: ADV su
ccessfully ended with a connection being created[0m

[0;33mW (4802658) BT_HCI: hcif disc complete: hdl 0x1, rsn 0x3e
[0m
[0;32mI (4802658) GATTC_GATTS_COEX: ESP_GAP_BLE_EXT_ADV_START_COMPLETE_EVT, status = 0[0m

[0;32mI (4833698) GATTC_GATTS_COEX: ESP_GAP_BLE_ADV_TERMINATED_EVT, status = 0[0m
[0;32mI (4833698) GATTC_GATTS_COEX: ADV successfully ended with a connection being created[0m

[0;33mW (4833928) BT_HCI: hcif disc complete: hdl 0x1, rsn 0x3e
[0m
[0;32mI (4833928) GATTC_GATTS_COEX: ESP_GAP_BLE_EXT_ADV_START_COMPLETE_EVT, status = 0[0m

[0;32mI (4957708) GATTC_GATTS_COEX: ESP_GAP_BLE_ADV_TERMINATED_
EVT, status = 0[0m
[0;32mI (4957708) GATTC_GATTS_COEX: ADV successfully ended with a connection being created[0m

[0;33mW (4957938) BT_HCI: hcif disc complete: hdl 0x1, rsn 0x3e
[0m
[0;32mI (4957938) GATTC_GATTS_COEX: ESP_GAP_BLE_EXT_ADV_START_COMPLETE_EVT, status = 0[0m

[0;32mI (4965898) GATTC_GATTS_C
OEX: ESP_GAP_BLE_ADV_TERMINATED_EVT, status = 0[0m
[0;32mI (4965898) GATTC_GATTS_COEX: ADV successfully ended with a connection being created[0m

[0;33mW (4966128) BT_HCI: hcif disc complete: hdl 0x1, rsn 0x3e
:[0m
[0;32mI (4966128) GATTC_GATTS_COEX: ESP_GAP_BLE_EXT_ADV_START_COMPLETE_EVT, status = 0[0m

[0;32mI (4978188) GATTC_GATTS_COEX: ESP_GAP_BLE_ADV_TERMINATED_EVT, status = 0[0m
[0;32mI (4978188) GATTC_GATTS_COEX: ADV successfully ended with a connection being created[0m

[0;33mW (4978418) BT_HCI: hcif disc complete: hdl 0x1, rsn 0x3e[0m
[0;32mI (4978418) GATTC_GATTS_COEX: ESP_GAP_BLE_EXT_ADV_S
TART_COMPLETE_EVT, status = 0[0m

[0;32mI (5002298) GATTC_GATTS_COEX: ESP_GAP_BLE_ADV_TERMINATED_EVT, status = 0[0m
[0;32mI (5002298) GATTC_GATTS_COEX: ADV successfully ended with a connection being created[0m

[0;33mW (5002528) BT_HCI: hcif disc complete: hdl 0x1, rsn 0x3e[0m
[0;32mI (5002528) GATTC_GATTS_COEX: ESP_GAP_BLE_EXT_ADV_START_COMPLETE_EVT, status = 0[0m

[0;32mI (5026208) GATTC_GATTS_COEX: ESP_GAP_BLE_ADV_TERMINATED_EVT, status = 0[0m
[0;32mI (5026208) GATTC_GATTS_COEX: ADV successfully ended with a connection being created[0m

[0;33mW (5026468) BT_HCI: hcif disc complete: hdl 0x1, rsn 0x3e[0m
[0;32mI (5026468) GATTC_GATTS_COEX: ESP_GAP_BLE_EXT_ADV_START_COMPLETE_EVT, status = 0[0m

[0;32mI (5172798) GATTC_GATTS_COEX: ESP_GAP_BLE_ADV_TERMINATED_EVT, status = 0[0m
[0;32mI (5172798) GATTC_GATTS_COEX: ADV successfully ended with a connection being created[0m

[0;33mW (5173068) BT_HCI: hcif d:isc complete: hdl 0x1, rsn 0x3e[0m
[0;32mI (5173068) GATTC_GATTS_COEX: ESP_GAP_BLE_EXT_ADV_START_COMPLETE_EVT, status = 0[0m

[0;32mI (5181088) GATTC_GATTS_COEX: ESP_GAP_BLE_ADV_TERMINATED_EVT, status = 0[0m
[0;32mI (5181088) GATTC_GATTS_COEX: ADV successfully ended with a connection being created[0m

[0;33mW (5181348) BT_HCI: hcif disc complete: hdl 0x1, rsn 0x3e[0m
[0;32mI (5181348) GATTC_GATTS_COEX: ESP_GAP_BLE_EXT_ADV_START_COMPLETE_EVT, status = 0[0m

[0;32mI (5238338) GATTC_GATTS_COEX: ESP_GAP_BLE_ADV_TERMINATED_EVT, status = 0[0m
[0;32mI (5238338) GATTC_GATTS_COEX: ADV successfully ended with a connection being created[0m

[0;33mW (5238578) BT_HCI: hcif :disc complete: hdl 0x1, rsn 0x3e[0m
[0;32mI (5238578) GATTC_GATTS_COEX: ESP_GAP_BLE_EXT_ADV_START_COMPLETE_EVT, status = 0[0m

[0;32mI (5285008) GATTC_GATTS_COEX: ESP_GAP_BLE_ADV_TERMINATED_EVT, status = 0[0m
[0;32mI (5285008) GATTC_GATTS_COEX: ADV su
ccessfully ended with a connection being created[0m

[0;33mW (5290188) BT_HCI: hcif disc complete: hdl 0x1, rsn 0x8[0m

zhp0406, 2024-05-06:
Note: When the connection is disconnected, both the gattc and the gatts report the disconnection event.

Please check whether re-broadcasting is successful after disconnection (in BLE50,esp_ble_gap_ext_adv_start can start broadcasting).

If you call the API to start the broadcast and the problem persists,
Please give me a demo to reproduce the problem, thank you.

zhp0406, 2024-05-06:
1-ESP_GAP_BLE_EXT_ADV_START_COMPLETE_EVT
    ESP_GAP_BLE_EXT_ADV_START_COMPLETE_EVT,                      /*!< when extended advertising start complete, the event comes */
    
    当 EXT ADV enable 后会上来该事件
 
 2-ESP_GAP_BLE_ADV_TERMINATED_EVT
     ESP_GAP_BLE_ADV_TERMINATED_EVT,                              /*!< when advertising terminate data complete, the event comes */
     当 EXT ADV timeout 时间(esp_ble_gap_ext_adv_start 中的 duration )到了或者连接建立,会上来该事件.

zhp0406, 2024-05-06:
"再次开启广播有没有作用":
你可以具体说一下吗,你调用 esp_ble_gap_ext_adv_start 并返回成功了吗?

zhp0406, 2024-05-06:
esp_ble_gap_ext_adv_start 
你设置了 duration 为 0，因此不存在 timeout 的情况，ESP_GAP_BLE_ADV_TERMINATED_EVT 说明有其他设备连接了
[image]

zhp0406, 2024-05-06:
另外，你使用的版本是多少？

monkey-qiu, 2024-05-07:
> 另外，你使用的版本是多少？

idf 5.1

monkey-qiu, 2024-05-07:
> "再次开启广播有没有作用": 你可以具体说一下吗,你调用 esp_ble_gap_ext_adv_start 并返回成功了吗?

广播返回是成功的，我想知道的是，如果出现这种情况，是不是连接被消耗完了，不能再次开启广播了？

zhp0406, 2024-05-07:
从你的 log 中可以看出， hcif disc complete: hdl 0x1, rsn 0x3e
下面是 spec 对 0x3e 的解释
[image]
0x3e 一般发生在连接没有建立的情况，并且 0x3e 多次上报，说明触发了底层的重连。
[image]

因此，我这面想到的一种情况是：
假设 device1 烧录了 GATTC_COEX example（附件中的 example），那么该 example 将会作为 client 连接 server （device2），此时连接事件会上来。此时， device1 还在作为 server 进行广播，并且广播的地址与发起连接时的地址不一样，如果 device2 在没有断开连接的情况又作为 client 连接了 device1，那么就会出现连接不上的情况。因为对于 device2 来说，不知道 device1 已经连接了，而 device1 直到与 device2 的连接还存在。这个时候 device2 就会重连，并且报错 0x3e。

你那面有可能发生这种情况吗？即两个设备已经有连接存在了，其中一个设备想再次与该设备建立一个连接。

zhp0406, 2024-05-07:
如果连接被消耗完了，不能再次开启广播了。这个时候你 enable ext adv 的时候会有报错的。从你的 log 中我没有找到相关的报错。

zhp0406, 2024-05-07:
如果你能够排除我说的情况（即两个设备已经有连接存在了，其中一个设备想再次与该设备建立一个连接），那么你可以给我一份更加完整的 log（从启动到出问题），让我进一步检查。

monkey-qiu, 2024-05-08:
> 从你的 log 中可以看出， hcif disc complete: hdl 0x1, rsn 0x3e 下面是 spec 对 0x3e 的解释 [...]

我这边使用环境是client是APP，设备端esp32作为server的，我们看能不能再次复现下， 0x3e是作为clinet时才会触发对不对？

zhp0406, 2024-05-08:
是的，0x3e 作为 client 才会触发。代表连接无法建立。一般出现在下面两种情况
1-server 端广播，client 端收到了 server 的广播，并发了连接请求，但是因为收包原因连接没有正常建立，就会出现 0x3e，并且会触发重连。
2-server 端广播，client 端收到了 server 的广播，并发了连接请求，但是 server 端已经和 client 有一个连接存在（该连接对于 server 来说是已知的，但是对于 client 来说是未知的设备），就会导致 client 出现 0x3e 并触发重连。

需要说明的是，如果 client 没有收到 server 的广播，是可能触发 0x3e 的，而是触发其他的错误码（具体来说：超时后触发0x100）

zhp0406, 2024-05-08:
你这里所说的不能被扫描到，是广播没有发出来,还是发出来了但是特定扫不到?你可以通过另一个设备来扫描试一下。
[image]

monkey-qiu, 2024-05-09:
> 你这里所说的不能被扫描到，是广播没有发出来,还是发出来了但是特定扫不到?你可以通过另一个设备来扫描试一下。 [...]

用的蓝牙调试工具，也扫描不到，那应该是广播没有发出来，准备在收到重连事件后，再主动触发一次广播看看

monkey-qiu, 2024-05-09:
> 是的，0x3e 作为 client 才会触发。代表连接无法建立。一般出现在下面两种情况 1-server 端广播，client 端收到了 server 的广播，并发了连接请求，但是因为收包原因连接没有正常建立，就会出现 0x3e，并且会触发重连。 这种情况在近距离情况（空旷环境 10 米内）一般不会出现，或者偶尔出现一次。 2-server 端广播，client 端收到了 server 的广播，并发了连接请求，但是 server 端已经和 client 有一个连接存在（该连接对于 server 来说是已知的，但是对于 client 来说是未知的设备），就会导致 client 出现 0x3e 并触发重连。
> 
> 需要说明的是，如果 client 没有收到 server 的广播，是可能触发 0x3e 的，而是触发其他的错误码（具体来说：超时后触发0x100）

和APP的同事沟通了，他们有设置快连接的，只要APP连接过设备（esp32）蓝牙，下次会自动连接，但是在处理业务的时候又会主动连接一次，和你提到的第二种类似

zhp0406, 2024-05-09:
@monkey-qiu 
好的，如果没有其他问题，该 github issue 可以先关闭、

closed 2024-12-17 by Alvin1Zhang