espressif/esp-idf #13720: In bt-spp example, how to type spp key xxxxxx?

zhou88mang, 2024-04-30:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

hello, i used esp-idf 5.2.1 to  run the bt-spp accept and bt-spp initiator demos, when the content shown in the image below appears, I am unable to input the spp key through the serial port. It seems that the console cannot obtain external input? Is this a bug or is there something wrong with my operation?
[image]

xiongweichao, 2024-05-06:
Hi @zhou88mang , May I ask whether you entered `spp key 718394;` ?

zhou88mang, 2024-05-06:
> Hi @zhou88mang , May I ask whether you entered `spp key 718394;` ?

yes，i wanted entered spp key 718394，but i was unable to input the content I want through the console

zhou88mang, 2024-05-07:
Any progress?

xiongweichao, 2024-05-07:
I tested it with no issues. May I ask if you have used other serial port tools to test it?
```
I (18083) SPP_INITIATOR_DEMO: ESP_BT_GAP_KEY_REQ_EVT Please enter passkey!
W (18093) SPP_INITIATOR_DEMO: To input the key, type `spp key xxxxxx;`
E (24023) CNSL: Command [spp key 331687;]
Input Paring Key: 331687
passkey:331687
I (24673) SPP_INITIATOR_DEMO: authentication success: Mi Phone10
I (24673) SPP_INITIATOR_DEMO: a4 4b d5 0d 80 98 
W (29013) BT_HCI: hci cmd send: disconnect: hdl 0x80, rsn:0x13
W (29093) BT_HCI: hcif disc complete: hdl 0x80, rsn 0x16
```

zhou88mang, 2024-05-08:
> yes，i wanted entered spp key 718394，but i was unable to input the content I want through the console

我使用了其他的串口工具但这个现象还是存在，你给的log我好像没看到你输入“spp key 331687”的操作，你能具体说说你测试时的详细步骤吗？

xiongweichao, 2024-05-08:
> 你给的log我好像没看到你输入“spp key 331687”的操作

是不会显示输入的字符串, 只有当输入`;` 之后才会有 `Command [spp key 331687;]` 的打印, 是以`;` 作为结束的判断

zhou88mang, 2024-05-08:
> > 你给的log我好像没看到你输入“spp key 331687”的操作
> 
> 是不会显示输入的字符串, 只有当输入`;` 之后才会有 `Command [spp key 331687;]` 的打印, 是以`;` 作为结束的判断

好的，加了；后可以了，谢谢

closed 2024-06-14 by Sherry616