espressif/esp-idf #13723: Nimble BLE Crash ASSERT - ESP-IDF 5.2.1

filzek, 2024-04-30:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.2.1-485-g9b3ae86417

### Espressif SoC revision.

esp32 v3

### Operating System used.

Windows

### How did you build your project?

Command line with Make

### If you are using Windows, please specify command line type.

None

### Development Kit.

esp32 v3 

### Power Supply used.

USB

### What is the expected behavior?

Nimble BLE running with a SPP, not connected to it, websocket running, than sudden crash with this error:

ASSERT_PARAM(1536 0), in rwble.c at line 384
Guru Meditation Error: Core  0 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x40090ff4: 0000f01d 00004136 f01d0000
0x40090ff4: btdm_sleep_check_duration at C:/Espressif/frameworks/esp-idf-v5.2/components/bt/controller/esp32/bt.c:919

Core  0 register dump:
PC      : 0x40090ffb  PS      : 0x00060c34  A0      : 0x8008be02  A1      : 0x3ffc2624
0x40090ffb: r_assert at C:/Espressif/frameworks/esp-idf-v5.2/components/bt/controller/esp32/bt.c:1926

A2      : 0x00000000  A3      : 0x00000600  A4      : 0x00000000  A5      : 0x3f43c735
A6      : 0x00000180  A7      : 0xfffffffc  A8      : 0x8000814b  A9      : 0x3ffc2594
A10     : 0x00000000  A11     : 0x3ffc25b7  A12     : 0x3ffc2563  A13     : 0x00000034
A14     : 0x00000000  A15     : 0x3ffc2568  SAR     : 0x00000004  EXCCAUSE: 0x00000000
EXCVADDR: 0x00000000  LBEG    : 0x400834fd  LEND    : 0x40083505  LCOUNT  : 0x00000027
0x400834fd: esp_timer_impl_get_counter_reg at C:/Espressif/frameworks/esp-idf-v5.2/components/esp_timer/src/esp_timer_impl_lac.c:118
0x40083505: esp_timer_impl_get_counter_reg at C:/Espressif/frameworks/esp-idf-v5.2/components/esp_timer/src/esp_timer_impl_lac.c:128

Backtrace: 0x40090ff8:0x3ffc2624 |<-CORRUPTED
0x40090ff8: r_assert at C:/Espressif/frameworks/esp-idf-v5.2/components/bt/controller/esp32/bt.c:1925

ELF file SHA256: e4d9f909a

Rebooting in  3 seconds...
Rebooting...

### What is the actual behavior?

Crashed

### Steps to reproduce.

Running normal until crash

### Debug Logs.

```plain
ASSERT_PARAM(1536 0), in rwble.c at line 384
Guru Meditation Error: Core  0 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x40090ff4: 0000f01d 00004136 f01d0000
0x40090ff4: btdm_sleep_check_duration at C:/Espressif/frameworks/esp-idf-v5.2/components/bt/controller/esp32/bt.c:919

Core  0 register dump:
PC      : 0x40090ffb  PS      : 0x00060c34  A0      : 0x8008be02  A1      : 0x3ffc2624
0x40090ffb: r_assert at C:/Espressif/frameworks/esp-idf-v5.2/components/bt/controller/esp32/bt.c:1926

A2      : 0x00000000  A3      : 0x00000600  A4      : 0x00000000  A5      : 0x3f43c735
A6      : 0x00000180  A7      : 0xfffffffc  A8      : 0x8000814b  A9      : 0x3ffc2594
A10     : 0x00000000  A11     : 0x3ffc25b7  A12     : 0x3ffc2563  A13     : 0x00000034
A14     : 0x00000000  A15     : 0x3ffc2568  SAR     : 0x00000004  EXCCAUSE: 0x00000000
EXCVADDR: 0x00000000  LBEG    : 0x400834fd  LEND    : 0x40083505  LCOUNT  : 0x00000027
0x400834fd: esp_timer_impl_get_counter_reg at C:/Espressif/frameworks/esp-idf-v5.2/components/esp_timer/src/esp_timer_impl_lac.c:118
0x40083505: esp_timer_impl_get_counter_reg at C:/Espressif/frameworks/esp-idf-v5.2/components/esp_timer/src/esp_timer_impl_lac.c:128

Backtrace: 0x40090ff8:0x3ffc2624 |<-CORRUPTED
0x40090ff8: r_assert at C:/Espressif/frameworks/esp-idf-v5.2/components/bt/controller/esp32/bt.c:1925
```

filzek, 2024-05-03:
Same happen another time with the sleep disable and also wifi to ps none, panic´ed

heap: string at ROData: 0x3f414648 PC: 0x401165ad

zhp0406, 2024-05-06:
The issue we have fixed in the latest master branch (fix commit), but the v5.2 branch is not synchronized yet, which will take some time.

You can use the latest master branch to test if the problem still exists, and please let me know if the problem still exists

zhp0406, 2024-05-06:
please replace components/bt/controller/lib_esp32/esp32/libbtdm_app.a 
[b2bdfeb.zip](https://github.com/espressif/esp-idf/files/15215589/b2bdfeb.zip)

filzek, 2024-05-09:
Hi @zhp0406  
> please replace components/bt/controller/lib_esp32/esp32/libbtdm_app.a [b2bdfeb.zip](https://github.com/espressif/esp-idf/files/15215589/b2bdfeb.zip)

I did replace it but now the problem occurs here not everytime, but occurs almost once a day.

ASSERT_PARAM(1536 0), in rwble.c at line 390
Guru Meditation Error: Core  0 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x4009103c: 0000f01d 00004136 f01d0000
0x4009103c: btdm_sleep_check_duration at C:/Espressif/frameworks/esp-idf-v5.2/components/bt/controller/esp32/bt.c:919

Core  0 register dump:
PC      : 0x40091043  PS      : 0x00060034  A0      : 0x8008bae6  A1      : 0x3ffc24b4
0x40091043: r_assert at C:/Espressif/frameworks/esp-idf-v5.2/components/bt/controller/esp32/bt.c:1926

A2      : 0x00000000  A3      : 0x00000600  A4      : 0x00000000  A5      : 0x3f41fbbc
A6      : 0x00000186  A7      : 0xfffffffc  A8      : 0x8000814b  A9      : 0x3ffc2424
A10     : 0x00000000  A11     : 0x3ffc2447  A12     : 0x3ffc23f3  A13     : 0x00000030
A14     : 0x00000000  A15     : 0x3ffc23f8  SAR     : 0x00000004  EXCCAUSE: 0x00000000
EXCVADDR: 0x00000000  LBEG    : 0x00000000  LEND    : 0x00000000  LCOUNT  : 0x00000000

Backtrace: 0x40091040:0x3ffc24b4 |<-CORRUPTED
0x40091040: r_assert at C:/Espressif/frameworks/esp-idf-v5.2/components/bt/controller/esp32/bt.c:1925

ELF file SHA256: 79744ba5e

filzek, 2024-05-09:
error in

/**
 * This function re-write controller's function,
 * As coredump can not show parameters in function which is in a .a file.
 *
 * After coredump fixing this issue, just delete this function.
 */
void IRAM_ATTR r_assert(const char *condition, int param0, int param1, const char *file, int line)
{
    __asm__ __volatile__("ill\n");
}

Even coredump is disable the problem happen anyway.

Also coredump is trashed in 5.2.1 in commit 9b3ae86417f5b812650e3382ad14d4139c563c60 (HEAD -> release/v5.2, origin/release/v5.2)
Merge: eb7c381761 373e585bb7
Author: morris <maoshengrong@espressif.com>
Date:   Thu Apr 25 16:31:22 2024 +0800

    Merge branch 'bugfix/fix_gpio_etm_multi_task_v5.2' into 'release/v5.2'

    fix(gpio_etm): allow one GPIO binds to multiple ETM tasks (v5.2)

    See merge request espressif/esp-idf!30456

zhp0406, 2024-05-09:
ok,thanks for your testing,
please replace components/bt/controller/lib_esp32/esp32/libbtdm_app.a
[4fb2129.zip](https://github.com/espressif/esp-idf/files/15259866/4fb2129.zip)

filzek, 2024-05-12:
still there BT crashes
Software reset due to exception/panic (ESP_RST_PANIC)
Reset (software or hardware) due to interrupt watchdog (ESP_RST_INT_WDT)

filzek, 2024-05-12:
it print this
ELx200
ELx200
ELx200
ELx200
ELx200
ELx200
ELx200
ELx200
ELx200
ELx200
ELx200
ELx200
ELx200
ELx200
ELx200
ELx200
ELx200
ELx600
ELx600
ELx600
ELx600
ELx600
ELx600
ELx600
ELx600
ELx600
ELx600
ELx600
ELx600
ELx600
ELx600
ELx600
ELx600
ELx600
ELx600
ELx600
ELx600
ELx600
ELx600
ELx600
ELx600
ELx600
ELx600
ELx600
ELx600
ELx600
ELx600
ELx600
ELx600
ELx600
ELx600
ELx600
ELx600
ELx600
ELx600
ELx600
ELx600
ELx600
ELx600
ELx600
ELx600
ELx600
ELx600
ELx600
ELx600
ELx600
ELx600
ELx600
and crash
Reset (software or hardware) due to interrupt watchdog (ESP_RST_INT_WDT)

filzek, 2024-05-13:
[4fb2129.zip](https://github.com/espressif/esp-idf/files/15259866/4fb2129.zip) this lib also corrupted the wifi breaking the TLS/Wifi connection, corrumpting the wifi task.

Now we have build and it is running the new firmware with [257d8b4.zip](https://github.com/espressif/esp-idf/files/15291896/257d8b4.zip) with DEBUG logs turned on.

filzek, 2024-05-13:
> Thank you for your feedback. You can use the following library to test and provide me with the complete log: make sure： BTDM_INIT: BT controller compile version [257d8b4] [257d8b4.zip](https://github.com/espressif/esp-idf/files/15291896/257d8b4.zip)
> 
> ps: If the new library still can't solve your problem, could you please provide more detailed information about the scenario where the issue occurs and the steps to reproduce it? Thank you.

@zhp0406 still broken

D (4734845) HTTP_CLIENT: Write header[5]: POST /prod/node/data?node_id=4C6C13005A78 HTTP/1.1

User-Agent: ESP32 HTTP Client/1.0

Host: client.insights.espressif.com

Authorization: eyJhbGciOiJSUzI1NiIsInR5cCI6IkpXVCJ9.eyJ1c2VyIjoiODRmZTBkZGUtOGZkMy00ZjAxLWFjZTctMjI5NmEyOTU2OGZhIiwiaXNzIjoiZTMyMmI1OWMtNjNjYy00ZTQwLThlYTItNGU3NzY2NTQ1Y2NhIiwic3ViIjoiNGFlMGU2MDYtMzQ1Ny00NjRiLTk2N2QtMzFkZDQ3MTM1ODk0IiwiZXhwIjoyMDI5OTg3NjgzLCJpYXQiOjE3MTQ2Mjc2ODN9.fgOhVzO-Qv8TKleBFFGgRvDTb_hs4IfoYgNbC47S9ZSggJdq1dQL3ckA72BmtSEPMgjJXUeW2suNkR6xJM5QO3_y0llnxewMwD_bfn6SNXDcUIEl6cL3FMZDY4sVI9Q_oTZweO4LWI3CxBvj7Xw9FhEf4hsJ_EEgPztHn4H03AVnASsLZyItiptvrKXpZU0Pr83KLe_bWFF3c2XA9Rcd5iL8VCF8K5Xyxd-hUA-FWSd6KH-egh0frTLFvfhALlNTNzKgHibIH3UOTVB63f3yeOFPBbaAaAUQ08Utm7dc2qaP6T1BpH-D8AKzL9QSU2F4LRvfE9tx8sD3bRKLudrjdQ

Content-type: application/octet-stream

Content-Length: 221

[0m
D (4734835) NimBLE: subevent: 0x2[0m
D (4735005) NimBLE: 
[0m
D (4734995) tport_https: HTTP_EVENT_HEADER_SENT[0m
D (4735005) NimBLE: ble_hs_event_rx_hci_ev; opcode=0x3e [0m
D (4735025) NimBLE: subevent: 0x2[0m
D (4735035) NimBLE: 
[0m
D (4735035) NimBLE: ble_hs_event_rx_hci_ev; opcode=0x3e [0m
D (4735055) NimBLE: subevent: 0x2[0m
D (4735055) NimBLE: 
[0m
D (4735065) NimBLE: ble_hs_event_rx_hci_ev; opcode=0x3e [0m
D (4735075) NimBLE: subevent: 0x2[0m
D (4735075) NimBLE: 
[0m
D (4735085) NimBLE: ble_hs_event_rx_hci_ev; opcode=0x3e [0m
D (4735095) NimBLE: subevent: 0x2[0m
D (4735105) NimBLE: 
[0m
D (4735105) NimBLE: ble_hs_event_rx_hci_ev; opcode=0x3e [0m
D (4735115) NimBLE: subevent: 0x2[0m
D (4735125) NimBLE: 
[0m
D (4735125) NimBLE: ble_hs_event_rx_hci_ev; opcode=0x3e [0m
D (4735135) NimBLE: subevent: 0x2[0m
D (4735145) NimBLE: 
[0m
D (4735155) NimBLE: ble_hs_event_rx_hci_ev; opcode=0x3e [0m
D (4735165) NimBLE: subevent: 0x2[0m
D (4735165) NimBLE: 
[0m
D (4734845) event: no handlers have been registered for event ESP_HTTP_CLIENT_EVENT:1 posted to loop 0x3ffbdadc[0m
D (4735175) NimBLE: ble_hs_event_rx_hci_ev; opcode=0x3e [0m
D (4735205) NimBLE: subevent: 0x2[0m
D (4735215) NimBLE: 
[0m
D (4735215) event: no handlers have been registered for event ESP_HTTP_CLIENT_EVENT:2 posted to loop 0x3ffbdadc[0m
D (4735275) NimBLE: ble_hs_event_rx_hci_ev; opcode=0x3e [0m
D (4735275) NimBLE: subevent: 0x2[0m
D (4735275) NimBLE: 
[0m
D (4735325) NimBLE: ble_hs_event_rx_hci_ev; opcode=0x3e [0m
D (4735325) NimBLE: subevent: 0x2[0m
D (4735325) NimBLE: 
[0m
D (4735365) NimBLE: ble_hs_event_rx_hci_ev; opcode=0x3e [0m
D (4735365) NimBLE: subevent: 0x2[0m
D (4735365) NimBLE: 
[0m
D (4735445) NimBLE: ble_hs_event_rx_hci_ev; opcode=0x3e [0m
[... 4647 lines cut ...]
ELx00000600
ELx00000600
ELx00000600
ELx00000600
ELx00000600
ELx00000600
ELx00000600
ELx00000600
ELx00000600
ELx00000600
ELx00000600
ELx00000600
ELx00000600
ELx00000600
ELx00000600
ELx00000600
ELx00000600
ELx00000600
ELx00000600
ELx00000600
ELx00000600
ELx00000600
ELx00000600
ELx00000600
ELx00000600

filzek, 2024-05-13:
@zhp0406  

Currently every 3 minutes we ran a insights post of memory and heap, also do a simple NIMBLE scan for all listened BLE for 30 seconds, just it, randomly it crashes wih the result as show below.

filzek, 2024-05-13:
I am at:

commit f8319f08cca75236c29eecfce01886a671f4fc17 (HEAD -> release/v5.2, origin/release/v5.2)
Merge: 83a73e994d dcf6b54e94
Author: Marius Vikhammer <marius.vikhammer@espressif.com>
Date:   Thu May 9 15:41:59 2024 +0800

    Merge branch 'fix/reduce-binary-size_v5.2' into 'release/v5.2'

    Reduce binary size (v5.2)

    See merge request espressif/esp-idf!30656

zhp0406, 2024-05-14:
@filzek 
Thank you for your feedback. We still need to do further debugging on our side. We need to reproduce the issue locally. Could you provide us with an example to replicate the problem?

filzek, 2024-05-14:
@zhp0406  the BLE code is:
BLE INIT:

```
if(debug_nimble){
          printf("INITIATING NIMBLE FRAMEWORK\n");
        }
        if (!nimble_started){
            ret = nimble_port_init();
            if (ret != ESP_OK) {
                printf("ERROR nimble_port_init [%s] [%s] [%d]\n",__FILE__,__func__, __LINE__ );
                if (debug_string_enable){
                    if (debug_string_ble) {
                      debug_add("%s %s initialize controller failed -err[%d][%s] WILL RESTART\n",__FILE__, __func__, ret, esp_err_to_name(ret));
                      wait(5000);
                    }
                }
                // esp_restart();
                return;
            }
            else {
              nimble_started = 1;
              if (debug_ble)
                  printf("BLE INICIALIZADO PERFEITAMENTE\n");
            }

        }

        esp_ble_tx_power_set(ESP_BLE_PWR_TYPE_ADV, ESP_PWR_LVL_P9);
        esp_ble_tx_power_set(ESP_BLE_PWR_TYPE_DEFAULT, ESP_PWR_LVL_P9);

        /* Initialize connection_handle array */
        for (int i = 0; i <= CONFIG_BT_NIMBLE_MAX_CONNECTIONS; i++) {
            conn_handle_subs[i] = false;
        }

        // /* Initialize uart driver and start uart task */
        // ble_spp_uart_init();

        ble_hs_cfg.reset_cb = ble_spp_server_on_reset;
        ble_hs_cfg.sync_cb = ble_spp_server_on_sync;
        ble_hs_cfg.gatts_register_cb = gatt_svr_register_cb;
        ble_hs_cfg.store_status_cb = ble_store_util_status_rr;

        ble_hs_cfg.sm_io_cap = 3;
    #ifdef CONFIG_EXAMPLE_BONDING
        ble_hs_cfg.sm_bonding = 1;
    #endif
    #ifdef CONFIG_EXAMPLE_MITM
        ble_hs_cfg.sm_mitm = 1;
    #endif
    #ifdef CONFIG_EXAMPLE_USE_SC
        ble_hs_cfg.sm_sc = 1;
    #else
        ble_hs_cfg.sm_sc = 0;
    #endif
    #ifdef CONFIG_EXAMPLE_BONDING
        ble_hs_cfg.sm_our_key_dist = 1;
        ble_hs_cfg.sm_their_key_dist = 1;
    #endif

        /* Register custom service */
        ble_svc_gap_init();
        ble_svc_gatt_init();

        ret = ble_gatts_count_cfg(new_ble_svc_gatt_defs);

        if (ret != 0) {
            printf("ERROR ble_gatts_count_cfg\n");
        }
[... 186 lines cut ...]
        strncpy(ble_devices[ble_device_count].name, name, MAX_DEVICE_NAME_LEN);
        ble_devices[ble_device_count].avg_rssi = rssi;
        ble_devices[ble_device_count].min_rssi = rssi;
        ble_devices[ble_device_count].max_rssi = rssi;
        ble_devices[ble_device_count].rssi_count = 1;
        ble_devices[ble_device_count].last_seen = now;
        ble_devices[ble_device_count].has_change = true;
        ble_device_count++;
    } else {
        // Handle the case where the device list is full
        printf("Device list is full. Unable to add new device.\n");
    }
}

```

called every 3 minutes:

ble_scan();

The other active taks in esp32 system is:
Zero crossing with timer dispatch to arm a triac with the correct power / dimer.
Websocket connection, with queue for receive and send data, so, once the ble list is finished it creates a packet and send to the websocket and clear the ble list sent

thats all the code.

filzek, 2024-05-14:
can you disclose how the nimble could lock it into ELx00000600 and also corrupt wifi task?

filzek, 2024-05-17:
update the 5.2 to the latest commit at
commit 70fc0e3037b240aacaf665d2c6f622ad54b510c0 (HEAD -> release/v5.2, origin/release/v5.2)
Merge: 3b15317109 2fc047d61d
Author: Jiang Jiang Jian <jack@espressif.com>
Date:   Mon May 13 22:13:32 2024 +0800

    Merge branch 'fix/some_fix_for_openthread_backport_5_2' into 'release/v5.2'

    Fix/some fix for openthread (backport v5.2)

    See merge request espressif/esp-idf!30796

filzek, 2024-05-19:
@zhp0406 still problem, 

Guru Meditation Error: Core  0 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x4009035c: 0000f01d 00004136 f01d0000
0x4009035c: btdm_sleep_check_duration at C:/Espressif/frameworks/esp-idf-v5.2/components/bt/controller/esp32/bt.c:919

Core  0 register dump:
PC      : 0x40090363  PS      : 0x00060d34  A0      : 0x8008adfa  A1      : 0x3ffc2554
0x40090363: r_assert at C:/Espressif/frameworks/esp-idf-v5.2/components/bt/controller/esp32/bt.c:1926

A2      : 0x00000000  A3      : 0x00000600  A4      : 0x00000000  A5      : 0x3f418669
A6      : 0x00000186  A7      : 0xfffffffc  A8      : 0x8000814b  A9      : 0x3ffc24c4
A10     : 0x00000000  A11     : 0x3ffc24e7  A12     : 0x3ffc2493  A13     : 0x00000030
A14     : 0x00000000  A15     : 0x3ffc2498  SAR     : 0x00000004  EXCCAUSE: 0x00000000
EXCVADDR: 0x00000000  LBEG    : 0x00000000  LEND    : 0x00000000  LCOUNT  : 0x00000000

Backtrace: 0x40090360:0x3ffc2554 |<-CORRUPTED
0x40090360: r_assert at C:/Espressif/frameworks/esp-idf-v5.2/components/bt/controller/esp32/bt.c:1925

filzek, 2024-05-21:
Hi @zhp0406, we can confirm that the problem are direct related to the Bluetooh/BLE framework.

If we don't call the NimBLE initialization (keeping it disabled in the software), the problem does not occur. However, if we enable NimBLE without establishing any connection or invoking any BLE-related functions, the problem arises.

zhp0406, 2024-05-22:
@filzek 
Could you test several other versions (for example, v5.2.0 or even earlier) to see if the issue persists? If the problem appeared in the recent versions, could you use the git bisect command to help identify the first commit where the issue occurred? I would greatly appreciate it.

filzek, 2024-05-23:
> @filzek Could you test several other versions (for example, v5.2.0 or even earlier) to see if the issue persists? If the problem appeared in the recent versions, could you use the git bisect command to help identify the first commit where the issue occurred? I would greatly appreciate it.

Hi @zhp0406,

we cant as even from the 5.2.0 the bug happen all the time, all version, until now no 5.2 can still without crash the ble with panic, wdt or a hard crash, major time the ble crash without any terminal output at all. So, all version are corrupted somehow and with problem.

filzek, 2024-05-23:
@zhp0406 

The problem doesn't happen at all with the old framework version 4.4.4. However, it does occur with the new versions 5.2, 5.2.1, and the latest git commit as described above.

One interesting thing we found out today is that if we enable verbose logging, the problem doesn't occur.

So, the problem are related to the BLE somehow creating some event that disturb the system into a crash state, sometime panic or wdt as ESP_RST_INT_WDT Reset (software or hardware) due to interrupt watchdog, so, we cant let the log run for release mode, and the BLE/Bluetooth driver doesnt show exact where the crash happen, maybe you could build a version with a __func__ and __LINE__ call to print there the problem calls the reset, is it possible?

zhp0406, 2024-05-24:
It seems that this issue was introduced recently, as v4.4.4 is quite an old version. Could you use bisect to find the first commit where the issue occurs? Additionally, could you provide a complete example to reproduce this issue?

filzek, 2024-06-05:
> Thank you for your feedback. You can use the following library to test and provide me with the complete log: make sure： BTDM_INIT: BT controller compile version [257d8b4] [...]

@zhp0406 the 257d8b4 also corrupts and prints the lines 

ELx00000200 
or
ELx00000600

filzek, 2024-06-05:
> Based on your description, the issue was introduced by commit [9f4f8e2](https://github.com/espressif/esp-idf/commit/9f4f8e24f2235098a022e57927fd1688f62578dc). However, this commit was merged into the IDF recently(Date: Fri May 24 21:20:22 2024 +0800), while your issue was reported a month ago(2024.5.1). Are you sure this commit is the cause?

not sure what is the commit that creates the problem, but it happens since the release 5.2, we will remove the lib and pull the lib from the repository as of 5.2.2. and test now.

zhp0406, 2024-06-06:
@filzek 
Thank you for your additional feedback.
Based on the current prints we have, **we are unable to determine the root cause of the issue at this moment**.
To further debug, we need you to provide a demo that reproduces the issue **or** identify the first commit where the problem appears.
We look forward to your further feedback.

filzek, 2024-06-06:
@zhp0406 

We have enable more debugs info and more protection on the stack for heap corruption, below are more info about the problem, it happen **D (ASSERT_PARAM(1536 0), in rwble.c at line 390**

It has doing a BLE Scan when the problem happen, we are using SPIRAM all code in SPIRAM LWIP/WiFi/NIMBLE

**Info:**
I (3118) mac_address_set: Base MAC Address read from EFUSE BLK3
I (3248) wifi:wifi driver task: 3ffd7470, prio:23, stack:6656, core=0
I (3248) wifi:wifi firmware version: 3e0076f
I (3258) wifi:wifi certification version: v7.0
I (3268) wifi:config NVS flash: disabled
I (3278) wifi:config nano formating: disabled
I (3278) wifi:Init data frame dynamic rx buffer num: 32
I (3288) wifi:Init static rx mgmt buffer num: 5
I (3298) wifi:Init management short buffer num: 32
I (3308) wifi:Init static tx buffer num: 16
I (3318) wifi:Init tx cache buffer num: 32
I (3328) wifi:Init static rx buffer size: 1600
I (3338) wifi:Init static rx buffer num: 10
I (3338) wifi:Init dynamic rx buffer num: 32
I (3408) wifi_init: rx ba win: 6
I (3418) wifi_init: tcpip mbox: 32
I (3428) wifi_init: udp mbox: 6
I (3428) wifi_init: tcp mbox: 6
I (3438) wifi_init: tcp tx win: 5760
I (3448) wifi_init: tcp rx win: 5760
I (3458) wifi_init: tcp mss: 1440
I (3468) wifi_init: WiFi/LWIP prefer SPIRAM
D (3478) adc_share_hw_ctrl: Wi-Fi takes adc2 lock.
I (3488) phy_init: phy_version 4791,2c4672b,Dec 20 2023,16:06:06
D (11528) BTDM_INIT: Release DRAM [0x3ffb2730] - [0x3ffb6388]
I (11528) BTDM_INIT: BT contD (11528) intr_alloc: Connected src -5 to int 29 (cpu 0)
Controller compile version [4012cfb]

**Commit build**

commit 9cf323d4c686883f3260c9f73edae9ce47ac81d6 (HEAD -> release/v5.2, origin/release/v5.2)
Merge: 8cd37950c7 f47a4796e5
Author: Jiang Jiang Jian <jack@espressif.com>
Date:   Fri May 31 17:46:43 2024 +0800

    Merge branch 'bugfix/add_deprecated_warning_for_wpa3_ext_v5.2' into 'release/v5.2'

    fix(wifi): Add back `WIFI_AUTH_WPA3_EXT_PSK` and `WIFI_AUTH_WPA3_EXT_PSK_MIXED_MODE` for v5.x (Backport v5.2)

    See merge request espressif/esp-idf!31177

**the debug problem below**

I (3279349) NimBLE: GAP procedure initiated: discovery;
I (3279349) NimBLE: own_addr_type=0 filter_policy=0 passive=0 limited=0 filter_duplicates=0
I (3279369) NimBLE: duration=30000ms
I (3279369) NimBLE:

D (3279369) NimBLE: ble_hs_hci_cmd_send: ogf=0x08 ocf=0x000b len=7

D (3279389) NimBLE: 0x0b
D (3279389) NimBLE: 0x20
D (3279399) NimBLE: 0x07
D (3279419) NimBLE: 0x01
D (3279419) NimBLE: 0x80
D (3279419) NimBLE: 0x00
D (3279419) NimBLE: 0x40
D (3279439) NimBLE: 0x00
D (3279439) NimBLE: 0x00
D (3279439) NimBLE: 0x00
D (3279439) NimBLE:

D (3279459) NimBLE: ble_hs_hci_cmd_send: ogf=0x08 ocf=0x000c len=2
[... 1819 lines cut ...]

D (3290719) NimBLE: ble_hs_event_rx_hci_ev; opcode=0x3e
D (3290739) NimBLE: subevent: 0x2
D (3290739) NimBLE:

D (3290739) NimBLE: ble_hs_event_rx_hci_ev; opcode=0x3e
D (ASSERT_PARAM(1536 0), in rwble.c at line 390
Guru Meditation Error: Core  0 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x40092650: f01d020c 00004136 f01d0000
0x40092650: btdm_sleep_check_duration at C:/Espressif/frameworks/esp-idf-v5.2/components/bt/controller/esp32/bt.c:914

Core  0 register dump:
PC      : 0x40092657  PS      : 0x00060f34  A0      : 0x8008d0c2  A1      : 0x3ffc2664
0x40092657: r_assert at C:/Espressif/frameworks/esp-idf-v5.2/components/bt/controller/esp32/bt.c:1926

A2      : 0x00000000  A3      : 0x00000600  A4      : 0x00000000  A5      : 0x3f459ce6
A6      : 0x00000186  A7      : 0xfffffffc  A8      : 0x8000814b  A9      : 0x3ffc25d4
A10     : 0x00000000  A11     : 0x3ffc25f7  A12     : 0x3ffc25a3  A13     : 0x00000030
A14     : 0x00000000  A15     : 0x3ffc25a8  SAR     : 0x00000004  EXCCAUSE: 0x00000000
EXCVADDR: 0x00000000  LBEG    : 0x00000000  LEND    : 0x00000000  LCOUNT  : 0x00000000

Backtrace: 0x40092654:0x3ffc2664 |<-CORRUPTED
0x40092654: r_assert at C:/Espressif/frameworks/esp-idf-v5.2/components/bt/controller/esp32/bt.c:1925

ELF file SHA256: a989b1a14

filzek, 2024-06-06:
@jack0c and @zhp0406 the last commit still with assert problems in BLE

**Updated to the new commit as of:**

commit cc9edf20859e13d3c10defaef02f77dd7080093a (HEAD -> release/v5.2, origin/release/v5.2)
Merge: f47e28c2c9 db8d2bc27f
Author: Roland Dobai <roland@espressif.com>
Date:   Thu Jun 6 02:18:31 2024 +0800

    Merge branch 'set_gdb_remotetimeout_v5.2' into 'release/v5.2'

    tools(gdbinit): set remote timeout for the gdb connection (v5.2)

    See merge request espressif/esp-idf!30805

**Crash info**

D (1321528) NimBLE: subeventASSERT_PARAM(1536 0), in rwble.c at line 390
Guru Meditation Error: Core  0 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x40092708: f01d020c 00004136 f01d0000
0x40092708: btdm_sleep_check_duration at C:/Espressif/frameworks/esp-idf-v5.2/components/bt/controller/esp32/bt.c:914

Core  0 register dump:
PC      : 0x4009270f  PS      : 0x00060434  A0      : 0x8008d142  A1      : 0x3ffc2664
0x4009270f: r_assert at C:/Espressif/frameworks/esp-idf-v5.2/components/bt/controller/esp32/bt.c:1926

A2      : 0x00000000  A3      : 0x00000600  A4      : 0x00000000  A5      : 0x3f459e06
A6      : 0x00000186  A7      : 0xfffffffc  A8      : 0x8000814b  A9      : 0x3ffc25d4
A10     : 0x00000000  A11     : 0x3ffc25f7  A12     : 0x3ffc25a3  A13     : 0x00000030
A14     : 0x00000000  A15     : 0x3ffc25a8  SAR     : 0x00000004  EXCCAUSE: 0x00000000
EXCVADDR: 0x00000000  LBEG    : 0x4000c46c  LEND    : 0x4000c477  LCOUNT  : 0x00000000
0x4000c46c: memset in ROM
0x4000c477: memset in ROM

Backtrace: 0x4009270c:0x3ffc2664 |<-CORRUPTED
0x4009270c: r_assert at C:/Espressif/frameworks/esp-idf-v5.2/components/bt/controller/esp32/bt.c:1925

tomasznowik, 2024-06-18:
I see the same issue with Nimble on esp-idf 5.2.2. I was testing at the same time v5.1.4 and v5.1.2 and there was no such issue on these versions. I logged 8 Guru Meditation Error during 10 days on 4 devices. Crash happens when device is not heavily used - just scanning BLE and sending these data through HTTP. 
```
dev4/2024-06-09T16:59:38.665099Z I 126155295 | scan             Sent BLE data to cloud
dev4/2024-06-09T16:59:39.198184Z W None |ASSERT_PARAM(576 0), in rwble.c at line 390
dev4/2024-06-09T16:59:39.212655Z W None |Guru Meditation Error: Core  0 panic'ed (IllegalInstruction). Exception was unhandled.
dev4/2024-06-09T16:59:39.214291Z W None |Memory dump at 0x4008d9dc: 00f01d12 00004136 f01d0000
dev4/2024-06-09T16:59:39.215273Z W None |Core  0 register dump:
dev4/2024-06-09T16:59:39.220456Z W None |PC      : 0x4008d9e3  PS      : 0x00060e34  A0      : 0x80088842  A1      : 0x3ffc2130  
dev4/2024-06-09T16:59:39.228729Z W None |A2      : 0x00000000  A3      : 0x00000240  A4      : 0x00000000  A5      : 0x3f4216b0  
dev4/2024-06-09T16:59:39.236838Z W None |A6      : 0x00000186  A7      : 0xfffffffc  A8      : 0x8000814b  A9      : 0x3ffc20a0  
dev4/2024-06-09T16:59:39.242983Z W None |A10     : 0x00000000  A11     : 0x3ffc20c3  A12     : 0x3ffc206f  A13     : 0x00000030  
dev4/2024-06-09T16:59:39.251141Z W None |A14     : 0x00000000  A15     : 0x3ffc2074  SAR     : 0x00000004  EXCCAUSE: 0x00000000  
dev4/2024-06-09T16:59:39.259267Z W None |EXCVADDR: 0x00000000  LBEG    : 0x00000000  LEND    : 0x00000000  LCOUNT  : 0x00000000  
dev4/2024-06-09T16:59:39.275825Z W None |Backtrace: 0x4008d9e0:0x3ffc2130 0x4008883f:0x3ffc2150 0x4008d099:0x3ffc2170 0x4008d68f:0x3ffc2190 0x40082cfa:0x3ffc21b0 0x40082b65:0x3ffc21d0 0x40082b47:0x00000000 |<-CORRUPTED
dev4/2024-06-09T16:59:39.278956Z W None |ELF file SHA256: 864e5fdcc73d910e
dev4/2024-06-09T16:59:39.279782Z W None |Rebooting...
dev4/2024-06-09T16:59:39.282574Z W None |ets Jul 29 2019 12:21:46
--
dev4/2024-06-11T05:45:44.903686Z I 66655834 | scan             Sent BLE data to cloud
dev4/2024-06-11T05:45:46.770406Z W None |ASSERT_PARAM(1536 0), in rwble.c at line 390
dev4/2024-06-11T05:45:46.778610Z W None |Guru Meditation Error: Core  0 panic'ed (IllegalInstruction). Exception was unhandled.
dev4/2024-06-11T05:45:46.781842Z W None |Memory dump at 0x4008d9dc: 00f01d12 00004136 f01d0000
dev4/2024-06-11T05:45:46.784534Z W None |Core  0 register dump:
dev4/2024-06-11T05:45:46.792672Z W None |PC      : 0x4008d9e3  PS      : 0x00060a34  A0      : 0x80088842  A1      : 0x3ffc2130  
dev4/2024-06-11T05:45:46.800904Z W None |A2      : 0x00000000  A3      : 0x00000600  A4      : 0x00000000  A5      : 0x3f4216b0  
dev4/2024-06-11T05:45:46.809052Z W None |A6      : 0x00000186  A7      : 0xfffffffc  A8      : 0x8000814b  A9      : 0x3ffc20a0  
dev4/2024-06-11T05:45:46.815196Z W None |A10     : 0x00000000  A11     : 0x3ffc20c3  A12     : 0x3ffc206f  A13     : 0x00000030  
dev4/2024-06-11T05:45:46.823458Z W None |A14     : 0x00000000  A15     : 0x3ffc2074  SAR     : 0x00000004  EXCCAUSE: 0x00000000  
dev4/2024-06-11T05:45:46.831707Z W None |EXCVADDR: 0x00000000  LBEG    : 0x00000000  LEND    : 0x00000000  LCOUNT  : 0x00000000  
dev4/2024-06-11T05:45:46.848156Z W None |Backtrace: 0x4008d9e0:0x3ffc2130 0x4008883f:0x3ffc2150 0x4008d099:0x3ffc2170 0x4008d68f:0x3ffc2190 0x40082cfa:0x3ffc21b0 0x40082b65:0x3ffc21d0 0x40082b47:0x00000000 |<-CORRUPTED
dev4/2024-06-11T05:45:46.851244Z W None |ELF file SHA256: 864e5fdcc73d910e
dev4/2024-06-11T05:45:46.852035Z W None |Rebooting...
--
dev4/2024-06-13T13:40:44.037674Z I 104215344 | scan             Sent BLE data to cloud
dev4/2024-06-13T13:40:46.311709Z W None |ASSERT_PARAM(1536 0), in rwble.c at line 390
dev4/2024-06-13T13:40:46.320011Z W None |Guru Meditation Error: Core  0 panic'ed (IllegalInstruction). Exception was unhandled.
dev4/2024-06-13T13:40:46.322974Z W None |Memory dump at 0x4008d9dc: 00f01d12 00004136 f01d0000
dev4/2024-06-13T13:40:46.325669Z W None |Core  0 register dump:
dev4/2024-06-13T13:40:46.333838Z W None |PC      : 0x4008d9e3  PS      : 0x00060634  A0      : 0x80088842  A1      : 0x3ffc2130  
dev4/2024-06-13T13:40:46.342138Z W None |A2      : 0x00000000  A3      : 0x00000600  A4      : 0x00000000  A5      : 0x3f4216b0  
dev4/2024-06-13T13:40:46.350487Z W None |A6      : 0x00000186  A7      : 0xfffffffc  A8      : 0x8000814b  A9      : 0x3ffc20a0  
dev4/2024-06-13T13:40:46.356227Z W None |A10     : 0x00000000  A11     : 0x3ffc20c3  A12     : 0x3ffc206f  A13     : 0x00000030  
dev4/2024-06-13T13:40:46.364545Z W None |A14     : 0x00000000  A15     : 0x3ffc2074  SAR     : 0x00000004  EXCCAUSE: 0x00000000  
dev4/2024-06-13T13:40:46.372867Z W None |EXCVADDR: 0x00000000  LBEG    : 0x00000000  LEND    : 0x00000000  LCOUNT  : 0x00000000  
dev4/2024-06-13T13:40:46.389422Z W None |Backtrace: 0x4008d9e0:0x3ffc2130 0x4008883f:0x3ffc2150 0x4008d099:0x3ffc2170 0x4008d68f:0x3ffc2190 0x40082cfa:0x3ffc21b0 0x40082b65:0x3ffc21d0 0x40082b47:0x00000000 |<-CORRUPTED
dev4/2024-06-13T13:40:46.392274Z W None |ELF file SHA256: 864e5fdcc73d910e
dev4/2024-06-13T13:40:46.392728Z W None |Rebooting...
--
dev2/2024-06-10T19:56:36.747647Z I 223114212 | scan             Sent BLE data to cloud
dev2/2024-06-10T19:56:38.202299Z W None |ASSERT_PARAM(1536 0), in rwble.c at line 390
dev2/2024-06-10T19:56:38.210511Z W None |Guru Meditation Error: Core  0 panic'ed (IllegalInstruction). Exception was unhandled.
[... cut ...]
dev1/2024-06-12T22:01:35.311120Z W None |A14     : 0x00000000  A15     : 0x3ffc2074  SAR     : 0x00000004  EXCCAUSE: 0x00000000  
dev1/2024-06-12T22:01:35.319385Z W None |EXCVADDR: 0x00000000  LBEG    : 0x00000000  LEND    : 0x00000000  LCOUNT  : 0x00000000  
dev1/2024-06-12T22:01:35.335828Z W None |Backtrace: 0x4008d9e0:0x3ffc2130 0x4008883f:0x3ffc2150 0x4008d099:0x3ffc2170 0x4008d68f:0x3ffc2190 0x40082cfa:0x3ffc21b0 0x40082b65:0x3ffc21d0 0x40082b47:0x00000000 |<-CORRUPTED
dev1/2024-06-12T22:01:35.338970Z W None |ELF file SHA256: 864e5fdcc73d910e
dev1/2024-06-12T22:01:35.339791Z W None |Rebooting...

```

esp-zhp, 2024-07-18:
@tomasznowik @filzek 
We recently fixed a bug related to the issue ASSERT_PARAM(1536 0), in rwble.c at line 384. The fix is included in the following commits:

master: 49c19fd85762110fea921a033328935b295d0e37
v5.3: aaf371027dcacce2be9d98d37959199038035e92
v5.2: 7c52034978b09d21f023147b94794ee129af125c
v5.1: 18cac360b92e80d2c8a5e240f2939defcba7c71a
v5.0: 67eb7a79bf8292569657e963f944e8fcd9130830
v4.4: 9f0e5aec6ccb8dc4a12b87d95549e8e9a985b835
Please update to the latest version. If you encounter any further issues, do not hesitate to contact us.

closed 2024-07-18 by filzek