espressif/esp-idf #13726: Adding new services to GATTS is invalid

Dylanliacc, 2024-05-01:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

V5.2.1

### Espressif SoC revision.

ESP32S3

### Operating System used.

Linux

### How did you build your project?

VS Code IDE

### If you are using Windows, please specify command line type.

None

### Development Kit.

Custom board

### Power Supply used.

USB

### What is the expected behavior?

Use The example  esp\v5.2.1\esp-idf\examples\bluetooth\bluedroid\ble\gatt_server
 use function esp_ble_gatts_add_char（） to add a new cha

Works correctly when using routine to add two characteristics

### What is the actual behavior?

When I try to use the function esp_ble_gatts_add_char（）to add one more characteristic value to the service Only one can be added

case ESP_GATTS_CREATE_EVT:
        ESP_LOGI(GATTS_TAG, "CREATE_SERVICE_EVT, status %d,  service_handle %d", param->create.status, param->create.service_handle);

        gl_profile_tab[PROFILE_A_APP_ID].service_handle = param->create.service_handle;
      
        gl_profile_tab[PROFILE_A_APP_ID].char_uuid_read.len = ESP_UUID_LEN_128;
       
        gl_profile_tab[PROFILE_A_APP_ID].char_uuid_write.len = ESP_UUID_LEN_128;
        uint8_t char_uuid_write[ESP_UUID_LEN_128] = {};
   

        uint8_t char_uuid_read[ESP_UUID_LEN_128] = {};
        for (int i = 0; i < ESP_UUID_LEN_128; i++)
        {
            gl_profile_tab[PROFILE_A_APP_ID].char_uuid_write.uuid.uuid128[i] = char_uuid_write[i];
        }
        for (int i = 0; i < ESP_UUID_LEN_128; i++)
        {
            gl_profile_tab[PROFILE_A_APP_ID].char_uuid_read.uuid.uuid128[i] = char_uuid_read[i];
        }

        esp_ble_gatts_start_service(gl_profile_tab[PROFILE_A_APP_ID].service_handle);

        a_property = ESP_GATT_CHAR_PROP_BIT_READ | ESP_GATT_CHAR_PROP_BIT_WRITE | ESP_GATT_CHAR_PROP_BIT_NOTIFY;

        esp_err_t add_char_ret = esp_ble_gatts_add_char(gl_profile_tab[PROFILE_A_APP_ID].service_handle, &gl_profile_tab[PROFILE_A_APP_ID].char_uuid_write,
                                                        ESP_GATT_PERM_READ | ESP_GATT_PERM_WRITE,
                                                        a_property,
                                                        &gatts_demo_char1_val, NULL);
        if (add_char_ret)
        {
            ESP_LOGE(GATTS_TAG, "add char1 failed, error code =%x", add_char_ret);
        }

        add_char_ret = esp_ble_gatts_add_char(gl_profile_tab[PROFILE_A_APP_ID].service_handle, &gl_profile_tab[PROFILE_A_APP_ID].char_uuid_read,
                                              ESP_GATT_PERM_READ | ESP_GATT_PERM_WRITE,
                                              a_property,
                                              &gatts_demo_char2_val, NULL);
        if (add_char_ret)
        {
            ESP_LOGE(GATTS_TAG, "add char2 failed, error code =%x", add_char_ret);
        }
        break;

### Steps to reproduce.

1. Use The example  esp\v5.2.1\esp-idf\examples\bluetooth\bluedroid\ble\gatt_server
2. use function esp_ble_gatts_add_char（） to add a new cha failed

zhp0406, 2024-05-06:
It looks like the value of "char_uuid_write" is incorrect
[image]

Dylanliacc, 2024-05-07:
This is not the main question, I filled it in casually .The main problem is that when adding the second characteristic value, the system does not report any errors, but the Bluetooth client does not show that the addition was successful.

zhp0406, 2024-05-07:
[image]

zhp0406, 2024-05-07:
[gatts_demo.zip](https://github.com/espressif/esp-idf/files/15232918/gatts_demo.zip)

Dylanliacc, 2024-05-08:
got it ! I succeeded according to your help