espressif/esp-idf #13727: Bootloader Overflow not Reported

diplfranzhoepfinger, 2024-05-01:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

d

0xjakob, 2024-05-02:
@diplfranzhoepfinger Thanks for noticing this! It looks a bit like the check during the build only tests the binary size of the bootloader, but not the D/IRAM size. We'll take a look at this.

ESP-Marius, 2024-05-08:
As long as the linker is not reporting an error here then I think we can assume that the actual binary is fine, and this is an error in `esp-idf-size`

With the latest version of idf size it does not show an overflow.

```
┏━━━━━━━━━━━━━━━━━━━━━┳━━━━━━━━━━━━━━┳━━━━━━━━━━┳━━━━━━━━━━━━━━━━┳━━━━━━━━━━━━━━━┓
┃ Memory Type/Section ┃ Used [bytes] ┃ Used [%] ┃ Remain [bytes] ┃ Total [bytes] ┃
┡━━━━━━━━━━━━━━━━━━━━━╇━━━━━━━━━━━━━━╇━━━━━━━━━━╇━━━━━━━━━━━━━━━━╇━━━━━━━━━━━━━━━┩
│ DIRAM               │        35465 │    57.72 │          25975 │         61440 │
│    .data            │        13332 │     21.7 │                │               │
│    .text            │        11675 │     19.0 │                │               │
│    .rodata          │         6336 │    10.31 │                │               │
│    .bss             │          272 │     0.44 │                │               │
│    .bootdesc        │           80 │     0.13 │                │               │
│    .text            │            3 │      0.0 │                │               │
└─────────────────────┴──────────────┴──────────┴────────────────┴───────────────┘
Total image size: 35190 bytes (.bin may be padded larger)
```

But with the old version (by passing `-l`) I do indeed see the same overflow that you see.

fhrbata, 2024-05-10:
Hello @diplfranzhoepfinger,

Thank you for your report and for bringing this issue to our attention. As @ESP-Marius mentioned, this is a bug in the esp-idf-size tool, and your binary is unaffected. A fix is currently in progress. Meanwhile, you can try using the refactored version, which should resolve this problem. It's included in the esp-idf-size since the 1.0.0 release and is now the default in the master and v5.3 branches. For older versions of esp-idf, you can set export ESP_IDF_SIZE_NG=1 to utilize the refactored version instead of the legacy one, which has this issue. Hope this helps, and once again, thank you.