espressif/esp-idf #13728: esp32s3，esp-idf5.2平台tdm tx格式错误。tdm协议时序都是错的。

crl6, 2024-05-01:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

代码：

/*
 * SPDX-FileCopyrightText: 2021-2023 Espressif Systems (Shanghai) CO LTD
 *
 * SPDX-License-Identifier: Unlicense OR CC0-1.0
 */

#include <stdint.h>
#include <stdlib.h>
#include "freertos/FreeRTOS.h"
#include "freertos/task.h"
#include "driver/i2s_tdm.h"
#include "driver/gpio.h"
#include "esp_check.h"
#include "sdkconfig.h"

	/* I2S port and GPIOs */
#define I2S_NUM         (0)
#define I2S_MCK_IO      (GPIO_NUM_0)
#define I2S_BCK_IO      (GPIO_NUM_10)//(GPIO_NUM_15)
#define I2S_WS_IO       (GPIO_NUM_16)
#define I2S_DO_IO       (GPIO_NUM_17)
#define I2S_DI_IO       (GPIO_NUM_18)

/* Set 1 to allocate rx & tx channels in duplex mode on a same I2S controller, they will share the BCLK and WS signal
 * Set 0 to allocate rx & tx channels in simplex mode, these two channels will be totally separated */
#define EXAMPLE_I2S_DUPLEX_MODE     1

#define EXAMPLE_TDM_BCLK_IO1        I2S_BCK_IO      // I2S bit clock io number
#define EXAMPLE_TDM_WS_IO1          I2S_WS_IO      // I2S word select io number
#define EXAMPLE_TDM_DOUT_IO1        I2S_DO_IO      // I2S data out io number
#define EXAMPLE_TDM_DIN_IO1         I2S_DI_IO      // I2S data in io number
#if !EXAMPLE_I2S_DUPLEX_MODE
#define EXAMPLE_TDM_BCLK_IO2    GPIO_NUM_6      // I2S bit clock io number
#define EXAMPLE_TDM_WS_IO2      GPIO_NUM_7      // I2S word select io number
#define EXAMPLE_TDM_DOUT_IO2    GPIO_NUM_8      // I2S data out io number
#define EXAMPLE_TDM_DIN_IO2     GPIO_NUM_9      // I2S data in io number
#endif

#define EXAMPLE_BUFF_SIZE               2048

static i2s_chan_handle_t                tx_chan;        // I2S tx channel handler
static i2s_chan_handle_t                rx_chan;        // I2S rx channel handler

static void i2s_example_read_task(void *args)
{
    uint8_t *r_buf = (uint8_t *)calloc(1, EXAMPLE_BUFF_SIZE);
    assert(r_buf); // Check if r_buf allocation success
    size_t r_bytes = 0;

    /* Enable the RX channel */
    ESP_ERROR_CHECK(i2s_channel_enable(rx_chan));

    /* ATTENTION: The print and delay in the read task only for monitoring the data by human,
     * Normally there shouldn't be any delays to ensure a short polling time,
     * Otherwise the dma buffer will overflow and lead to the data lost */
    while (1) {
        /* Read i2s data */
        if (i2s_channel_read(rx_chan, r_buf, EXAMPLE_BUFF_SIZE, &r_bytes, 1000) == ESP_OK) {
            printf("Read Task: i2s read %d bytes\n-----------------------------------\n", r_bytes);
            printf("[0] %x [1] %x [2] %x [3] %x\n[4] %x [5] %x [6] %x [7] %x\n\n",
                   r_buf[0], r_buf[1], r_buf[2], r_buf[3], r_buf[4], r_buf[5], r_buf[6], r_buf[7]);
        } else {
            printf("Read Task: i2s read failed\n");
        }
        vTaskDelay(pdMS_TO_TICKS(200));
    }
    free(r_buf);
    vTaskDelete(NULL);
}

static void i2s_example_write_task(void *args)
{
    uint8_t *w_buf = (uint8_t *)calloc(1, EXAMPLE_BUFF_SIZE);
    assert(w_buf); // Check if w_buf allocation success

    /* Assign w_buf */
    for (int i = 0; i < EXAMPLE_BUFF_SIZE; i += 8) {
        w_buf[i]     = 0x12;
        w_buf[i + 1] = 0x34;
        w_buf[i + 2] = 0x56;
        w_buf[i + 3] = 0x78;
        w_buf[i + 4] = 0x9A;
        w_buf[i + 5] = 0xBC;
        w_buf[i + 6] = 0xDE;
        w_buf[i + 7] = 0xF0;
    }

    size_t w_bytes = EXAMPLE_BUFF_SIZE;

    /* (Optional) Preload the data before enabling the TX channel, so that the valid data can be transmitted immediately */
    while (w_bytes == EXAMPLE_BUFF_SIZE) {
        /* Here we load the target buffer repeatedly, until all the DMA buffers are preloaded */
        ESP_ERROR_CHECK(i2s_channel_preload_data(tx_chan, w_buf, EXAMPLE_BUFF_SIZE, &w_bytes));
    }

    /* Enable the TX channel */
    ESP_ERROR_CHECK(i2s_channel_enable(tx_chan));
    while (1) {
        /* Write i2s data */
        if (i2s_channel_write(tx_chan, w_buf, EXAMPLE_BUFF_SIZE, &w_bytes, 1000) == ESP_OK) {
            printf("Write Task: i2s write %d bytes\n", w_bytes);
        } else {
            printf("Write Task: i2s write failed\n");
        }
        //vTaskDelay(pdMS_TO_TICKS(200));
    }
    free(w_buf);
    vTaskDelete(NULL);
}

// here
static void i2s_example_init_tdm_duplex(void)
{
    /* Setp 1: Determine the I2S channel configuration and allocate both channels
     * The default configuration can be generated by the helper macro,
     * it only requires the I2S controller id and I2S role */
    i2s_chan_config_t chan_cfg = I2S_CHANNEL_DEFAULT_CONFIG(I2S_NUM_AUTO, I2S_ROLE_MASTER);
    ESP_ERROR_CHECK(i2s_new_channel(&chan_cfg, &tx_chan, &rx_chan));

    /* Step 2: Setting the configurations of TDM mode, and initialize rx & tx channels
     * The slot configuration and clock configuration can be generated by the macros
     * These two helper macros is defined in 'i2s_tdm.h' which can only be used in TDM mode.
     * They can help to specify the slot and clock configurations for initialization or re-configuring */
    i2s_tdm_config_t tdm_cfg = {
        .clk_cfg  = I2S_TDM_CLK_DEFAULT_CONFIG(16000),
        /* For the target that not support full data bit-width in multiple slots (e.g. ESP32C3, ESP32S3, ESP32C6)
         * The maximum bits in one frame is limited by the hardware, the number of bit clock can't exceed 128 in one frame,
         * which is to say, TDM mode can only support 32 bit-width data upto 4 slots,
         * 16 bit-width data upto 8 slots and 8 bit-width data upto 16 slots
         * But for the target that support full data bit-width in multiple slots (e.g. ESP32H2)
         * There is no such limitation, it can support up to 32 bit-width with 16 slots */
        .slot_cfg = I2S_TDM_MSB_SLOT_DEFAULT_CONFIG(I2S_DATA_BIT_WIDTH_32BIT, I2S_SLOT_MODE_STEREO,
                                                    I2S_TDM_SLOT0 | I2S_TDM_SLOT1 | I2S_TDM_SLOT2 | I2S_TDM_SLOT3),
        .gpio_cfg = {
            .mclk = I2S_GPIO_UNUSED,    // some codecs may require mclk signal, this example doesn't need it
            .bclk = EXAMPLE_TDM_BCLK_IO1,
            .ws   = EXAMPLE_TDM_WS_IO1,
            .dout = EXAMPLE_TDM_DOUT_IO1,
            .din  = EXAMPLE_TDM_DOUT_IO1, // In duplex mode, bind output and input to a same gpio can loopback internally
            .invert_flags = {
                .mclk_inv = false,
                .bclk_inv = false,
                .ws_inv   = false,
            },
        },
    };
    /* While there are more than two slots activated, receiving data might be wrong if the mclk multiple is too small
     * The driver will increasing the multiple automatically to ensure the bclk_div is greater than 2.
     * Modify the mclk_multiple to 512 directly here to avoid the warning */
    tdm_cfg.clk_cfg.mclk_multiple = I2S_MCLK_MULTIPLE_512;

    /* Initialize the channels */
    //ESP_ERROR_CHECK(i2s_channel_init_tdm_mode(tx_chan, &tdm_cfg));
    ESP_ERROR_CHECK(i2s_channel_init_tdm_mode(rx_chan, &tdm_cfg));
}

void i2s_tdm_test(void)
{

    i2s_example_init_tdm_duplex();
    /* Step 3: Create writing and reading task, enable and start the channels */
    //xTaskCreate(i2s_example_read_task, "i2s_example_read_task", 4096, NULL, 5, NULL);
    xTaskCreate(i2s_example_write_task, "i2s_example_write_task", 4096, NULL, 5, NULL);
}

[image]

配置的是四通道32bit，为啥逻辑分析仪抓出来才2通道？（虽然这个逻辑分析仪无法解析tdm信号，但是可以看的出数据宽度不对）

正确的tdm格式时序是这样的：

[image]

其它芯片tdm 8通道发出来的时序：

[image]

Indastri, 2024-05-02:
Hi @crl6 were you able to resolve this issue?

L-KAYA, 2025-01-10:
@crl6 
> 配置的是四通道32bit，为啥逻辑分析仪抓出来才2通道？（虽然这个逻辑分析仪无法解析tdm信号，但是可以看的出数据宽度不对）
因为你配置的是MSB格式，MSB格式的WS信号占空比就是50%的。TDM模式一共有四种格式，分别是PHILIPS，MSB，PCM_SHORT，PCM_LONG。图中的效果用的是 PCM_SHORT，可以用`I2S_TDM_PCM_SHORT_SLOT_DEFAULT_CONFIG`宏来生成。 具体的格式区别请参考 [i2s tdm 编程手册]https://docs.espressif.com/projects/esp-idf/zh_CN/latest/esp32s3/api-reference/peripherals/i2s.html#tdm

L-KAYA, 2025-01-10:
@Indastri 
The configuration in the example is using MSB format, which is 50% duty. TDM mode has four formats, they are PHILIPS, MSB, PCM_SHORT, PCM_LONG. The format shown in the picture is PCM_SHORT, you can use this format by setting `slot_cfg` with the macro `I2S_TDM_PCM_SHORT_SLOT_DEFAULT_CONFIG`. For the detailed difference, please refer to the [i2s tdm programming guide](https://docs.espressif.com/projects/esp-idf/zh_CN/latest/esp32s3/api-reference/peripherals/i2s.html#tdm)