espressif/esp-idf #13730: Documentation issue on xRingbufferSendAcquire()

huming2207, 2024-05-02:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

Hi all,

I was trying to use the FreeRTOS ring buffer for one of our storage library's write buffer. I had a look on the API `xRingbufferSendAcquire()` and my understanding for this is like a "malloc" but instead of allocates on heap, it "prepares" a contiguous RAM for the consumer to modify (e.g. `memcpy()` from other stuff) on it, and then call `xRingbufferSendComplete()` to tell the ring buffer my modification is done.

But the ring buffer documentation below declares `dma_item_t item` instead of `dma_item_t *item`. Have I misunderstood anything?

[image]

Regards,
Jackson

0xjakob, 2024-05-02:
@huming2207 I think you're right. The documentation looks incorrect. Indeed, the code would give a warning when it is compiled. We'll take a look and fix this.

0xjakob, 2024-05-06:
The fix has been merged to our internal code base. It should appear soon on the master branch here on github.

huming2207, 2024-05-06:
> The fix has been merged to our internal code base. It should appear soon on the master branch here on github.

Nice, thanks for the fix!

closed 2024-05-16 by espressif-bot