espressif/esp-idf #13731: Boot directly into Test Partition

law-ko, 2024-05-02:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

There are examples on how to boot into factory test partition using a GPIO pin, but is there any way that the device can directly boot into test partition without bootstrapping any pin and later rewrite otadata partition once factory procedure completes?

https://docs.espressif.com/projects/esp-idf/en/stable/esp32/api-guides/bootloader.html#boot-from-test-firmware

law-ko, 2024-05-02:
We found out that we can use `otatool.py -p /dev/tty.usbserial-0001 switch_ota_partition --name fctryapp` to switch to the fctry partition, but this does not support Flash Encryption and Securt Boot devices. Any way to resolve this?

0xjakob, 2024-05-06:
@law-ko We're currently checking this issue and will come back once we have an answer.

mahavirj, 2024-05-09:
@law-ko 

There is no direct way to boot from the test app partition. However, you may consider using `factory` partition here (https://docs.espressif.com/projects/esp-idf/en/stable/esp32/api-guides/bootloader.html#factory-reset).

If the `otadata` partition is empty then the boot sequence will fallback to `factory` partition. From the `factory` partition, appropriate `app` partition can be activated using API `esp_ota_set_boot_partition` to proceed for the normal boot flow.

closed 2024-05-20 by mahavirj