espressif/esp-idf #13734: WiFi still crash and doesnt come back, reboot is needed to work again.

filzek, 2024-05-02:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.2.1-485-g9b3ae86417

### Espressif SoC revision.

esp32 wrover ie 3.0

### Operating System used.

Windows

### How did you build your project?

Command line with Make

### If you are using Windows, please specify command line type.

None

### Development Kit.

esp32 wrover ie 3.0

### Power Supply used.

USB

### What is the expected behavior?

WiFi still crash and doesnt come back, reboot is needed to work again.

This happen randomly and there is no log to know what is happen to the wifi driver, even if we try to destroy and recreate the wifi sta connect it does nothing, we are unable to catch this problem.

We have set some debug boards to esp-insights to try to log it.

### What is the actual behavior?

Randonly crash the wifi and looses connection.

The NIMBLE keep working and 100% functional.

### Steps to reproduce.

just have websocket, mdns, wifi running and crash occurs randomly.

AxelLin, 2024-05-03:
@filzek 
1. Does this issue happen if BLE is not used?
2. Could you share sdkconfig?

filzek, 2024-05-03:
> @filzek
> 
> 1. Does this issue happen if BLE is not used?
> 2. Could you share sdkconfig?

1) didnt try without nimble as we need this working, soo have no idea so far, we can disable the nimble init at least and try tomorrow.

2) sure, 

```
#
# Automatically generated file. DO NOT EDIT.
# Espressif IoT Development Framework (ESP-IDF) 5.2.1 Project Configuration
#
CONFIG_SOC_BROWNOUT_RESET_SUPPORTED="Not determined"
CONFIG_SOC_TWAI_BRP_DIV_SUPPORTED="Not determined"
CONFIG_SOC_DPORT_WORKAROUND="Not determined"
CONFIG_SOC_CAPS_ECO_VER_MAX=301
CONFIG_SOC_ADC_SUPPORTED=y
CONFIG_SOC_DAC_SUPPORTED=y
CONFIG_SOC_UART_SUPPORTED=y
CONFIG_SOC_MCPWM_SUPPORTED=y
CONFIG_SOC_GPTIMER_SUPPORTED=y
CONFIG_SOC_SDMMC_HOST_SUPPORTED=y
CONFIG_SOC_BT_SUPPORTED=y
CONFIG_SOC_PCNT_SUPPORTED=y
CONFIG_SOC_WIFI_SUPPORTED=y
CONFIG_SOC_SDIO_SLAVE_SUPPORTED=y
CONFIG_SOC_TWAI_SUPPORTED=y
CONFIG_SOC_EFUSE_SUPPORTED=y
CONFIG_SOC_EMAC_SUPPORTED=y
CONFIG_SOC_ULP_SUPPORTED=y
CONFIG_SOC_CCOMP_TIMER_SUPPORTED=y
CONFIG_SOC_RTC_FAST_MEM_SUPPORTED=y
CONFIG_SOC_RTC_SLOW_MEM_SUPPORTED=y
CONFIG_SOC_RTC_MEM_SUPPORTED=y
CONFIG_SOC_I2S_SUPPORTED=y
CONFIG_SOC_RMT_SUPPORTED=y
CONFIG_SOC_SDM_SUPPORTED=y
CONFIG_SOC_GPSPI_SUPPORTED=y
CONFIG_SOC_LEDC_SUPPORTED=y
CONFIG_SOC_I2C_SUPPORTED=y
CONFIG_SOC_SUPPORT_COEXISTENCE=y
CONFIG_SOC_AES_SUPPORTED=y
CONFIG_SOC_MPI_SUPPORTED=y
CONFIG_SOC_SHA_SUPPORTED=y
CONFIG_SOC_FLASH_ENC_SUPPORTED=y
CONFIG_SOC_SECURE_BOOT_SUPPORTED=y
CONFIG_SOC_TOUCH_SENSOR_SUPPORTED=y
CONFIG_SOC_BOD_SUPPORTED=y
CONFIG_SOC_ULP_FSM_SUPPORTED=y
CONFIG_SOC_CLK_TREE_SUPPORTED=y
CONFIG_SOC_MPU_SUPPORTED=y
CONFIG_SOC_WDT_SUPPORTED=y
CONFIG_SOC_SPI_FLASH_SUPPORTED=y
CONFIG_SOC_DPORT_WORKAROUND_DIS_INTERRUPT_LVL=5
CONFIG_SOC_XTAL_SUPPORT_26M=y
CONFIG_SOC_XTAL_SUPPORT_40M=y
CONFIG_SOC_XTAL_SUPPORT_AUTO_DETECT=y
CONFIG_SOC_ADC_RTC_CTRL_SUPPORTED=y
CONFIG_SOC_ADC_DIG_CTRL_SUPPORTED=y
CONFIG_SOC_ADC_DMA_SUPPORTED=y
CONFIG_SOC_ADC_PERIPH_NUM=2
CONFIG_SOC_ADC_MAX_CHANNEL_NUM=10
CONFIG_SOC_ADC_ATTEN_NUM=4
CONFIG_SOC_ADC_DIGI_CONTROLLER_NUM=2
CONFIG_SOC_ADC_PATT_LEN_MAX=16
CONFIG_SOC_ADC_DIGI_MIN_BITWIDTH=9
CONFIG_SOC_ADC_DIGI_MAX_BITWIDTH=12
CONFIG_SOC_ADC_DIGI_RESULT_BYTES=2
[... 2372 lines cut ...]
# CONFIG_PPP_SUPPORT is not set
CONFIG_ESP32_TIME_SYSCALL_USE_RTC_HRT=y
CONFIG_ESP32_TIME_SYSCALL_USE_RTC_FRC1=y
# CONFIG_ESP32_TIME_SYSCALL_USE_RTC is not set
# CONFIG_ESP32_TIME_SYSCALL_USE_HRT is not set
# CONFIG_ESP32_TIME_SYSCALL_USE_FRC1 is not set
# CONFIG_ESP32_TIME_SYSCALL_USE_NONE is not set
CONFIG_ESP32_PTHREAD_TASK_PRIO_DEFAULT=5
CONFIG_ESP32_PTHREAD_TASK_STACK_SIZE_DEFAULT=3072
CONFIG_ESP32_PTHREAD_STACK_MIN=768
CONFIG_ESP32_DEFAULT_PTHREAD_CORE_NO_AFFINITY=y
# CONFIG_ESP32_DEFAULT_PTHREAD_CORE_0 is not set
# CONFIG_ESP32_DEFAULT_PTHREAD_CORE_1 is not set
CONFIG_ESP32_PTHREAD_TASK_CORE_DEFAULT=-1
CONFIG_ESP32_PTHREAD_TASK_NAME_DEFAULT="pthread"
CONFIG_SPI_FLASH_WRITING_DANGEROUS_REGIONS_ABORTS=y
# CONFIG_SPI_FLASH_WRITING_DANGEROUS_REGIONS_FAILS is not set
# CONFIG_SPI_FLASH_WRITING_DANGEROUS_REGIONS_ALLOWED is not set
# CONFIG_ESP32_ULP_COPROC_ENABLED is not set
CONFIG_SUPPRESS_SELECT_DEBUG_OUTPUT=y
CONFIG_SUPPORT_TERMIOS=y
CONFIG_SEMIHOSTFS_MAX_MOUNT_POINTS=1
# End of deprecated options

```

vik-gokhale, 2024-05-07:
Hi @filzek,
Could you please share the logs with debug enabled?
Here are the sdkconfig parameters to be set for debug logs : 
----
CONFIG_WPA_DEBUG_PRINT=y
CONFIG_LOG_DEFAULT_LEVEL_DEBUG=y
CONFIG_LOG_DEFAULT_LEVEL=4

filzek, 2024-05-13:
WIFI_EVENT_STA_DISCONNECTED - retry to connect to the AP
[0;31mE (45937595) transport_base: poll_read select error 113, errno = Software caused connection abort, fd = 57[0m
[0;31mE (45937605) websocket_client: esp_transport_poll_read() returned -1, transport_error=ESP_OK, tls_error_code=0, tls_flags=0, errno=119[0m
WIFI_EVENT_STA_DISCONNECTED - retry to connect to the AP
WIFI_EVENT_STA_DISCONNECTED - retry to connect to the AP
WIFI_EVENT_STA_DISCONNECTED - retry to connect to the AP
WIFI_EVENT_STA_DISCONNECTED - retry to connect to the AP
GOING TO DESTROY WI-FI and NETIF INTERFACE 
E (45956955) wifi:NAN WiFi stop
E (45956995) wifi:sta is connecting, return error
WI-FI and NETIF INTERFACE RECREATED
E (45956995) wifi:sta is connecting, return error
E (45956995) wifi:sta is connecting, return error
WIFI_EVENT_STA_DISCONNECTED - retry to connect to the AP
[0;31mE (45966635) esp-tls: couldn't get hostname for :CANTDISCLOSETHESITE: getaddrinfo() returns 202, addrinfo=0x0[0m
[0;31mE (45966635) esp-tls: Failed to open new connection[0m
[0;31mE (45966645) transport_base: Failed to open a new connection[0m
[0;31mE (45966655) transport_ws: Error connecting to host CANTDISCLOSETHESITE:443[0m
WIFI_EVENT_STA_DISCONNECTED - retry to connect to the AP
[0;31mE (45966675) websocket_client: esp_transport_connect() failed with -1, transport_error=ESP_ERR_ESP_TLS_CANNOT_RESOLVE_HOSTNAME, tls_error_code=0, tls_flags=0, esp_ws_handshake_status_code=101, errno=128[0m
WIFI_EVENT_STA_DISCONNECTED - retry to connect to the AP
WIFI_EVENT_STA_DISCONNECTED - retry to connect to the AP
WIFI_EVENT_STA_DISCONNECTED - retry to connect to the AP
GOING TO DESTROY WI-FI and NETIF INTERFACE 
E (45981205) wifi:NAN WiFi stop
E (45981255) wifi:sta is connecting, return error
WI-FI and NETIF INTERFACE RECREATED
E (45981255) wifi:sta is connecting, return error
E (45981255) wifi:sta is connecting, return error
WIFI_EVENT_STA_DISCONNECTED - retry to connect to the AP
WIFI_EVENT_STA_DISCONNECTED - retry to connect to the AP
[0;31mE (45995635) esp-tls: couldn't get hostname for :CANTDISCLOSETHESITE: getaddrinfo() returns 202, addrinfo=0x0[0m
[0;31mE (45995635) esp-tls: Failed to open new connection[0m
[0;31mE (45995645) transport_base: Failed to open a new connection[0m
[0;31mE (45995655) transport_ws: Error connecting to host CANTDISCLOSETHESITE:443[0m
[0;31mE (45995675) websocket_client: esp_transport_connect() failed with -1, transport_error=ESP_ERR_ESP_TLS_CANNOT_RESOLVE_HOSTNAME, tls_error_code=0, tls_flags=0, esp_ws_handshake_status_code=101, errno=128[0m
[0;31mE (46010705) esp-tls: couldn't get hostname for :CANTDISCLOSETHESITE: getaddrinfo() returns 202, addrinfo=0x0[0m
[0;31mE (46010705) esp-tls: Failed to open new connection[0m
[0;31mE (46010715) transport_base: Failed to open a new connection[0m
[0;31mE (46010725) transport_ws: Error connecting to host CANTDISCLOSETHESITE:443[0m
[0;31mE (46010745) websocket_client: esp_transport_connect() failed with -1, transport_error=ESP_ERR_ESP_TLS_CANNOT_RESOLVE_HOSTNAME, tls_error_code=0, tls_flags=0, esp_ws_handshake_status_code=101, errno=128[0m
WIFI_EVENT_STA_DISCONNECTED - retry to connect to the AP
[0;31mE (46065655) transport_base: poll_read select error 113, errno = Software caused connection abort, fd = 56[0m
[0;31mE (46065665) websocket_client: esp_transport_poll_read() returned -1, transport_error=ESP_OK, tls_error_code=0, tls_flags=0, errno=119[0m
[0;31mE (76718705) task_wdt: Task watchdog got triggered. The following tasks/users did not reset the watchdog in time:[0m
[0;31mE (76718705) task_wdt:  - IDLE0 (CPU 0)[0m
[0;31mE (76718705) task_wdt: Tasks currently running:[0m
[0;31mE (76718705) task_wdt: CPU 0: wifi[0m
[0;31mE (76718705) task_wdt: CPU 1: IDLE1[0m
[0;31mE (76718705) task_wdt: Print CPU 0 (current core) backtrace[0m

Backtrace: 0x401313FF:0x3FFC0220 0x40085F25:0x3FFC0250 0x401D1CAF:0x3FFD88D0 0x401D5AFB:0x3FFD88F0 0x401750E2:0x3FFD8910 0x40175437:0x3FFD8930 0x40185BA7:0x3FFD8950 0x4017844D:0x3FFD89B0 0x401816C5:0x3FFD89D0 0x40177C5C:0x3FFD8A50 0x4017EFE4:0x3FFD8A70 0x4017F113:0x3FFD8A90 0x401D0CA5:0x3FFD8AB0
0x401313ff: task_wdt_timeout_handling at C:/Espressif/frameworks/esp-idf-v5.2/components/esp_system/task_wdt/task_wdt.c:441
 (inlined by) task_wdt_isr at C:/Espressif/frameworks/esp-idf-v5.2/components/esp_system/task_wdt/task_wdt.c:515
0x40085f25: _xt_lowint1 at C:/Espressif/frameworks/esp-idf-v5.2/components/xtensa/xtensa_vectors.S:1240
0x401d1caf: hal_mac_deinit at ??:?
0x401d5afb: ic_mac_deinit at ??:?
0x401750e2: ieee80211_crypto_gmac_decrypt at ??:?
0x40175437: chm_set_current_channel at ??:?
0x40185ba7: ieee80211_update_channel at ??:?
0x4017844d: cnx_sta_leave at ??:?
0x401816c5: ieee80211_sta_new_state at ??:?
0x40177c5c: mgd_probe_send_timeout_process at ??:?
0x4017efe4: ieee80211_ioctl at ??:?
0x4017f113: ieee80211_timer_do_process at ??:?
0x401d0ca5: ppTask at ??:?

[0;31mE (76738705) task_wdt: Task watchdog got triggered. The following tasks/users did not reset the watchdog in time:[0m
[0;31mE (76738705) task_wdt:  - IDLE0 (CPU 0)[0m
[0;31mE (76738705) task_wdt: Tasks currently running:[0m
[... 719 lines cut ...]
[0;31mE (77296985) transport_base: Failed to open a new connection[0m
[0;31mE (77296995) transport_ws: Error connecting to host CANTDISCLOSETHESITE:443[0m
[... cut ...]
0x40085f25: _xt_lowint1 at C:/Espressif/frameworks/esp-idf-v5.2/components/xtensa/xtensa_vectors.S:1240
0x401d1cb4: hal_mac_deinit at ??:?
0x401d5afb: ic_mac_deinit at ??:?
0x401750e2: ieee80211_crypto_gmac_decrypt at ??:?
0x40175437: chm_set_current_channel at ??:?
0x40185ba7: ieee80211_update_channel at ??:?
0x4017844d: cnx_sta_leave at ??:?
0x401816c5: ieee80211_sta_new_state at ??:?
0x40177c5c: mgd_probe_send_timeout_process at ??:?
0x4017efe4: ieee80211_ioctl at ??:?
0x4017f113: ieee80211_timer_do_process at ??:?
0x401d0ca5: ppTask at ??:?

filzek, 2024-05-13:
debug turned on as request, logging all

AxelLin, 2024-06-05:
@vik-gokhale Any finding in https://github.com/espressif/esp-idf/issues/13734#issuecomment-2108806099 ?

vik-gokhale, 2024-06-05:
Hi,
this is being looked at internally.
Once the problem is resolved, an update would be shared on the thread.

Regards,
Vikram

filzek, 2024-06-05:
Maybe this could be the crash reason of nimble stack being corrupted?

AxelLin, 2024-07-27:
> Hi, this is being looked at internally. Once the problem is resolved, an update would be shared on the thread.

@vik-gokhale
It has been a while since your last reply, any update?

filzek, 2024-09-23:
complete solved at commit c8c6fd141377b9a48b29c8ebb51fb7efadde0566 (HEAD -> release/v5.2, origin/release/v5.2)

closed 2024-09-23 by filzek