espressif/esp-idf #13737: Crash on Coredump and doesnt even boot anymore

filzek, 2024-05-03:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.2.1-485-g9b3ae86417

### Espressif SoC revision.

Esp32 Wrover 3,0

### Operating System used.

Windows

### How did you build your project?

Command line with Make

### If you are using Windows, please specify command line type.

None

### Development Kit.

Esp32 Wrover

### Power Supply used.

USB

### What is the expected behavior?

run!

### What is the actual behavior?

not working, 

now with the dump to flash and with info, the coredump enable, when it fills the system doenst boot anymore.

### Steps to reproduce.

Boot it!

### Debug Logs.

```plain
Chip is ESP32-D0WD-V3 (revision v3.0)
Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
Crystal is 40MHz
MAC: 08:3a:f2:04:a3:c8
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 921600
Changed.
Configuring flash size...
Flash will be erased from 0x00020000 to 0x001bffff...
Flash will be erased from 0x00009000 to 0x00009fff...
Flash will be erased from 0x00010000 to 0x00011fff...

WARNING: - compress and encrypt options are mutually exclusive
Will flash testingSwitch.bin uncompressed
Writing at 0x00020000... (0 %)
Writing at 0x00024000... (1 %)
Writing at 0x00028000... (2 %)
[... 533 lines cut ...]
D (3142) intr_alloc: Connected src 17 to int 3 (cpu 0)
D (3147) cpu_start: calling init function: 0x400d52c8 on core: 0
0x400d52c8: __esp_system_init_fn_init_components0 at C:/Espressif/frameworks/esp-idf-v5.2/components/esp_system/startup.c:493

V (3162) mmap: actual_mapped_len is 0x10000
lots of this until crash....
V (3162) mmap: actual_mapped_len is 0x10000

ets Jul 29 2019 12:21:46

rst:0x10 (RTCWDT_RTC_RESET),boot:0x1b (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:4
load:0x3fff0034,len:6608
ho 0 tail 12 room 4
load:0x40078000,len:16140
load:0x40080400,len:4048
0x40080400: _init at ??:
```

### More Information.

ok

erhankur, 2024-05-03:
There is no init log for the coredump. If coredump config is enabled, we should see `esp_core_dump_flash: Init core dump to flash`  after `__esp_system_init_fn_init_components0`

What is your IDF version? I see from the boot log 4.2 application is loaded into the ESP32.

chipweinberger, 2024-05-03:
RTCWDT Reset means your device is taking too long to boot

removed the watchdog, or increase the timeout 

if you remove the verbose logs it'll boot faster

filzek, 2024-05-03:
@erhankur 

> There is no init log for the coredump. If coredump config is enabled, we should see `esp_core_dump_flash: Init core dump to flash` after `__esp_system_init_fn_init_components0`
> 
> What is your IDF version? I see from the boot log 4.2 application is loaded into the ESP32.
v5.2.1-485-g9b3ae86417

its on the header.

filzek, 2024-05-03:
@chipweinberger 

> RTCWDT Reset means your device is taking too long to boot
> 
> remove the watchdog, or increase the timeout
> 
> if you remove the verbose logs it'll boot faster

It was not booting either, thats why I enable all kind of logs to try to fetch the problem, but did not change anything.

The reset occurs either way enable or disable.

I have now increased the full allowed to both and will test now, later will post results.

filzek, 2024-05-13:
we has been unable to use coredump on 5.2.1 still bugged

HDLA-BG, 2024-09-02:
Issue exist on 5.0.6

Has anyone been able to find an workaround on this problem or we should disable coredump in production?

Guru Meditation Error: Core  0 panic'ed (InstructionFetchError). Exception was unhandled.

Core  0 register dump:
PC      : 0x3fcd5ac4  PS      : 0x00060430  A0      : 0x820b8aac  A1      : 0x3fcea720  
A2      : 0x3fcd5ac4  A3      : 0x3fcc781c  A4      : 0x00000012  A5      : 0x00000000  
A6      : 0x00000001  A7      : 0x00000000  A8      : 0x821a7260  A9      : 0x3fcea840  
A10     : 0x00000018  A11     : 0x3fcc781c  A12     : 0x00000012  A13     : 0x00000000  
A14     : 0x00060523  A15     : 0x00000000  SAR     : 0x00000006  EXCCAUSE: 0x00000002  
EXCVADDR: 0x3fcd5ac4  LBEG    : 0x40056f5c  LEND    : 0x40056f72  LCOUNT  : 0xffffffff  

Backtrace: 0x3fcd5ac1:0x3fcea720 |<-CORRUPTED

ELF file SHA256: 7699768fd333bff0

[0;32mI (3178) esp_core_dump_flash: Save core dump to flash...[0m
[0;32mI (3184) esp_core_dump_flash: Erase flash 40960 bytes @ 0x794000[0m

filzek, 2025-04-04:
Since 5.4.1 is working.

closed 2025-04-04 by filzek