espressif/esp-idf #13738: HFP_HF example project produces static in PCM mode

devrintalen, 2024-05-03:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

I am running the hfp_hf example project on my ESP32-PICO-KIT-1_V1.0 board and I am unable to get the PCM output to produce anything other than static.

I am building the project with the attached sdkconfig file (I had to rename to sdkconfig.txt in order to attach to this case).
[sdkconfig.txt](https://github.com/espressif/esp-idf/files/15200930/sdkconfig.txt)

I am able to pair successfully to an iPhone 14. When I start a phone call, I see the I2S/PCM output begin clocking on the default GPIOs assigned in gpio_pcm_config.c.  I capture this output with a logic analyzer:

[image]

We have connected this output to both an I2S DAC as well as parsing the logic analyzer capture and porting this to a WAV file. The output is just static.

What is the specific format of the data on the pins? The technical reference manual lists three modes:
* Philips Standard
* MSB Alignment Standard
* PCM Standard

When I parse the data according to the MSB alignment standard I only hear static.

We have tried HCI mode with the wideband speech codec and the audio feedback is working as expected. Our issue is with the PCM output.

devrintalen, 2024-05-03:
We hooked the PCM_DOUT GPIO to the PCM_DIN GPIO and we can hear audio loopback in PCM mode. So this must be an issue of us not understanding the data format on the PCM pins. Can you please clarify what the format is?

esp-qing, 2024-05-07:
Hi, sorry for replying so late. 
In esp-idf, the default synchronization is 8kHZ; the default data clock is 256kHZ; Stereo mode, 16 bits samples; right channel when sync polarity = 0 , left channel when sync polarity = 1 ; sample are sent / received MSB first; pcmd_out pad is push-pull.
The rest configuration can be checked in esp-idf. (including rising edge / falling edge and master / slave role).
If you have any questions please feel free to ask.

devrintalen, 2024-07-25:
Qing, we are using a MAX98357A DAC to play the speaker output. This device expects the DIN pin to follow the Phillips Standard:

[image]

The data that I am tracing on the GPIO pins on our ESP32-PICO-KIT-1_V1.0 board looks like this, which shows that it is following the MSB alignment standard, as you said.

[image]
[image]

Is it possible to modify the I2S output to follow the Phillips Standard?

esp-qing, 2024-07-26:
Hi, @devrintalen ,Sorry, the hfp-hf example cannot modify the output for now. But I have an idea for your previous problem, it seems to be this reason. https://github.com/espressif/esp-idf/issues/14231 Could you check if this is the reason according to the comments?

jtwaleson, 2024-10-15:
Hi @esp-qing, I have a microphone INMP441 emitting its signal with Philips Standard as well, so including a bit shift where it outputs low during the first clock cycle. I'm getting a LOT of static using this microphone.

I now have a theory: I'm pretty sure that the closed source bt stack is interpreting this first clock cycle as a 0. The values are cycling around the 0 point, flipping between positive and negative as the signal tracks the audio waveform. Because the signal is a signed integer, the microphone sends positive values like 000001... and negative like 1111110... . But because the bt stack is including an additional 0, it interprets the negative values as 01111110... , that is, as a very high positive number. So the microphone data is basically cycling between the neutral 0 point and maximum positive numbers all the time.

The microphone is sending 24bit data (from the top of my head), but the bt stack only picks the first 16 bits. Because it's MSB encoded this is fine.

Could you see if my theory is correct and maybe suggest a solution? I think specifying "philips" format with bit shift for microphone input would solve my issue.

In the meantime I'll try and send a modified clock signal via a different pin to the microphone, to see if I can trick it in sending the data earlier.

jtwaleson, 2024-10-16:
Ok, I've confirmed that with the timing offset by one bit it works :) I used bit banging with another esp32 to fake the timing signal, so it still contains noise, but it confirms my hypothesis. (This was fun!)

Please let me know if there's a way to change the input format to be Philips compatible. If not I'll have to go the HCI path I guess, but I still have no clue how that works and whether I can get it performant enough for voice.

esp-qing, 2024-10-17:
Hi, @jtwaleson I think your theory is correct.The default PCM output waveform is stereo mode(dual channels). This is similar to the MSB Alignment Standard ,and it is currently not possible to change to the Philips mode.
[image]
 However, PCM frame synchronization signal can also be configured as Mono mode(single channel) similar to this.
[image]
FSYNC signal starts to change a CLK clock cycle earlier than the DOUT signal. The FSYNC signal continues for one extra CLK clock cycle. I don't know if it can meet the your needs. This mode can be configured in the menuconfig in master branch (introduced in README).

jtwaleson, 2024-10-17:
Great, I'll give that a try tomorrow! I'm using mono for both in / output so it meets my needs. Thank you!

esp-qing, 2024-10-17:
You're welcome, if you have any questions, feel free to ask.

jtwaleson, 2024-10-19:
Ok, so far no luck. I managed to compile with esp-idf 5.4, but both my i2s amp and microphone don't seem to understand the custom(?) mono 1 or mono 2 i2s variant. I've tried both left and right channels but no luck. I noticed that CLK goes down during the second half of the mono 1/2 slot. Is that by design?

[image]

By the way, I'm planning to use the ADC for the amp+microphone as I'm working with an existing headset that only has two wires going to to mouthpiece. I don't that hfp_hf via PCM would support using the ADC right? As a work-around, I'm considering using the second I2S slot, setting it up as a slave to the first slot, and connecting that to the ADC. Hacky,  but I think it could work.

jtwaleson, 2024-10-27:
Update: I gave up on using the INMP441 or even the ADC config as that also [doesn't seem to work](https://esp32.com/viewtopic.php?p=139197), and went with two dedicated i2s DAC / ADC boards instead. I made sure these allowed for "standard" PCM mode (not 1 bit shifted like Philips) by looking at the data sets. This worked beautifully.

esp-qing, 2024-10-28:
>  I noticed that CLK goes down during the second half of the mono 1/2 slot. Is that by design?

Yes, and this situation cannot be changed for the time being.

> Update: I gave up on using the INMP441 or even the ADC config as that also doesn't seem to work, and went with two dedicated i2s DAC / ADC boards instead. I made sure these allowed for "standard" PCM mode (not 1 bit shifted like Philips) by looking at the data sets. This worked beautifully.

I haven't tested this with the INMP441, so I'm not sure if it works. I have tried using PCM5100 with hfp_hf via PCM, it works well. 
Anyway, glad you finally found a way to make it work beautifully.  If you have any other questions, feel free to ask.

closed 2025-04-03 by Alvin1Zhang