espressif/esp-idf #13741: Provisioning gets failed

adhiqmax, 2024-05-04:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

How do I run my provisioner, built using **ESP Idf  v4.4.4**, to provision a node that is using Mesh 1.1? Provisioning a device with Mesh 1.0 works fine, but I encounter issues when provisioning a Mesh 1.1 device.
how to make it work in old provisiner itself?

Below is the provisioner log

I (2601) EXAMPLE: ESP_BLE_MESH_PROVISIONER_PROV_ENABLE_COMP_EVT, err_code 0
I (2611) EXAMPLE: ESP_BLE_MESH_PROVISIONER_ADD_LOCAL_APP_KEY_COMP_EVT, err_code 0
I (133351) EXAMPLE: Received Provisioning frame!!ed!!!Y_TO_MODEL_COMP_EVT, err~�Q�E��
I (133351) EXAMPLE: MACADDR - d1:2d:c5:64:3c:8d:
I (133351) EXAMPLE: ESP_BLE_MESH_PROVISIONER_SET_DEV_UUID_MATCH_COMP_EVT, err_code 0
I (140691) EXAMPLE: ESP_BLE_MESH_PROVISIONER_PROV_LINK_OPEN_EVT, bearer PB-GATT
E (140751) BLE_MESH: Invalid algorithms 0x0003
W (140761) BT_HCI: hci cmd send: disconnect: hdl 0x0, rsn:0x13
W (140761) BT_APPL: gattc_conn_cb: if=3 st=0 id=3 rsn=0x16
I (140761) EXAMPLE: ESP_BLE_MESH_PROVISIONER_PROV_LINK_CLOSE_EVT, bearer PB-GATT, reason 0x16
W (140781) BT_HCI: hcif disc complete: hdl 0x0, rsn 0x16
I (142521) EXAMPLE: ESP_BLE_MESH_PROVISIONER_PROV_LINK_OPEN_EVT, bearer PB-GATT
E (142581) BLE_MESH: Invalid algorithms 0x0003
W (142591) BT_HCI: hci cmd send: disconnect: hdl 0x0, rsn:0x13
W (142591) BT_APPL: gattc_conn_cb: if=3 st=0 id=3 rsn=0x16
I (142591) EXAMPLE: ESP_BLE_MESH_PROVISIONER_PROV_LINK_CLOSE_EVT, bearer PB-GATT, reason 0x16
W (142611) BT_HCI: hcif disc complete: hdl 0x0, rsn 0x16
I (144261) EXAMPLE: ESP_BLE_MESH_PROVISIONER_PROV_LINK_OPEN_EVT, bearer PB-GATT
E (144321) BLE_MESH: Invalid algorithms 0x0003
W (144331) BT_HCI: hci cmd send: disconnect: hdl 0x0, rsn:0x13
W (144331) BT_APPL: gattc_conn_cb: if=3 st=0 id=3 rsn=0x16
I (144331) EXAMPLE: ESP_BLE_MESH_PROVISIONER_PROV_LINK_CLOSE_EVT, bearer PB-GATT, reason 0x16
W (144351) BT_HCI: hcif disc complete: hdl 0x0, rsn 0x16
I (146181) EXAMPLE: ESP_BLE_MESH_PROVISIONER_PROV_LINK_OPEN_EVT, bearer PB-GATT
E (146241) BLE_MESH: Invalid algorithms 0x0003
W (146251) BT_HCI: hci cmd send: disconnect: hdl 0x0, rsn:0x13
W (146251) BT_APPL: gattc_conn_cb: if=3 st=0 id=3 rsn=0x16
I (146251) EXAMPLE: ESP_BLE_MESH_PROVISIONER_PROV_LINK_CLOSE_EVT, bearer PB-GATT, reason 0x16
W (146271) BT_HCI: hcif disc complete: hdl 0x0, rsn 0x16
I (147891) EXAMPLE: ESP_BLE_MESH_PROVISIONER_PROV_LINK_OPEN_EVT, bearer PB-GATT
E (147951) BLE_MESH: Invalid algorithms 0x0003
W (147961) BT_HCI: hci cmd send: disconnect: hdl 0x0, rsn:0x13
W (147961) BT_APPL: gattc_conn_cb: if=3 st=0 id=3 rsn=0x16
I (147961) EXAMPLE: ESP_BLE_MESH_PROVISIONER_PROV_LINK_CLOSE_EVT, bearer PB-GATT, reason 0x16
W (147981) BT_HCI: hcif disc complete: hdl 0x0, rsn 0x16
I (149751) EXAMPLE: ESP_BLE_MESH_PROVISIONER_PROV_LINK_OPEN_EVT, bearer PB-GATT
E (149811) BLE_MESH: Invalid algorithms 0x0003
W (149821) BT_HCI: hci cmd send: disconnect: hdl 0x0, rsn:0x13
W (149821) BT_APPL: gattc_conn_cb: if=3 st=0 id=3 rsn=0x16
I (149821) EXAMPLE: ESP_BLE_MESH_PROVISIONER_PROV_LINK_CLOSE_EVT, bearer PB-GATT, reason 0x16
W (149841) BT_HCI: hcif disc complete: hdl 0x0, rsn 0x16
I (151521) EXAMPLE: ESP_BLE_MESH_PROVISIONER_PROV_LINK_OPEN_EVT, bearer PB-GATT
E (151581) BLE_MESH: Invalid algorithms 0x0003
W (151591) BT_HCI: hci cmd send: disconnect: hdl 0x0, rsn:0x13
W (151591) BT_APPL: gattc_conn_cb: if=3 st=0 id=3 rsn=0x16
I (151591) EXAMPLE: ESP_BLE_MESH_PROVISIONER_PROV_LINK_CLOSE_EVT, bearer PB-GATT, reason 0x16
W (151611) BT_HCI: hcif disc complete: hdl 0x0, rsn 0x16
I (153321) EXAMPLE: ESP_BLE_MESH_PROVISIONER_PROV_LINK_OPEN_EVT, bearer PB-GATT
E (153411) BLE_MESH: Invalid algorithms 0x0003
W (153421) BT_HCI: hci cmd send: disconnect: hdl 0x0, rsn:0x13
W (153421) BT_APPL: gattc_conn_cb: if=3 st=0 id=3 rsn=0x16
I (153421) EXAMPLE: ESP_BLE_MESH_PROVISIONER_PROV_LINK_CLOSE_EVT, bearer PB-GATT, reason 0x16
W (153441) BT_HCI: hcif disc complete: hdl 0x0, rsn 0x16
I (155211) EXAMPLE: ESP_BLE_MESH_PROVISIONER_PROV_LINK_OPEN_EVT, bearer PB-GATT
E (155271) BLE_MESH: Invalid algorithms 0x0003
W (155281) BT_HCI: hci cmd send: disconnect: hdl 0x0, rsn:0x13
W (155281) BT_APPL: gattc_conn_cb: if=3 st=0 id=3 rsn=0x16
I (155281) EXAMPLE: ESP_BLE_MESH_PROVISIONER_PROV_LINK_CLOSE_EVT, bearer PB-GATT, reason 0x16
W (155301) BT_HCI: hcif disc complete: hdl 0x0, rsn 0x16
I (157341) EXAMPLE: ESP_BLE_MESH_PROVISIONER_PROV_LINK_OPEN_EVT, bearer PB-GATT
E (157401) BLE_MESH: Invalid algorithms 0x0003
W (157411) BT_HCI: hci cmd send: disconnect: hdl 0x0, rsn:0x13
W (157411) BT_APPL: gattc_conn_cb: if=3 st=0 id=3 rsn=0x16
I (157411) EXAMPLE: ESP_BLE_MESH_PROVISIONER_PROV_LINK_CLOSE_EVT, bearer PB-GATT, reason 0x16
W (157431) BT_HCI: hcif disc complete: hdl 0x0, rsn 0x16
I (159001) EXAMPLE: ESP_BLE_MESH_PROVISIONER_PROV_LINK_OPEN_EVT, bearer PB-GATT
E (159051) BLE_MESH: Invalid algorithms 0x0003
W (159061) BT_HCI: hci cmd send: disconnect: hdl 0x0, rsn:0x13
W (159061) BT_APPL: gattc_conn_cb: if=3 st=0 id=3 rsn=0x16
I (159061) EXAMPLE: ESP_BLE_MESH_PROVISIONER_PROV_LINK_CLOSE_EVT, bearer PB-GATT, reason 0x16
W (159081) BT_HCI: hcif disc complete: hdl 0x0, rsn 0x16
I (160791) EXAMPLE: ESP_BLE_MESH_PROVISIONER_PROV_LINK_OPEN_EVT, bearer PB-GATT
E (160851) BLE_MESH: Invalid algorithms 0x0003
W (160861) BT_HCI: hci cmd send: disconnect: hdl 0x0, rsn:0x13
W (160861) BT_APPL: gattc_conn_cb: if=3 st=0 id=3 rsn=0x16
I (160861) EXAMPLE: ESP_BLE_MESH_PROVISIONER_PROV_LINK_CLOSE_EVT, bearer PB-GATT, reason 0x16
W (160881) BT_HCI: hcif disc complete: hdl 0x0, rsn 0x16
I (162691) EXAMPLE: ESP_BLE_MESH_PROVISIONER_PROV_LINK_OPEN_EVT, bearer PB-GATT
E (162741) BLE_MESH: Invalid algorithms 0x0003
W (162751) BT_HCI: hci cmd send: disconnect: hdl 0x0, rsn:0x13
W (162751) BT_APPL: gattc_conn_cb: if=3 st=0 id=3 rsn=0x16
I (162751) EXAMPLE: ESP_BLE_MESH_PROVISIONER_PROV_LINK_CLOSE_EVT, bearer PB-GATT, reason 0x16
W (162771) BT_HCI: hcif disc complete: hdl 0x0, rsn 0x16
I (164451) EXAMPLE: ESP_BLE_MESH_PROVISIONER_PROV_LINK_OPEN_EVT, bearer PB-GATT
W (165451) BT_HCI: hcif disc complete: hdl 0x0, rsn 0x8
W (165451) BT_APPL: gattc_conn_cb: if=3 st=0 id=3 rsn=0x8

forx157, 2024-05-06:
Hi, @adhiqmax 
You can disable the `BLE Mesh enhanced provisioning authentication` configuration in menuconfig.

adhiqmax, 2024-05-06:
"BLE Mesh enhanced provisioning authentication" is not found in the menuconfig. Could you please assist in locating it ,below  screenshot attached
[image]

forx157, 2024-05-06:
@adhiqmax 
You can follow the path below:
`(Top) -> Component config -> ESP BLE Mesh Support -> BLE Mesh Provisioning support`.

adhiqmax, 2024-05-06:
@forx157 
Already went through this path for me it is showing like this.

[image]

forx157, 2024-05-07:
@adhiqmax 
You should change the node settings.

closed 2025-03-07 by Alvin1Zhang