espressif/esp-idf #13742: ESP32C3 glitch on GPIO 4

FL0WL0W, 2024-05-04:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

during power up, I get a 20-30us low glitch on GPIO 4. On my board i have a reset delay IC (5230G). As soon as that releases the CHIP_EN pin I get the glitch. I simplified my circuit to isolate the problem. It is just a pullup resistor on GPIO 5, with a diode connecting GPIO 5 to GPIO 4. The cathode is connected to GPIO 4.

I have simplified the firmware down to only this being in app_main
```
void app_main()
{
    while(1) {
        vTaskDelay(1);
    }
}
```

I do not see any mention of this in the datasheet.

suda-morris, 2024-05-06:
With ESP32-C3-DevKitM board, I didn't see a glitch on the GPIO4, after manually pushed the "RST" button. Not sure if it's related to chip revision, could you tell what chip revision of the esp32c3 you're using? 

```
➤ esptool.py chip_id
esptool.py v4.8.dev3
Found 33 serial ports
Serial port /dev/ttyUSB0
Connecting....
Detecting chip type... ESP32-C3
Chip is ESP32-C3 (QFN32) (revision v0.3)
```

FL0WL0W, 2024-05-06:
revision v0.3
```
esptool.py vv4.8.dev1
Serial port /dev/ttyACM0
Connecting...
Detecting chip type... ESP32-C3
Chip is ESP32-C3 (QFN32) (revision v0.3)
Features: WiFi, BLE, Embedded Flash 4MB (XMC)
Crystal is 40MHz
MAC: 34:b4:72:b5:6a:2c
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 460800
Changed.
Warning: ESP32-C3 has no Chip ID. Reading MAC instead.
MAC: 34:b4:72:b5:6a:2c
Hard resetting via RTS pin...
```

Alvin1Zhang, 2025-03-07:
Thanks for reporting and sorry for the slow turnaround, will close due to that it failed to be reproduced, feel free to reopen.

closed 2025-03-07 by Alvin1Zhang