espressif/esp-idf #13744: Allow disabling RS coding on BLK3

ricmoo, 2024-05-05:
### Is your feature request related to a problem?

I use my own encoding scheme to validate the bits of the eFuse have not been tampered with; RS doesn't actually prevent clever alterations that satisfy the RS checksum, and bits can be forcibly set through other means. A zero-bit-counter provides a guarantee to detect tampering, requires very few bits per range (albeit more than RS) and can be applied on sub-block-ranges.

The consequence of the RS encoding being (the unchangeable) default, is that a block cannot be partially written with unused bits being used in for future purposes, since the RS checksum is cemented once the burn chip is called.

The error I get is `E (5461) efuse: Bits are not empty. Write operation is forbidden.` despite DATA being empty. Although, this part I am confused by, because I can only find that error related to 3/4 coding in the code base, but the coding returned for block 3 is RS.

### Describe the solution you'd like.

I would love a way (perhaps via menuconfig) to disable the [RS Coding logic](https://github.com/espressif/esp-idf/blob/master/components/efuse/src/efuse_controller/keys/with_key_purposes/esp_efuse_api_key.c#L110) for eFuse block 3, just allowing `None` to be returned if configured.

### Describe alternatives you've considered.

Another option would be to have a config option that skips applying the coding scheme, but that seems less obvious as to the consequences.

### Additional context.

Related docs that makes it seem intentional, but believe it should not be forced:

- [this note at the bottom of the Custom MAC section](https://docs.espressif.com/projects/esp-idf/en/latest/esp32c3/api-reference/system/misc_system_api.html#custom-mac-address-in-efuse)
- [this section](https://docs.espressif.com/projects/esptool/en/latest/esp32c3/espefuse/index.html#coding-scheme) in the ESP32-c3 docs.

Thanks! :)

0xjakob, 2024-05-06:
Hi @ricmoo!

it might not be obvious from the TRM and documentation, but the eFuse controller hardware expects the data in RS coding, otherwise it won't burn the eFuses.

ricmoo, 2024-05-07:
Does that mean at the *hardware level*, block 0 is treated differently than block 3?

I was also hoping to allows the eFuses to hold an attestation key, but if a developer wished to jail-break their device, the firmware would allow it, but only after burning 1’s across the board so that the attention key was no longer readable by their own home brew firmware.

0xjakob, 2024-05-07:
> Does that mean at the hardware level, block 0 is treated differently than block 3?

Yes. Except for block 0, all blocks have to be written with RS encoding.

mahavirj, 2024-05-15:
For C3 case:

- BLOCK0 allows each bit to be programmed separately
- BLOCK1 is not user programmable, fixed during manufacturing time
- BLOCK2 ~ BLOCK10 can be programmed only once, repeated programming is not allowed. It is not possible to disabled the encoding scheme here.

Please refer to section 4.3.1.3 and 4.3.2 in ESP32-C3 TRM for more information.

Hope this helps!

closed 2024-05-15 by mahavirj