espressif/esp-idf #13745: esp32-c3, can't disable flash encryption

zzzh, 2024-05-05:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v4.4.2

### Espressif SoC revision.

ESP32-C3 (revision 4)

### Operating System used.

Linux

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### Development Kit.

ESP32C3-CORE

### Power Supply used.

External 5V

### What is the expected behavior?

after running "espefuse.py burn_efuse SPI_BOOT_CRYPT_CNT", SPI_BOOT_CRYPT_CNT become 0x10 from 0x01

### What is the actual behavior?

SPI_BOOT_CRYPT_CNT become 0x11

### Steps to reproduce.

1. In [Project Configuration Menu], disable [Enable flash encryption on boot], then save and exit.

2. Open project configuration menu again and double-check have disabled this option

3. With flash encryption disabled, build and flash the new bootloader and application by running idf.py flash.

4. Use espefuse.py to disable the SPI_BOOT_CRYPT_CNT by running:
espefuse.py burn_efuse SPI_BOOT_CRYPT_CNT

### Debug Logs.

```plain
before disable:
SPI_BOOT_CRYPT_CNT (BLOCK0)                        Enables flash encryption when 1 or 3 bits are set  = Disable R/W (0b001)

after disable:
SPI_BOOT_CRYPT_CNT (BLOCK0)                        Enables flash encryption when 1 or 3 bits are set  = Disable R/W (0b011)
```

### More Information.

It can only be done one time per chip for esp32-c3?
I remember it's 3 times for esp32

mahavirj, 2024-05-06:
@zzzh 

> It can only be done one time per chip for esp32-c3?

Yes. ESP32-C3 has only 3 bits for the flash crypt count field and hence the flash encryption can only be disabled once (in the development mode).

mahavirj, 2024-05-06:
@zzzh 

> SPI_BOOT_CRYPT_CNT (BLOCK0)                        Enables flash encryption when 1 or 3 bits are set  = Disable R/W (0b011)

As per this log that you have shared, the flash encryption is disabled on the device. So, it will expect plaintext artifacts on the flash. If you are flashing encrypted artifacts (e.g., bootloader, application) then device won't boot with invalid header/image related error logs.

Can you please share your detailed instructions, device side logs here?

zzzh, 2024-05-06:
@mahavirj,

So, 0b011 means disabled?  
I see, it's my wrong understanding.
No need to paste more info here, thanks for the help.

closed 2024-05-06 by zzzh