espressif/esp-idf #13747: ASSERT_PARAM(-218959118 0), in arch_main.c at line 365 while using BLE

Koxx3, 2024-05-05:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

# Environment

    Development Kit: none (this is a custom board, but could reproduce the issue with a ESP32-WROOM-based board too)
    Module or chip used: ESP32-D0WD
    IDF version: v4.4.4 from Tasmota 2023.06.02
    Build System: Platformio/Make
    Compiler version: xtensa-esp32-elf-gcc (crosstool-NG esp-2021r2-patch5) 8.4.0
    Operating System: Windows
    Power Supply: external 5V or USB

# Problem Description

My software use BLE, Wifi (disabled during this test, but enable in sdkconfig), TWAI, UART, SPI, U2C and UART...

It scans for BLE beacons while it is connected in BLE a 1 smartphone.
Advertising is enabled while the smartphone is connected.
It uses NimBLE.

I crash after 5 to 8h if a smartphone is connected. I don't see any crash if the smartphone isn't connected.

I've tried many many configurations following previous BLE stack errors reported on this github : 
- customize bt controller adv report 
- disable BLE sleep
- remove _btdm_sleep_check_duration callback from bt.c
- increase watchdog timeout
- change scan stragegies and filtering

Without success ...

I can't move to higher IDF without a large impact on my software... which is not possible considering we are launching the production soon.
IDF 4.4.5 is still possible, but it crashes too (I didn't note the exception/backtrace)

Any help please ?

### **Expected Behaviour**

The BLE stack should not panic when scanning.

### **Actual Behaviour**

The application will panic after seemingly non-deterministic intervals, generally during BLE scans.

### **Steps to ### reproduce**

N/A considering the complexity of the application.

# Debug Logs

```
ASSERT_PARAM(-218959118 0), in arch_main.c at line 365
Guru Meditation Error: Core  0 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x40090b88: 000000f0 00004136 f01d0000
Core  0 register dump:
PC      : 0x40090b8f  PS      : 0x00060734  A0      : 0x80086505  A1      : 0x3ffb6020  
A2      : 0x00000000  A3      : 0xf2f2f2f2  A4      : 0x00000000  A5      : 0x3f43627e  
A6      : 0x0000016d  A7      : 0xfffffffc  A8      : 0x8000814b  A9      : 0x3ffb5f90  
A10     : 0x00000000  A11     : 0x3ffb5fb3  A12     : 0x3ffb5f5f  A13     : 0x00000035  
A14     : 0x00000000  A15     : 0x3ffb5f64  SAR     : 0x00000004  EXCCAUSE: 0x00000000  
EXCVADDR: 0x00000000  LBEG    : 0x4008640d  LEND    : 0x40086415  LCOUNT  : 0x00000000  

Backtrace: 0x40090b8c:0x3ffb6020 0x40086502:0x3ffb6040 0x401a1691:0x3ffb6060 0x40019fb5:0x3ffb6080 0x4001a1f2:0x3ffb60b0 0x401afe5a:0x3ffb60d0 0x401b15e1:0x3ffb6140 0x401ae5ff:0x3ffb6190 0x401ab7d2:0x3ffb61e0 0x40019d11:0x3ffb6220 0x40055b4d:0x3ffb6240 0x401a1a83:0x3ffb6260 0x401a20e5:0x3ffb6280 0x40096cad:0x3ffb62b0

Decoded stack trace :
r_assert at C:\Users\koxx3\.platformio\packages\framework-espidf\components\bt\controller\esp32/bt.c:1862
r_assert_param at ??:?
r_platform_reset at ??:?
?? ??:0
?? ??:0
r_llm_le_adv_report_ind at ??:?
r_llm_pdu_defer at ??:?
r_lld_pdu_check at ??:?
r_lld_evt_deffered_elt_handler at ??:?
?? ??:0
?? ??:0
r_rw_schedule at ??:?
btdm_controller_task at ??:?
vPortTaskWrapper at C:\Users\koxx3\.platformio\packages\framework-espidf\components\freertos\port\xtensa/port.c:142

MEM left 75468 idle = 5064     tft = 2236      wifi = 4492     btn = 2440      canrx = 1572    cantx = 1760    sound = 220     serial = 4488   btCntr = 1580           btHost = 4932   CAN_WD_BI = 1616        CAN_LORX = 3524  Tmr = 1384      esp_timer = 3140        loopTask = 4200 ipc0 = 860      ipc1 = 992      sys_evt = 4200  arduino_events = 4200

```

zhp0406, 2024-05-07:
Thank you for reporting the issue. It seems to be related to the BLE controller's memory. The reason for this issue is that the controller failed to allocate memory. We will attempt to replicate the problem. Could you provide a demo that reproduces this issue? This would be helpful in resolving the problem.

zhp0406, 2024-05-07:
The BLE controller allocates a block of memory during initialization, which is exclusively for the controller's use. There's no way to determine from the outside whether there has been a memory leak in the memory allocated for the controller.

Koxx3, 2024-05-07:
thanks a lot for the answer. I've spent a lot of time tracking this issue...

I switch to IDF 4.4.5, but no improvement.

I'll try to patch IDF-4.4.5 BT controller with 4.4.7 BT controller to ensure this issue isn't already fixed in the latest version.

Some questions :
1/ is it possible to observe the memory leak from outside of the BT controller ?
2/ do you think disconnecting the client can release the memory ?
3/ would is possible to allocate more memory to this module by any mean ?
4/ considering it seems to be "r_llm_le_adv_report_ind" the issue, is there any way to prevent it ? with advertising configuration, scan configuration, sdk configuration or anything else ?

A said in the first post, it will be very complicated to release a test module if other software parts are involved, but I'll try during next weeks.

Koxx3, 2024-05-07:
patching BT controller 4.4.5 with 4.4.7 doesn't work unfortunatly  : 
```
c:/users/koxx3/.platformio/packages/toolchain-xtensa-esp32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: C:/Users/koxx3/.platformio/packages/framework-espidf-4.4.5-tasmo/components/bt/controller/lib_esp32/esp32\libbtdm_app.a(ld_acl.o): in function `r_ld_acl_start':
ld_acl.c:(.text+0x117c): undefined reference to `ld_acl_sniff_exit'
c:/users/koxx3/.platformio/packages/toolchain-xtensa-esp32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: ld_acl.c:(.text+0x125c): undefined reference to `ld_acl_sniff_exit'
c:/users/koxx3/.platformio/packages/toolchain-xtensa-esp32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: C:/Users/koxx3/.platformio/packages/framework-espidf-4.4.5-tasmo/components/bt/controller/lib_esp32/esp32\libbtdm_app.a(rf_espressif.o):(.iram1.26+0x0): undefined reference to `phy_bt_ifs_set'
c:/users/koxx3/.platformio/packages/toolchain-xtensa-esp32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: C:/Users/koxx3/.platformio/packages/framework-espidf-4.4.5-tasmo/components/bt/controller/lib_esp32/esp32\libbtdm_app.a(rf_espressif.o):(.iram1.26+0x13): undefined reference to `phy_bt_ifs_set'
c:/users/koxx3/.platformio/packages/toolchain-xtensa-esp32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: C:/Users/koxx3/.platformio/packages/framework-espidf-4.4.5-tasmo/components/bt/controller/lib_esp32/esp32\libbtdm_app.a(lm_sco.o):(.text+0x4): undefined reference to `lm_nb_sync_active'
c:/users/koxx3/.platformio/packages/toolchain-xtensa-esp32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: C:/Users/koxx3/.platformio/packages/framework-espidf-4.4.5-tasmo/components/bt/controller/lib_esp32/esp32\libbtdm_app.a(lm_sco.o):(.text+0x8): undefined reference to `lm_sync_nego'
c:/users/koxx3/.platformio/packages/toolchain-xtensa-esp32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: C:/Users/koxx3/.platformio/packages/framework-espidf-4.4.5-tasmo/components/bt/controller/lib_esp32/esp32\libbtdm_app.a(lm_sco.o):(.text+0xc): undefined reference to `lm_sync_conf'
c:/users/koxx3/.platformio/packages/toolchain-xtensa-esp32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: C:/Users/koxx3/.platformio/packages/framework-espidf-4.4.5-tasmo/components/bt/controller/lib_esp32/esp32\libbtdm_app.a(lm_sco.o):(.text+0x10): undefined reference to `lm_nego_cnt'
c:/users/koxx3/.platformio/packages/toolchain-xtensa-esp32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: C:/Users/koxx3/.platformio/packages/framework-espidf-4.4.5-tasmo/components/bt/controller/lib_esp32/esp32\libbtdm_app.a(lm_sco.o):(.text+0x14): undefined reference to `lm_nego_cntl'
c:/users/koxx3/.platformio/packages/toolchain-xtensa-esp32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: C:/Users/koxx3/.platformio/packages/framework-espidf-4.4.5-tasmo/components/bt/controller/lib_esp32/esp32\libbtdm_app.a(lm_sco.o):(.text+0x18): undefined reference to `lm_nego_max_cnt'
c:/users/koxx3/.platformio/packages/toolchain-xtensa-esp32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: C:/Users/koxx3/.platformio/packages/framework-espidf-4.4.5-tasmo/components/bt/controller/lib_esp32/esp32\libbtdm_app.a(lm_sco.o):(.text+0x1c): undefined reference to `lm_nego_pkt_used'
```

zhp0406, 2024-05-07:
1-Is it possible to observe memory leaks from outside of the BT controller?
Answer: Currently, it's not possible to detect memory leaks externally. The memory allocation for the controller is predetermined, and there's no API available for external monitoring of its memory usage. However, for debugging purposes, I can provide you with a new library and internal interfaces to access the controller's memory usage.Do you require access to this interface?
2-Do you think disconnecting the client can release the memory?
Answer: When there are no connections, a portion of the memory is indeed released. This is because the memory size is fixed after the controller's initialization.
3-Is it possible to allocate more memory to this module by any means?
Answer: The memory size allocated during controller initialization is calculated based on factors such as duplicate list size and maximum connection count. In theory, there should not be any assertion errors as sufficient memory is already allocated. Currently, there's no external interface to adjust memory allocation size, but internal adjustments can be made.
4-Considering "r_llm_le_adv_report_ind" seems to be the issue, is there any way to prevent it? Whether through advertising configuration, scan configuration, reducing MTU (255 bytes negotiated by the iOS client in my case), SDK configuration, or anything else?
Answer: The consumption of memory decreases when there are no established connections, and it also reduces when there are fewer devices in the air (due to duplicate list reasons). Other scenarios require further assessment.
5-As mentioned in the first post, it might be challenging to release a test module if other software components are involved, but I'll attempt it in the next few weeks.
Answer: Currently, it seems that this issue is solely related to BLE and not connected to other components. I'm still attempting to replicate the problem here. If you can replicate the issue using BLE, it would greatly aid in resolving the problem.

Koxx3, 2024-05-07:
ok, thanks again.
I'll keep you updated of the 4.4.5/4.4.7 patch tests.
if it fails, yes, it would be really interesting to have few APIs to monitor the memory and debug.

zhp0406, 2024-05-07:
[d166426.zip](https://github.com/espressif/esp-idf/files/15235517/d166426.zip)
Please replace the esp-idf/components/bt/controller/lib_esp32/esp32/libbtdm_app.a file. In the new lib file, I've added a new API that can retrieve the memory usage of the controller. Below is an example and the printed log.
When the BT controller compile version is printed as [d166426], it indicates successful application.

```
I (586) BTDM_INIT: BT controller compile version [d166426]
```

code：

```
extern uint16_t ke_get_heap_free_size(void);
uint16_t free_size;
while (1) {
free_size = ke_get_heap_free_size();
printf("free size %d byte\n",free_size);
vTaskDelay(10000 / portTICK_PERIOD_MS);
}
```

log：

```
I (29) boot: ESP-IDF v4.4.4-dirty 2nd stage bootloader
I (29) boot: compile time 20:28:41
I (29) boot: chip revision: v3.1
I (33) boot_comm: chip revision: 3, min. bootloader chip revision: 0
I (40) boot.esp32: SPI Speed      : 40MHz
I (44) boot.esp32: SPI Mode       : DIO
I (49) boot.esp32: SPI Flash Size : 2MB
I (53) boot: Enabling RNG early entropy source...
I (59) boot: Partition Table:
I (62) boot: ## Label            Usage          Type ST Offset   Length
I (70) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (77) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (85) boot:  2 factory          factory app      00 00 00010000 00100000
I (92) boot: End of partition table
I (96) boot_comm: chip revision: 3, min. application chip revision: 0
I (103) esp_image: segment 0: paddr=00010020 vaddr=3f400020 size=1d050h (118864) map
I (155) esp_image: segment 1: paddr=0002d078 vaddr=3ffbdb60 size=02fa0h ( 12192) load
I (160) esp_image: segment 2: paddr=00030020 vaddr=400d0020 size=7ca74h (510580) map
I (346) esp_image: segment 3: paddr=000aca9c vaddr=3ffc0b00 size=017f8h (  6136) load
I (348) esp_image: segment 4: paddr=000ae29c vaddr=40080000 size=17658h ( 95832) load
I (402) boot: Loaded app from partition at offset 0x10000
I (402) boot: Disabling RNG early entropy source...
I (414) cpu_start: Pro cpu up.
I (414) cpu_start: Starting app cpu, entry point is 0x400811d8
0x400811d8: call_start_cpu1 at /home/zhanghaipeng/esp_v4/esp-idf/components/esp_system/port/cpu_start.c:148

I (0) cpu_start: App cpu up.
I (430) cpu_start: Pro cpu start user code
I (430) cpu_start: cpu freq: 160000000
I (431) cpu_start: Application information:
I (435) cpu_start: Project name:     gatt_client_demo
I (441) cpu_start: App version:      v4.4.4-dirty
I (446) cpu_start: Compile time:     May  7 2024 20:28:36
I (452) cpu_start: ELF file SHA256:  c3c24db549fa719a...
I (458) cpu_start: ESP-IDF:          v4.4.4-dirty
I (464) heap_init: Initializing. RAM available for dynamic allocation:
I (471) heap_init: At 3FFAFF10 len 000000F0 (0 KiB): DRAM
I (477) heap_init: At 3FFB6388 len 00001C78 (7 KiB): DRAM
I (483) heap_init: At 3FFB9A20 len 00004108 (16 KiB): DRAM
I (489) heap_init: At 3FFC8AC8 len 00017538 (93 KiB): DRAM
I (495) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (502) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (508) heap_init: At 40097658 len 000089A8 (34 KiB): IRAM
I (516) spi_flash: detected chip: generic
I (519) spi_flash: flash io: dio
W (523) spi_flash: Detected size(4096k) larger than the size in the binary image header(2048k). Using the size in the binary image header.
I (537) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (586) BTDM_INIT: BT controller compile version [d166426]
I (586) system_api: Base MAC address is not set
I (586) system_api: read default base MAC address from EFUSE
I (596) phy_init: phy_version 4670,719f9f6,Feb 18 2021,17:07:07
I (1016) GATTC_DEMO: REG_EVT
free size 9556 byte
I (1016) GATTC_DEMO: scan start success
free size 9364 byte
free size 9364 byte
free size 9364 byte
free size 9540 byte
```

Koxx3, 2024-05-07:
ok. I am currently trying with IDF 4.4.7 version.
I'll be out of the office for few days, I'll be back in few days to continue tests.
If it's not too much to ask, can you generate it for IDF 4.4.5 and 4.4.7 please ?

Koxx3, 2024-05-07:
ok, let's go for a test in IDF 4.4.4 + libbtdm_app.a patch before the long weekend.

compilation test : 
```
15:29:32.558 > I (216) BTDM_INIT: BT controller compile version [d166426]
15:29:32.558 > ESP_ERROR_CHECK failed: esp_err_t 0x101 (ESP_ERR_NO_MEM) at 0x40092ab0
15:29:32.559 > file: ".pio/libdeps/smartdisplay_vesc_35v2_pcb210/nimble/src/NimBLEDevice.cpp" line 879
15:29:32.560 > func: static void NimBLEDevice::init(const string&)
15:29:32.560 > expression: esp_bt_controller_init(&bt_cfg)
15:29:32.561 >
15:29:32.561 > abort() was called at PC 0x40092ab3 on core 1
15:29:32.561 >
15:29:32.561 >
15:29:32.561 > Backtrace: 0x40083842:0x3ffd93c0 0x40092abd:0x3ffd93e0 0x400991da:0x3ffd9400 0x40092ab3:0x3ffd9470 0x40115622:0x3ffd9490 0x400d8517:0x3ffd94e0 0x4010726f:0x3ffd96c0 0x401471da:0x3ffd97c0 0x40096a69:0x3ffd97e0
15:29:32.564 > 
15:29:32.564 >
15:29:32.564 >
15:29:32.564 >
15:29:32.564 > ELF file SHA256: 22f76682b73e94f5
15:29:32.564 >
15:29:32.564 > Rebooting...
```

NimBle function at this line : 

```
esp_err_t esp_bt_controller_init(esp_bt_controller_config_t *cfg)
{
    esp_err_t err;
    uint32_t btdm_cfg_mask = 0;

#if CONFIG_BTDM_CTRL_HLI
    hli_queue_setup_pinned_to_core(CONFIG_BTDM_CTRL_PINNED_TO_CORE);
#endif /* CONFIG_BTDM_CTRL_HLI */

    //if all the bt available memory was already released, cannot initialize bluetooth controller
    if (btdm_dram_available_region[0].mode == ESP_BT_MODE_IDLE) {
        return ESP_ERR_INVALID_STATE;
    }

    osi_funcs_p = (struct osi_funcs_t *)malloc_internal_wrapper(sizeof(struct osi_funcs_t));
    if (osi_funcs_p == NULL) {
        return ESP_ERR_NO_MEM;
    }

    memcpy(osi_funcs_p, &osi_funcs_ro, sizeof(struct osi_funcs_t));
    if (btdm_osi_funcs_register(osi_funcs_p) != 0) {
        return ESP_ERR_INVALID_ARG;
    }

    if (btdm_controller_status != ESP_BT_CONTROLLER_STATUS_IDLE) {
        return ESP_ERR_INVALID_STATE;
    }

    if (cfg == NULL) {
        return ESP_ERR_INVALID_ARG;
    }

    if (cfg->controller_task_prio != ESP_TASK_BT_CONTROLLER_PRIO
            || cfg->controller_task_stack_size < ESP_TASK_BT_CONTROLLER_STACK) {
        return ESP_ERR_INVALID_ARG;
    }

    //overwrite some parameters
    cfg->bt_max_sync_conn = CONFIG_BTDM_CTRL_BR_EDR_MAX_SYNC_CONN_EFF;
    cfg->magic  = ESP_BT_CONTROLLER_CONFIG_MAGIC_VAL;

    if (((cfg->mode & ESP_BT_MODE_BLE) && (cfg->ble_max_conn <= 0 || cfg->ble_max_conn > BTDM_CONTROLLER_BLE_MAX_CONN_LIMIT))
            || ((cfg->mode & ESP_BT_MODE_CLASSIC_BT) && (cfg->bt_max_acl_conn <= 0 || cfg->bt_max_acl_conn > BTDM_CONTROLLER_BR_EDR_MAX_ACL_CONN_LIMIT))
            || ((cfg->mode & ESP_BT_MODE_CLASSIC_BT) && (cfg->bt_max_sync_conn > BTDM_CONTROLLER_BR_EDR_MAX_SYNC_CONN_LIMIT))) {
[... 106 lines cut ...]
        goto error;
    }

    btdm_controller_status = ESP_BT_CONTROLLER_STATUS_INITED;

    return ESP_OK;

error:

    bt_controller_deinit_internal();

    return err;
}

```

build test with ke_get_heap_free_size : 
```
Linking .pio\build\smartdisplay_vesc_35v2_pcb210\firmware.elf
c:/users/koxx3/.platformio/packages/toolchain-xtensa-esp32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: .pio/build/smartdisplay_vesc_35v2_pcb210/src/main.o:(.literal._Z4loopv+0x2c): undefined reference to `ke_get_heap_free_size()'
c:/users/koxx3/.platformio/packages/toolchain-xtensa-esp32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: .pio/build/smartdisplay_vesc_35v2_pcb210/src/main.o: in function `loop()':
C:\Users\koxx3\Documents\Coding\SmartController_SmartDisplay_ESP32_priv3/src/main.cpp:1075: undefined reference to `ke_get_heap_free_size()'
collect2.exe: error: ld returned 1 exit status
*** [.pio\build\smartdisplay_vesc_35v2_pcb210\firmware.elf] Error 1
```

Koxx3, 2024-05-07:
in debug level : 

```
D (1299) BTDM_INIT: Release DRAM [0x3ffb2730] - [0x3ffb6388]
D (1300) intr_alloc: Connected src -5 to int 29 (cpu 1)
I (1300) BTDM_INIT: BT controller compile version [d166426]
D (1301) BTDM_INIT: .data initialise [0x3ffae6e0] <== [0x4000d890]
D (1301) BTDM_INIT: .bss initialise [0x3ffb0000] - [0x3ffb09a8]
D (1302) BTDM_INIT: .bss initialise [0x3ffb09a8] - [0x3ffb1ddc]
D (1303) BTDM_INIT: .bss initialise [0x3ffb1ddc] - [0x3ffb2730]
D (1304) BTDM_INIT: .bss initialise [0x3ffb8000] - [0x3ffb9a20]
D (1304) BTDM_INIT: .bss initialise [0x3ffbdb28] - [0x3ffbdb5c]
ESP_ERROR_CHECK failed: esp_err_t 0x101 (ESP_ERR_NO_MEM) at 0x40092ad4
file: ".pio/libdeps/smartdisplay_vesc_35v2_pcb210/nimble/src/NimBLEDevice.cpp" line 879
func: static void NimBLEDevice::init(const string&)
expression: esp_bt_controller_init(&bt_cfg)

abort() was called at PC 0x40092ad7 on core 1
```

zhp0406, 2024-05-08:
The reason for this issue is the mismatch between the "magic value" externally and internally in the controller. The external controller value is configured by ESP_BT_CONTROLLER_CONFIG_MAGIC_VAL and needs to be consistent internally.

It's worth noting that in version v4.4.4, the internal "magic value" is 0x20200622. If you encounter this problem, you need to verify whether you are using version v4.4.4.
[image]

zhp0406, 2024-05-08:
replace esp-idf/components/bt/controller/lib_esp32/esp32/libbtdm_app.a
and 
extern uint16_t ke_get_heap_free_size(void);
[image]

Koxx3, 2024-05-11:
I managed to remove Tasmota dependencies and work with "pure" packages now: 

```
PACKAGES:
 - framework-arduinoespressif32 @ 3.20016.0 (2.0.16)
 - framework-espidf @ 3.40407.0 (4.4.7)
 - tool-cmake @ 3.16.4
 - tool-esptoolpy @ 1.40501.0 (4.5.1)
 - tool-idf @ 1.0.1
 - tool-mconf @ 1.4060000.20190628 (406.0.0)
 - tool-mkspiffs @ 2.230.0 (2.30)
 - tool-ninja @ 1.9.0
 - toolchain-esp32ulp @ 1.23500.220830 (2.35.0)
 - toolchain-xtensa-esp32 @ 8.4.0+2021r2-patch5
 - 
```

here are the IDF 4.4.7 results with stock "libbtdm_app.a" : 

```
r_assert at C:\Users\koxx3\.platformio\packages\framework-espidf\components\bt\controller\esp32/bt.c:1902
r_assert_param at ??:?
r_platform_reset at ??:?
?? ??:0
?? ??:0
r_llm_le_adv_report_ind at ??:?
r_llm_pdu_defer at ??:?
r_lld_pdu_check at ??:?
r_lld_evt_deffered_elt_handler at ??:?
?? ??:0
?? ??:0
r_rw_schedule at ??:?
btdm_controller_task at ??:?
vPortTaskWrapper at C:\Users\koxx3\.platformio\packages\framework-espidf\components\freertos\port\xtensa/port.c:142

```

board 1 crashed after 52 min, then 11h43m
board 2 crashed after 12h10m

I tried you patched "libbtdm_app.a" with IDF 4.4.7, but it doesn't work ;)

zhp0406, 2024-05-11:
@Koxx3 
can you use ke_get_heap_free_size now?

Koxx3, 2024-05-11:
ok, I'll try to switch back to officiel 4.4.4 and use ke_get_heap_free_size

zhp0406, 2024-05-11:
I guess there might be a subtle memory leak in the controller. Perhaps some patterns could be discovered by using ke_get_heap_free_size... thanks for your testing.

Koxx3, 2024-05-11:
argh. I cleaned everything (build folder & cache).
I made sure only 1 esp-idf remained.
... but it still doesn't link.

[image]

```
Linking .pio\build\smartdisplay_vesc_35v2_pcb210\firmware.elf
c:/users/koxx3/.platformio/packages/toolchain-xtensa-esp32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: .pio/build/smartdisplay_vesc_35v2_pcb210/src/main.o:(.literal._Z4loopv+0x30): undefined reference to `ke_get_heap_free_size()'
c:/users/koxx3/.platformio/packages/toolchain-xtensa-esp32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: .pio/build/smartdisplay_vesc_35v2_pcb210/src/main.o: in function `loop()':
C:\Users\koxx3\Documents\Coding\SmartController_SmartDisplay_ESP32_priv3/src/main.cpp:1075: undefined reference to `ke_get_heap_free_size()'
collect2.exe: error: ld returned 1 exit status
*** [.pio\build\smartdisplay_vesc_35v2_pcb210\firmware.elf] Error 1

```

zhp0406, 2024-05-11:
Can you provide me with your current commit ID? I'd like to perform some local testing.

Koxx3, 2024-05-11:
sorry updated too late : 
At execution (without using ke_get_heap_free_size), it clearly state I am using d166426 :
I (1553) BTDM_INIT: BT controller compile version [d166426]

Koxx3, 2024-05-11:
> Can you provide me with your current commit ID? I'd like to perform some local testing.

how do I do that ?

Koxx3, 2024-05-11:
```
; IDF 4.4.4 / arduino 2.0.16 / building fine
platform = espressif32 @ ^6.6.0
platform_packages = 
	tool-mkspiffs @ 2.230.0
	platformio/framework-espidf @ ~3.40404.0
	platformio/framework-arduinoespressif32 @ ~3.20016.0

framework = arduino, espidf
board = esp32dev

```

zhp0406, 2024-05-11:
If you don't call ke_get_heap_free_size, the compilation should still proceed without issues, right?

Koxx3, 2024-05-11:
> If you don't call ke_get_heap_free_size, the compilation should still proceed without issues, right?

yes

Koxx3, 2024-05-11:
running logs in debug level : 
https://koxx3.fr.eu.org:8086/device-monitor-240511-095738.zip

zhp0406, 2024-05-11:
Are you building your project based on Arduino? I didn't encounter any issues while testing on ESP-IDF, and I can use the controller's internal functions externally without problems. Perhaps I could place this function inside the controller, so you wouldn't need to use it externally (ke_get_heap_free_size) anymore.

[image]

Koxx3, 2024-05-11:
oh !!! you gave me an idea !
I forced the declaration in esp_bt.h.

and it works now !

[image]

```
10:54:16.786 > main : free size = 6760 bytes
10:54:22.126 > main : free size = 6760 bytes
10:54:27.461 > main : free size = 6760 bytes
10:54:32.778 > main : free size = 6748 bytes
```

Koxx3, 2024-05-11:
after only few minutes, we can see the leak !

```
10:54:16.786 > main : free size = 6760 bytes
10:54:22.126 > main : free size = 6760 bytes
10:54:27.461 > main : free size = 6760 bytes
10:54:32.778 > main : free size = 6748 bytes
10:54:38.107 > main : free size = 6768 bytes
10:54:43.449 > main : free size = 6744 bytes
10:54:48.793 > main : free size = 6760 bytes
10:54:54.135 > main : free size = 6944 bytes
10:54:59.476 > main : free size = 6760 bytes
10:55:04.818 > main : free size = 6760 bytes
10:55:10.156 > main : free size = 6748 bytes
10:55:15.527 > main : free size = 6760 bytes
10:55:20.861 > main : free size = 6760 bytes
10:55:26.196 > main : free size = 6764 bytes
10:55:31.523 > main : free size = 6744 bytes
10:55:36.860 > main : free size = 6760 bytes
10:55:42.191 > main : free size = 6768 bytes
10:55:47.530 > main : free size = 6748 bytes
10:55:52.854 > main : free size = 6748 bytes
10:55:58.180 > main : free size = 6768 bytes
10:56:03.504 > main : free size = 6760 bytes
10:56:08.836 > main : free size = 6744 bytes
10:56:14.177 > main : free size = 6744 bytes
10:56:19.519 > main : free size = 6760 bytes
10:56:24.847 > main : free size = 6768 bytes
10:56:30.171 > main : free size = 6760 bytes
10:56:35.532 > main : free size = 6944 bytes
10:56:40.872 > main : free size = 6740 bytes
10:56:46.223 > main : free size = 6748 bytes
10:56:51.581 > main : free size = 6760 bytes
10:56:56.925 > main : free size = 6760 bytes
10:57:02.262 > main : free size = 6760 bytes
10:57:07.604 > main : free size = 6744 bytes
10:57:12.935 > main : free size = 6744 bytes
10:57:18.281 > main : free size = 6744 bytes
10:57:23.634 > main : free size = 6744 bytes
10:57:28.973 > main : free size = 6732 bytes
10:57:34.309 > main : free size = 6744 bytes
10:57:39.651 > main : free size = 6744 bytes
10:57:45.000 > main : free size = 6736 bytes
10:57:50.340 > main : free size = 6732 bytes
10:57:55.664 > main : free size = 6744 bytes
10:58:01.005 > main : free size = 6728 bytes
```

Koxx3, 2024-05-11:
yes, it really seems to leak :

```
10:54:16.786 > main : free size = 6760 bytes
10:54:22.126 > main : free size = 6760 bytes
10:54:27.461 > main : free size = 6760 bytes
10:54:32.778 > main : free size = 6748 bytes
10:54:38.107 > main : free size = 6768 bytes
10:54:43.449 > main : free size = 6744 bytes
10:54:48.793 > main : free size = 6760 bytes
10:54:54.135 > main : free size = 6944 bytes
10:54:59.476 > main : free size = 6760 bytes
10:55:04.818 > main : free size = 6760 bytes
10:55:10.156 > main : free size = 6748 bytes
10:55:15.527 > main : free size = 6760 bytes
10:55:20.861 > main : free size = 6760 bytes
10:55:26.196 > main : free size = 6764 bytes
10:55:31.523 > main : free size = 6744 bytes
10:55:36.860 > main : free size = 6760 bytes
10:55:42.191 > main : free size = 6768 bytes
10:55:47.530 > main : free size = 6748 bytes
10:55:52.854 > main : free size = 6748 bytes
10:55:58.180 > main : free size = 6768 bytes
10:56:03.504 > main : free size = 6760 bytes
10:56:08.836 > main : free size = 6744 bytes
10:56:14.177 > main : free size = 6744 bytes
10:56:19.519 > main : free size = 6760 bytes
10:56:24.847 > main : free size = 6768 bytes
10:56:30.171 > main : free size = 6760 bytes
10:56:35.532 > main : free size = 6944 bytes
10:56:40.872 > main : free size = 6740 bytes
10:56:46.223 > main : free size = 6748 bytes
10:56:51.581 > main : free size = 6760 bytes
10:56:56.925 > main : free size = 6760 bytes
10:57:02.262 > main : free size = 6760 bytes
10:57:07.604 > main : free size = 6744 bytes
10:57:12.935 > main : free size = 6744 bytes
10:57:18.281 > main : free size = 6744 bytes
10:57:23.634 > main : free size = 6744 bytes
10:57:28.973 > main : free size = 6732 bytes
10:57:34.309 > main : free size = 6744 bytes
10:57:39.651 > main : free size = 6744 bytes
10:57:45.000 > main : free size = 6736 bytes
10:57:50.340 > main : free size = 6732 bytes
10:57:55.664 > main : free size = 6744 bytes
10:58:01.005 > main : free size = 6728 bytes
10:58:06.342 > main : free size = 6744 bytes
10:58:11.688 > main : free size = 6728 bytes
10:58:17.031 > main : free size = 6928 bytes
10:58:22.377 > main : free size = 6744 bytes
10:58:27.711 > main : free size = 6744 bytes
10:58:33.060 > main : free size = 6744 bytes
10:58:38.405 > main : free size = 6764 bytes
10:58:43.738 > main : free size = 6744 bytes
10:58:49.068 > main : free size = 6744 bytes
10:58:54.401 > main : free size = 6728 bytes
10:58:59.731 > main : free size = 6744 bytes
10:59:05.089 > main : free size = 6744 bytes
10:59:10.424 > main : free size = 6728 bytes
10:59:15.767 > main : free size = 6724 bytes
10:59:21.146 > main : free size = 6724 bytes
10:59:26.511 > main : free size = 6724 bytes
10:59:31.861 > main : free size = 6712 bytes
```

Koxx3, 2024-05-11:
also, disconnecting and reconnecting BLE doesn't release the memory.

zhp0406, 2024-05-11:
If you can analyze the scenarios of memory leaks (broadcasting, scanning, connecting, etc.), it would be very helpful in resolving the issue.

zhp0406, 2024-05-11:
I recently discovered an issue with bidirectional frequent initiation of connection parameter updates leading to memory leaks. Are both ends frequently initiating updates of connection parameters over there? It seems like it might not be the same issue at the moment.

Koxx3, 2024-05-11:
> If you can analyze the scenarios of memory leaks (broadcasting, scanning, connecting, etc.), it would be very helpful in resolving the issue.

yes, I'll try.

during this basic test, my IDLE loop : 
- send 3 notifications 
- check is scan is running, if not : 
  - set filter policy
  - reset device whitelist if needed
  - start a 1s scan
- the scan results processing is made inside the callback

```

void BluetoothHandler::processBLE()
{

    // notify changed value
    if (deviceStatus == BLE_STATUS_CONNECTED_AND_AUTHENTIFIED)
    {
        uint16_t period = 250;
        if ((shrd->fastUpdate) && (shrd->navMode == 0))
            period = 100;

        /*
        Serial.print(millis());
        Serial.print("BLH :  / ");
        Serial.print(shrd->timeLastNotifyBle);
        Serial.print("BLH :  / ");
        Serial.print(period);
        Serial.println(" / ");
        */

        if (millis() > shrd->timeLastNotifyBle + period)
        {

            notifyMeasurements();

            notifyBleLockAndErrors();

            shrd->timeLastNotifyBle = millis();
        }

        if (millis() > shrd->timeLastNotifyBleBms + 500)
        {
            notifyMeasurementsBms();

            shrd->timeLastNotifyBleBms = millis();
        }
    }

    // reset BLE stack if impossible to connect to BLE anymore
    if (bleNbFailure > 2)
    {
        deinit();
        delay(100);

        init();
        delay(100);

        bleScanEnabled = 1;

        bleNbFailure = 0;
    }

    // fail-safe restart for scan
    if ((!pBLEScan->isScanning()) && (bleScanEnabled >= 0) /*&& (bleScanStartFailedInMillis != 0) && (millis() - bleScanStartFailedInMillis > BEACON_SCAN_PERIOD_IN_SECONDS * 1000)*/)
    {
        // Serial.println("BLH : millis() - bleScanStartFailedInMillis = " + (String)(millis() - bleScanStartFailedInMillis));

/*
        char print_buffer[200];
        sprintf(print_buffer, "BLH : backup scan restart / isScanning = %d / bleScanEnabled = %d", pBLEScan->isScanning(), bleScanEnabled);
        Serial.println(print_buffer);
        // notifyBleLogs(print_buffer);
*/
        startBleScan();
    }

    oldDeviceStatus = deviceStatus;
}

void BluetoothHandler::startBleScan()
{

/*
    Serial.println("BLH : startBleScan / beacon scan : enabled = " + (String)bleScanEnabled +
                   " / vescToolMode = " + (String)shrd->vescToolMode +
                   " / bleScanMode = " + (String)shrd->bleScanMode +
                   " / beacon1 = " + (String)settings->get_Ble_beacon_mac_address1() +
                   " / beacon2 = " + (String)settings->get_Ble_beacon_mac_address2() +
                   " / tpms1 = " + (String)settings->get_Tpms_mac_address1() +
                   " / tpms2 = " + (String)settings->get_Tpms_mac_address2());
*/

    // start scan only if different from dummy mac address
    if ((bleScanEnabled >= 0) && (shrd->vescToolMode == 0) &&
        (((settings->get_Ble_beacon_mac_address1() != "aa:bb:cc:dd:ee:ff") || //
          (settings->get_Ble_beacon_mac_address2() != "aa:bb:cc:dd:ee:ff") || //
          (settings->get_Tpms_mac_address1() != "aa:bb:cc:dd:ee:ff") ||       //
          (settings->get_Tpms_mac_address2() != "aa:bb:cc:dd:ee:ff")) ||      //
         (shrd->bleScanMode != 0)))
    {
        // Serial.println("BLH : startBleScan / trying to start scan");

        if ((pBLEScan != NULL) && (pBLEScan->isScanning()))
            pBLEScan->stop();

        // Start scanning
        pBLEScan = BLEDevice::getScan(); // create new scan

        /*
        pBLEScan->setActiveScan(true);   // active scan uses more power, but get results faster
        pBLEScan->setInterval(100);
        pBLEScan->setWindow(98); // less or equal setInterval value
        */

#if SCAN_USE_WHITELIST
        if (shrd->bleScanMode == 0)
        {
            // Serial.println("BLH : startBleScan / set white list");
            pBLEScan->setFilterPolicy(BLE_HCI_SCAN_FILT_USE_WL);
            for (auto i = 0; i < NimBLEDevice::getWhiteListCount(); ++i)
                NimBLEDevice::whiteListRemove(NimBLEDevice::getWhiteListAddress(i));
            NimBLEDevice::whiteListAdd(NimBLEAddress(settings->get_Ble_beacon_mac_address1().c_str()));
            NimBLEDevice::whiteListAdd(NimBLEAddress(settings->get_Ble_beacon_mac_address2().c_str()));
            NimBLEDevice::whiteListAdd(NimBLEAddress(settings->get_Tpms_mac_address1().c_str()));
            NimBLEDevice::whiteListAdd(NimBLEAddress(settings->get_Tpms_mac_address2().c_str()));
        }
        else
        {
            // Serial.println("BLH : startBleScan / clear white list");
            pBLEScan->setFilterPolicy(BLE_HCI_SCAN_FILT_NO_WL);
        }
#endif

        // NEW TEST - no improvement
        // pBLEScan->clearResults();

        bool bleScanStarted = pBLEScan->start(BEACON_SCAN_PERIOD_IN_SECONDS, &bleOnScanResults, false);
        if (!bleScanStarted)
        {
            Serial.println("BLH : startBleScan / scan start failed");
            notifyBleLogs("BLH : scan start failed");
            bleScanStartFailedInMillis = millis();
        }
        else
        {
            bleScanStartFailedInMillis = 0;
        }
        // Serial.println("BLH : startBleScan / Beacon scan start res = " + (String)bleScanStarted);
    }
    else
    {
        Serial.println("BLH : startBleScan / Beacon scan disabled - conditions not ok - addresses or bleScanMode or vesctool");
    }
}

```

Koxx3, 2024-05-11:
> I recently discovered an issue with bidirectional frequent initiation of connection parameter updates leading to memory leaks. Are both ends frequently initiating updates of connection parameters over there? It seems like it might not be the same issue as yours ...

hum ... no, the connection is stable in my test.

here is the nimble service class :

```
    class BLEServerCallback : public NimBLEServerCallbacks
    {

        void onConnect(NimBLEServer *pServer)
        {

            deviceStatus = BLE_STATUS_CONNECTED_AND_AUTHENTIFYING;

...

            /*
            // TODO : add again
            // ble_svc_gatt_changed(0x0000, 0xFFFF);
            */

            // TODO : add again
            // NimBLEDevice::startAdvertising();

            Serial.println("BLH : onConnect1 - getConnectedCount() : " + (String)pServer->getConnectedCount());

            Serial.println("BLH : onConnect1 - nbFailure : " + (String)bleNbFailure);
            bleNbFailure++;
        };

        void onConnect(NimBLEServer *pServer, ble_gap_conn_desc *desc)
        {
            Serial.println("BLH : onConnect2 - getPeerMTU() : " + (String)pServer->getPeerMTU(desc->conn_handle));

            mtuReal = pServer->getPeerMTU(desc->conn_handle) - 3;
            Serial.println("BLH : onConnect2 - mtuReal - 3 : " + (String)mtuReal);
        }

        void onAuthenticationComplete(ble_gap_conn_desc *desc)
        {

            Serial.println(F("BLH : onAuthenticationComplete - BEGIN"));
            if (!desc->sec_state.encrypted)
            {

                Serial.println(F("BLH : onAuthenticationComplete : Encrypt connection failed - disconnecting"));
                NimBLEDevice::getServer()->disconnect(desc->conn_handle);
                deviceStatus = BLE_STATUS_DISCONNECTED;
                deviceStatusConnectedStartTime = 0;
            }
            else
            {
                Serial.println(F("BLH : onAuthenticationComplete : success"));
                deviceStatus = BLE_STATUS_CONNECTED_AND_AUTHENTIFIED;
                deviceStatusConnectedStartTime = millis();
            }

            ble_svc_gatt_changed(0x0000, 0xFFFF);

            if (bleNbFailure > 0)
                bleNbFailure--;

            Serial.println(F("BLH : onAuthenticationComplete - END"));
        }

        bool onConfirmPIN(uint32_t pin)
        {
            Serial.println(F("BLH : onConfirmPIN - BEGIN"));

            uint32_t pinCode = std::stoi(settings->get_Ble_pin_code2().c_str());
            Serial.println(pinCode);
            Serial.print(F("BLH : onConfirmPIN - END"));
            Serial.println(pinCode == pin);

            return pinCode == pin;
        }

        void onDisconnect(NimBLEServer *pServer)
        {
            Serial.println(F("BLH : onDisconnect - BEGIN"));
            deviceStatus = BLE_STATUS_DISCONNECTED;

            // shrd->isRecording = 1;

            if (shrd->bleLockForced == 0)
            {
                if (settings->get_Bluetooth_lock_mode() == settings->LIST_Bluetooth_lock_mode_Smartphone_connected)
                {
                    shrd->isLocked = 1;
                    Serial.println(F("BLH :  ==> device disconnected ==> LOCK decision"));
                    Serial.println(F("BLH : -------------------------------------"));
                }
                if (settings->get_Bluetooth_lock_mode() == settings->LIST_Bluetooth_lock_mode_Smartphone_connected_or_beacon_visible)
                {
                    if ((!shrd->bleBeaconVisible1) && (!shrd->bleBeaconVisible2))
                    {
                        shrd->isLocked = 1;
                        Serial.println(F(" ==> device disconnected / Beacon not visible ==> LOCK decision"));
                        Serial.println(F("-------------------------------------"));
                    }
                }
            }

            shrd->dateIsSyncedInBle = false;

            NimBLEDevice::startAdvertising();

            Serial.println("BLH : onDisconnect - getConnectedCount = " + (String)pServer->getConnectedCount());

            Serial.println(F("BLH : onDisconnect - END"));
        }

        void onMTUChange(uint16_t mtu, ble_gap_conn_desc *desc)
        {
            Serial.println("BLH : onMTUChange : " + (String)mtu);
            mtuReal = mtu - 3;
        }
    };

```

Koxx3, 2024-05-11:
it seems the leak....

test 1 :
only notifyMeasurements (3 notifications every 100ms)
6436 lowest available mem after beginning
12:02-12:07 => perfectly stable at 6964

test 2 :
notifyMeasurements/notifyBleLockAndErrors (4 notifications every 100ms) + notifyMeasurementsBms (1 notification every 500ms)
6832 lowest available mem after beginning
12:08-12:13 => stable at 6964-6980

test 3 :
test 2 + start scan from idle loop WITHOUT whitelist and scan results are sent in a dummy callback
lowest during : 6292
max during scan : 6864
12:18-12:48 => seems to leak, I never see memory above 6800 after 20min.

test 3 logs : 
```
12:18:46.460 > main : btContrl / free size = 6752 bytes  / new low !
12:18:46.460 > main : btContrl / free size = 6744 bytes  / new low !
12:18:46.460 > main : btContrl / free size = 6644 bytes  / new low !
12:18:46.460 > main : btContrl / free size = 6636 bytes  / new low !
12:18:46.460 > BLH : onAuthenticationComplete - BEGIN
12:18:46.460 > BLH : onAuthenticationComplete : success
12:18:46.460 > BLH : onAuthenticationComplete - END
12:18:46.460 > main : btContrl / free size = 6624 bytes  / new low !
12:18:46.460 > main : btContrl / free size = 6564 bytes  / new low !
12:18:46.460 > main : btContrl / free size = 6560 bytes  / new low !
12:18:46.460 > main : btContrl / free size = 6528 bytes  / new low !
12:18:46.460 > BLH : onMTUChange : 255
12:18:46.460 > main : btContrl / free size = 6484 bytes  / new low !
12:18:46.460 > BLH : onRead : beb5483e-36e1-4688-b7f5-ea07361b26a3
12:18:46.460 > BLH : onWrite : beb5483e-36e1-4688-b7f5-ea07361b26b7
12:18:46.460 > BLH : onRead : beb5483e-36e1-4688-b7f5-ea07361b26a2
12:18:46.460 > BLH : setEbFirmwaresDataPacket
12:18:46.460 > BLH : onRead : beb5483e-36e1-4688-b7f5-ea07361b26a5
12:18:47.023 > BLH : onRead : beb5483e-36e1-4688-b7f5-ea07361b26a7
12:18:47.114 > BLH : onRead : beb5483e-36e1-4688-b7f5-ea07361b26a8
12:18:47.204 > BLH : onRead : beb5483e-36e1-4688-b7f5-ea07361b26fc
12:18:47.325 > BLH : onWrite : beb5483e-36e1-4688-b7f5-ea07361b26fe
12:18:47.329 > W NimBLECharacteristic: Sending indication to client subscribed to notification, sending notification instead
12:18:47.351 > W NimBLECharacteristic: Sending indication to client subscribed to notification, sending notification instead
12:18:47.372 > W NimBLECharacteristic: Sending indication to client subscribed to notification, sending notification instead
12:18:47.744 > BLH : onWrite : beb5483e-36e1-4688-b7f5-ea07361b26a5
12:18:48.321 > main : btContrl / free size = 6480 bytes  / new low !
12:18:48.333 > main : btContrl / free size = 6468 bytes  / new low !
12:18:49.978 > main : btContrl / free size = 6608 bytes
12:18:50.587 > main : btContrl / free size = 6464 bytes  / new low !
12:18:55.366 > main : btContrl / free size = 6600 bytes
12:19:00.759 > main : btContrl / free size = 6596 bytes
12:19:03.641 > main : btContrl / free size = 6456 bytes  / new low !
12:19:03.751 > main : btContrl / free size = 6444 bytes  / new low !
12:19:06.172 > main : btContrl / free size = 6668 bytes
12:19:11.566 > main : btContrl / free size = 6596 bytes
12:19:12.175 > main : btContrl / free size = 6436 bytes  / new low !
12:19:16.968 > main : btContrl / free size = 6644 bytes
12:19:18.331 > main : btContrl / free size = 6424 bytes  / new low !
12:19:22.343 > main : btContrl / free size = 6572 bytes
12:19:27.723 > main : btContrl / free size = 6700 bytes
12:19:30.594 > main : btContrl / free size = 6408 bytes  / new low !
12:19:33.115 > main : btContrl / free size = 6688 bytes
12:19:38.510 > main : btContrl / free size = 6584 bytes
12:19:43.912 > main : btContrl / free size = 6584 bytes
12:19:47.975 > main : btContrl / free size = 6404 bytes  / new low !
12:19:49.307 > main : btContrl / free size = 6668 bytes
12:19:52.087 > main : btContrl / free size = 6396 bytes  / new low !
12:19:54.708 > main : btContrl / free size = 6596 bytes
12:20:00.109 > main : btContrl / free size = 6584 bytes
[... 316 lines cut ...]
12:47:08.142 > main : btContrl / free size = 6480 bytes
12:47:13.528 > main : btContrl / free size = 6396 bytes
12:47:18.901 > main : btContrl / free size = 6440 bytes
12:47:24.276 > main : btContrl / free size = 6772 bytes
12:47:29.670 > main : btContrl / free size = 6472 bytes
12:47:35.050 > main : btContrl / free size = 6452 bytes
12:47:39.481 > main : btContrl / free size = 6180 bytes  / new low !
12:47:39.588 > main : btContrl / free size = 6156 bytes  / new low !
12:47:40.435 > main : btContrl / free size = 6328 bytes
12:47:45.850 > main : btContrl / free size = 6528 bytes
12:47:51.278 > main : btContrl / free size = 6496 bytes
12:47:56.673 > main : btContrl / free size = 6364 bytes
12:48:02.086 > main : btContrl / free size = 6408 bytes
12:48:07.482 > main : btContrl / free size = 6472 bytes
12:48:12.874 > main : btContrl / free size = 6332 bytes
12:48:14.296 > main : btContrl / free size = 6144 bytes  / new low !
12:48:18.285 > main : btContrl / free size = 6248 bytes
12:48:18.624 > main : btContrl / free size = 6136 bytes  / new low !
12:48:23.691 > main : btContrl / free size = 6344 bytes
12:48:29.097 > main : btContrl / free size = 6452 bytes
12:48:34.495 > main : btContrl / free size = 6320 bytes
```

FYI, I use the h2zero Nimble version :
	nimble=https://github.com/h2zero/NimBLE-Arduino.git@1.4.1+sha.e46123a

zhp0406, 2024-05-11:
1-When issues arise, are there many devices advertising around you (for example, more than 20 devices broadcasting simultaneously)? 
2-is duplicate scan being utilized?

zhp0406, 2024-05-11:
Based on your description, is it possible for this issue to occur when performing notify and scan simultaneously?
[image]

zhp0406, 2024-05-11:
1-I will provide more debugging methods to pinpoint the location of the memory leak (by supply you with a new controller library)
This will involve logging the memory allocated and deallocated, and checking the malloc positions of any memory that hasn't been deallocated. 
2-Additionally, I will locally reproduce the scenarios causing the issue.
If you make any new discoveries, please let me know."

Koxx3, 2024-05-11:
> duplicate

1/
with a BLE scanner, I detect ~25 devices after 1 min.

2/
CONFIG_BLE_SCAN_DUPLICATE=y
CONFIG_SCAN_DUPLICATE_BY_DEVICE_ADDR=y
# CONFIG_SCAN_DUPLICATE_BY_ADV_DATA is not set
# CONFIG_SCAN_DUPLICATE_BY_ADV_DATA_AND_DEVICE_ADDR is not set
CONFIG_SCAN_DUPLICATE_TYPE=0
CONFIG_DUPLICATE_SCAN_CACHE_SIZE=100

> 1-When issues arise, are there many devices advertising around you (for example, more than 20 devices broadcasting simultaneously)? 2-is duplicate scan being utilized?

Koxx3, 2024-05-11:
> Based on your description, is it possible for this issue to occur when performing notify and scan simultaneously? [...]

yes, I send notifications while it scans.

Koxx3, 2024-05-11:
> 1-I will provide more debugging methods to pinpoint the location of the memory leak (by supply you with a new controller library) This will involve logging the memory allocated and deallocated, and [...]

1/ ok, thanks a lot
2/ yes, I'll let it run during few hours to see how fast it decrease

question : can nimble version have any impact ? (even if the bt controller should be bullet proof...)

zhp0406, 2024-05-11:
The issue appears to be related to the controller at the moment, not the host.

Koxx3, 2024-05-11:
If it can help, I noticed when I have a "new low", it's often by 8 bytes : 

```
14:38:40.110 > main : btContrl / free size = 6564 bytes  / new low !
...
14:39:56.245 > main : btContrl / free size = 6556 bytes  / new low !
14:39:56.508 > main : btContrl / free size = 6548 bytes  / new low !
...
14:40:02.035 > main : btContrl / free size = 6540 bytes  / new low !
...
14:41:27.760 > main : btContrl / free size = 6528 bytes  / new low !
14:41:28.822 > main : btContrl / free size = 6520 bytes  / new low !
14:41:28.828 > main : btContrl / free size = 6512 bytes  / new low !
```

Koxx3, 2024-05-11:
another interesting thing.
If I scan but do not send any notification, the free memory seems to remain stable.

test 6 :
scan without any notification, using nimble whitelist (1 device registered in the whitelist), and still dummy scan processing callback. the remote BLE device is connected.
14:45-15:10 => stabilized at 6776 during 15 minutes

FIY, I use one notification with a payload of 30 bytes (the MTU has been negociated greater than that). but it doesn't seems to have a real impact.

test 8 :
scan, nimble whitelist (1 device registered), 1 device connected, 4 notifications with 1ms delay between notification sending
15:25-15:35 => no real change but sometimes, I have a very close group (in time) of memory leaks (look at 15:29:28.377) 

```
15:25:29.943 > main : btContrl / free size = 6788 bytes
15:25:30.320 > BLH : onWrite : beb5483e-36e1-4688-b7f5-ea07361b26a5
15:25:35.308 > main : btContrl / free size = 6792 bytes
15:25:40.700 > main : btContrl / free size = 6772 bytes
15:25:46.081 > main : btContrl / free size = 6772 bytes
15:25:51.441 > main : btContrl / free size = 6788 bytes
15:25:56.836 > main : btContrl / free size = 6800 bytes
15:26:02.216 > main : btContrl / free size = 6768 bytes
15:26:07.588 > main : btContrl / free size = 6752 bytes
15:26:12.974 > main : btContrl / free size = 6768 bytes
15:26:18.345 > main : btContrl / free size = 6752 bytes
15:26:23.741 > main : btContrl / free size = 6752 bytes
15:26:29.130 > main : btContrl / free size = 6764 bytes
15:26:33.991 > main : btContrl / free size = 6656 bytes  / new low !
15:26:34.512 > main : btContrl / free size = 6752 bytes
15:26:39.876 > main : btContrl / free size = 6768 bytes
15:26:46.124 > main : btContrl / free size = 6636 bytes  / new low !
15:26:50.702 > main : btContrl / free size = 6752 bytes
15:26:56.085 > main : btContrl / free size = 6764 bytes
15:27:01.471 > main : btContrl / free size = 6736 bytes
15:27:06.844 > main : btContrl / free size = 6740 bytes
15:27:12.215 > main : btContrl / free size = 6752 bytes
15:27:17.579 > main : btContrl / free size = 6752 bytes
15:27:22.965 > main : btContrl / free size = 6748 bytes
15:27:28.348 > main : btContrl / free size = 6736 bytes
15:27:33.729 > main : btContrl / free size = 6752 bytes
15:27:39.120 > main : btContrl / free size = 6752 bytes
15:27:44.475 > main : btContrl / free size = 6768 bytes
15:27:49.860 > main : btContrl / free size = 6932 bytes  / recover !!!
15:27:55.232 > main : btContrl / free size = 6752 bytes
15:28:00.602 > main : btContrl / free size = 6752 bytes
15:28:05.977 > main : btContrl / free size = 6736 bytes
15:28:11.357 > main : btContrl / free size = 6724 bytes
15:28:13.357 > main : btContrl / free size = 6616 bytes  / new low !
15:28:16.720 > main : btContrl / free size = 6732 bytes
15:28:22.111 > main : btContrl / free size = 6716 bytes
15:28:27.491 > main : btContrl / free size = 6716 bytes
15:28:32.878 > main : btContrl / free size = 6732 bytes
15:28:38.266 > main : btContrl / free size = 6732 bytes
15:28:43.641 > main : btContrl / free size = 6720 bytes
15:28:45.932 > main : btContrl / free size = 6604 bytes  / new low !
15:28:49.045 > main : btContrl / free size = 6708 bytes
15:28:54.447 > main : btContrl / free size = 6716 bytes
15:28:59.832 > main : btContrl / free size = 6708 bytes
15:29:03.302 > main : btContrl / free size = 6596 bytes  / new low !
15:29:05.219 > main : btContrl / free size = 6708 bytes
15:29:10.593 > main : btContrl / free size = 6708 bytes
15:29:15.977 > main : btContrl / free size = 6700 bytes
15:29:21.355 > main : btContrl / free size = 6712 bytes
15:29:26.740 > main : btContrl / free size = 6692 bytes
15:29:28.377 > main : btContrl / free size = 6588 bytes  / new low !
15:29:28.808 > main : btContrl / free size = 6580 bytes  / new low !
15:29:31.176 > main : btContrl / free size = 6556 bytes  / new low !
15:29:31.654 > main : btContrl / free size = 6552 bytes  / new low !
15:29:31.660 > main : btContrl / free size = 6540 bytes  / new low !
15:29:32.122 > main : btContrl / free size = 6832 bytes  / recover !!!
[... 60 lines cut ...]
15:34:01.013 > main : btContrl / free size = 6528 bytes
15:34:06.411 > main : btContrl / free size = 6512 bytes
15:34:11.800 > main : btContrl / free size = 6540 bytes
15:34:17.185 > main : btContrl / free size = 6516 bytes
15:34:22.573 > main : btContrl / free size = 6512 bytes
15:34:27.973 > main : btContrl / free size = 6528 bytes
15:34:33.354 > main : btContrl / free size = 6512 bytes
15:34:35.098 > main : btContrl / free size = 6404 bytes  / new low !
15:34:36.606 > main : btContrl / free size = 6384 bytes  / new low !
15:34:38.744 > main : btContrl / free size = 6692 bytes
15:34:44.123 > main : btContrl / free size = 6492 bytes
15:34:49.507 > main : btContrl / free size = 6492 bytes
15:34:49.740 > main : btContrl / free size = 6380 bytes  / new low !
15:34:54.869 > main : btContrl / free size = 6476 bytes
15:35:00.250 > main : btContrl / free size = 6504 bytes
15:35:05.640 > main : btContrl / free size = 6492 bytes
15:35:11.017 > main : btContrl / free size = 6492 bytes
15:35:16.390 > main : btContrl / free size = 6476 bytes
15:35:21.770 > main : btContrl / free size = 6492 bytes
15:35:27.161 > main : btContrl / free size = 6492 bytes
15:35:32.554 > main : btContrl / free size = 6480 bytes
15:35:37.938 > main : btContrl / free size = 6476 bytes
15:35:43.324 > main : btContrl / free size = 6492 bytes
15:35:48.709 > main : btContrl / free size = 6492 bytes
```

Koxx3, 2024-05-11:
I confrm the asser is due to memory issue : 

```
main : btContrl / free size = 400 bytes
main : btContrl / free size = 296 bytes  / new low !
main : btContrl / free size = 384 bytes
main : btContrl / free size = 384 bytes
main : btContrl / free size = 400 bytes
main : btContrl / free size = 384 bytes
main : btContrl / free size = 384 bytes
main : btContrl / free size = 384 bytes
main : btContrl / free size = 384 bytes
main : btContrl / free size = 384 bytes
main : btContrl / free size = 376 bytes
main : btContrl / free size = 288 bytes  / new low !
main : btContrl / free size = 560 bytes
main : btContrl / free size = 380 bytes
main : btContrl / free size = 380 bytes
main : btContrl / free size = 380 bytes
main : btContrl / free size = 380 bytes
main : btContrl / free size = 380 bytes
main : btContrl / free size = 380 bytes
main : btContrl / free size = 380 bytes
main : btContrl / free size = 380 bytes
main : btContrl / free size = 396 bytes
main : btContrl / free size = 380 bytes
main : btContrl / free size = 380 bytes
main : btContrl / free size = 380 bytes
main : btContrl / free size = 384 bytes
main : btContrl / free size = 380 bytes
main : btContrl / free size = 380 bytes
main : btContrl / free size = 380 bytes
main : btContrl / free size = 392 bytes
main : btContrl / free size = 380 bytes
main : btContrl / free size = 560 bytes
main : btContrl / free size = 380 bytes
main : btContrl / free size = 380 bytes
main : btContrl / free size = 280 bytes  / new low !
main : btContrl / free size = 264 bytes  / new low !
main : btContrl / free size = 376 bytes
ASSERT_PARAM(-218959118 0), in arch_main.c at line 364

Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.

Memory dump at 0x4009090c: f01d020c 00004136 f01d0000

Core  1 register dump:

PC      : 0x40090913  PS      : 0x00060b34  A0      : 0x8008655d  A1      : 0x3ffc4c64  

A2      : 0x00000000  A3      : 0xf2f2f2f2  A4      : 0x00000000  A5      : 0x3f434dce  

A6      : 0x0000016c  A7      : 0xfffffffc  A8      : 0x8000814b  A9      : 0x3ffc4bd4  

A10     : 0x00000000  A11     : 0x3ffc4bf7  A12     : 0x3ffc4ba3  A13     : 0x00000034  

A14     : 0x00000000  A15     : 0x3ffc4ba8  SAR     : 0x00000004  EXCCAUSE: 0x00000000  

EXCVADDR: 0x00000000  LBEG    : 0x40086465  LEND    : 0x4008646d  LCOUNT  : 0x00000000  

Backtrace: 0x40090910:0x3ffc4c64 |<-CORRUPTED

```

Koxx3, 2024-05-12:
I have build a "kinda minimal" app, but it still contains some sensitive code.
How can I send you the code privatly ?

To reproduce the issue with "NRF connect" android app :
- connect
- in service 914c -> read charac 26a3
- bond with code 147258
- request mtu 255
- read all characteritics
- request priority high
- in service 914b -> enable notifications/indicates for 26a0, 26a1, 26a5, 26b4

zhp0406, 2024-05-13:
sent me by email:
zhanghaipeng@espressif.com

zhp0406, 2024-05-13:
If you've created your program using a platform other than ESP-IDF, such as Arduino or PlatformIO, please explain the process required for building and flashing.

Koxx3, 2024-05-13:
at the same time, if you want to me test another lib, let me know.

zhp0406, 2024-05-14:
Where can I configure the path for the IDF and replace the controller lib?
[image]

Koxx3, 2024-05-14:
ok ! almost good !

on windows : C:\Users\<USERNAME>\.platformio\packages\framework-espidf (replace USERNAME)

on Linux, I think it's the same /home/<USERNAME>/.platformio/packages/framework-espidf

zhp0406, 2024-05-17:
Currently, I can build and flash your application onto the ESP32, but replacing the controller library doesn't seem to take effect. I will try to set up the environment on a Windows platform.

After flashing, I noticed that the ESP32 continuously sends notifications at a high frequency. Additionally, based on your description, the ESP32 is still scanning. I suspect the issue might be that the ESP32 is unable to send data quickly enough, causing the internal queues to grow and eventually leading to memory exhaustion.
To verify this hypothesis, you can disconnect the connections before the memory is exhausted and then check if the memory has been freed.

zhp0406, 2024-05-17:
extern void ke_mem_dbg_dump();
[eb32dbe.zip](https://github.com/espressif/esp-idf/files/15343429/eb32dbe.zip)

Koxx3, 2024-05-17:
> Currently, I can build and flash your application onto the ESP32, but replacing the controller library doesn't seem to take effect. I will try to set up the environment on a Windows platform.
>
> After [...]

make sure you clean the ".pio" folder inside the project folder. it should force a complete rebuild.

Koxx3, 2024-05-17:
> extern void ke_mem_dbg_dump(); [eb32dbe.zip](https://github.com/espressif/esp-idf/files/15343429/eb32dbe.zip) I have provided a new library for you to test. Please ensure that the BT controller compiles with version [eb32dbe].
> 
> In this new library, I have implemented tracking for memory allocation and release. However, since some of the code resides in ROM, I have temporarily omitted tracking the exact locations of memory allocations (I will add this functionality later using other methods).
> 
> Additionally, the new library tracks the usage of queues. Detailed information can be printed using ke_mem_dbg_dump.
> 
> P.S.: If the memory leak occurs at the same location each time, I have set up an internal pointer, void *ptr_rec. If the address of the allocated memory matches ptr_rec, an assert will be triggered, allowing you to determine the assert location via backtrace. However, since the memory allocation location might vary, this method might not always be effective.

ok. I'll try !

zhp0406, 2024-05-17:
> > Currently, I can build and flash your application onto the ESP32, but replacing the controller library doesn't seem to take effect. I will try to set up the environment on a Windows platform.
> > After flashing, I noticed that the ESP32 continuously sends notifications at a high frequency. Additionally, based on your description, the ESP32 is still scanning. I suspect the issue might be that the ESP32 is unable to send data quickly enough, causing the internal queues to grow and eventually leading to memory exhaustion. To verify this hypothesis, you can disconnect the connections before the memory is exhausted and then check if the memory has been freed.
> 
> make sure you clean the ".pio" folder inside the project folder. it should force a complete rebuild. Even with a lower notifications frequency (250ms instead of 100ms), I still have the issue.

great,I will try again.

Koxx3, 2024-05-18:
Here are the logs...

before connect : 

```
====== KE MEM DUMP START ======
mem 0 6804 used 612 total 1964
queue 0  0
node 0x3ffb2fa8 free 1352
mem 1 6864 used 0 total 892
queue 0  0
node 0x3ffb2c28 free 892
mem 2 6860 used 0 total 4620
queue 0  0
node 0x3ffb3758 free 4620
mem leak 22
p=0x3ffb3734 s=32
p=0x3ffb3690 s=160
p=0x3ffb3674 s=12
p=0x3ffb35d0 s=160
p=0x3ffb2f98 s=12
p=0x3ffb35b8 s=8
p=0x3ffb35ac s=8
p=0x3ffb35c4 s=8
p=0x3ffb3588 s=8
p=0x3ffb354c s=8
p=0x3ffb357c s=8
p=0x3ffb35a0 s=8
p=0x3ffb3594 s=8
p=0x3ffb3564 s=8
p=0x3ffb353c s=8
p=0x3ffb3570 s=8
p=0x3ffb3510 s=8
p=0x3ffb3558 s=8
p=0x3ffb3500 s=8
p=0x3ffb3684 s=8
p=0x3ffb3520 s=8
p=0x3ffb352c s=8
====== KE MEM DUMP END ======
```

after connect :

```
====== KE MEM DUMP START ======
mem 0 6400 used 912 total 1964
queue 0  0
node 0x3ffb2fa8 free 996
node 0x3ffb3504 free 56
mem 1 6460 used 104 total 892
queue 0  0
node 0x3ffb2c28 free 764
node 0x3ffb2f44 free 24
mem 2 6460 used 0 total 4620
queue 0  0
node 0x3ffb3758 free 4620
mem leak 32
p=0x3ffb3734 s=32
p=0x3ffb35d0 s=160
p=0x3ffb3674 s=12
p=0x3ffb2f70 s=12
p=0x3ffb369c s=8
p=0x3ffb36b8 s=8
p=0x3ffb36a8 s=8
p=0x3ffb371c s=8
p=0x3ffb2f60 s=12
p=0x3ffb370c s=8
p=0x3ffb3728 s=8
p=0x3ffb36fc s=8
p=0x3ffb36c4 s=8
p=0x3ffb2f38 s=12
p=0x3ffb2f94 s=12
p=0x3ffb3684 s=16
p=0x3ffb2f28 s=12
p=0x3ffb36d0 s=8
p=0x3ffb36ec s=8
p=0x3ffb35c0 s=8
p=0x3ffb36e0 s=8
p=0x3ffb3464 s=160
p=0x3ffb344c s=20
p=0x3ffb3390 s=184
p=0x3ffb2f84 s=12
p=0x3ffb3594 s=16
p=0x3ffb35b0 s=8
p=0x3ffb3588 s=8
p=0x3ffb3574 s=16
p=0x3ffb3560 s=16
p=0x3ffb354c s=16
p=0x3ffb3540 s=8
====== KE MEM DUMP END ======
```

after few minutes : 

```
====== KE MEM DUMP START ======
mem 0 6152 used 1028 total 1964
queue 0  0
node 0x3ffb2fa8 free 936
mem 1 6152 used 296 total 892
queue 0  0
node 0x3ffb2c28 free 552
node 0x3ffb2e98 free 44
mem 2 6152 used 0 total 4620
queue 0  0
node 0x3ffb3758 free 4620
mem leak 48
p=0x3ffb3734 s=32
p=0x3ffb35d0 s=160
p=0x3ffb3674 s=12
p=0x3ffb2f70 s=12
p=0x3ffb2f48 s=12
p=0x3ffb2ed8 s=12
p=0x3ffb2f5c s=12
p=0x3ffb36ec s=8
p=0x3ffb3724 s=8
p=0x3ffb3714 s=8
p=0x3ffb2e54 s=12
p=0x3ffb2f10 s=12
p=0x3ffb2e84 s=12
p=0x3ffb2f38 s=12
p=0x3ffb2f94 s=12
p=0x3ffb3684 s=16
p=0x3ffb2f28 s=12
p=0x3ffb2f00 s=12
p=0x3ffb36cc s=8
p=0x3ffb36a8 s=8
p=0x3ffb3704 s=8
p=0x3ffb3464 s=160
p=0x3ffb344c s=20
p=0x3ffb2ec8 s=12
p=0x3ffb3390 s=184
p=0x3ffb2f84 s=12
p=0x3ffb35c0 s=8
p=0x3ffb36dc s=8
p=0x3ffb2ee8 s=12
p=0x3ffb35b0 s=8
p=0x3ffb353c s=8
p=0x3ffb36b8 s=8
p=0x3ffb369c s=8
p=0x3ffb2e64 s=12
p=0x3ffb2e74 s=12
p=0x3ffb36f8 s=8
p=0x3ffb3508 s=8
p=0x3ffb3518 s=8
p=0x3ffb35a4 s=8
p=0x3ffb3594 s=8
p=0x3ffb354c s=8
p=0x3ffb352c s=8
p=0x3ffb3584 s=8
p=0x3ffb3568 s=16
p=0x3ffb355c s=8
p=0x3ffb337c s=16
p=0x3ffb3368 s=16
p=0x3ffb3354 s=16
====== KE MEM DUMP END ======
```

I'll let it run until crash and send you the full log.

is there anything I can do to use the "void *ptr_rec" ? sorry but I don't see how to use it. can you provide an example ?

Koxx3, 2024-05-18:
> After flashing, I noticed that the ESP32 continuously sends notifications at a high frequency. Additionally, based on your description, the ESP32 is still scanning. I suspect the issue might be that [...]

by the way, no disconnect & reconnect doesn't release the memory. you'll see in the full log.
The only way to release the memory is to deinit & reinit the BT controller (see BluetoothHandler::init() and BluetoothHandler::deinit() )

Koxx3, 2024-05-18:
if crashed before the memory exhaustion : 

```
====== KE MEM DUMP START ======
mem 0 5144 used 980 total 1964
queue 0  0
node 0x3ffb2fa8 free 972
node 0x3ffb3528 free 12
mem 1 5144 used 880 total 892
queue 0  0
node 0x3ffb2c28 free 12
mem 2 5156 used 448 total 4620
queue 0  0
node 0x3ffb3758 free 4044
node 0x3ffb4734 free 52
node 0x3ffb4798 free 24
node 0x3ffb2f28 free 16379
Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.

Core  1 register dump:
PC      : 0x4008656c  PS      : 0x00060530  A0      : 0x800d2d78  A1      : 0x3ffc9940  
A2      : 0x00060101  A3      : 0x3ffb93d0  A4      : 0x00000002  A5      : 0x3ffb93d4  
A6      : 0x00000001  A7      : 0x00000000  A8      : 0x8008657a  A9      : 0x3ffc98b0  
A10     : 0x00000019  A11     : 0x00000019  A12     : 0x0000000c  A13     : 0xffffffff  
A14     : 0x00000000  A15     : 0xfffffffa  SAR     : 0x00000004  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x00060103  LBEG    : 0x4000c349  LEND    : 0x4000c36b  LCOUNT  : 0xffffffff  

Backtrace: 0x40086569:0x3ffc9940 0x400d2d75:0x3ffc9960 0x400ea079:0x3ffc9980 0x40094d7d:0x3ffc99a0

ELF file SHA256: 6bf33f137f032efd

```

```
C:\Users\koxx3\.platformio\packages\toolchain-xtensa-esp32\bin\xtensa-esp32-elf-addr2line.exe -i -p -fe  0x40086569:0x3ffc9940 0x400d2d75:0x3ffc9960 0x400ea07
[device-monitor-240518-090238.log](https://github.com/espressif/esp-idf/files/15360572/device-monitor-240518-090238.log)
9:0x3ffc9980 0x40094d7d:0x3ffc99a0
C:\Users\koxx3\.platformio\packages\toolchain-xtensa-esp32\bin\xtensa-esp32-elf-addr2line.exe: '0x40086569:0x3ffc9940': No such file
```

Koxx3, 2024-05-18:
2nd run.

it also crashed during the ke_mem_dbg_dump call.

```

====== KE MEM DUMP START ======
mem 0 5544 used 924 total 1964
queue 0  0
node 0x3ffb306c free 988
node 0x3ffb35d4 free 36
node 0x3ffb3654 free 16
mem 1 5544 used 880 total 892
queue 0  0
node 0x3ffb2cec free 12
mem 2 5556 used 104 total 4620
queue 0  0
node 0x3ffb381c free 4448
node 0x3ffb498c free 24
node 0x3ffb3038 free 16379
Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.

Core  1 register dump:
PC      : 0x4008656c  PS      : 0x00060530  A0      : 0x800d2d7c  A1      : 0x3ffc9940  
A2      : 0x00060101  A3      : 0x3ffb93d0  A4      : 0x00000002  A5      : 0x3ffb93d4  
A6      : 0x00000001  A7      : 0x00000000  A8      : 0x8008657a  A9      : 0x3ffc98b0  
A10     : 0x00000019  A11     : 0x00000019  A12     : 0x0000000c  A13     : 0xffffffff  
A14     : 0x00000000  A15     : 0xfffffffa  SAR     : 0x00000004  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x00060103  LBEG    : 0x4000c349  LEND    : 0x4000c36b  LCOUNT  : 0xffffffff  

Backtrace: 0x40086569:0x3ffc9940 0x400d2d79:0x3ffc9960 0x400ea091:0x3ffc9980 0x40094d7d:0x3ffc99a0

ELF file SHA256: c9ab3846bc496b67

C:\Users\koxx3\.platformio\packages\toolchain-xtensa-esp32\bin\xtensa-esp32-elf-addr2line.exe -i -p -fe  c:\Users\koxx3\Documents\Coding\IDFGH-12768\.pio\build\IDFGH-12768\firmware.elf 0x40086569:0x3ffc9940 0x400d2d79:0x3ffc9960 0x400ea091:0x3ffc9980 0x40094d7d:0x3ffc99a0
ke_mem_dbg_dump at ??:?
_Z4loopv at C:\Users\koxx3\Documents\Coding\IDFGH-12768/src/main.cpp:201
_Z8loopTaskPv at C:/Users/koxx3/.platformio/packages/framework-arduinoespressif32/cores/esp32/main.cpp:50
vPortTaskWrapper at C:\Users\koxx3\.platformio\packages\framework-espidf\components\freertos\port\xtensa/port.c:142
```

by the way, you can see the effect of the deinit/init (search for "BLH : deinit" in the log).

full log : 
https://koxx3.fr.eu.org:8086/device-monitor-240518-094903.zip

Koxx3, 2024-05-20:
> Thank you for your feedback. With your help, I was able to reproduce the issue you encountered locally.

ohhhh ! excellent !

[27 more replies]

closed 2024-06-01 by Koxx3