espressif/esp-idf #13748: 刚开始学习esp32，使用modbus-rtu-master例程，一直报超时。

faqqqsn, 2024-05-05:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

使用modbus-rtu-master例程，一直报超时，按照文档提示，更改了超时时间，并用RS485例程检测连接，未发现异常，slave通信截图如下。
[image]
[image]
[image]

alisitsyn, 2024-05-06:
@faqqqsn ,

As per your limited amount of information you are going to use the RTU communication but did not configured the master to use the RTU communication properly. The RTU can be enabled [here](https://github.com/espressif/esp-idf/blob/master/examples/protocols/modbus/serial/mb_master/main/master.c#L268) using the `CONFIG_MB_COMM_MODE_RTU` in menuconfig.

Please try the above and let me know the result.

alisitsyn, 2024-05-06:
@faqqqsn , ok. Feel free to ask again. Please close the issue as completed. Thanks.

closed 2024-05-06 by faqqqsn