espressif/esp-idf #13756: esp32c6-NimBLE power_save sample code, the measured power consumption is too high.

hiroshi3117, 2024-05-07:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

Hi, everyone,
I used nimble's power_save sample program to confirm the power consumption of esp32-c6's ble in idle state. My development board added an external 32kHz crystal oscillator and matching capacitor 12pF, but the power consumption is not as good as the 34uA written in the README.md.MD file, and the actual measured result is about "9mA".
[image]

I haven't made any modifications to this sample program and configuration file, and I have double confirmed some necessary configuration:
1. RTC clock source use external 32kHz crystal.
2. Enable power management.
3. configUSE_TICKLESS_IDLE.
4. Enable BLE sleep.
5. BLE low power clock source use system RTC slow clock source.

and the log also shows that an external 32kHz is used, but I still can't get the power consumption of 34uA.
[image]

I referred to this post but still not helpful. https://github.com/espressif/esp-idf/issues/947#issuecomment-500312453
Are there any other settings required?
Or is the power saving mode of esp32-c6 ble not yet completed?
Does anyone have any successful examples that can be used as a reference?
Thank you very much.

IDF version: (master)5.4.0
Development board: ESP32-C6-DevKitM-1
Development Platform: Windows, VS Code
Below are the programs I use.
[sdkconfig.txt](https://github.com/espressif/esp-idf/files/15233141/sdkconfig.txt)
[main.c.txt](https://github.com/espressif/esp-idf/files/15233162/main.c.txt)
[gatt_svr.c.txt](https://github.com/espressif/esp-idf/files/15233164/gatt_svr.c.txt)

hiroshi3117, 2024-05-08:
Hi, @ChrysalisChenJ ,
Thank you for your reply. I downloaded the last release version of ESP-IDF v5.2.1, and re-created a new NimBLE power_save sample code project. The power consumption is still not as good as described in README.md.
The SDK configuration is set as follows, and the measured power consumption is about "11mA". Not the expected 3.3mA.
1. BLE low power clock source -> Use main XTAL as RTC clock source
2. RTC clock source -> Internal 150kHz RC oscillator

The SDK configuration is set as follows, and there is an external 32.768kHz crystal oscillator. The measured power consumption is about "9mA". No expected 34uA.
1. BLE low power clock source -> Use system RTC slow clock source
2. RTC clock source -> external 32kHz crystal

I have 2 pieces of ESP32-C6-DevKitM-1, and the test with another piece has the same result.
One of the development boards did not process an external 32.768kHz crystal oscillator, so the 3.3mA result was not measured when the main crystal was used as the RTC clock source. So it should not be the high current caused by the external 32.768kHz crystal oscillator.

Environment:
The external 32.768kHz I use is a passive crystal oscillator.
The GPIO of the development board is not connected to other circuits, and the GPIO status uses the default value of the sample program.
The power uses USB and power supply, and the test results are the same.
The ammeter is connected in series to J5 on the development board for measurement.
[image]

As for whether the chip actually goes into light sleep mode, that's a good question.
But I don't know how to confirm. The terminal that outputs the log has been disconnected. I guess it must have entered power saving mode.
```
I (221) esp_image: segment 3: paddr=0009f9ec vaddr=408050W (597) clk: esp_perip_clk_init() has not been implemented yet
I (604) cpu_start: Pro cpu start user code
I (604) cpu_start: cpu freq: 160000000 Hz
I (604) cpu_start: Application information:
I (607) cpu_start: Project name:     power_save
I (612) cpu_start: App version:      1
I (617) cpu_start: Compile time:     May  8 2024 17:01:01
I (623) cpu_start: ELF file SHA256:  054d828e1...
I (628) cpu_start: ESP-IDF:          v5.2.1
I (633) cpu_start: Min chip rev:     v0.0
I (638) cpu_start: Max chip rev:     v0.99
I (642) cpu_start: Chip rev:         v0.0
I (647) heap_init: Initializing. RAM available for dynamic allocation:
I (654) heap_init: At 40819400 len 00063210 (396 KiB): RAM
I (660) heap_init: At 4087C610 len 00002F54 (11 KiB): RAM
I (667) heap_init: At 50000058 len 00003F90 (15 KiB): RTCRAM
I (674) spi_flash: detected chip: generic
I (678) spi_flash: flash io: dio
W (681) spi_flash: Detected size(4096k) larger than the size in the binary image header(2048k). Using the size in the binary image header.
I (695) sleep: Enable automatic switching of GPIO sleep configuration
I (702) sleep_clock: System Power, Clock and Reset sleep retention initialization
I (710) sleep_clock: Modem Power, Clock and Reset sleep retention initialization
I (718) sleep_sys_periph: Interrupt Matrix sleep retention initialization
I (726) sleep_sys_periph: HP System sleep retention initialization
I (733) sleep_sys_periph: TEE/APM sleep retention initialization
I (739) sleep_sys_periph: UART sleep retention initialization
I (746) sleep_sys_periph: Timer Group sleep retention initialization
I (753) sleep_sys_periph: IO Matrix sleep retention initialization
I (760) sleep_sys_periph: SPI Mem sleep retention initialization
I (766) sleep_sys_periph: SysTimer sleep retention initialization
I (784) pm: Frequency switching config: CPU_MAX: 160, APB_MAX: 80, APB_MIN: 40, Light sleep: DISABLED
I (784) coexist: coex firmware version: 77cd7f8
I (788) coexist: coexist rom version 5b8dcfa
I (793) main_task: Started on CPU0
I (796) main_task: Calling app_main()
I (809) pm: Frequency switching config: CPU_MAX: 160, APB_MAX: 80, APB_MIN: 40, Light sleep: ENABLED
I (811) BLE_INIT: Using external 32.768 kHz XTAL as clock source
I (822) BLE_INIT: ble controller commit:[217f1bf]
W (822) BLE_INIT: BLE modem sleep is enabled
I (828) BLE_INIT: Modem BLE MAC retention initialization
W (833) BLE_INIT: Enable light sleep, the wake up source is BLE timer
I (841) phy_init: phy_version 250,e14681b,Jan 24 2024,17:43:11
I (890) phy: libbtbb version: 939f79c, Jan 24 2024, 17:43:26
I (891) btbb_init: btbb sleep retention initialization
I (892) NimBLE_BLE_PRPH: BLE Host Task Started
I (900) NimBLE: Device Address:
I (901) NimBLE: 54:32:04:07:14:76
I (904) NimBLE: 

I (908) NimBLE: GAP procedure initiated: extended advertise; instance=0

I (916) uart: queue free spaces: 8
I (919) main_task: Returned from app_main()
None
Waiting for the device to reconnect.................................................................................................................
```
Or is there a development board that can measure signals that have entered power saving mode? Or is there any other way?
The following is my project, built using IDF v5.2.1. Could you please help me confirm again, does it have the same result in your environment? Thank you very much.

The following are the burning files and source code.
[esp32c6_ble_power_save_bin.zip](https://github.com/espressif/esp-idf/files/15248199/esp32c6_ble_power_save_bin.zip)
[BLE_power_save.zip](https://github.com/espressif/esp-idf/files/15248203/BLE_power_save.zip)

ChrysalisChenJ, 2024-05-10:
Hi hiroshi3117,

You mentioned you are trying to measure power consumption of esp32-c6's BLE in idle state. However, in the BLE power save example, BLE is actually advertising on a certain interval(), during which the current consumption will be much higher than that in IDLE state. If you are using an oscilloscope or a power analyzer to measure the current, you can see the whole current profile, it was something like this:

 
[image]

That's why you are getting around 9 mA from the ampere meter, I believe the sampling rate is too low to capture the light sleep current and you ended up getting like the average current.

If it's the light sleep current you are measuring, you can comment out the following lines in `bleprph_on_sync` function in main.c, this will stop the BLE advertising process and keeps the chip in light sleep, then you can measure the current.

```
 

#if CONFIG_EXAMPLE_EXTENDED_ADV
 ext_bleprph_advertise();
#else
 bleprph_advertise();
#endif
```

hiroshi3117, 2024-05-10:
Hi @ChrysalisChenJ ,
I see what you mean. The power consumption of 34uA written in README.md refers to the period during which only light sleep does with out RF transmission, just like the power consumption indicated in the specification. 
Thank you very much for your explanation.

closed 2024-05-10 by hiroshi3117