espressif/esp-idf #13757: Support for 802.11h "Transmit Power Control" (TPC)?

br101, 2024-05-07:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

Does ESP32, especially ESP32-C3 support 802.11h, especially "Transmit Power Control" (TPC)?

br101, 2024-05-10:
Let me add that TPC was originally specified in 802.11h but it has been included in the 802.11 standard since 2007 (IEEE 802.11-2007). So if the wifi stack of the ESP32 is conforming to the standard, it should have implemented it, right?

MaxwellAlan, 2024-05-11:
Hi @br101 

Current chips do not yet support complete TPC functions, which means that STA cannot adjust the transmit power according to the requirements of the AP.

However, our current chips all support dynamic adjustment of transmit power through IDF API: https://github.com/espressif/esp-idf/blob/d4cd437ede613fffacc06ac6d6c93a083829022f/components/esp_wifi/include/esp_wifi.h#L1075

br101, 2024-05-21:
Thank you, setting the TX power by code is not enough. Please let me add that this feature is often a requirement in strictly managed corporate networks, so very important in order to use ESP32 based systems in larger scale professional environments like big offices or production sites.

MaxwellAlan, 2024-06-14:
Hi @br101 

Thank you for the clarification. Due to the dynamic adjustment of transmission power for each channel and rate required by the TPC feature, it conflicts with our existing chip (32/s2/s3/c2/c3/c6) software and hardware architecture design, resulting in the inability to support this feature on the current chip. For future chips, we will consider redesigning the software and hardware architecture to meet this functional requirement.

closed 2024-09-12 by MaxwellAlan