espressif/esp-idf #13759: USB Audio device on ESP32-S3 : Hitting upper limit of open endpoints when I should not; issue in portable/synopsys/dcd_dwc2.c?

giri-a, 2024-05-07:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

I am implementing a headset (speaker+mic) on ESP32-S3. I notice that when I pause the playback or record 3 times, playback or record does not happen any more.
I traced the issue to a couple of TU_ASSERT in dcd_edpt_counter_release(). When I pause playback, the endpoint is released and opened again when I start playing.

```
// Release an endpoint.
static bool dcd_edpt_counter_release(uint8_t rhport, uint8_t dir)
{
  if (dir) {
    TU_ASSERT(dwc2_ep_counters[rhport].active_in == 0);    // Check if number of opened EPs is not zero
    dwc2_ep_counters[rhport].active_in--;                  // Decrease endpoint counter
  } else {
    TU_ASSERT(dwc2_ep_counters[rhport].active_out == 0);
    dwc2_ep_counters[rhport].active_out--;
  }
  return true;
}

```
Obviously the ASSERT fails when active_in and active_out are >0 and they are not decremented. Do this 3 times and total number of open endpoints are 6, which the limit. Subsequently, dcd_edpt_counter_available() returns false.

When I changed '==' to '>', the problem went away. It makes sense to me. Also the comment implies that it should be '!=' or '>'.

Anybody else faced this issue? Am I missing something or thinking it wrong?

roma-jam, 2024-05-07:
@giri-a, 
Thanks for reporting the issue.
 
The fix was already merged (https://github.com/espressif/tinyusb/pull/25), we will publish the new version soon. 

Sorry for inconvenience.

roma-jam, 2024-05-08:
Hi @giri-a, 

There is a new version of espressif/tinyusb is available (v0.15.0.9). 
If there will be anything else or you will run into any other issue, fell free to reopen this ticket or create a new one. 

Thanks for your help.

closed 2024-05-08 by roma-jam

giri-a, 2024-05-08:
Perfect. That was quick. Thanks.