espressif/esp-idf #13760: Absolute volume doesn't work by default on Android

qombajn, 2024-05-07:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v4.4

### Espressif SoC revision.

esp32

### Operating System used.

Windows

### How did you build your project?

VS Code IDE

### If you are using Windows, please specify command line type.

None

### Development Kit.

Custom Board

### Power Supply used.

External 3.3V

### What is the expected behavior?

I expected that when I connect device to A2DP Sink, esp is sending 40% of volume. When I adjust volume on the streaming device, the volume is being send as "absolute volume" to esp from streaming device.

### What is the actual behavior?

Actual, when I connect windows, it works perfectly, when I connect Android I have to go to "settings -> bluetooth -> speaker -> set absolute volume". I have to uncheck this feature, and check this feature to make it work.

### Steps to reproduce.

1. Modify bt_app_av.c :
static void bt_av_hdl_a2d_evt(uint16_t event, void *p_param)
{
    ESP_LOGD(BT_AV_TAG, "%s event: %d", __func__, event);

    esp_a2d_cb_param_t *a2d = NULL;

    switch (event) {
    /* when connection state changed, this event comes */
    case ESP_A2D_CONNECTION_STATE_EVT: {
        a2d = (esp_a2d_cb_param_t *)(p_param);
        uint8_t *bda = a2d->conn_stat.remote_bda;
        ESP_LOGI(BT_AV_TAG, "A2DP connection state: %s, [%02x:%02x:%02x:%02x:%02x:%02x]",
            s_a2d_conn_state_str[a2d->conn_stat.state], bda[0], bda[1], bda[2], bda[3], bda[4], bda[5]);
        if (a2d->conn_stat.state == ESP_A2D_CONNECTION_STATE_DISCONNECTED) {
            esp_bt_gap_set_scan_mode(ESP_BT_CONNECTABLE, ESP_BT_GENERAL_DISCOVERABLE);
            volume_set_by_controller((uint8_t)0); //MUTE
            bt_i2s_task_shut_down();
        } else if (a2d->conn_stat.state == ESP_A2D_CONNECTION_STATE_CONNECTED){
            esp_bt_gap_set_scan_mode(ESP_BT_NON_CONNECTABLE, ESP_BT_NON_DISCOVERABLE);
            volume_set_by_controller((uint8_t)51); //128*40% = 51,2
            bt_i2s_task_start_up();
        }
        break;
    }
2. Connect Windows and observe how it works perfectly
3. Connect Android and observe how it doesn't work and you have to uncheck and check "absolute volume" manually.

### Debug Logs.

```plain
Windows:
I (699185) RC_TG: Volume is set by remote controller to: 40%
I (699355) RC_CT: AVRC conn_state event: state 1, [dc:53:60:fd:3b:ca]
I (699355) RC_CT: AVRC remote features 24b, TG features 11
I (699355) RC_TG: AVRC remote features: 24b, CT features: 2
I (699405) RC_CT: remote rn_cap: count 6, bitmask 0xf06
Android:
I (739355) RC_TG: Volume is set by remote controller to: 40%
I (739505) RC_CT: AVRC conn_state event: state 1, [64:a2:00:0f:59:88]
I (739505) RC_CT: AVRC remote features 249, TG features 1
I (739535) RC_CT: remote rn_cap: count 5, bitmask 0xc26
```

### More Information.

As you can see there is no "I (699355) RC_TG: AVRC remote features: 24b, CT features: 2" this line in logs while connecting to android.

qombajn, 2024-05-07:
When I call "esp_avrc_ct_send_get_rn_capabilities_cmd" manually after connection, I get from 
Windows:
I (180685) RC_CT: remote rn_cap: count 6, bitmask 0xf06
Android:
I (206145) RC_CT: remote rn_cap: count 5, bitmask 0xc26

qombajn, 2024-07-17:
@espressif-bot help :(

xiongweichao, 2024-08-14:
Hi @qombajn ,

I'm very sorry to reply you so late. May I ask what model this Android phone is?

BTW, have you used other Android phones to test it?

Thanks

qombajn, 2024-08-14:
Hello @xiongweichao ,

Thanks for reply. My phone is POCO X3 NFC. About other phones, I've checked but I don't remember results right now. As I remember, when I connected Iphone, volume was set after opening App but it was long time ago and I'm not 100% sure. I can check with other phones tonight. 

I am not into avrcp and I don't know how exactly it should work. I've read about some flags and possibility that device can provide. Other headphones like Edifier w830bt and xiaomi redmi buds 5 pro, works without any settings. But still I don't know if they are muting music by cutting data or by mutting DAC. 

Project is still being in progress :) 

Thanks

qombajn, 2024-08-14:
I closed by mistake, sorry

closed 2024-08-14 by qombajn

xiongweichao, 2025-04-14:
@qombajn  Please test this issue on the latest master branch to see if it still exists. Thanks