espressif/esp-idf #13761: esp_sleep_ext1 doesn't work correctly with ESP_EXT1_WAKEUP_ANY_HIGH mode in release/v5.1

NightSkySK, 2024-05-07:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

Here is code which I run and with which I found that bug. The chip which I use is EPS32-WROOM-32UE

```c
void example_deep_sleep_register_ext1_wakeup(void) {
  const int ext_wakeup_pin_1 = 13;
  const int ext_wakeup_pin_2 = 14;
  const uint64_t ext_wakeup_pin_1_mask = 1ULL << ext_wakeup_pin_1;
  const uint64_t ext_wakeup_pin_2_mask = 1ULL << ext_wakeup_pin_2;
  printf("Enabling EXT1 wakeup on pins GPIO%d, GPIO%d\n", ext_wakeup_pin_1,
         ext_wakeup_pin_2);

  ESP_ERROR_CHECK(esp_sleep_enable_ext1_wakeup(
      ext_wakeup_pin_1_mask | ext_wakeup_pin_2_mask, ESP_EXT1_WAKEUP_ANY_HIGH));
}

void app_main(void) {

  switch (esp_sleep_get_wakeup_cause()) {
  case ESP_SLEEP_WAKEUP_EXT1: {
    uint64_t wakeup_pin_mask = esp_sleep_get_ext1_wakeup_status();
    if (wakeup_pin_mask != 0) {
      int pin = __builtin_ffsll(wakeup_pin_mask) - 1;
      printf("Wake up from GPIO %d\n", pin);
    } else {
      printf("Wake up from GPIO\n");
    }
    break;
  }
  case ESP_SLEEP_WAKEUP_UNDEFINED:
  default:
    printf("Not a deep sleep reset\n");
  }

  vTaskDelay(1000 / portTICK_PERIOD_MS);

  example_deep_sleep_register_ext1_wakeup();
  printf("Entering deep sleep\n");
  esp_deep_sleep_start();
}
```
Code always work as mode would be defined as `ESP_EXT1_WAKEUP_ALL_LOW`. I have boards with two different configuration of hardware:
1. Board where both pins 13 & 14 are by default HIGH and by pressing the button it goes LOW (with external PULLUP's)
 a. Case one `ESP_EXT1_WAKEUP_ANY_HIGH` compiled:
     13 HIGH & 14 HIGH - DEEP SLEEP
     13 LOW & 14 HIGH  - DEEP SLEEP
     13 HIGH & 14 LOW  - DEEP SLEEP
     13 LOW & 14 LOW   - WAKE UP
b.  Case one `ESP_EXT1_WAKEUP_ALL_LOW` compiled:
     13 HIGH & 14 HIGH - DEEP SLEEP
     13 LOW & 14 HIGH  - DEEP SLEEP
     13 HIGH & 14 LOW  - DEEP SLEEP
     13 LOW & 14 LOW   - WAKE UP

2. Board where both pins 13 & 14 are by default LOW and by pressing the button it goes HIGH (with external PULLDOWN's)
 a. Case one `ESP_EXT1_WAKEUP_ANY_HIGH` compiled:
     13 HIGH & 14 HIGH - DEEP SLEEP
     13 LOW & 14 HIGH  - DEEP SLEEP
     13 HIGH & 14 LOW  - DEEP SLEEP
     13 LOW & 14 LOW   - WAKE UP
b.  Case one `ESP_EXT1_WAKEUP_ALL_LOW` compiled:
     13 HIGH & 14 HIGH - DEEP SLEEP
     13 LOW & 14 HIGH  - DEEP SLEEP
     13 HIGH & 14 LOW  - DEEP SLEEP
     13 LOW & 14 LOW   - WAKE UP

so it always behaves like it would be set to `ESP_EXT1_WAKEUP_ALL_LOW`

in my opinion the method definition for [esp-idf](https://github.com/espressif/esp-idf/tree/release/v5.1)/[components](https://github.com/espressif/esp-idf/tree/release/v5.1/components)/[esp_hw_support](https://github.com/espressif/esp-idf/tree/release/v5.1/components/esp_hw_support)/sleep_modes.c have error
```c
esp_err_t esp_sleep_enable_ext1_wakeup(uint64_t io_mask, esp_sleep_ext1_wakeup_mode_t level_mode)
{
    if (level_mode > ESP_EXT1_WAKEUP_ANY_HIGH) {
        return ESP_ERR_INVALID_ARG;
    }
    // Translate bit map of GPIO numbers into the bit map of RTC IO numbers
    uint32_t rtc_gpio_mask = 0;
    for (int gpio = 0; io_mask; ++gpio, io_mask >>= 1) {
        if ((io_mask & 1) == 0) {
            continue;
        }
        if (!esp_sleep_is_valid_wakeup_gpio(gpio)) {
            ESP_LOGE(TAG, "Not an RTC IO: GPIO%d", gpio);
            return ESP_ERR_INVALID_ARG;
        }
        rtc_gpio_mask |= BIT(rtc_io_number_get(gpio));
    }
    s_config.ext1_rtc_gpio_mask = rtc_gpio_mask;
    if (level_mode) {
        s_config.ext1_trigger_mode = io_mask;
    } else {
        s_config.ext1_trigger_mode = 0;
    }
    s_config.wakeup_triggers |= RTC_EXT1_TRIG_EN;
    return ESP_OK;
}
```
in line `s_config.ext1_trigger_mode = io_mask;` shouldn't be just `1` instead of `io_mask`?
For comparison I've checked the same file from release/v4.4 and that definition make more sens:

```c
esp_err_t esp_sleep_enable_ext1_wakeup(uint64_t mask, esp_sleep_ext1_wakeup_mode_t mode)
{
    if (mode > ESP_EXT1_WAKEUP_ANY_HIGH) {
        return ESP_ERR_INVALID_ARG;
    }
    // Translate bit map of GPIO numbers into the bit map of RTC IO numbers
    uint32_t rtc_gpio_mask = 0;
    for (int gpio = 0; mask; ++gpio, mask >>= 1) {
        if ((mask & 1) == 0) {
            continue;
        }
        if (!esp_sleep_is_valid_wakeup_gpio(gpio)) {
            ESP_LOGE(TAG, "Not an RTC IO: GPIO%d", gpio);
            return ESP_ERR_INVALID_ARG;
        }
        rtc_gpio_mask |= BIT(rtc_io_number_get(gpio));
    }
    s_config.ext1_rtc_gpio_mask = rtc_gpio_mask;
    s_config.ext1_trigger_mode = mode;
    s_config.wakeup_triggers |= RTC_EXT1_TRIG_EN;
    return ESP_OK;
}
```
as `s_config.ext1_trigger_mode = mode;` is just equal `mode`

10086loutianhao, 2024-05-08:
@NightSkySK  hello， please try git cherry-pick 640ce31daebe3aed913ec7d95decb9927c8b9cdd to sovle this issue.

10086loutianhao, 2024-05-08:
It's a known issue, and we have an unmerged MR

NightSkySK, 2024-05-08:
@10086loutianhao Thank you! I couldn't find any bug report regarding this issue. As I see the `release/v5.2` has already merged. is it only `release/v5.1` which is still affected? [release/v5.0](https://github.com/espressif/esp-idf/blob/release/v5.0/components/esp_hw_support/sleep_modes.c) seems to be good with the same definition which was used in `release/v4.4`, isn't it?
```c
esp_err_t esp_sleep_enable_ext1_wakeup(uint64_t mask, esp_sleep_ext1_wakeup_mode_t mode)
{
    if (mode > ESP_EXT1_WAKEUP_ANY_HIGH) {
        return ESP_ERR_INVALID_ARG;
    }
    // Translate bit map of GPIO numbers into the bit map of RTC IO numbers
    uint32_t rtc_gpio_mask = 0;
    for (int gpio = 0; mask; ++gpio, mask >>= 1) {
        if ((mask & 1) == 0) {
            continue;
        }
        if (!esp_sleep_is_valid_wakeup_gpio(gpio)) {
            ESP_LOGE(TAG, "Not an RTC IO: GPIO%d", gpio);
            return ESP_ERR_INVALID_ARG;
        }
        rtc_gpio_mask |= BIT(rtc_io_number_get(gpio));
    }
    s_config.ext1_rtc_gpio_mask = rtc_gpio_mask;
    s_config.ext1_trigger_mode = mode;
    s_config.wakeup_triggers |= RTC_EXT1_TRIG_EN;
    return ESP_OK;
}
```
I'm asking to ensure which of my CI/CD builds are going to be save to use.

10086loutianhao, 2024-05-08:
@NightSkySK  yes, only release/5.1 need to fix this issue, now we need to wait for the MR to merge

10086loutianhao, 2024-05-13:
hello ， release/5.1 has fixed this bug, pls check @NightSkySK

NightSkySK, 2024-05-13:
@10086loutianhao which commit should contain that fix? 
I checked [commit history](https://github.com/espressif/esp-idf/commits/release/v5.1) of file [components/esp_hw_support/sleep_modes.c](https://github.com/espressif/esp-idf/commits/release/v5.1/components/esp_hw_support/sleep_modes.c) and nothing changes there since Apr-3rd

For me still doesn't work without  cherry-pick https://github.com/espressif/esp-idf/commit/640ce31daebe3aed913ec7d95decb9927c8b9cdd

10086loutianhao, 2024-05-14:
@NightSkySK  hello, b8c617935589e1367dbb42ac21fa511099a65c92 can fix it

NightSkySK, 2024-05-22:
@10086loutianhao I can confirm that solution implemented in https://github.com/espressif/esp-idf/commit/b8c617935589e1367dbb42ac21fa511099a65c92 fixed my issue with ext1_wakeup

closed 2024-05-22 by NightSkySK