espressif/esp-idf #13765: Bluetooth HFP Audio Gate Philips mode I2S

Ageofenigma, 2024-05-08:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

 Is it possible to enable "philips" mode (data one clock shifted) for PCM  Bluetooth HFP Audio Gate example?

Ageofenigma, 2024-05-13:
Is it possible?

esp-qing, 2024-05-14:
Hi,  sorry to reply you so late. Maybe the _"philips" mode_ you mentioned is one of the three modes of I2S in the esp32 document. The default PCM output waveform of hfp_ag example is similar to the _MSB Alignment Standard_ in the document and currently cannot be changed to the _philips standard_.