espressif/esp-idf #13766: esp_err.h ESP_ERROR_CHECK definition error

PaulskPt, 2024-05-09:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

ESP-IDF v5.4-dev-78-gd4cd437ede-dirty

### Operating System used.

Linux

### How did you build your project?

Command line with CMake

### If you are using Windows, please specify command line type.

None

### What is the expected behavior?

I expected the build not showing errors from esp-idf components parts.

### What is the actual behavior?

ninja output at end of build:
/home/paulsk/esp/esp-idf/components/esp_common/include/esp_err.h:115:28: error: expected identifier or '(' before 'do'
  115 | #define ESP_ERROR_CHECK(x) do {                                         \
      |                            ^~
/home/paulsk/esp/temphum_master/main/main_master_v1.h:25:1: note: in expansion of macro 'ESP_ERROR_CHECK'
   25 | ESP_ERROR_CHECK(i2c_new_master_bus(&i2c_mst_config, &bus_handle));
      | ^~~~~~~~~~~~~~~
/home/paulsk/esp/esp-idf/components/esp_common/include/esp_err.h:121:7: error: expected identifier or '(' before 'while'
  121 |     } while(0)
      |       ^~~~~
/home/paulsk/esp/temphum_master/main/main_master_v1.h:25:1: note: in expansion of macro 'ESP_ERROR_CHECK'
   25 | ESP_ERROR_CHECK(i2c_new_master_bus(&i2c_mst_config, &bus_handle));
      | ^~~~~~~~~~~~~~~
/home/paulsk/esp/esp-idf/components/esp_common/include/esp_err.h:115:28: error: expected identifier or '(' before 'do'
  115 | #define ESP_ERROR_CHECK(x) do {                                         \
      |                            ^~
/home/paulsk/esp/temphum_master/main/main_master_v1.h:34:1: note: in expansion of macro 'ESP_ERROR_CHECK'
   34 | ESP_ERROR_CHECK(i2c_master_bus_add_device(bus_handle, &dev_cfg, &dev_handle));
      | ^~~~~~~~~~~~~~~
/home/paulsk/esp/esp-idf/components/esp_common/include/esp_err.h:121:7: error: expected identifier or '(' before 'while'
  121 |     } while(0)
      |       ^~~~~
/home/paulsk/esp/temphum_master/main/main_master_v1.h:34:1: note: in expansion of macro 'ESP_ERROR_CHECK'
   34 | ESP_ERROR_CHECK(i2c_master_bus_add_device(bus_handle, &dev_cfg, &dev_handle));

### Steps to reproduce.

1. from within the build folder: cmake .. -G ninja
2. ninja

### Build or installation Logs.

```plain
the ninja log is very long.
```

### More Information.

This is the part of esp_err.h containing the macro definition:

/**
 * Macro which can be used to check the error code,
 * and terminate the program in case the code is not ESP_OK.
 * Prints the error code, error location, and the failed statement to serial output.
 *
 * Disabled if assertions are disabled.
 */
#ifdef NDEBUG
#define ESP_ERROR_CHECK(x) do {                                         \
        esp_err_t err_rc_ = (x);                                        \
        (void) sizeof(err_rc_);                                         \
    } while(0)
#elif defined(CONFIG_COMPILER_OPTIMIZATION_ASSERTIONS_SILENT)
#define ESP_ERROR_CHECK(x) do {                                         \
        esp_err_t err_rc_ = (x);                                        \
        if (unlikely(err_rc_ != ESP_OK)) {                              \
            abort();                                                    \
        }                                                               \
    } while(0)
#else
#define ESP_ERROR_CHECK(x) do {                                         \
        esp_err_t err_rc_ = (x);                                        \
        if (unlikely(err_rc_ != ESP_OK)) {                              \
            _esp_error_check_failed(err_rc_, __FILE__, __LINE__,        \
                                    __ASSERT_FUNC, #x);                 \
        }                                                               \
    } while(0)
#endif

ESP-Marius, 2024-05-09:
How are you actually using `ESP_ERROR_CHECK(i2c_new_master_bus(&i2c_mst_config, &bus_handle));` here? From the log it looks like you are trying to put it in a header file?

PaulskPt, 2024-05-09:
Yes, I see I also used ESP_ERROR_CHECK() in a header file. I am going to remove that and test again.

PaulskPt, 2024-05-09:
I removed it from the header file, however the errors continue:

```

In file included from /home/paulsk/esp/esp-idf/components/esp_hw_support/include/esp_intr_alloc.h:12,
                 from /home/paulsk/esp/esp-idf/components/esp_hw_support/include/esp_cpu.h:20,
                 from /home/paulsk/esp/esp-idf/components/esp_hw_support/include/spinlock.h:11,
                 from /home/paulsk/esp/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/include/freertos/portmacro.h:49,
                 from /home/paulsk/esp/esp-idf/components/freertos/FreeRTOS-Kernel/include/freertos/portable.h:57,
                 from /home/paulsk/esp/esp-idf/components/freertos/FreeRTOS-Kernel/include/freertos/FreeRTOS.h:69,
                 from /home/paulsk/esp/temphum_master/components/i2cdev/i2cdev.c:35:
/home/paulsk/esp/esp-idf/components/esp_common/include/esp_err.h:115:28: error: expected identifier or '(' before 'do'
  115 | #define ESP_ERROR_CHECK(x) do {                                         \
      |                            ^~
/home/paulsk/esp/temphum_master/components/i2cdev/i2cdev.c:58:1: note: in expansion of macro 'ESP_ERROR_CHECK'
   58 | ESP_ERROR_CHECK(i2c_new_master_bus(&i2c_mst_config, &bus_handle));
      | ^~~~~~~~~~~~~~~
/home/paulsk/esp/esp-idf/components/esp_common/include/esp_err.h:121:7: error: expected identifier or '(' before 'while'
  121 |     } while(0)
      |       ^~~~~
/home/paulsk/esp/temphum_master/components/i2cdev/i2cdev.c:58:1: note: in expansion of macro 'ESP_ERROR_CHECK'
   58 | ESP_ERROR_CHECK(i2c_new_master_bus(&i2c_mst_config, &bus_handle));
      | ^~~~~~~~~~~~~~~
/home/paulsk/esp/esp-idf/components/esp_common/include/esp_err.h:115:28: error: expected identifier or '(' before 'do'
  115 | #define ESP_ERROR_CHECK(x) do {                                         \
      |                            ^~
/home/paulsk/esp/temphum_master/components/i2cdev/i2cdev.c:71:1: note: in expansion of macro 'ESP_ERROR_CHECK'
   71 | ESP_ERROR_CHECK(i2c_master_bus_add_device(bus_handle, &dev_cfg, &dev_handle));
      | ^~~~~~~~~~~~~~~
/home/paulsk/esp/esp-idf/components/esp_common/include/esp_err.h:121:7: error: expected identifier or '(' before 'while'
  121 |     } while(0)
      |       ^~~~~
/home/paulsk/esp/temphum_master/components/i2cdev/i2cdev.c:71:1: note: in expansion of macro 'ESP_ERROR_CHECK'
   71 | ESP_ERROR_CHECK(i2c_master_bus_add_device(bus_handle, &dev_cfg, &dev_handle));
      | ^~~~~~~~~~~~~~~
/home/paulsk/esp/esp-idf/components/esp_common/include/esp_err.h:115:28: error: expected identifier or '(' before 'do'
  115 | #define ESP_ERROR_CHECK(x) do {                                         \
      |                            ^~
/home/paulsk/esp/temphum_master/components/i2cdev/i2cdev.c:76:1: note: in expansion of macro 'ESP_ERROR_CHECK'
   76 | ESP_ERROR_CHECK(i2c_new_master_bus(&i2c_mst_config, &bus_handle));
      | ^~~~~~~~~~~~~~~
/home/paulsk/esp/esp-idf/components/esp_common/include/esp_err.h:121:7: error: expected identifier or '(' before 'while'
  121 |     } while(0)
      |       ^~~~~
/home/paulsk/esp/temphum_master/components/i2cdev/i2cdev.c:76:1: note: in expansion of macro 'ESP_ERROR_CHECK'
   76 | ESP_ERROR_CHECK(i2c_new_master_bus(&i2c_mst_config, &bus_handle));
      | ^~~~~~~~~~~~~~~
/home/paulsk/esp/esp-idf/components/esp_common/include/esp_err.h:115:28: error: expected identifier or '(' before 'do'
  115 | #define ESP_ERROR_CHECK(x) do {                                         \
      |                            ^~
/home/paulsk/esp/temphum_master/components/i2cdev/i2cdev.c:79:1: note: in expansion of macro 'ESP_ERROR_CHECK'
   79 | ESP_ERROR_CHECK(i2c_master_bus_add_device(bus_handle, &dev_cfg, &dev_handle));
      | ^~~~~~~~~~~~~~~
/home/paulsk/esp/esp-idf/components/esp_common/include/esp_err.h:121:7: error: expected identifier or '(' before 'while'
  121 |     } while(0)
      |       ^~~~~
/home/paulsk/esp/temphum_master/components/i2cdev/i2cdev.c:79:1: note: in expansion of macro 'ESP_ERROR_CHECK'
   79 | ESP_ERROR_CHECK(i2c_master_bus_add_device(bus_handle, &dev_cfg, &dev_handle));
```

ESP-Marius, 2024-05-10:
Can you please show the piece of code where you are actually using this macro?

PaulskPt, 2024-05-10:
Here is an extract of the code where ESP_ERROR_CHECK is used:

```
/*
 * The MIT License (MIT)
 *
 * Copyright (c) 2018 Ruslan V. Uss <unclerus@gmail.com>
 *
 * Permission is hereby granted, free of charge, to any person obtaining a copy
 * of this software and associated documentation files (the "Software"), to deal
 * in the Software without restriction, including without limitation the rights
 * to use, copy, modify, merge, publish, distribute, sublicense, and/or sell
 * copies of the Software, and to permit persons to whom the Software is
 * furnished to do so, subject to the following conditions:
 * The above copyright notice and this permission notice shall be included in all
 * copies or substantial portions of the Software.
 *
 * THE SOFTWARE IS PROVIDED "AS IS", WITHOUT WARRANTY OF ANY KIND, EXPRESS OR
 * IMPLIED, INCLUDING BUT NOT LIMITED TO THE WARRANTIES OF MERCHANTABILITY,
 * FITNESS FOR A PARTICULAR PURPOSE AND NONINFRINGEMENT. IN NO EVENT SHALL THE
 * AUTHORS OR COPYRIGHT HOLDERS BE LIABLE FOR ANY CLAIM, DAMAGES OR OTHER
 * LIABILITY, WHETHER IN AN ACTION OF CONTRACT, TORT OR OTHERWISE, ARISING FROM,
 * OUT OF OR IN CONNECTION WITH THE SOFTWARE OR THE USE OR OTHER DEALINGS IN THE
 * SOFTWARE.
 */

/**
 * @file i2cdev.c
 *
 * ESP-IDF I2C master thread-safe functions for communication with I2C slave
 *
 * Copyright (c) 2018 Ruslan V. Uss <unclerus@gmail.com>
 *
 * MIT Licensed as described in the file LICENSE
 * 
 * 2024-05 Changes by Paulus Schulinck (@PaulskPt on Github & Gist)
 * to adapt this file for migration to driver/i2c_master.h
 * after a run of the original app gave the following warning:
 * W (270) i2c: This driver is an old driver, please migrate your application code to adapt `driver/i2c_master.h`
 * 
 *  */
#include <string.h>
#include <inttypes.h>
#include <freertos/FreeRTOS.h>
#include <freertos/task.h>
#include <esp_log.h>

#include "i2cdev.h"

static const char *TAG = "i2cdev";

/*
#if !defined(i2c_mst_config)
i2c_master_bus_config_t i2c_mst_config = {
    .clk_source = I2C_CLK_SRC_DEFAULT,
    .i2c_port = 0,
    .scl_io_num = CONFIG_EXAMPLE_I2C_MASTER_SCL,
    .sda_io_num = CONFIG_EXAMPLE_I2C_MASTER_SDA,
    .glitch_ignore_cnt = 7,
    .flags.enable_internal_pullup = true,
};
#endif
*/

#if !defined(bus_handle)
i2c_master_bus_handle_t bus_handle;
ESP_ERROR_CHECK(i2c_new_master_bus(&i2c_mst_config, &bus_handle));
#endif 

#if !defined(dev_cfg)
i2c_device_config_t dev_cfg = {
    .dev_addr_length = I2C_ADDR_BIT_LEN_7,
    .device_address = 0x38,
    .scl_speed_hz = 100000,
};
#endif

#if !defined(dev_handle)
i2c_master_dev_handle_t dev_handle;
ESP_ERROR_CHECK(i2c_master_bus_add_device(bus_handle, &dev_cfg, &dev_handle));
#endif 

#if (!defined(bus_handle) || !defined(dev_handle))
i2c_master_bus_handle_t bus_handle;
ESP_ERROR_CHECK(i2c_new_master_bus(&i2c_mst_config, &bus_handle));

i2c_master_dev_handle_t dev_handle;
ESP_ERROR_CHECK(i2c_master_bus_add_device(bus_handle, &dev_cfg, &dev_handle));
#endif
[...]
```

adokitkat, 2024-05-12:
Hello.

ESP_ERROR_CHECK needs to be inside of a function (`app_main` for example).

This is not valid:
```c
#if !defined(bus_handle)
i2c_master_bus_handle_t bus_handle;
ESP_ERROR_CHECK(i2c_new_master_bus(&i2c_mst_config, &bus_handle));
#endif 
```

You need to put `ESP_ERROR_CHECK(i2c_new_master_bus(&i2c_mst_config, &bus_handle));` in some initializer function which is called from somewhere else. Same with everything else.

Tbh any function call should be done from an another function, regardless if it's wrapped in ESP_ERROR_CHECK macro or not, initializer elements for global variables should be constant as far as I understand, it shouldn't compile anyway.

ESP-Marius, 2024-05-13:
Yes as @adokitkat says, you cannot simply call functions/run code "statically" like that in C. 

Closing this one as there doesnt seem to be any IDF related issues (not related to `ESP_ERROR_CHECK`).

For generic questions which are not related to ESP-IDF you can also try our [forum](https://www.esp32.com/)

closed 2024-05-13 by ESP-Marius

PaulskPt, 2024-05-13:
Thank you both!