espressif/esp-idf #13768: During subscribing notification on characteristic -  Error 6 (0x6) GATT REQ NOT SUPPORTED

veneno-529, 2024-05-09:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

Hello, 

I'm using ESP - IDF 5.1.1 and NIMBLE. I have recently added notify to one my characteristics. Normally when I erase the esp32 and flash the code at first I'm able to successfully subscribe and unsubscribe notifications. Once the notification is subscribed it stays subscribed till the device disconnects. But in after certain hours when I try to subscribe the notification it gives me Error 6 (0x6) GATT REQ NOT SUPPORTED. I am trying to do this by NRF app. It subscribes and automatically unsubscribes too in case of auto connect. 
PFB code below in which I have implemented the notify part. Also screenshot of the error and some logs of the device.
[image]
[image]

under gapp events function below event has been added for notifications

`case BLE_GAP_EVENT_SUBSCRIBE:
        f_pairing = FALSE;
        f_ble_paired = TRUE;
        //vTaskResume(pirStatusHandle);
        MODLOG_DFLT(INFO, "subscribe event; conn_handle=%d attr_handle=%d "
                    "reason=%d prevn=%d curn=%d previ=%d curi=%d\n",
                    event->subscribe.conn_handle,
                    event->subscribe.attr_handle,
                    event->subscribe.reason,
                    event->subscribe.prev_notify,
                    event->subscribe.cur_notify,
                    event->subscribe.prev_indicate,
                    event->subscribe.cur_indicate);
        if (event->subscribe.attr_handle == notification_handle)
        {
            printf("\nSubscribed with notification_handle =%d\n", event->subscribe.attr_handle);
            notify_state = event->subscribe.cur_notify; //!! As the client is now subscribed to notifications, the value is set to 1
            printf("notify_state=%d\n", notify_state);
        }
        return 0;`

below is the gatt svcs code
`const struct ble_gatt_svc_def gatt_svcs[] = {
    {.type = BLE_GATT_SVC_TYPE_PRIMARY,
     .uuid = &gatt_svr_svc_uuid.u,                 // Define UUID for device type
     .characteristics = (struct ble_gatt_chr_def[]){
         {.uuid = &gatt_svr_chartx_rx_uuid.u,           // Define UUID for reading from  esp32 and writing to esp32
          .flags = BLE_GATT_CHR_F_READ | BLE_GATT_CHR_F_WRITE 
                    | BLE_GATT_CHR_F_READ_ENC | BLE_GATT_CHR_F_WRITE_ENC,
          .access_cb = device_char_read_write},
          {
            .uuid = &gatt_svr_strtx_rx_uuid.u,
            .val_handle = &notification_handle,
            .flags = BLE_GATT_CHR_F_READ | BLE_GATT_CHR_F_WRITE | BLE_GATT_CHR_F_WRITE_ENC | BLE_GATT_CHR_F_NOTIFY, 
            .access_cb = device_string_read_write},
         {0}}},
    {0}};`

The error is occuring randomly and once it occurs I'm not able to subscribe to notifications until I erase the esp32 and again reflash it.
I have configured nimble for below settings as follows - 
1. Maximum number of concurrent connects = 1
2. Maximum number of bonds to save across boots = 1
3. Maximum number of CCC descriptors to save across reboots = 12
4. Host based privacy for random address = enabled
5. BLE MAX connections  =  3

Additionally, when one event occurs I'm unpairing the device using below function which unpairs the device.
`struct ble_hs_dev_records * peer_dev_records = ble_rpa_get_peer_dev_records();
            ble_gap_unpair(&peer_dev_records[0].peer_sec.peer_addr);`

veneno-529, 2024-05-21:
Hello @rahult-github 
PFB debug level logs when i try to subscribe to notification.

I (62286) NimBLE: subscribe event; conn_handle=0 attr_handle=27 reason=1 prevn=0 curn=1 previ=0 curi=0
Subscribed with notification_handle =27
notify_state=1
D (62301) NimBLE: error persisting cccd; too many entries (12)

D (62307) NimBLE: looking up our sec;
D (62310) NimBLE:

D (62313) NimBLE: looking up our sec;
D (62316) NimBLE:

D (62319) NimBLE: host tx hci data; handle=0 length=9

D (62324) NimBLE: ble_hs_hci_acl_tx():
D (62328) NimBLE: 0x00
D (62330) NimBLE: 0x00
D (62333) NimBLE: 0x09
D (62335) NimBLE: 0x00
D (62338) NimBLE: 0x05
D (62340) NimBLE: 0x00
D (62343) NimBLE: 0x04
D (62345) NimBLE: 0x00
D (62348) NimBLE: 0x01
D (62351) NimBLE: 0x12
D (62353) NimBLE: 0x1c
D (62356) NimBLE: 0x00
D (62358) NimBLE: 0x06
D (62361) NimBLE:

D (62508) NimBLE: ble_hs_hci_evt_acl_process(): conn_handle=0 pb=2 len=7 data=
D (62509) NimBLE: 0x03
D (62509) NimBLE: 0x00
D (62510) NimBLE: 0x04
D (62512) NimBLE: 0x00
D (62515) NimBLE: 0x0a
D (62517) NimBLE: 0x19
D (62520) NimBLE: 0x00
D (62522) NimBLE:

D (62525) NimBLE: host tx hci data; handle=0 length=6

D (62530) NimBLE: ble_hs_hci_acl_tx():
D (62534) NimBLE: 0x00
D (62536) NimBLE: 0x00
D (62539) NimBLE: 0x06
D (62541) NimBLE: 0x00
D (62544) NimBLE: 0x02
D (62546) NimBLE: 0x00
D (62549) NimBLE: 0x04
D (62552) NimBLE: 0x00
D (62554) NimBLE: 0x0b
D (62556) NimBLE: 0x63
D (62559) NimBLE:

D (62654) NimBLE: ble_hs_hci_evt_acl_process(): conn_handle=0 pb=2 len=9 data=
D (62655) NimBLE: 0x05
D (62655) NimBLE: 0x00
D (62656) NimBLE: 0x04
D (62658) NimBLE: 0x00
D (62661) NimBLE: 0x12
D (62663) NimBLE: 0x1c
D (62666) NimBLE: 0x00
D (62669) NimBLE: 0x00
D (62671) NimBLE: 0x00
D (62674) NimBLE:

I (62676) NimBLE: subscribe event; conn_handle=0 attr_handle=27 reason=1 prevn=1 curn=0 previ=0 curi=0

Subscribed with notification_handle =27
notify_state=0

Please help me out with it.

rahult-github, 2024-05-21:
error persisting cccd; too many entries (12)   ..

looks like too many cccds are getting written . Please increase the value of BT_NIMBLE_MAX_CCCDS from menuconfig and try once.

veneno-529, 2024-05-21:
@rahult-github this will only temporarily resolve the issue.
After sometime, again it will occur.

I have found a similar issue here 
[=](https://github.com/espressif/esp-idf/issues/4413)

I have seen that patch provided in this issue is not matching with the current files of the nimble version.
I have implemented the patch but still not able to resolve the issue.
It seems that it is not deleting old peer ccds that is stored.

rahult-github, 2024-05-22:
Hi @veneno-529 , 

Can you please:

1. Try attached patch and see if it helps. The patch is to be applied in $IDF_PATH/components/bt/host/nimble/nimble path. 
2. Please share steps on how your are reproducing the issue, which can be tried by us to reproduce issue , in case above patch is not helping the case. 
[check_for_size.txt](https://github.com/espressif/esp-idf/files/15400500/check_for_size.txt)

veneno-529, 2024-05-22:
Hello @rahult-github 

So first I tried the patch still no luck. Issue is occurring. To regenerate the issue please follow below steps.

Configure nimble for below settings as follows -

Maximum number of concurrent connects = 1
Maximum number of bonds to save across boots = 3
Maximum number of CCC descriptors to save across reboots = 4
Host based privacy for random address = enabled
BLE MAX connections = 3

`ble_hs_cfg.store_status_cb = ble_store_util_status_rr;`
RPA is enabled exactly how it is mentioned in examples.

First erase flash completely to ensure memory is clean and no previous data of connections is present.
Burn the firmware file. 

1. Connect android device, pair it, subscribe to notifications.
2. Disconnect android device. 
3. Restart ESP32, you get log of RL count = 1
4. Now, connect iphone device or any other android device i did it with iphone, subscribe to notifications.
5. Disconnect device.
6. Restart ESP32, you get log of RL count = 2 meaning it has stored info of both the devices.
7. Try to connect iphone again. It works fine. Disconnect it.
8. Restart the esp32 you will see that the RL count has changed to 1. It deleted info of android device.
9. Now connect the android device again, it fails at first. Removes ios device info and post disconnecting and connecting again it stores the android device info. Ios/or any other device's info is deleted.

Now in the logs I'm clearing getting error persisting cccd. Did not remove peer device. Removing peer device from index 0.
So, this is how im able to regenerate but here I observer another two things.
1. It stores info of devices of only two devices when max ble restore and device count is 3.
2. When error persisting cccd error triggers, after this error it is only able to store info of only one device.

veneno-529, 2024-05-27:
@rahult-github 
Any update on the issue?
Were you able to regenerate it.
As far as my observations go I think it is deleting all the peer info except the one it is currently connected.
I think changing the delete except function to delete the oldest peer should have worked but again the patch which was shared in earlier thread didn't work.

rahult-github, 2024-05-27:
Hi @veneno-529  , I need sometime for this. This code is taken as is from upstream nimble and not updated by espressif. But will try to provide a fix soon.

veneno-529, 2024-05-30:
@rahult-github Okay will wait for the fix. Temporarily I've increased the number of Max CCCDs  to store count. But don't know when it may overflow.

veneno-529, 2024-06-10:
Hello @rahult-github 
Any update on the issue?

veneno-529, 2024-07-09:
Hi @rahult-github its been over a month have you found any solution for this? because even increasing the count of max cccd is not helping at one point it just overflows and the issue again arises.

jrahlf, 2024-10-04:
We have also seen `Error 6 (0x6) GATT REQ NOT SUPPORTED` in our setup and could "fix" it by increasing the maximum number of stored CCCDs.

@rahult-github 
Can you explain
a) how large the config value should be? Should it be `number of charactersistics` times `maximum number of bonds in nvs`?
b) Why are the CCCDs stored in nvs in the first place? As I understand it, they only store the info if notification and indication is enabled and this should not be persisted anyway.

jrahlf, 2024-10-21:
Hi @SumeetSingh19 can you answer my questions regarding the CCCDs in nvs?

SumeetSingh19, 2025-04-08:
Hi @veneno-529 and @jrahlf , a similar issue was found and fixed recently in our internal testing. Can you please confirm if this issue still occurs for you?

veneno-529, 2025-05-05:
@SumeetSingh19 can you share the patched files?

jrahlf, 2025-05-07:
> Hi [@veneno-529](https://github.com/veneno-529) and [@jrahlf](https://github.com/jrahlf) , a similar issue was found and fixed recently in our internal testing. Can you please confirm if this issue still occurs for you?

We have disabled storing CCCDs in NVS. We do not need it, our central device (smartphone) sets up the subscriptions on every connect.

jrahlf, 2025-09-05:
By the way, the Bluetooth spec says "The Client Characteristic Configuration descriptor value shall be
persistent across connections for bonded devices" in section 3.3.3.3 Client Characteristic Configuration.
So choosing a value `BT_NIMBLE_MAX_CCCDS` smaller than number of supported bonds times subscriptable characteristics would violate that.
I would still assume that application will normally subscribe on every connect and **not** rely on the server to restore to the notification states, though.