espressif/esp-idf #13772: socketpair support for posix compliance

sramrajkar, 2024-05-10:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

I am trying to compile the following library for ESP32S3 using 5.x.x branches 
https://github.com/libplctag/libplctag
and need some implementation for  int socketpair(int domain, int type, int protocol, int sv[2])
Looks like ESP IDF is missing this posix API.

ESP-Marius, 2024-05-13:
Thanks for the request, we do indeed not support his. It is something we can consider adding support for in the future.

From looking at their github I could see some [discussion about porting the library](https://github.com/libplctag/libplctag/issues/68) though, and it seems that even if we had `socketpair()` you might still run into problems trying to run this.

david-cermak, 2024-09-20:
@sramrajkar I think you're just missing the non-blocking mode, everything else looks good.
here's the updated code, I've only briefly tested:
```cpp
#include <netdb.h>
#include <sys/socket.h>
#include "esp_check.h"

#define INVALID_SOCKET (-1)

static const char *TAG = "socket_helpers";

static int set_nonblocking(int sock)
{
    int opt;
    opt = fcntl(sock, F_GETFL, 0);
    if (opt == -1){
        return -1;
    }
    if (fcntl(sock, F_SETFL, opt | O_NONBLOCK) == -1) {
        return -1;
    }
    return 0;
}

int socketpair(int domain, int type, int protocol, int sv[2])
{
    struct sockaddr_storage ss;
    struct sockaddr_in *sa = (struct sockaddr_in *)&ss;
    socklen_t ss_len = sizeof(struct sockaddr_in);
    int fd1 = INVALID_SOCKET;
    int fd2 = INVALID_SOCKET;
    int listenfd = INVALID_SOCKET;
    int ret = 0; // Success

    sa->sin_family = AF_INET;
    sa->sin_addr.s_addr = htonl(INADDR_LOOPBACK);
    sa->sin_port = 0;
    listenfd = socket(AF_INET, SOCK_STREAM, IPPROTO_TCP);
    ESP_GOTO_ON_FALSE(listenfd != INVALID_SOCKET, -1, err, TAG, "Cannot create listening socket");
    ESP_GOTO_ON_FALSE(listen(listenfd, 1) == 0, -1, err, TAG, "Failed to listen");

    memset(&ss, 0, sizeof(ss));
    ss_len = sizeof(ss);
    ESP_GOTO_ON_FALSE(getsockname(listenfd, (struct sockaddr *)&ss, &ss_len) >= 0, -1, err, TAG, "getsockname failed");

    sa->sin_family = AF_INET;
    sa->sin_addr.s_addr = htonl(INADDR_LOOPBACK);

    fd1 = socket(AF_INET, SOCK_STREAM, IPPROTO_TCP);
    ESP_GOTO_ON_FALSE(fd1 != INVALID_SOCKET, -1, err, TAG, "Cannot create read side socket");
    ESP_GOTO_ON_FALSE(set_nonblocking(fd1) == 0, -1, err, TAG, "Failed to set socket to nonblocking mode");
    if (connect(fd1, (struct sockaddr *)&ss, ss_len) < 0) {
        ESP_GOTO_ON_FALSE(errno == EINPROGRESS || errno == EWOULDBLOCK, -1, err, TAG, "Failed to connect fd1");
    }
    fd2 = accept(listenfd, NULL, 0);
    if (fd2 == -1) {
        ESP_GOTO_ON_FALSE(errno == EINPROGRESS || errno == EWOULDBLOCK, -1, err, TAG, "Failed to accept fd2");
    }
    ESP_GOTO_ON_FALSE(set_nonblocking(fd2) == 0, -1, err, TAG, "Failed to set socket to nonblocking mode");

    close(listenfd);
    sv[0] = fd1;
    sv[1] = fd2;
    return ret;

err:
    if (listenfd != INVALID_SOCKET) {
        close(listenfd);
    }
    if (fd1 != INVALID_SOCKET) {
        close(fd1);
    }
    if (fd2 != INVALID_SOCKET) {
        close(fd2);
    }
    return ret;
}
```

david-cermak, 2024-10-30:
@sramrajkar  We will soon publish `socket-helper` component containing the above implementation of socketpair and some other utilities: https://github.com/espressif/esp-protocols/pull/671
You can check if you're able to compile (and run) some `libplctag` project with this.

sramrajkar, 2024-11-01:
@david-cermak yes I can try this over the weekend with the new module. Thank you for looking into this.

david-cermak, 2024-12-16:
Here's an example of using sock-utils with libplctag library: https://github.com/david-cermak/esp-network-examples/tree/main/plctag/simple

There's still one problem with declaration of `socketpair()` from IDF --  I will address this in https://github.com/espressif/esp-protocols/pull/713 and close this issue. Please comment if you have any concern.

closed 2024-12-20 by espressif-bot