espressif/esp-idf #13773: Weird memory random consumption increase about 6k without any reason.

filzek, 2024-05-10:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

Using a esp32 with esp-insight enable, the tasks are already using a limited resoruces very well tuned, but often the internal heap size drops about 6.5k

 free 10852 allocated 264232 min_free 9944 largest_free_block 8176

and suddenly rertuns to 17.5k

 free 17404 allocated 258520 min_free 9944 largest_free_block 10224

this happen every 3 of 64 times of 3 minutes each.

ESP-Marius, 2024-05-13:
Hmm, without some more info it is hard to say why.

Maybe you could try enabling [Heap-tracing](https://docs.espressif.com/projects/esp-idf/en/stable/esp32/api-reference/system/heap_debug.html#heap-tracing) or [Heap task tracing(https://docs.espressif.com/projects/esp-idf/en/stable/esp32/api-reference/system/heap_debug.html#heap-task-tracking) and dump this info when this memory "disappears" to help you trace down where it went.

ESP-Marius, 2024-05-31:
Closing this due to lack of information, feel free to re-open if there is any more information about this issue.

closed 2024-05-31 by ESP-Marius