espressif/esp-idf #13774: demo代码效果异常

crl6, 2024-05-10:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

sdasddsdsdsds

crl6, 2024-05-10:
i2s_pdm_example_main.c
一块板子做master接收，一块板子做slave发送。

接收到的数据保存在sd卡文件。文件如附件。
[录音文件.zip](https://github.com/espressif/esp-idf/files/15275294/default.zip)

L-KAYA, 2024-05-20:
PDM TX 和 PDM RX 的时钟方向都是输出，所以现在不支持把 PDM RX 和 PDM TX 作为主从机来传输数据的用法。

另外 PDM 作为脉冲密度调制，不建议作为设备间通信协议传输数字信号，因为通过PDM传输的信号会经过硬件滤波器调制和还原，和原始数据会有一定差异。所以例程中也并没有演示这种用法。

crl6, 2024-05-20:
“PDM TX 和 PDM RX 的时钟方向都是输出”   我可以理解成： esp32s3 不支持pdm slave模式吗

crl6, 2024-05-26:
done