espressif/esp-idf #13776: gattc_multi_connect The number of connections can only be five

laopi333, 2024-05-11:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

ESP-IDF v5.1.2-1-g2bbd2ec531

### Espressif SoC revision.

esp32s3

### Operating System used.

Linux

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### Development Kit.

ESP32-S3-WROOM-1

### Power Supply used.

USB

### What is the expected behavior?

我尝试使用开发板gattc连接最大数量的蓝牙设备，在menuconfig里面设置了 `(9) BT/BLE MAX ACL CONNECTIONS(1~9) `

### What is the actual behavior?

我期待最大能连接到9个或者7个设备，但是只能连接到5个，当连接第6个设备的时候，在 ESP_GATTC_OPEN_EVT 中产生了这样的错误：
`W (290423) BT_L2CAP: L2CA_CancelBleConnectReq - no connection pending
E (290423) GATTC_MULTIPLE_DEMO: connect device failed, status 133`
开发板一直在尝试连接第6个设备，但是一直打印这样的错误，我找不到原因

### Steps to reproduce.

修改 gattc_multi_connect demo，将连接的数量和相关参数增大

### Debug Logs.

```plain
W (49023) BT_L2CAP: L2CA_CancelBleConnectReq - no connection pending
E (49033) GATTC_MULTIPLE_DEMO: connect device failed, status 133
```

### More Information.

#include <stdint.h>
#include <string.h>
#include <stdbool.h>
#include <stdio.h>
#include "nvs.h"
#include "nvs_flash.h"

#include "esp_bt.h"
#include "esp_gap_ble_api.h"
#include "esp_gattc_api.h"
#include "esp_gatt_defs.h"
#include "esp_bt_main.h"
[... 438 lines cut ...]

    //register the callback function to the gattc module
    ret = esp_ble_gattc_register_callback(esp_gattc_cb);
    if(ret){
        ESP_LOGE(GATTC_TAG, "gattc register error, error code = %x", ret);
        return;
    }

    ret = esp_ble_gattc_app_register(PROFILE_A_APP_ID);
    if (ret){
        ESP_LOGE(GATTC_TAG, "gattc app register error, error code = %x", ret);
        return;
    }

    ret = esp_ble_gatt_set_local_mtu(500);
    if (ret){
        ESP_LOGE(GATTC_TAG, "set local  MTU failed, error code = %x", ret);
    }

    for(;;)
    {
        printf("conn_device_num:%d\n",conn_device_num);
        vTaskDelay(3000 / portTICK_PERIOD_MS);
    }
}

laopi333, 2024-05-20:
@esp-cjh 非常感谢你的回复，这解决了我的问题。另外我还有一个问题，我不太清楚 BT_BTC_TASK_STACK_SIZE 和 BT_BTU_TASK_STACK_SIZE 这两个栈空间有什么区别，如果我想提升蓝牙多连接的稳定性，增大这两个栈空间会有帮助嘛，如果能回答我会非常感谢。

esp-cjh, 2024-05-21:
@laopi333 
BT_BTC_TASK_STACK_SIZE 和 BT_BTU_TASK_STACK_SIZE 分别代表 Bluedroid 协议栈的 BTC task 和 BTU task 的栈空间大小。BTC task 主要用于处理上层应用消息和事件，BTU task 主要用于处理蓝牙协议（Host 部分）。你在测试多连接过程中有遇到过这两个栈溢出的情况吗？如果没有的话增大栈空间可能作用不大。
关于蓝牙多连接稳定性，你主要遇到哪些问题呢？是连接断开还是其他问题？

laopi333, 2024-05-22:
@esp-cjh 谢谢回复，我遇到的问题主要是在gatt客户端连接超过4个以上设备的时候，连接会不稳定，有时候会出现断开连接的现象，出现了这样的警告
`W (95703) BT_APPL: gattc_conn_cb: if=1 st=0 id=1 rsn=0x8
W (95703) BT_HCI: hcif disc complete: hdl 0x1, rsn 0x8
`，同样的连接多个设备后有时候断开连接的问题我在使用nimble协议栈进行蓝牙多连接时也产生了，我不知道是因为资源受限还是怎样的原因，因为在连接数量较少时候是很难产生断开连接的问题，所以我觉得应该不是我的gatt服务器产生的问题。

esp-cjh, 2024-05-22:
@laopi333 GATT 服务器端用的是哪款芯片？连接上之后双方会不会进行数据交互？大概多久会出现连接断开的问题？我这边会尝试复现并找到问题原因。

laopi333, 2024-05-22:
@esp-cjh gatt客户端用的是ESP32-S3，gatt服务端用的都是NRF52840开发板，测试中连接后没有进行数据交互，但是我想要在确保连接稳定性的情况下进行数据交互。连接数量超过4个后，可能5分钟之内就会出现断连现象，连接数量越多出现的频率越高。
这是我用gattc_multi_connect demo简单修改后的测试代码:
[ble_multi_conn_problem.zip](https://github.com/espressif/esp-idf/files/15402335/ble_multi_conn_problem.zip)
非常感谢您的热心帮助。

esp-cjh, 2024-05-22:
@laopi333 我使用 ESP32-S3 作为 client，连接 7 个 ESP32-C3（作为 server，使用 gatt_server example），测试了大概半小时左右，没有出现连接断开的问题。
[image]

所以我这边还需要再确认以下问题：
1 NRF52840 作为 server 时是否发起了连接参数更新？
请在 example 中添加以下代码然后复现，并提供出现问题时的完整 log 文件。
[image]
2 NRF52840 和 ESP32-S3 之间的距离大概多远？

laopi333, 2024-05-23:
@esp-cjh 非常感谢你的测试，首先:1.nrf52840在连接的时候发起了连接参数更新，这是我复现的日志：
[ble_multi_conn_log.txt](https://github.com/espressif/esp-idf/files/15413498/ble_multi_conn_log.txt)
2.我的s3和nrf52840都距离在1m内，不过我测试的环境中周围2.4g干扰较多，可能是有对这个连接测试产生影响

esp-cjh, 2024-05-23:
@laopi333 我查看了 log，你当前使用的连接参数是 interval = 50ms，timeout = 420ms，这导致如果连续 8 个连接事件都收不到包或者收错包就会导致超时断开。对于多连接和复杂环境下，这个 timeout 太小了，建议至少设置为 500（即 5 秒）。
[image]

laopi333, 2024-05-27:
@esp-cjh 非常感谢你的建议，在我修改我的广播端参数后，我分别使用nimble协议栈和bluedroid协议栈都进行了长时间的测试，并且都可以进行稳定的连接，解决了我的问题，这个issue可以关闭了，再次感谢你的帮助。

Alvin1Zhang, 2024-05-30:
Thanks for sharing the updates and glad to hear that the issue gets resolved, feel free to reopen if the issue still happens. Thanks.

closed 2024-05-30 by Alvin1Zhang