espressif/esp-idf #13780: Strange behavior of the SD_MMС example on 30 pins and 38 pins boards

Discorez, 2024-05-11:
Hello!
I use boards to which nothing is connected and upload a sketch of the SD_MMС example.
On a 30-pin board the example does not work and does not show any errors. I tried it on 5 different 30-pins boards.
Sreenshot
[image]

But if I do the same with the 38-pin board, then everything works as expected and I get an error message.
I tried it on 5 different 38-pins boards too.
Screenshot
[image]
Through experimentation, I found out that for boards with 30-pin, the code hangs in the SD_MMC.cpp file and is not executed after the line
`esp_err_t ret = esp_vfs_fat_sdmmc_mount(mountpoint, &host, &slot_config, &mount_config, &_card);`

Why does this happen and can it be fixed?

atanisoft, 2024-05-11:
This should likely be reviewed over on https://github.com/espressif/arduino-esp32 as a first pass.

Discorez, 2024-05-11:
Thank You.
Sorry, but I didn’t understand what to watch there. 
I'm using the standard example and it doesn't work. 
This example is expected to show an error because there is nothing connected to the board, but nothing happens.
If I connect an SD card, then nothing happens either... 
Any ideas?

atanisoft, 2024-05-11:
It's impossible to tell exactly why you are facing what you are facing as you did not fill in the issue template or provide a link to the example you are running. There have been a number of issues across both ESP-IDF and arduino-esp32 related to SD cards, both have different reasons.

Discorez, 2024-05-11:
The example which I'm uploading is located here - 
https://github.com/espressif/arduino-esp32/blob/master/libraries/SD_MMC/examples/SDMMC_Test/SDMMC_Test.ino

atanisoft, 2024-05-11:
In that case, I would suggest close this issue and open a new one over on https://github.com/espressif/arduino-esp32. Be sure to fill in the full issue template when you log the new issue, most importantly which esp32 variant you are using and which version of arduino-esp32 you are using. Also note that using examples from "master" branch are not guaranteed to work on the arduino-esp32 2.x releases and are targeting only the 3.x releases which are only available as pre-release and not guaranteed to work.

Discorez, 2024-05-11:
Thank you very much. This really isn't the right place...

closed 2024-05-11 by Discorez